avrdudes/avrdude #1720: avr_write_mem() does not really write something

dl8dtl, 2024-03-21:
When calling `avr_write_mem()` from my Python wrapper, it never writes something.

```
>>> m=ad.avr_locate_mem(p, 'eeprom')
>>> amnt=ad.avr_read_mem(pgm, p, m)
Reading 100 %
>>> print(amnt)
512
>>> m.get(10)
b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'
```
=> EEPROM is empty, OK
Now, fill the buffer, and try writing:
```
>>> amnt=ad.avr_write_mem(pgm, p, m, 512, True)
>>> ing   0 %
>>> print(amnt)
512
```
Note the "ing  0 %" - the remainder of the progress report. It returns so fast, no progress visible at all.
Now, read it back:
```
>>> amnt=ad.avr_read_mem(pgm, p, m)
Reading 100 %
>>> m.get(10)
b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'
```
=> Nothing written

Seems the logic inside `avr_write_mem()` somehow doesn't trigger any write.

stefanrueger, 2024-03-21:
I suspect the corresponding `m->tag[]` bytes are not set.

(off-topic: I also suspect the python implementation/wrapper of progress reporting does something else than the AVRDUDE progress reporting as I haven't seen the behaviour that "ing  0 %" is written for the latter.)

dl8dtl, 2024-03-21:
As I've got that `m->put` method, it could certainly somehow handle setting the tags.

stefanrueger, 2024-03-21:
> Who is responsible for setting the tags? (and how is it done? :)

Reading in a file for writing sets the tags (to manage holes in files in the Intel hex or Motorola S record format)

dl8dtl, 2024-03-21:
Yes, the progress reporter is a much simplified version in my test (it's a plain Python callback now).
I guess I have to assign `m->tag[]` in my `m.put()` helper then, yes.

dl8dtl, 2024-03-21:
Yep, works now:
```
>>> m=ad.avr_locate_mem(p, 'eeprom')
>>> ad.avr_read_mem(pgm, p, m)
Reading 100 %
512
>>> m.get(10)
b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff'
>>> m.put(b'The quick brown fox jumps over the lazy dog.')
44
>>> ad.avr_write_mem(pgm, p, m, 44)
Writing 100 %
44
>>> ad.avr_read_mem(pgm, p, m)
Reading 100 %
512
>>> m.get(44)
b'The quick brown fox jumps over the lazy dog.'
```

dl8dtl, 2024-03-21:
All is working.

closed 2024-03-21 by dl8dtl

stefanrueger, 2024-03-21:
> `m.put` and `m.get` are Python access methods to `m->buf`, accepting or returning Python bytes objects

`m.put` and `m.get` ought to take an address within buffer argument as well in order to provide random access to `m->buf`

dl8dtl, 2024-03-21:
They do, it is called `offset`. It just defaults to 0.

dl8dtl, 2024-03-21:
And for `m.put`, the length is taken from the input byte string, while for `m.get`, the length needs to be specified.

dl8dtl, 2024-03-21:
Remember, this is Python, we can have default values for arguments. :-) (The SWIG wrapper can handle them, even though the C backend functions cannot.)

stefanrueger, 2024-03-21:
> They do, it is called offset. It just defaults to 0.

Ahh, great.

> Yes, the progress reporter is a much simplified version in my test (it's a plain Python callback now).

In order to support the logic of libavrdude progress reporting, the application-specific `update_progress()` functions should not do anything before a header was provided and they should stop updating on progress as soon as a terminating progress report call was issued. Basically, it's a logic that dances around 
https://github.com/avrdudes/avrdude/blob/34366a6b9d4b489099e0cd9853d1e60098b44a41/src/term.c#L2685-L2686

This needs documentation, and this could be provided at the same time when Issue #1721is dealt with.

dl8dtl, 2024-03-21:
I think the existing `progress_callback` in my Python test code works basically like that:
```
# very simplified progress callback
global prog_hdr
prog_hdr = ""
def progress_callback(percent: int, etime: float, hdr: str, finish: int):
    if hdr:
        global prog_hdr
        prog_hdr = hdr
    print(f"{prog_hdr} {percent:3d} %", end='\r', file=sys.stderr, flush=True)
    if (percent == 100):
        print("", file=sys.stderr)
```
But that's merely a simplified example right now. There's room for fine-tuning. :-)