avrdudes/avrdude #1722: avr*timestamp() oddities

dl8dtl, 2024-03-21:
```
// Return us since program start, rolls over after ca 1h 12min
unsigned long avr_ustimestamp() {
```
…
```
// Return ms since program start, rolls over after ca 49d 17h
unsigned long avr_mstimestamp() {
  return avr_ustimestamp()/1000;
```
How could `avr_ustimestamp` possibly roll over after only 49 days when `avr_ustimestamp`, which is used as the base for its computation, already rolls over after 1 hour? I see that `avr_ustimestamp` maintains an internal start time as `unsigned long long`, but as the result is shortened to `unsigned long` (and `avr_mstimestamp` cannot access its internal state), the extra long intermediate state doesn't help much.
Besides, I think the figures listed there depend on the CPU architecture:
```
% cat size.c
#include <stdio.h>

int main(void) {
  printf("sizeof(unsigned long): %zu\n", sizeof(unsigned long));
  printf("sizeof(unsigned long long): %zu\n", sizeof(unsigned long long));
  return 0;
}
% cc -m32 -O -o size size.c
j@uriah 125% ./size
sizeof(unsigned long): 4
sizeof(unsigned long long): 8
% cc -m64 -O -o size size.c
% ./size
sizeof(unsigned long): 8
sizeof(unsigned long long): 8
```
I'd suggest to move the "epoch" state to outside of these functions (maybe even with the option to trigger an external reset of them), and then have the `avr_ustimestamp` and `avr_mstimestamp` routines access that state individually.

stefanrueger, 2024-03-22:
Yes, the comments are hogwash, and the actual rollover depends on architecture. I'd suggest to move to uint64_t for the integral us and ms timestamps, which at 64 bit has a rollover we don't need to worry about (or comment).

I don't know about the epoch; users can set their own quite easily by subtracting what they want to read as zero.

dl8dtl, 2024-03-22:
Yes, moving to `uint64_t` sounds like a clean solution to me.
The "epoch" is a variable only visible inside `avr_ustimestamp()` right now, so users cannot do anything with it.

stefanrueger, 2024-03-22:
> The "epoch" is a variable only visible inside avr_ustimestamp() right now, so users cannot do anything with it.

Sure, but if they need a different epoch (the most likely use case is *now*), then they can easily do so:
```
  uint64_t my_epoch = avr_ustimestamp(); // or some other time
  // ... do something
  uint64_t my_time = avr_ustimestamp() - my_epoch;
```

That is not radically more code than with an additional function that allows to set the epoch
```
  avr_set_epoch(avr_ustimestamp()); // or some other time
  // ... do something
  uint64_t my_time = avr_ustimestamp();
```

stefanrueger, 2024-03-29:
@dl8dtl Please check whether this change/update is sufficient for libavrdude. I realise that the code often assigns `uint64_t avr_ustimestamp()` to an `unsigned long` variable, and not an `uint64_t` variable, but the way these variables are used to determine the time between two calls is still correct owing to `unsigned` modulo arithmetic assuming the time between two calls is less than 2<sup>32</sup> µs (more than an hour).

PR #1729 (2024-04-05): Change return type for avr_(ms|us)timestamp() to uint64_t
Fixes #1722

@ndim Does commit https://github.com/avrdudes/avrdude/commit/2283b970ecda3da4f2824a56fcb26905e3dd32f7 increment the libavrdude version correctly?
closed 2024-04-05 by stefanrueger