avrdudes/avrdude #1723: Help wanted: CMake to compile Qt .ui files

dl8dtl, 2024-03-23:
I'd like to add a GUI example for the Python wrapper, using Qt / PySide.
For that, I started a `src/python` subdirectory. I'm about to create a number of `.ui` files (using Qt Designer), and would like to automatically compile them into `.py` implementation files that can be loaded into the Python code.
This is done using a command like

`uic -g python -o ui_foobar.py foobar.ui`

I'm stumbling across how to tell CMake about a generic compile command for an arbitrary number of `.ui` source files.
My rough idea:
```
SET(SOURCES
    file1.ui file2.ui file3.ui)
ADD_CUSTOM_COMMAND(
    OUTPUT ui?.py
    COMMAND "uic -g python -o ${PROJECT_BINARY_DIR}/src/python/? ?"
    ${SOURCES})
```
I have placed some question marks where I don't see how to do it.

Any helpers?

mcuee, 2024-03-24:
@Youw 

Just wondering if you can provide a quick tip here. Thanks.

Youw, 2024-03-24:
Either set [CMAKE_AUTOUIC](https://cmake.org/cmake/help/latest/variable/CMAKE_AUTOUIC.html#variable:CMAKE_AUTOUIC) to `TRUE` before creating a target (a library or executable) or set the [AUTOUIC](https://cmake.org/cmake/help/latest/prop_tgt/AUTOUIC.html) property of the target to `TRUE`.
Then add your `.ui` files to your target as if it is a regular cpp source.

Sample:
```cpp
set(CMAKE_AUTOUIC ON)
add_library(testLib
    mainwindow.cpp
    mainwindow.h
    mainwindow.ui
)
```
or
```cpp
add_library(testLib
    mainwindow.cpp
    mainwindow.h
    mainwindow.ui
)
set_target_properties(testLib
    PROPERTIES AUTOUIC ON
)
```

Youw, 2024-03-24:
there're other ways to achieve the same, but the one above is the most convenient

and the one you've tried to use (manyally invoking the `uic`) is probably the least convenient/recommended

dl8dtl, 2024-03-24:
I tried `CMAKE_AUTOUIC`, to no avail.
Problem here: it's not C++, it's Python. I do not need to compile anything but only the .ui files. As far as I could see, `CMAKE_AUTOUIC` only seems to parse C++ implementation files – which I don't have.
I don't even know how I would setup a CMake target for something that isn't going to be a classic compilation result. (A couple of steps later, it might become one: when trying to create a portable bundle out of all the Python files.)
It's not that I'd object to using C++, but the purpose of all this is to demonstrate a Python wrapper around our backend library.

Youw, 2024-03-25:
In that case you'd need to have a separate custom command for each `.ui` file, and a single custom target to aggregate all of those, something like:
```cmake
SET(SOURCES file1.ui file2.ui file3.ui)
set(UI_GENERATION_COMMANDS)
foreach(SOURCE SOURCES)
    add_custom_command(
        OUTPUT "${CMAKE_CURRENT_BINARY_DIR}/${SOURCE}.py"
        COMMAND "uic -g python -o ${CMAKE_CURRENT_BINARY_DIR}/${SOURCE}.py ${SOURCE}"
        COMMENT "Generating ${SOURCE} -> ${SOURCE}.py"
    )
    list(APPEND UI_GENERATION_COMMANDS "${CMAKE_CURRENT_BINARY_DIR}/${SOURCE}.py")
endforeach()
add_custom_target(GenerateUIToPy ALL DEPENDS ${UI_GENERATION_COMMANDS})
```

dl8dtl, 2024-03-25:
Besides, your example is also valuable in case we'd have a need for running any similar custom command in future.