avrdudes/avrdude #1732: 0xff optimization when reading

dl8dtl, 2024-03-30:
```
int avr_mem_hiaddr(const AVRMEM * mem)
{
  int i, n;
  static int disableffopt;

  /* calling once with NULL disables any future trailing-0xff optimisation */
  if(!mem) {
    disableffopt = 1;
    return 0;
  }
```
There is no way to turn it on again? This is not good.

dl8dtl, 2024-03-30:
Also, what I don't understand, from my protocol:
```
2024-03-30T09:55:55 Read 8666  bytes
2024-03-30T09:56:05 Wrote 16384 bytes to /home/joerg/src/avrdude/foo.hex
```
It clearly optimized the trailing 0xFF when reading (through `avr_read_mem()`, as can be seen from the 8666 bytes. Yet, when writing the memory buffer through `fileio()`, the entire memory region is saved. The code suggests to me that only the "relevant" bytes ought to be written.

dl8dtl, 2024-03-30:
OK, I understand the latter:
```
  if(size < 0 || op == FIO_READ || op == FIO_READ_FOR_VERIFY)
    size = mem->size;
```
So the upper layer is supposed to track the size.

stefanrueger, 2024-03-30:
> There is no way to turn it on again? This is not good.

I agree. AVRDUDE doesn't need to switch `-A` off again, but a library might want to. I suggest to make the `-A` flag a global variable that can be accessed by the library. 

`-A` is meant to *not* remove trailing 0xff sequences from avr_reads() or file input, as they could be relevant data and the programmer could see AVR flash as normal memory (as certain bootloaders do or AVR with page erase) rather than NOR memory.

dl8dtl, 2024-03-31:
Global variable sounds OK to me.
Other users of the library might turn `-A` into their own option system.
Btw. even the CLI might have a use for it: in terminal mode.

dl8dtl, 2024-03-31:
Rather than making the variable global, it could remain `static` (module-internal, not exposed in the library). That enforces the use of access functions.
A completely different option would be a kind of library context structure that holds all that kind of options (also `verbose`, and `quell_progress`). It could start out as a global, and eventually moved as a pointer into API calls – that could even make the library API reentrant. :-)

stefanrueger, 2024-03-31:
> even the CLI might have a use for it: in terminal mode

AVRDUDE only ever considers one part/programmer combo in each incarnation; once that's fixed `-A` either makes sense or not, but it wouldn't be required (or, in my understanding, make sense) to toggle that option back and forth.

stefanrueger, 2024-03-31:
> [global, access functions or context pointer]

As a quick shot I'd make this option global in the first instance, and then at some point change to context pointer for all global variables.

closed 2024-05-17 by stefanrueger