avrdudes/avrdude #1734: Error when using linuxgpio

JanLuudi, 2024-04-05:
I am trying to use avrdude in a similar way as @MCUdude described it in PR #1299 .

I am using a Raspberry Pi 4B (40 pin header)
Unfortunately, there seems to be something wrong and I need your help to solve it.

### This is the error I am getting
```
janno@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm

janno@raspberrypi:~ $ uname -a
Linux raspberrypi 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux
janno@raspberrypi:~ $ sudo avrdude -c spi_mfm -p m328p -P gpiochip0 -v

avrdude: Version 7.3-20240330 (cb611715)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /usr/local/etc/avrdude.conf
         User configuration file is /root/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using port            : gpiochip0
         Using programmer      : spi_mfm
avrdude linuxgpio_sysfs_open() OS error: cannot export GPIO 25, already exported/busy?: Invalid argument
avrdude main() error: unable to open port gpiochip0 for programmer spi_mfm

avrdude done.  Thank you.

```
It's the same for -P GPIO

### Snippet of my avrdude.conf
```
avrdude_conf_version = "7.3-20240330 (cb611715)";
...
default_linuxgpio  = "gpiochip0";
...

#------------------------------------------------------------
# Program from a Raspberry Pi GPIO port using linuxgpio
#------------------------------------------------------------

programmer # raspberry_pi_gpio
    id                     = "raspberry_pi_gpio";
    desc                   = "Bitbang Raspberry Pi GPIO via linuxgpio (sysfs or libgpiod)";
    type                   = "linuxgpio";
    prog_modes             = PM_ISP;
    connection_type        = linuxgpio;
    reset                  = 22;
    sck                    = 24;
    sdo                    = 23;
    sdi                    = 18;
;

programmer
    id                   = "spi_mfm";
    desc                 = "Use RPi GPIO for mains frequency monitor unit";
    type                 = "linuxgpio";
    prog_modes           = PM_ISP;
    connection_type      = linuxgpio;
    reset                = 25;
    sck                  = 11;
    sdo                 = 10;
    sdi                 = 9;
;
```

### And the content of /sys/class/gpio/
```
janno@raspberrypi:~ $ ls /sys/class/gpio
export  gpiochip512  gpiochip570  unexport
```

I did not export or use GPIO25 before, so I don't thinks that's the issue.

Does anyone have a clue what is going wrong?

mcuee, 2024-04-06:
Not so sure what is the issue.

But I just tested linuxgpio and linuxspi under my Raspberry Pi 400 and both are okay.

mcuee, 2024-04-06:
@JanLuudi 

Hmm, you seem to still use the older sysfs API, please install libgpiod-dev and the rebuild avrdue to use libgpiod instead.
https://github.com/avrdudes/avrdude/pull/1299

mcuee, 2024-04-06:
My config.

```
# #------------------------------------------------------------
# # linuxgpio
# #------------------------------------------------------------
#
programmer
    id                   = "linuxgpio";
    desc                 = "Use the Linux sysfs interface to bitbang GPIO lines";
    type                 = "linuxgpio";
    prog_modes           = PM_ISP;
    connection_type      = linuxgpio;
    reset                = 25;
    sck                  = 11;
    sdo                  = 10;
    sdi                  = 9;
```

I am testing PR https://github.com/avrdudes/avrdude/pull/1714 and it works.
```
mcuee@raspberrypi400arm64:~/build/avrdude_jw/build_linux/src $ ./avrdude -c linuxgpio -P gpiochip0 -p m328p
using libgpiod for linuxgpio
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)

avrdude done.  Thank you.
```