avrdudes/avrdude #1742: Add support for more AVR-DU parts

MCUdude, 2024-04-09:
Microchip has just released a preliminary datasheet revealing more AVR-DU chips.

https://onlinedocs.microchip.com/oxy/GUID-C431CD9B-65A0-4015-871C-58A444612066-en-US-2/GUID-C1FC3A0A-5479-44A0-979B-ACDD6118FAA0.html

[image]

CPN | Flash/SRAM | Pin Count | Package Type | Temperature Range | Carrier Type
-- | -- | -- | -- | -- | --
AVR32DU14T-I/SL | 32 KB/4 KB | 14 | SOIC | -40°C to +85°C | Tape & Reel
AVR32DU14-I/SL | 32 KB/4 KB | 14 | SOIC | -40°C to +85°C | Tube
AVR32DU20T-I/REB | 32 KB/4 KB | 20 | VQFN | -40°C to +85°C | Tape & Reel
AVR32DU20-I/REB | 32 KB/4 KB | 20 | VQFN | -40°C to +85°C | Tube
AVR32DU20T-I/SS | 32 KB/4 KB | 20 | SSOP | -40°C to +85°C | Tape & Reel
AVR32DU20-I/SS | 32 KB/4 KB | 20 | SSOP | -40°C to +85°C | Tube
AVR32DU28-I/SP | 32 KB/4 KB | 28 | SPDIP | -40°C to +85°C | Tube
AVR32DU28T-I/SS | 32 KB/4 KB | 28 | SSOP | -40°C to +85°C | Tape & Reel
AVR32DU28-I/SS | 32 KB/4 KB | 28 | SSOP | -40°C to +85°C | Tube
AVR32DU28T-I/STX | 32 KB/4 KB | 28 | VQFN | -40°C to +85°C | Tape & Reel
AVR32DU28-I/STX | 32 KB/4 KB | 28 | VQFN | -40°C to +85°C | Tube
AVR32DU32T-I/PT | 32 KB/4 KB | 32 | TQFP | -40°C to +85°C | Tape & Reel
AVR32DU32-I/PT | 32 KB/4 KB | 32 | TQFP | -40°C to +85°C | Tray
AVR32DU32T-I/RXB | 32 KB/4 KB | 32 | VQFN | -40°C to +85°C | Tape & Reel
AVR32DU32-I/RXB | 32 KB/4 KB | 32 | VQFN | -40°C to +85°C | Tray
AVR16DU14T-I/SL | 16 KB/2 KB | 14 | SOIC | -40°C to +85°C | Tape & Reel
AVR16DU14-I/SL | 16 KB/2 KB | 14 | SOIC | -40°C to +85°C | Tube
AVR16DU20T-I/REB | 16 KB/2 KB | 20 | VQFN | -40°C to +85°C | Tape & Reel
AVR16DU20-I/REB | 16 KB/2 KB | 20 | VQFN | -40°C to +85°C | Tube
AVR16DU20T-I/SS | 16 KB/2 KB | 20 | SSOP | -40°C to +85°C | Tape & Reel
AVR16DU20-I/SS | 16 KB/2 KB | 20 | SSOP | -40°C to +85°C | Tube
AVR16DU28-I/SP | 16 KB/2 KB | 28 | SPDIP | -40°C to +85°C | Tube
AVR16DU28T-I/SS | 16 KB/2 KB | 28 | SSOP | -40°C to +85°C | Tape & Reel
AVR16DU28-I/SS | 16 KB/2 KB | 28 | SSOP | -40°C to +85°C | Tube
AVR16DU28T-I/STX | 16 KB/2 KB | 28 | VQFN | -40°C to +85°C | Tape & Reel
AVR16DU28-I/STX | 16 KB/2 KB | 28 | VQFN | -40°C to +85°C | Tube
AVR16DU32T-I/PT | 16 KB/2 KB | 32 | TQFP | -40°C to +85°C | Tape & Reel
AVR16DU32-I/PT | 16 KB/2 KB | 32 | TQFP | -40°C to +85°C | Tray
AVR16DU32T-I/RXB | 16 KB/2 KB | 32 | VQFN | -40°C to +85°C | Tape & Reel
AVR16DU32-I/RXB | 16 KB/2 KB | 32 | VQFN | -40°C to +85°C | Tray

stefanrueger, 2024-04-13:
Best to wait for the .atdf files. We have a couple of ghost parts in `avrdude.conf` that were documented but never materialised:
```
$ grep //AVR src/avrintel.c  | grep -v atdf
  //AVR8EA28                                                  avrdude // Sources
  //AVR8EA32                                                  avrdude // Sources
```

PR #1747 updates the avrintel files to the most recent .atdf files but unfortunately, above AVR-DU parts are not yet there.

MCUdude, 2024-05-08:
A new AVR-Dx pack has _just_ been released! lots of new DU chips added

https://packs.download.microchip.com/

[image]

PR #1804 (2024-05-17): Add new AVRnnDUnn parts
Fixes #1742 

@askn37 Would you be interested in reviewing this entry? You know *a lot* about the new parts. `avrdude -p AVR32DU32/S` shows the full definition without parenting. Added `16du14`, `16du20`, `16du28`, `16du32`, `32du14`, `32du20`, `32du28` and `32du32`
closed 2024-05-17 by stefanrueger