avrdudes/avrdude #1743: Add support for emscripten

koen1711, 2024-04-12:
I made a fork of avrdude that works with emscripten, would you be interested in a fix upstream that allows support for this and maybe be used within Arduino?

mcuee, 2024-04-12:
Thanks, please help to provide more details about the fork and why you need to use emscripten. Arduino already supports the use of avrdude.

koen1711, 2024-04-12:
Using this technique I managed to get an uploader for an Arduino in the browser without any necessary installs, like Arduino Create Agent. Please note that the compilation still runs on a backend.

mcuee, 2024-04-13:
@koen1711 

You can always create a pull request so that the reviewers can review. PRs will be welcome.

koen1711, 2024-04-13:
Yes I will, but I first wanted to investigate if avrdude/Arduino would be interested in such additions.q

stefanrueger, 2024-04-13:
@koen1711 Yes, the project is interested in user contributions. Whether or not a PR is going to be accepted will depend on its scope, complexity, maintainability, code quality and whether it might be used reasonably often. But yes, we're look at it with an open mind.

stefanrueger, 2024-05-08:
Just checking whether you want to keep this issue open @koen1711

stefanrueger, 2024-08-01:
Closing as we have answered the question

closed 2024-08-01 by MCUdude