avrdudes/avrdude #1748: Trouble using Raspberry pi 5

BastienCagna, 2024-04-14:
Hello,
I was using avrdude to flash my MCU from an old Raspberry Pi 3 and recently move to the new Pi 5.
I was used to flash using this command:
```shell
sudo avrdude -p m162 -c linuxgpio -U flash:w:build/firmwares/main.hex 
```
But now, it return that: 
 ```shell
avrdude OS error: cannot export GPIO 22, already exported/busy?: Invalid argumentavrdude error: unable to open programmer linuxgpio on port unknown

avrdude done.  Thank you.
```
I know that this new version of the raspberry use the new RP1 chip to control GPIO. Could it be the source of this trouble? May exist a workarround?

Regards,
Bastien

stefanrueger, 2024-04-14:
[Try installing libgpiod-dev and the rebuild avrdue to use libgpiod instead](https://github.com/avrdudes/avrdude/discussions/1737)

stefanrueger, 2024-04-17:
Apparently new kernel versions base the GPIO numbers at 512, not 0. So, another workaround is to add 512 to the gpio numbers of your linuxgpio entry in ~/.avrduderc.

stefanrueger, 2024-04-25:
@BastienCagna It's been some time. Has your question been answered? Can the issue be closed?

BastienCagna, 2024-05-07:
Hey,
Thanks for your answers @stefanrueger !
Sorry I didn't work oon this project since my post. I guess I fixed the issue I was describing using libgpio. But now, it's looks like avrdude can not communicate with the MCU. Whe I execute the following command (which just worked few minutes ago on raspberry pi 3B), it says that the MCU is not responding :(

```shell
sudo /usr/local/bin/avrdude -p t2313 -c linuxgpio -P gpiochip0 -U flash:w:test.hex -v

avrdude: Version 7.3-20240428 (37ac6eba)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /usr/local/etc/avrdude.conf
         User configuration file is /root/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

using libgpiod for linuxgpio
         Using port            : gpiochip0
         Using programmer      : linuxgpio
avrdude: input file test.hex auto detected as Intel Hex
         AVR Part              : ATtiny2313
         Programming modes     : ISP, HVPP, debugWIRE, SPM
         Programmer Type       : linuxgpio
         Description           : Use the Linux sysfs interface to bitbang GPIO lines
         Pin assignment        : libgpiod
           RESET   =  22
           SCK     =  11
           SDO     =  10
           SDI     =  9
avrdude bitbang_initialize() error: AVR device not responding
avrdude main() error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower the bit clock frequency, e.g. -B 125kHz
        - use -F to override this check

avrdude done.  Thank you.
```

It tried to change baudrate (-B 100)  but it also failed.
I also tried with gpiochip4 but same error.

I have no ~/.avrduderc but I guess it is equivalent to the global configuration made in avrdude.conf. So your workaround is to specify SCK = 512 + 11 = 523 ? For me it failed (but as I said, the first problem looks already fixed now for me).

stefanrueger, 2024-05-08:
@BastienCagna The short of it is that I don't have an answer. This requires more debugging. The posted run looks like `libgpiod` is compiled in and working; you are using the latest git-main. You tried `-P gpiochip0` and also `-P gpiochip4` (which seems to be suggested by the build system judging from your system avrdude.conf). You say it works with an RPi 3b. Is the failing run on an RPi 5? Have you installed `libgpiod-dev` and compiled avrdude on the RPi 5? @mcuee can you reproduce above? @BastienCagna Can you post a `-vv` output and upload the output of a `-vvvv` run?

>  the empty default spi suprised me

@BastienCagna If you know a method to set `default_spi` from the build system please let us know. @mcuee? @MCUdude?

MCUdude, 2024-05-08:
It looks like neither the auto tools nor Cmake deals with the `default_spi` port.

It may be as simple as adding `set(DEFAULT_SPI_PORT "/dev/spidev0.0")` here:

https://github.com/MCUdude/avrdude/blob/37ac6ebaa0330470d14cff37256dfd94293fda83/src/CMakeLists.txt#L90

And `DEFAULT_SPI_PORT="/dev/spidev0.0"` here:

https://github.com/MCUdude/avrdude/blob/37ac6ebaa0330470d14cff37256dfd94293fda83/src/configure.ac#L452-L462

mcuee, 2024-05-08:
@stefanrueger and @BastienCagna 

I do not have Raspberry Pi 5, but libgpiod and git main work well for me on my Raspberry Pi 400. I have no idea why it does not work on Raspberry Pi 5.

I can only suggest @BastienCagna to check the connection. The other suggestion is to try linuxspi as well to see if that works or not. Or try another MCU to see if that makes a difference or not.

stefanrueger, 2024-05-08:
> It may be as simple as adding ...

@MCUdude Great idea! Do you want to do a PR?

mcuee, 2024-06-25:
@BastienCagna 

Not so sure if this helps.
* https://github.com/openocd-org/openocd/commit/ad87fbd1cf28760795c4e18f3318a2d720e5a8a6
* https://github.com/openocd-org/openocd/blob/ad87fbd1cf28760795c4e18f3318a2d720e5a8a6/tcl/interface/raspberrypi-gpio-connector.cfg

mcuee, 2024-07-15:
@BastienCagna 
Any updates on this issue? Thanks.

mcuee, 2024-07-15:
> > It may be as simple as adding ...
> 
> @MCUdude Great idea! Do you want to do a PR?

Adding `enhancement` label to capture this one.

stefanrueger, 2024-08-01:
Do we have an update on this? @BastienCagna Have you solved the problem and want to share the solution? 

Other than that I suggest closing the issue in a week or so

mcuee, 2024-08-09:
@BastienCagna 

Any updates? If not we will close the issue.

closed 2024-08-14 by mcuee

liamw9534, 2024-10-24:
I think this issue should be reopened.  I did manage to get the `linuxgpio` method to work on rpi5 but there was a modification I had to make to the source code for this to work.  The bottom line is that rpi5 has relocated its expansion header GPIO pins to `gpiochip4` as opposed to `gpiochip0` in previous releases.  Here is the GPIO mappings as reported by the command `gpioinfo gpiochip4` on rpi5:

```
gpiochip4 - 54 lines:
        line   0:     "ID_SDA"       unused   input  active-high
        line   1:     "ID_SCL"       unused   input  active-high
        line   2:      "GPIO2"       unused   input  active-high
        line   3:      "GPIO3"       unused   input  active-high
        line   4:      "GPIO4" "wiringpi_gpio_irq" input active-high [used]
        line   5:      "GPIO5"       unused   input  active-high
        line   6:      "GPIO6"       unused   input  active-high
        line   7:      "GPIO7"       unused   input  active-high
        line   8:      "GPIO8"       unused   input  active-high
        line   9:      "GPIO9"       unused   input  active-high
        line  10:     "GPIO10"       unused   input  active-high
        line  11:     "GPIO11"       unused   input  active-high
        line  12:     "GPIO12"       unused   input  active-high
        line  13:     "GPIO13"       unused   input  active-high
        line  14:     "GPIO14"       unused   input  active-high
        line  15:     "GPIO15"       unused   input  active-high
        line  16:     "GPIO16"       unused   input  active-high
        line  17:     "GPIO17"       unused   input  active-high
        line  18:     "GPIO18"       unused   input  active-high
        line  19:     "GPIO19"       unused   input  active-high
        line  20:     "GPIO20"       unused   input  active-high
        line  21:     "GPIO21"       unused   input  active-high
        line  22:     "GPIO22"       unused   input  active-high
        line  23:     "GPIO23"       unused   input  active-high
        line  24:     "GPIO24"       unused   input  active-high
        line  25:     "GPIO25"       unused   input  active-high
        line  26:     "GPIO26"       unused   input  active-high
        line  27:     "GPIO27"       unused   input  active-high
        line  28: "PCIE_RP1_WAKE" unused output active-high
        line  29:   "FAN_TACH"       unused   input  active-high
        line  30:   "HOST_SDA"       unused   input  active-high
        line  31:   "HOST_SCL"       unused   input  active-high
        line  32:  "ETH_RST_N"  "phy-reset"  output   active-low [used]
        line  33:          "-"       unused   input  active-high
        line  34: "CD0_IO0_MICCLK" "cam0_reg" output active-high [used]
        line  35: "CD0_IO0_MICDAT0" unused input active-high
        line  36: "RP1_PCIE_CLKREQ_N" unused input active-high
        line  37:          "-"       unused   input  active-high
        line  38:    "CD0_SDA"       unused  output  active-high
        line  39:    "CD0_SCL"       unused   input  active-high
        line  40:    "CD1_SDA"       unused   input  active-high
        line  41:    "CD1_SCL"       unused   input  active-high
        line  42: "USB_VBUS_EN" unused output active-high
        line  43:   "USB_OC_N"       unused   input  active-high
        line  44: "RP1_STAT_LED" "PWR" output active-low [used]
        line  45:    "FAN_PWM"       unused  output  active-high
        line  46: "CD1_IO0_MICCLK" "cam1_reg" output active-high [used]
        line  47:  "2712_WAKE"       unused   input  active-high
        line  48: "CD1_IO1_MICDAT1" unused input active-high
        line  49: "EN_MAX_USB_CUR" unused output active-high
        line  50:          "-"       unused   input  active-high
        line  51:          "-"       unused   input  active-high
        line  52:          "-"       unused   input  active-high
        line  53:          "-"       unused   input  active-high
```

To be clear you have to open the device `/dev/gpiochip4` and use lines in the range of 0-53 when using `libgpiod` on rpi5.  The current `/dev/gpiochipXXX` device is currently hardcoded in the file `linuxgpio.c`.  See https://github.com/avrdudes/avrdude/blob/0b92721af9aaa686095542cc54715a3fa06b5895/src/linuxgpio.c#L509.

This is somewhat unfortunate because the configuration file has the setting `default_linuxgpio` which ideally would be set by the user to `gpiochip4`.  The most sensible modification would be to allow the user-defined `default_linuxgpio` to be used by `linuxgpio.c` and then it is just a case of the user specifying which GPIOs they wish to use in the range 0-53 in their configuration file.

Note: this `libgpiod` mapping is different to that reported by `/sys/kernel/debug/gpio` which numbers GPIOs in the range 512-624 -- this mapping is internal to the kernel and not exposed through the user space API so you can ignore this.

BastienCagna, 2025-05-05:
Hi there,
Thanks a lot for the energy you spend on it. I just tried again and notified that the MOSI pin of the RPI has a bad response (don't know if it is just mine or general). 

Solution: use an other pin (like GPIO 25) for MOSI worked for me!

But i also found the -B option to reduce the baudrate don't look to works and it looks to cause failure often. The clock is locked to around 360kHz which looks to be a bit too high.

MCUdude, 2025-05-05:
If you're using linuxgpio or raspberry_pi_gpio, you'll have to use the `-i` flag instead to set the bit-bang delay

BastienCagna, 2025-05-05:
OOhh yes it's works fine now! Thank you so much!

mcuee, 2025-05-06:
Using libgpiod latest git version and the pin number assignment does not seem to change for my Raspberry Pi 500.

```
mcuee@raspberrypi:~/build/avr/avrdude $ gpioinfo 
gpiochip0 - 54 lines:
	line   0:	"ID_SDA"        	input
	line   1:	"ID_SCL"        	input
	line   2:	"GPIO2"         	input
	line   3:	"GPIO3"         	input
	line   4:	"GPIO4"         	input
	line   5:	"GPIO5"         	input
	line   6:	"GPIO6"         	input
	line   7:	"GPIO7"         	output active-low consumer="spi0 CS1"
	line   8:	"GPIO8"         	output active-low consumer="spi0 CS0"
	line   9:	"GPIO9"         	input
	line  10:	"GPIO10"        	input
	line  11:	"GPIO11"        	input
	line  12:	"GPIO12"        	input
	line  13:	"GPIO13"        	input
	line  14:	"GPIO14"        	input
	line  15:	"GPIO15"        	input
	line  16:	"GPIO16"        	input
	line  17:	"GPIO17"        	input
	line  18:	"GPIO18"        	input
	line  19:	"GPIO19"        	input
	line  20:	"GPIO20"        	input
	line  21:	"GPIO21"        	input
	line  22:	"GPIO22"        	input
	line  23:	"GPIO23"        	input
	line  24:	"GPIO24"        	input
	line  25:	"GPIO25"        	input
	line  26:	"GPIO26"        	input
	line  27:	"GPIO27"        	input
	line  28:	"PCIE_RP1_WAKE" 	input
	line  29:	"-"             	input
	line  30:	"HOST_SDA"      	input
	line  31:	"HOST_SCL"      	input
	line  32:	"ETH_RST_N"     	output active-low consumer="phy-reset"
	line  33:	"PCIE_DET_WAKE" 	input
	line  34:	"-"             	output consumer="cam0_reg"
	line  35:	"-"             	input
	line  36:	"RP1_PCIE_CLKREQ_N"	input
	line  37:	"-"             	input
	line  38:	"-"             	input
	line  39:	"-"             	input
	line  40:	"CD1_SDA"       	input
	line  41:	"CD1_SCL"       	input
	line  42:	"USB_VBUS_EN"   	output
	line  43:	"USB_OC_N"      	input
	line  44:	"RP1_STAT_LED"  	output active-low consumer="PWR"
	line  45:	"-"             	input
	line  46:	"-"             	output consumer="cam1_reg"
	line  47:	"HOST_WAKE"     	input
	line  48:	"-"             	input
	line  49:	"EN_MAX_USB_CUR"	output
	line  50:	"-"             	input
	line  51:	"-"             	input
	line  52:	"-"             	input
	line  53:	"-"             	input
gpiochip10 - 32 lines:
	line   0:	unnamed         	input
	line   1:	"2712_BOOT_CS_N"	output active-low consumer="spi10 CS0"
	line   2:	"2712_BOOT_MISO"	input
	line   3:	"2712_BOOT_MOSI"	input
	line   4:	"2712_BOOT_SCLK"	input
	line   5:	unnamed         	input
	line   6:	unnamed         	input
	line   7:	unnamed         	input
	line   8:	unnamed         	input
	line   9:	unnamed         	input
	line  10:	"-"             	input
	line  11:	"-"             	input
	line  12:	"-"             	input
	line  13:	"-"             	input
	line  14:	"M2_DET_WAKE"   	input
	line  15:	"M2_PWR_EN"     	input
	line  16:	unnamed         	input
	line  17:	unnamed         	input
	line  18:	"KEYB_BOOTSEL"  	input
	line  19:	"-"             	input
	line  20:	"PWR_GPIO"      	input active-low consumer="pwr_button"
	line  21:	"KEYB_RUN"      	input
	line  22:	"-"             	input
	line  23:	"USER_LED"      	input
	line  24:	"BT_RTS"        	input
	line  25:	"BT_CTS"        	input
	line  26:	"BT_TXD"        	input
	line  27:	"BT_RXD"        	input
	line  28:	"WL_ON"         	output consumer="wl-on-reg"
	line  29:	"BT_ON"         	output consumer="shutdown"
	line  30:	"WIFI_SDIO_CLK" 	input
	line  31:	"WIFI_SDIO_CMD" 	input
gpiochip11 - 15 lines:
	line   0:	"RP1_SDA"       	input
	line   1:	"RP1_SCL"       	input
	line   2:	"RP1_RUN"       	output consumer="RP1 RUN pin"
	line   3:	"SD_IOVDD_SEL"  	output consumer="vdd-sd-io"
	line   4:	"SD_PWR_ON"     	output consumer="sd-vcc-reg"
	line   5:	"SD_CDET_N"     	input active-low consumer="cd"
	line   6:	"SD_FLG_N"      	input
	line   7:	unnamed         	input
	line   8:	"2712_WAKE"     	input
	line   9:	"2712_STAT_LED" 	output active-low consumer="ACT"
	line  10:	unnamed         	input
	line  11:	unnamed         	input
	line  12:	"PMIC_INT"      	input
	line  13:	"UART_TX_FS"    	input
	line  14:	"UART_RX_FS"    	input
gpiochip12 - 6 lines:
	line   0:	"HDMI0_SCL"     	input
	line   1:	"HDMI0_SDA"     	input
	line   2:	"HDMI1_SCL"     	input
	line   3:	"HDMI1_SDA"     	input
	line   4:	"PMIC_SCL"      	input
	line   5:	"PMIC_SDA"      	input
gpiochip13 - 4 lines:
	line   0:	"WIFI_SDIO_D0"  	input
	line   1:	"WIFI_SDIO_D1"  	input
	line   2:	"WIFI_SDIO_D2"  	input
	line   3:	"WIFI_SDIO_D3"  	input
```

mcuee, 2025-05-06:
Latest git works fine.

```
mcuee@raspberrypi:~/build/avr/avrdude $ cat ~/.avrduderc 
# #------------------------------------------------------------
# # linuxgpio
# #------------------------------------------------------------
#
programmer
    id                   = "linuxgpio";
    desc                 = "Use the Linux libgpiod interface to bitbang GPIO lines";
    type                 = "linuxgpio";
    prog_modes           = PM_ISP;
    connection_type      = linuxgpio;
    reset                = 25;
    sck                  = 11;
    sdo                  = 10;
    sdi                  = 9;
;

mcuee@raspberrypi:~/build/avr/avrdude $ avrdude -c linuxgpio -p m328p -E reset -v
Avrdude version 8.0-20250504 (58408592)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file is /home/mcuee/.avrduderc

using libgpiod for linuxgpio
Using port            : gpiochip0
Using programmer      : linuxgpio
AVR part              : ATmega328P
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : linuxgpio
Description           : Use the Linux libgpiod interface to bitbang GPIO lines
Pin assignment        : libgpiod
  RESET  = 25
  SCK    = 11
  SDO    = 10
  SDI    = 9

AVR device initialized and ready to accept instructions
Device signature = 1E 95 0F (ATmega328P, ATA6614Q, LGT8F328P)

Avrdude done.  Thank you.

mcuee@raspberrypi:~/build/avr/avrdude $ gpioget --version
gpioget (libgpiod) v2.3-devel
Copyright (C) 2017-2023 Bartosz Golaszewski
License: GPL-2.0-or-later
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

mcuee@raspberrypi:~/build/avr/avrdude $ gpioget -c gpiochip0 GPIO25
"GPIO25"=active

mcuee@raspberrypi:~/build/avr/avrdude $ gpioget GPIO25
"GPIO25"=active

mcuee@raspberrypi:~/build/avr/avrdude $ avrdude -c linuxgpio -p m328p -E noreset -v
Avrdude version 8.0-20250504 (58408592)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file is /home/mcuee/.avrduderc

using libgpiod for linuxgpio
Using port            : gpiochip0
Using programmer      : linuxgpio
AVR part              : ATmega328P
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : linuxgpio
Description           : Use the Linux libgpiod interface to bitbang GPIO lines
Pin assignment        : libgpiod
  RESET  = 25
  SCK    = 11
  SDO    = 10
  SDI    = 9

AVR device initialized and ready to accept instructions
Device signature = 1E 95 0F (ATmega328P, ATA6614Q, LGT8F328P)

Avrdude done.  Thank you.

mcuee@raspberrypi:~/build/avr/avrdude $ gpioget -c gpiochip0 GPIO25
"GPIO25"=inactive

mcuee@raspberrypi:~/build/avr/avrdude $ gpioget GPIO25
"GPIO25"=active
```

mcuee, 2025-05-06:
Raspberry Pi Linux may come with old version of libgpiod which does not seem to work too well with Raspberry Pi 5/500. My recommendation will be to use newer version of libgpiod.
https://github.com/brgl/libgpiod.git
https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/