avrdudes/avrdude #1751: linuxgpio sysfs fallback improvement to deal with shifted GPIO numbering in new kernel version

mcuee, 2024-04-15:
From here.
* https://github.com/avrdudes/avrdude/pull/1725#issuecomment-2054201555

From @stefanrueger 

>The AVRDUDE project likes to be backward compatible, and as such is unlikely to remove support for libgpiod v1 or sysfs for that matter. I understand your point of view.
>
> I think(!) I have read up sufficiently on sysfs and gpiod to hazard a guess that it ought to be possible to try a workaround once AVRDUDE realises that the current config doesn't work with the current kernel. Knowing that some kernels shifted the GPIO numbering from one version to the next, AVRDUDE could try to undo that shift by carefully looking at the config GPIO numbers and the GPIO numbers offered by the kernel.

mcuee, 2024-04-15:
* https://github.com/avrdudes/avrdude/pull/1725#issuecomment-2053639316

Reference for the pin number change.
* https://github.com/raspberrypi/linux/issues/5668

stefanrueger, 2024-08-01:
It's been a long time, and judging by the the number of complaints around this ebbing away, I am guessing users have gotten accustomed to using libgpiod ... I don't think spending effort on this is worth our while. So closing....

closed 2024-08-01 by stefanrueger