avrdudes/avrdude #1793: macOS github action failed.

mcuee, 2024-04-27:
Reference github action build for git main.
https://github.com/avrdudes/avrdude/actions/runs/8844905570/job/24287714207

```
Run cmake --build build
[  1%] Generating avrdude.conf
[  1%] Built target conf
[  2%] [FLEX][Parser] Building scanner with flex 2.6.4
[  3%] [BISON][Parser] Building parser with bison 2.3
[  5%] Building C object src/CMakeFiles/libavrdude.dir/arduino.c.o
[  6%] Building C object src/CMakeFiles/libavrdude.dir/avr.c.o
[  7%] Building C object src/CMakeFiles/libavrdude.dir/avr910.c.o
[  9%] Building C object src/CMakeFiles/libavrdude.dir/avrcache.c.o
[ 10%] Building C object src/CMakeFiles/libavrdude.dir/avrftdi.c.o
In file included from /Users/runner/work/avrdude/avrdude/src/avrftdi.c:42:
/Users/runner/work/avrdude/avrdude/src/avrftdi_private.h:10:12: fatal error: 'libusb.h' file not found
#  include <libusb.h>
           ^~~~~~~~~~
1 error generated.
make[2]: *** [src/CMakeFiles/libavrdude.dir/avrftdi.c.o] Error 1
make[1]: *** [src/CMakeFiles/libavrdude.dir/all] Error 2
make: *** [all] Error 2
Error: Process completed with exit code 2.
```

mcuee, 2024-04-27:
The issue seems to be in the configure stage that `libusb-1.0\libusb.h` is not found.

```
Run cmake -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/local/Cellar -D DEBUG_CMAKE=1 -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build
-- The C compiler identification is AppleClang 15.0.0.15000040
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode_15.0.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /opt/homebrew/bin/git (found version "2.44.0")
-- Found FLEX: /usr/bin/flex (found version "2.[6](https://github.com/avrdudes/avrdude/actions/runs/8844905570/job/24287714207#step:4:7).4")
-- Found BISON: /usr/bin/bison (found version "2.3")
-- Looking for libelf.h
-- Looking for libelf.h - not found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - not found
-- Looking for usb.h
-- Looking for usb.h - not found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - not found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - not found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - not found
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Darwin-23.4.0
-- CMAKE_SYSTEM: Darwin-23.4.0
-- CMAKE_FIND_ROOT_PATH: 
-- CMAKE_C_COMPILER: /Applications/Xcode_15.0.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- CONFIG_DIR: /usr/local/etc
-- AVRDUDE_FULL_VERSION: [7](https://github.com/avrdudes/avrdude/actions/runs/8844905570/job/24287714207#step:4:8).3-20240426 (db2580d)
-- USE_EXTERNAL_LIBS: OFF
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: /opt/homebrew/lib/libelf.a
-- HAVE_LIBUSB: HAVE_LIBUSB-NOTFOUND
-- HAVE_LIBUSB_1_0: /opt/homebrew/lib/libusb-1.0.dylib
-- HAVE_LIBUSB_WIN32: HAVE_LIBUSB_WIN32-NOTFOUND
-- HAVE_LIBHIDAPI: /opt/homebrew/lib/libhidapi.dylib
-- HAVE_LIBFTDI: HAVE_LIBFTDI-NOTFOUND
-- HAVE_LIBFTDI1: /opt/homebrew/lib/libftdi1.dylib
-- HAVE_LIBREADLINE: /Applications/Xcode_15.0.1.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX[14](https://github.com/avrdudes/avrdude/actions/runs/8844905570/job/24287714207#step:4:15).0.sdk/usr/lib/libreadline.tbd
-- HAVE_LIBSERIALPORT: 1
-- HAVE_LIBELF_H: 
-- HAVE_LIBELF_LIBELF_H: 
-- HAVE_USB_H: 
-- HAVE_LUSB0_USB_H: 
-- HAVE_LIBUSB_H: 
-- HAVE_LIBUSB_1_0_LIBUSB_H: 
-- HAVE_HIDAPI_HIDAPI_H: 
-- LIBUSB_COMPAT_DIR: 
-- LIBGPIODV2_FOUND: 
-- LIBGPIOD_FOUND: 
-- ----------------------
-- DO HAVE    libelf
-- DON'T HAVE libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DO HAVE    libserialport
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- ----------------------
-- Configuring done (4.0s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/runner/work/avrdude/avrdude/build
```

mcuee, 2024-04-27:
I have no issues to build git main under my macOS machine (Mac Mini M1 2020, latest macOS 14.4.1 and up-to-date Homebrew).

@MCUdude and @dl8dtl 

Please help to check as well on your macOS machine if you have some time. Thanks.

mcuee, 2024-04-27:
@Youw 

Sorry to bother you again, just wondering if you have some ideas why `libusb.h` was not found properly. Thanks.

mcuee, 2024-04-27:
I guess the following header file needs some fix to be more robust. For libusb project, we usually recommend using `pkg-config` to find `libusb.h` and not to use `libusb-1.0/libusb.h` in any case.
https://github.com/avrdudes/avrdude/blob/main/src/avrftdi_private.h

```
#if defined(HAVE_LIBFTDI1) && defined(HAVE_LIBUSB_1_0)
# if defined(HAVE_LIBUSB_1_0_LIBUSB_H)
#  include <libusb-1.0/libusb.h>
# else
#  include <libusb.h>
# endif
# include <libftdi1/ftdi.h>
# undef HAVE_LIBFTDI_TYPE_232H
# define HAVE_LIBFTDI_TYPE_232H 1
#elif defined(HAVE_LIBFTDI)
#include <ftdi.h>
#else
```

mcuee, 2024-04-28:
Since autotools build seems to work from github action build of PR #1796, looks like the issue is with the changes in CMake related files. Still kind of strange though.

* #1796

mcuee, 2024-04-28:
Finally figured out the issue from the github action build log.

github has switched to use `/opt/homebrew` for homebrew. They no longer use `/usr/local`.
https://github.com/avrdudes/avrdude/actions/runs/8866899974/job/24344872706

mcuee, 2024-04-28:
This issue is now fixed by commit https://github.com/avrdudes/avrdude/commit/dceb55da6a386510721f3361bad0343d10dd1a32.

closed 2024-04-28 by mcuee

Youw, 2024-04-28:
>  just wondering if you have some ideas why libusb.h was not found properly

Yeap, this is the reason:

> For libusb project, we usually recommend using pkg-config to find libusb.h and not to use libusb-1.0/libusb.h in any case.
https://github.com/avrdudes/avrdude/blob/main/src/avrftdi_private.h

---

> still happening even if I add pkg-config as a macOS dependancy

It is not enough to install pkg-config. One must explicitly use it to find a package.

> This issue is now fixed by commit https://github.com/avrdudes/avrdude/commit/dceb55da6a386510721f3361bad0343d10dd1a32.

I'd call it a workaround.

The propper way would be to remove things like `find_library(HAVE_LIBUSB_1_0 ...)` and similar nonsense with a propper use of cmake/pkg-config, e.g. in a similar way like HIDAPI [does it](https://github.com/libusb/hidapi/blob/bda710178dfe9650bdf76aa02e3671e17e3a8a27/libusb/CMakeLists.txt#L14-L16).

mcuee, 2024-04-28:
@Youw 

Good point. I will create a new issue for CMake improvement.