avrdudes/avrdude #1797: CMake improvement -- using pkg-find to find the libraries

mcuee, 2024-04-28:
From the comment by @Youw here.
* https://github.com/avrdudes/avrdude/issues/1793#issuecomment-2081429919

++++++++++++++++++
>  just wondering if you have some ideas why libusb.h was not found properly

Yeap, this is the reason:

> For libusb project, we usually recommend using pkg-config to find libusb.h and not to use libusb-1.0/libusb.h in any case.
https://github.com/avrdudes/avrdude/blob/main/src/avrftdi_private.h
> ---
> still happening even if I add pkg-config as a macOS dependancy

It is not enough to install pkg-config. One must explicitly use it to find a package.

The propper way would be to remove things like `find_library(HAVE_LIBUSB_1_0 ...)` and similar nonsense with a propper use of cmake/pkg-config, e.g. in a similar way like HIDAPI [does it](https://github.com/libusb/hidapi/blob/bda710178dfe9650bdf76aa02e3671e17e3a8a27/libusb/CMakeLists.txt#L14-L16).

mcuee, 2024-04-28:
@ndim 

In case you continue the work on the CMake improvement along with autotools improvement, maybe you want to take a look at the above recommendations. Thanks.

ndim, 2024-07-31:
Now that the gpiod support requires pkg-config logic anyway, we can jump on board completely, both on the cmake and on the automake side.

Both for finding our own dependencies and for installing a `libavrdude-2.pc` alongside libavrdude.

mcuee, 2025-06-07:
This may also help for the MSVC build with vcpkg.

For example, we have the following two long standing issues and the current recommended work-around is to use MSYS2 mingw build.
* https://github.com/avrdudes/avrdude/issues/1253
* https://github.com/avrdudes/avrdude/issues/968

It may also help the following issue.
* https://github.com/avrdudes/avrdude/issues/1552

mcuee, 2025-06-07:
In the end we may be able to deprecate the following repo.
https://github.com/avrdudes/libhidapi

The following repo is not used anyway -- we can probably archive it.
https://github.com/avrdudes/libreadline

The following is still needed.
https://github.com/avrdudes/libelf

We may still need the following (to add CMake support).
https://github.com/avrdudes/libserialport

The following two can be optional -- we should probably let the user choose whether to use them or the generic library.
https://github.com/avrdudes/libftdi --> this is the root cause of Issue 1253
https://github.com/avrdudes/libusb --> this is the root cause of Issue #968.

mcuee, 2025-06-20:
Example:

To find libusb-1.0.

**avrdude**
https://github.com/avrdudes/avrdude/blob/main/CMakeLists.txt
```
find_library(HAVE_LIBUSB_1_0 NAMES ${PREFERRED_LIBUSB_1_0})
if(HAVE_LIBUSB_1_0)
    set(LIB_LIBUSB_1_0 ${HAVE_LIBUSB_1_0})
endif()
```

**hidapi**
https://github.com/libusb/hidapi/blob/master/libusb/CMakeLists.txt
```
if(TARGET usb-1.0)
    target_link_libraries(hidapi_libusb PRIVATE usb-1.0)
else()
    include(FindPkgConfig)
    pkg_check_modules(libusb REQUIRED IMPORTED_TARGET libusb-1.0>=1.0.9)
    target_link_libraries(hidapi_libusb PRIVATE PkgConfig::libusb)
endif()
```

mcuee, 2025-06-20:
libftdi
http://developer.intra2net.com/git/?p=libftdi;a=blob;f=cmake/FindLibUSB.cmake;h=8bb3b1176470e0ba78b0264896875e3eab49c32b;hb=HEAD

```
# - Try to find the libusb library
# Once done this defines
#
#  LIBUSB_FOUND - system has libusb
#  LIBUSB_INCLUDE_DIR - the libusb include directory
#  LIBUSB_LIBRARIES - Link these to use libusb

# Copyright (c) 2006, 2008  Laurent Montel, <montel@kde.org>
#
# Redistribution and use is allowed according to the terms of the BSD license.
# For details see the accompanying COPYING-CMAKE-SCRIPTS file.

if (LIBUSB_INCLUDE_DIR AND LIBUSB_LIBRARIES)

  # in cache already
  set(LIBUSB_FOUND TRUE)

else (LIBUSB_INCLUDE_DIR AND LIBUSB_LIBRARIES)
  # use pkg-config to get the directories and then use these values
  # in the FIND_PATH() and FIND_LIBRARY() calls
  find_package(PkgConfig)
  pkg_check_modules(PC_LIBUSB libusb-1.0)

  find_path(LIBUSB_INCLUDE_DIR libusb.h
    PATH_SUFFIXES libusb-1.0
    PATHS ${PC_LIBUSB_INCLUDEDIR} ${PC_LIBUSB_INCLUDE_DIRS})

  find_library(LIBUSB_LIBRARIES NAMES ${PC_LIBUSB_LIBRARIES}
    PATHS ${PC_LIBUSB_LIBDIR} ${PC_LIBUSB_LIBRARY_DIRS})

  include(FindPackageHandleStandardArgs)
  find_package_handle_standard_args(
    LibUSB DEFAULT_MSG LIBUSB_LIBRARIES LIBUSB_INCLUDE_DIR
  )

  mark_as_advanced(LIBUSB_INCLUDE_DIR LIBUSB_LIBRARIES)

endif (LIBUSB_INCLUDE_DIR AND LIBUSB_LIBRARIES)
```