avrdudes/avrdude #1799: format mismatch warnings

dl8dtl, 2024-04-28:
```
/home/joerg/src/avrdude/src/dfu.c:398:5: warning: format specifies type 'unsigned short' but the argument has type 'int' [-Wformat]
    ((unsigned short) dfu->dev_desc.bcdDevice >> 8) & 0xFF,
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/joerg/src/avrdude/src/avrdude.h:59:97: note: expanded from macro 'msg_info'
#define msg_info(...)       avrdude_message2(stderr, __LINE__, __FILE__, __func__, 0, MSG_INFO, __VA_ARGS__)
                                                                                                ^~~~~~~~~~~
/home/joerg/src/avrdude/src/dfu.c:399:5: warning: format specifies type 'unsigned short' but the argument has type 'int' [-Wformat]
    ((unsigned short) dfu->dev_desc.bcdDevice >> 4) & 0xF,
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/joerg/src/avrdude/src/avrdude.h:59:97: note: expanded from macro 'msg_info'
#define msg_info(...)       avrdude_message2(stderr, __LINE__, __FILE__, __func__, 0, MSG_INFO, __VA_ARGS__)
                                                                                                ^~~~~~~~~~~
/home/joerg/src/avrdude/src/dfu.c:400:5: warning: format specifies type 'unsigned short' but the argument has type 'int' [-Wformat]
    ((unsigned short) dfu->dev_desc.bcdDevice >> 0) & 0xF);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/joerg/src/avrdude/src/avrdude.h:59:97: note: expanded from macro 'msg_info'
#define msg_info(...)       avrdude_message2(stderr, __LINE__, __FILE__, __func__, 0, MSG_INFO, __VA_ARGS__)
…
/home/joerg/src/avrdude/src/flip2.c:559:5: warning: format specifies type 'unsigned short' but the argument has type 'int' [-Wformat]
    ((unsigned short) flip2->boot_ver >> 4) & 0xF,
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/joerg/src/avrdude/src/avrdude.h:59:97: note: expanded from macro 'msg_info'
#define msg_info(...)       avrdude_message2(stderr, __LINE__, __FILE__, __func__, 0, MSG_INFO, __VA_ARGS__)
                                                                                                ^~~~~~~~~~~
/home/joerg/src/avrdude/src/flip2.c:560:5: warning: format specifies type 'unsigned short' but the argument has type 'int' [-Wformat]
    ((unsigned short) flip2->boot_ver >> 0) & 0xF);
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/joerg/src/avrdude/src/avrdude.h:59:97: note: expanded from macro 'msg_info'
#define msg_info(...)       avrdude_message2(stderr, __LINE__, __FILE__, __func__, 0, MSG_INFO, __VA_ARGS__)
                                                                                                ^~~~~~~~~~~
/home/joerg/src/avrdude/src/flip2.c:781:70: warning: format specifies type 'unsigned short' but the argument has type 'int' [-Wformat]
      pmsg_error("address out of range [0x%04hX,0x%04hX]\n", offset, offset+size-1);
                                                  ~~~~~              ^~~~~~~~~~~~~
                                                  %04X
/home/joerg/src/avrdude/src/avrdude.h:67:176: note: expanded from macro 'pmsg_error'
#define pmsg_error(...)     avrdude_message2(stderr, __LINE__, __FILE__, __func__, MSG2_PROGNAME|MSG2_FUNCTION|MSG2_FILELINE|MSG2_TYPE|MSG2_FLUSH|MSG2_LEFT_MARGIN, MSG_ERROR, __VA_ARGS__)
                                                                                                                                                                               ^~~~~~~~~~~
/home/joerg/src/avrdude/src/flip2.c:839:70: warning: format specifies type 'unsigned short' but the argument has type 'int' [-Wformat]
      pmsg_error("address out of range [0x%04hX,0x%04hX]\n", offset, offset+size-1);
                                                  ~~~~~              ^~~~~~~~~~~~~
                                                  %04X
/home/joerg/src/avrdude/src/avrdude.h:67:176: note: expanded from macro 'pmsg_error'
#define pmsg_error(...)     avrdude_message2(stderr, __LINE__, __FILE__, __func__, MSG2_PROGNAME|MSG2_FUNCTION|MSG2_FILELINE|MSG2_TYPE|MSG2_FLUSH|MSG2_LEFT_MARGIN, MSG_ERROR, __VA_ARGS__)
                                                                                                                                                                               ^~~~~~~~~~~
```
There would at least be an `(unsigned short)` typecast missing.

Anyway, I wonder what's the point of forcing all that into unsigned short, rather than keeping it simply `unsigned`, and remove all the `h` format modificators?

stefanrueger, 2024-04-28:
The format is wrong, it should not say `h`, as C cannot pass a short to a function without first promoting to int (likewise unsigned). However, the printf implementation should also know that C passes an int, so should silently ignore the format (which it probably does).

dl8dtl, 2024-04-28:
I am no longer so sure about that statement (and even reading the C standard doesn't make it better :).

I'd say, let's drop the `h` modifiers, and change all `(unsigned short)` typecasts when passing arguments (in the lines before those complained about above) into `(unsigned)` typecasts.

Do you want me to prepare a PR?

stefanrueger, 2024-04-28:
> The format is wrong, it should not say `h`

The utility of %h[dxu] is to convert the int/unsigned argument (automatically promoted!) to short or unsigned short before printing.

PR #1800 (2024-05-17): Silence compiler warnings
Fixes #1799 

The `%hd` converts the int argument to short before printing, so I suspect #1799 is a matter of overzealous compiler warning.
closed 2024-05-17 by stefanrueger