avrdudes/avrdude #1807: custom programmer broken in v7.3

jmgurney, 2024-05-16:
If I checkout v7.2 and build it on MacOSX, I can have the following in `~/.avrduderc`:
```
programmer
  id    = "ft232r-cts";
  desc  = "FT232R Synchronous BitBang, CTS for RESET";
  type  = "ftdi_syncbb";
  connection_type = usb;
  sck   = 0;  # TxD
  #sdi   = 1;  # RxD
  #sdo   = 2;  # RTS
  miso  = 1;  # RxD
  mosi  = 2;  # RTS
  reset = 3;  # CTS
  #reset = 4;  # DTR
;

#      PIN NO.     DATA LINE
#       0           TXD
#       1           RXD
#       2           RTS
#       3           CTS
#       4           DTR
#       5           DSR
#       6           DCD
#       7           RI
```
and 7.2 sees it fine:
```
$ git checkout v7.2
[...]
$ sh build.sh -j 4 && ./build_darwin/src/avrdude -v -p attiny85 -c ft232r-cts
[...]
avrdude: Version 7.2-20230719 (cb9e7e49)
[...]
         System wide configuration file is /Users/jmg/github/avrdude/build_darwin/src/avrdude.conf
         User configuration file is /Users/jmg/.avrduderc
avrdude yywarning() warning: miso is deprecated, will be removed in v8.0, use sdi [/Users/jmg/.avrduderc:9]
avrdude yywarning() warning: mosi is deprecated, will be removed in v8.0, use sdo [/Users/jmg/.avrduderc:10]

         Using Port                    : usb
         Using Programmer              : ft232r-cts
[...]
avrdude ft245r_program_enable() error: device is not responding to program enable; check connection
avrdude main() error: initialization failed, rc=-1
        - double check the connections and try again
```

But if I do the same for 7.3, the programmer is not found:
```
$ git checkout v7.3
Previous HEAD position was cb9e7e49 Prepare for version 7.2
HEAD is now at e599214c Update version info for 7.3
$ sh build.sh && ./build_darwin/src/avrdude -v -p attiny85 -c ft232r-cts
[...]
avrdude: Version 7.3
[...]
         System wide configuration file is /Users/jmg/github/avrdude/build_darwin/src/avrdude.conf
         User configuration file is /Users/jmg/.avrduderc
avrdude yywarning() warning: miso is deprecated, will be removed in v8.0, use sdi [/Users/jmg/.avrduderc:9]
avrdude yywarning() warning: mosi is deprecated, will be removed in v8.0, use sdo [/Users/jmg/.avrduderc:10]

avrdude programmer_not_found() error: cannot find programmer id ft232r-cts
use -c? to see all possible programmers
```

and the programmer ft232r-cts is not seen in `-c?` for 7.3, but it is for 7.2.

After a bit of digging:
```
avrdude: Version 7.2-20230827 (b8c47f39)
```
"works" in that the programmer is seen, but it fails with:
```
avrdude main() error: programmer ft232r-cts cannot program part ATtiny85 as they
        lack a common programming mode; use -F to override this check
```

It identifies commit 91ffab5fc6f22 as the bad one, but 91ffab5fc6f22 fails to compile:
```
CMake Error at src/CMakeLists.txt:140 (add_library):
  Cannot find source file:

    serialadapter.c
```

so the breakage was introduced in commits 91ffab5fc6f2 and aaddbbf18731.

stefanrueger, 2024-05-17:
Thanks for reporting, @jmgurney. It will be the missing `prog_modes` feature that's needed in newer AVRDUDE versions. Try
```
#------------------------------------------------------------
# ft232r-cts
#------------------------------------------------------------

programmer # ft232r-cts
    id                     = "ft232r-cts";
    desc                   = "FT232R Synchronous BitBang, CTS for RESET";
    type                   = "ftdi_syncbb";
    prog_modes             = PM_TPI | PM_ISP;
    connection_type        = usb;
    reset                  = 3;  # CTS
    sck                    = 0;  # TxD
    sdo                    = 2;  # RTS
    sdi                    = 1;  # RxD
;
```

jmgurney, 2024-05-17:
ahh, this also likely explains the other error that I was seeing when bisecting the error.  I'd have expected an error message since this is a change in behavior in a point release, either that prog_modes is missing, or that programmer ft232r-cts not used because of it missing.

stefanrueger, 2024-05-18:
That's a fair point. With PR #1810 it should look like
```
$ avrdude -qq -cdryrun -pm328p
avrdude warning: programmer ft232r-cts fails to specify prog_modes = PM_...; [/home/srueger/.avrduderc:57]
```

closed 2024-05-20 by stefanrueger

jmgurney, 2024-05-21:
Thanks, works great.

PR #1810 (2024-05-20): Warn when programmer failes to define `prog_modes`
Fixes #1807