avrdudes/avrdude #1813: Misleading error message when specifying wrong AVR-Ex part

stefanrueger, 2024-05-27:
I have a perfectly set up programmer (a shiny new [SerialUPDI](https://www.tindie.com/products/mcudude/serialupdi-programmer/) since you asked) that's connected to an AVR-Ex part.

If I mistype which one it is, I get the following misleading error message:
```
$ avrdude -qc serialupdi -p 16eb28 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0xffffff (probably .avr8x) (retrying)
avrdude: device signature = 0xffffff (probably .avr8x) (retrying)
avrdude: device signature = 0xffffff (probably .avr8x)
avrdude error: Yikes!  Invalid device signature.
avrdude error: expected signature for AVR16EB28 is 1E 94 3F
        Double check connections and try again, or use -F to override
        this check.

avrdude done.  Thank you.
```

Actually, it's a different part, and as soon as I type the correct line it works:
```
$ avrdude -qqc serialupdi -p 64ea28 -t
avrdude>
```

Notwithstanding that the error was between the chair and the keyboard, why could AVRDUDE not read the signature and could AVRDUDE have done any better in its error message? @dbuchwald might have some insights.

dbuchwald, 2024-05-28:
Hello, as usual, I would ask for verbose output ('-vvvv') for each of these commands, I need to see the exact communication happening. I must say this looks very, very weird and I don't think I could replicate it easily...

stefanrueger, 2024-05-28:
OK, here the output of `$ avrdude -vvvq -p 64ea28 -c serialupdi ` when the target is an AVR16EB28 
<details>
avrdude: Version 7.3-20240520 (a3e4c808)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /usr/local/etc/avrdude.conf
         User configuration file is /home/srueger/.avrduderc

         Using port            : /dev/ttyUSB0
         Using programmer      : serialupdi
avrdude: opening serial port ...
avrdude: sending 1 bytes [0x00]
         AVR Part              : AVR64EA28
         Programming modes     : SPM, UPDI

         Memory            Size  Pg size    Offset
         -----------------------------------------
         eeprom             512        8    0x1400 
         flash            65536      128  0x800000 
         fuses               16        1    0x1050 
         fuse0/wdtcfg         1        1    0x1050 
         fuse1/bodcfg         1        1    0x1051 
         fuse2/osccfg         1        1    0x1052 
         fuse5/syscfg0        1        1    0x1055 
         fuse6/syscfg1        1        1    0x1056 
         fuse7/codesize       1        1    0x1057 
         fuse8/bootsize       1        1    0x1058 
         lock                 4        1    0x1040 
         prodsig/sigrow     128      128    0x1100 
         signature            3        1    0x1100 
         tempsense            4        1    0x1104 
         sernum              16        1    0x1110 
         userrow/usersig     64       64    0x1080 
         io                4160        1         0 
         sram              6144        1    0x6800 
         sib                 32        1         0 

         Variants        Package  F max   T range         V range       
         ---------------------------------------------------------------
         AVR64EA28       SOIC28   20 MHz  [-40 C, 125 C]  [1.8 V, 5.5 V]
         AVR64EA28-I/SP  SPDIP28  20 MHz  [-40 C,  85 C]  [1.8 V, 5.5 V]
         AVR64EA28-I/SS  SSOP28   20 MHz  [-40 C,  85 C]  [1.8 V, 5.5 V]

         Programmer Type       : serialupdi
         Description           : SerialUPDI
avrdude: STCS 0x08 to address 0x03
avrdude: sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: STCS 0x80 to address 0x02
avrdude: sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: LDCS from 0x00
avrdude: sending 2 bytes [0x55, 0x80]
avrdude: received 1 bytes [0x30]
avrdude: UDPI init OK
avrdude: UPDI link initialization OK
avrdude: LDCS from 0x0B
avrdude: sending 2 bytes [0x55, 0x8b]
avrdude: received 1 bytes [0x82]
avrdude: sending 2 bytes [0x55, 0xe6]
avrdude: received 32 bytes [0x41, 0x56, 0x52, 0x20, 0x20, 0x20, 0x20, 0x20, 0x50, 0x3a, 0x35, 0x44, 0x3a, 0x31, 0x2d, 0x33, 0x4d, 0x32, 0x20, 0x28, 0x41, 0x30, 0x2e, 0x35, 0x39, 0x42, 0x31, 0x44, 0x2e, 0x30, 0x29, 0x00]
avrdude: received SIB: [AVR     P:5D:1-3M2 (A0.59B1D.0)]
avrdude: Device family ID: AVR     
avrdude: NVM interface: P:5
avrdude: Debug interface: D:1
avrdude: PDI oscillator: 3M2 
avrdude: Extra information: (A0.59B1D.0)
avrdude: NVM type 5: 24-bit, page oriented
avrdude: STCS 0x08 to address 0x03
avrdude: sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: STCS 0x80 to address 0x02
[... 337 lines cut ...]
avrdude main() [main.c:1655] error: Yikes!  Invalid device signature.
avrdude main() [main.c:1657] error: expected signature for AVR64EA28 is 1E 96 20
        Double check connections and try again, or use -F to override
        this check.

avrdude: leaving NVM programming mode
avrdude: sending reset request
avrdude: STCS 0x59 to address 0x08
avrdude: sending 3 bytes [0x55, 0xc8, 0x59]
avrdude: ser_send: U [55] . [c8] Y [59]
avrdude: ser_recv: U [55] . [c8] Y [59]
avrdude: sending release reset request
avrdude: STCS 0x00 to address 0x08
avrdude: sending 3 bytes [0x55, 0xc8, 0x00]
avrdude: ser_send: U [55] . [c8] . [00]
avrdude: ser_recv: U [55] . [c8] . [00]
avrdude: STCS 0x0C to address 0x03
avrdude: sending 3 bytes [0x55, 0xc3, 0x0c]
avrdude: ser_send: U [55] . [c3] . [0c]
avrdude: ser_recv: U [55] . [c3] . [0c]

avrdude done.  Thank you.
</details>

Thanks for looking into this

stefanrueger, 2024-05-28:
And, just for comparison, the output when the command line is corrected to `-c 16eb28` and therefore matches the target:
`avrdude -vvvq -p 16eb28 -c serialupdi`
<details>
avrdude: Version 7.3-20240520 (a3e4c808)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /usr/local/etc/avrdude.conf
         User configuration file is /home/srueger/.avrduderc

         Using port            : /dev/ttyUSB0
         Using programmer      : serialupdi
avrdude: opening serial port ...
avrdude: sending 1 bytes [0x00]
         AVR Part              : AVR16EB28
         Programming modes     : SPM, UPDI

         Memory            Size  Pg size    Offset
         -----------------------------------------
         eeprom             512        8    0x1400 
         flash            16384       64  0x800000 
         fuses               16        1    0x1050 
         fuse0/wdtcfg         1        1    0x1050 
         fuse1/bodcfg         1        1    0x1051 
         fuse2/osccfg         1        1    0x1052 
         fuse5/syscfg0        1        1    0x1055 
         fuse6/syscfg1        1        1    0x1056 
         fuse7/codesize       1        1    0x1057 
         fuse8/bootsize       1        1    0x1058 
         fusea/pdicfg         2        1    0x105a 
         lock                 4        1    0x1040 
         prodsig/sigrow     128      128    0x1080 
         signature            3        1    0x1080 
         tempsense            4        1    0x1084 
         sernum              16        1    0x1090 
         bootrow             64       64    0x1100 
         userrow/usersig     64       64    0x1200 
         io                4160        1         0 
         sram              2048        1    0x7800 
         sib                 32        1         0 

         Variants              Package  F max   T range         V range       
         ---------------------------------------------------------------------
         AVR16EB28-SSOP/SPDIP  SOIC28   20 MHz  [-40 C, 125 C]  [1.8 V, 5.5 V]
         AVR16EB28-VQFN        VQFN28   20 MHz  [-40 C, 125 C]  [1.8 V, 5.5 V]

         Programmer Type       : serialupdi
         Description           : SerialUPDI
avrdude: STCS 0x08 to address 0x03
avrdude: sending 3 bytes [0x55, 0xc3, 0x08]
avrdude: STCS 0x80 to address 0x02
avrdude: sending 3 bytes [0x55, 0xc2, 0x80]
avrdude: LDCS from 0x00
avrdude: sending 2 bytes [0x55, 0x80]
avrdude: received 1 bytes [0x30]
avrdude: UDPI init OK
avrdude: UPDI link initialization OK
avrdude: LDCS from 0x0B
avrdude: sending 2 bytes [0x55, 0x8b]
avrdude: received 1 bytes [0x82]
avrdude: sending 2 bytes [0x55, 0xe6]
avrdude: received 32 bytes [0x41, 0x56, 0x52, 0x20, 0x20, 0x20, 0x20, 0x20, 0x50, 0x3a, 0x35, 0x44, 0x3a, 0x31, 0x2d, 0x33, 0x4d, 0x32, 0x20, 0x28, 0x41, 0x30, 0x2e, 0x35, 0x39, 0x42, 0x31, 0x44, 0x2e, 0x30, 0x29, 0x00]
avrdude: received SIB: [AVR     P:5D:1-3M2 (A0.59B1D.0)]
avrdude: Device family ID: AVR     
avrdude: NVM interface: P:5
avrdude: Debug interface: D:1
avrdude: PDI oscillator: 3M2 
avrdude: Extra information: (A0.59B1D.0)
avrdude: NVM type 5: 24-bit, page oriented
avrdude: STCS 0x08 to address 0x03
[... 184 lines cut ...]
avrdude: LDCS from 0x0B
avrdude: sending 2 bytes [0x55, 0x8b]
avrdude: received 1 bytes [0x08]
avrdude: LD from 0x001100
avrdude: sending 5 bytes [0x55, 0x08, 0x00, 0x11, 0x00]
avrdude: received 1 bytes [0x1e]
avrdude: LD from 0x001101
avrdude: sending 5 bytes [0x55, 0x08, 0x01, 0x11, 0x00]
avrdude: received 1 bytes [0x96]
avrdude: LD from 0x001102
avrdude: sending 5 bytes [0x55, 0x08, 0x02, 0x11, 0x00]
avrdude: received 1 bytes [0x20]
avrdude: device signature = 0x1e9620 (probably 64ea28)
avrdude: leaving NVM programming mode
avrdude: sending reset request
avrdude: STCS 0x59 to address 0x08
avrdude: sending 3 bytes [0x55, 0xc8, 0x59]
avrdude: sending release reset request
avrdude: STCS 0x00 to address 0x08
avrdude: sending 3 bytes [0x55, 0xc8, 0x00]
avrdude: STCS 0x0C to address 0x03
avrdude: sending 3 bytes [0x55, 0xc3, 0x0c]

avrdude done.  Thank you.
</details>

dbuchwald, 2024-05-28:
@stefanrueger sorry, but I don't have good news for you, this is something that can't really be fixed. 

See, the problem is that when initiating connection with the MCU using SerialUPDI the signature is read from arbitrary location in memory. For AVR16EBxx chips SIGROW memory is at 0x1080, while for AVR64EAxx chips it's 0x1100. This is why when you specify incorrect part but with identical memory map (and most of these guys use 0x1100: https://github.com/avrdudes/avrdude/blob/main/src/avrdude.conf.in#L20263) then AVRDUDE reads the correct signature and responds with reasonable error proposing correct part.

In your case it reads from a totally different memory (0x1100 in AVR16EBxx is BOOTROW), gets meaningless data and doesn't know how to recover from this, nor is able to suggest anything else. If there was any way to ask the chip for its memory map, we could do it instead, but unfortunately, no such option is available AFAIK.

Let me know if my explanation is not clear, I will try to explain it better.

MCUdude, 2024-05-28:
AFAIK, only the AVR-DU and AVR-EB has their signature at 0x1080, and these are the only chips with NVMv4 and NVMv5. So the SIB could be read to determine the NVM version, and then figure out the correct address if Avrdude is reading the signature as 0xffffff.

BTW this is also a problem for jtag3 based programmers as well:

```
$ ./avrdude -cpkobn_updi -pavr64dd32 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0xffffff (probably .avr8x) (retrying)
avrdude: device signature = 0xffffff (probably .avr8x) (retrying)
avrdude: device signature = 0xffffff (probably .avr8x)
avrdude error: Yikes!  Invalid device signature.
avrdude error: expected signature for AVR64DD32 is 1E 96 1A
        Double check connections and try again, or use -F to override
        this check.

avrdude done.  Thank you.

Hanss-MacBook-Pro:src hans$ ./avrdude -cpkobn_updi -pavr16eb32 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e943e (probably avr16eb32)

avrdude: processing -t interactive terminal
avrdude> read sib
```

MCUdude, 2024-05-28:
When taking a closer look at the jtag3 and serialupdi source code, it appears that these programmers reads the SIB before reading the device signature. So after a failed signature read attempt, we may use the NVM version number to find the correct offset for the signature/prodsig memory.

askn37, 2024-05-29:
This problem of poor part selection cannot be solved unless you read UPDI_SIB before reading PROD_SIGNATURE to determine the NVM version.

Currently, the only protocols (as far as I know) that actually support this are SerialUPDI and JTAGmk3. Microchip has added modern UPDI support to these, but it is not documented. On the other hand, older protocols such as JTAGmk2 and STK600 do not define a mechanism to return UPDI_SIB to AVRDUDE and are not maintained by Microchip at all.

Older programmers will need their own extensions to add modern support. For example, JTAG2UPDI and UPDI4AVR firmware read UPDI_SIB before reading PROD_SIG to assist with subsequent operations. This is because there is no publicly known mechanism for AVRDUDE to inform firmware of the NVM version.

For UPDI4AVR only, with a small change to avrdude.conf, you can signal the NVM version and control information via unused areas in the device descriptor to support correct high-voltage programming. This is really something special.

dbuchwald, 2024-05-29:
My take on the idea to base the error message on the NVM controller version is that **we shouldn't do it**. Given all the inconsistencies between latest AVR chips and their respective NVM controllers I wouldn't be surprised if that wasn't the last case we saw. All of these "ifs/elses" will end up being unmanageable spaghetti that nobody understands and everyone is afraid to remove even after NVM v3/v5 chips are no longer produced nor sold. I see that kind of tech debt in my work daily and I see how it gradually slows down value delivery to a grinding halt.

That being said, if you want me to implement this feature I will do so, it's just that my instinct tells me it's not a good idea.

stefanrueger, 2024-05-29:
Thanks all round @dbuchwald @MCUdude @askn37 for input. I feel we *definitely* have to do something, if only to change the error message to add "check whether the part is correctly specified". Right now, one of the recommendations is `-F` which can lead to tears as programming would proceed under wrong assumptions.

It is amazing that Microchip finds new and innovative ways beyond [the known reuse](https://github.com/avrdudes/avrdude/issues/1478#issuecomment-1656791055) of signatures  to make it hard to use them for part identification. I also feel that AVRDUDE could give better advice than telling the user to specify the correct part number (the gadget housing the MCU may be difficult to open, the printing on the MCU illegible, the user may simply misremember but be convinced it *is* the wrong part).

When given the wrong `-p` UPDI part, the UPDI  programmers read (correctly, without problems and without encountering an error) three bytes, but possibly from the wrong memory (0x1100 vs 0x1080). Fun fact: as the other memory is at least in one case writable (bootrow) one could therefore spoof this MCU to be a different one. Normally, however, one wouldn't expect a valid UPDI signature in the wrong. So, some instance outside the individual `-c` programmer code (say, the `avr_signature()` function in `avr.c` might well check whether what the `-c` programmer code read without error is a plausible signature, eg, check against known full signatures, check first byte against all known first bytes (currently only `1e` and `ed`). If not, see what would happen if the specified UPDI part was a different one with *another* signature memory address.

Another fun fact: There must have been a similar problem in the past b/c there is code in `main.c` that reads `sib` and makes a comparison with the `AVR_FAMILY`:
https://github.com/avrdudes/avrdude/blob/a3e4c8083e38b257915881999c16e2e4438ec6d1/src/main.c#L1575-L1603

Anyone know what this was about?

MCUdude, 2024-05-29:
> Anyone know what this was about?

The family ID is pretty much useless. It started out with `tinyAVR` and `megaAVR`, but when the AVR-DA were introduced they used "    AVR" (that's four leading spaces). Later they "fixed" this error, and the later generations are now "AVR    " (that's four trailing spaces). Why they couldn't use the actual family name is unclear, but it's way too late to change now.

I suggest mentioning that the user should double check the `-p` part name and try again if Avrdude is able to read a signature, but it's incorrect, if trying to communicate with a UPDI part.

PR #1840 (2024-07-15): Improve signature check for UPDI parts
Fixes #1813.

Microchip decided to put the `signature` area for UPDI at two different memory offsets, so you now need to know which part you have in order to be able to read the `signature`. This PR checks all different known offsets if an attempt to read a UPDI-part signature draws a blank. It's not fool-proof as, conceivably, if there *was* a valid signature at the wrong location that byte sequence would mistakenly be understood as the signature of the part.

Without this PR:
```
$ avrdude -cserialupdi -p64dd28 -t
avrdude: AVR device initialized and rea [... cut]
closed 2024-07-15 by stefanrueger