avrdudes/avrdude #1823: UPDI with AtmelICE, how to?

dl8dtl, 2024-06-17:
I tried to connect my AtmelICE to one of the AVRxEx chips.

However, connection to the chip fails, and I don't see any signal toggling on the oscilloscope.

Can anyone tell what's the minimum firmware requirement on the ICE for UPDI?

Alternatively, what would you recommend me to connect to the chip using UPDI? One of the standard FTDI demo boards maybe?

MCUdude, 2024-06-17:
Not sure which version is the minimum one to support chips with UPDI, but can't you just get the latest version from https://packs.download.microchip.com ?

> Alternatively, what would you recommend me to connect to the chip using UPDI?

The SerialUPDI programmer. Simple, fast, and reliable.

stefanrueger, 2024-06-17:
> Stefan got [this dual-purpose one](https://www.tindie.com/products/mcudude/serialupdi-programmer/)

It's amazing: take it out of the box, turn the microswitch to UPDI, connect Gnd, Vcc and UPDI and off you go (`-c serialupdi`). Worth every penny. Great work by @MCUdude and @dbuchwald.

dl8dtl, 2024-06-17:
> but can't you just get the latest version from https://packs.download.microchip.com ?

Is the firmware in the device pack of any help to me without first installing Microchip Studio into some Windows? ;-)

I'll see how to quickly create a SerialUPDI one, thanks for the hint!

MCUdude, 2024-06-17:
> Is the firmware in the device pack of any help to me without first installing Microchip Studio into some Windows? ;-)

You can probably use MPLAB. Should work on a wide variety of Linux distros.

dl8dtl, 2024-06-17:
ok, good idea

dl8dtl, 2024-06-21:
OK, I downloaded MPLAB-X for MacOS. They don't have it natively for ARM CPUs, but the emulation works well enough, and just for a firmware upgrade, that's fine with me.

Now, I can connect successfully to my AVR16EB14:

```
% ./build_freebsd/src/avrdude -c atmelice_updi -p avr16eb14 -v

avrdude: Version 7.3-20240428 (fcfe327c)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /home/joerg/src/avrdude/build_freebsd/src/avrdude.conf
         User configuration file is /home/joerg/.avrduderc

         Using port            : usb
         Using programmer      : atmelice_updi
         AVR Part              : AVR16EB14
         Programming modes     : UPDI, SPM
         Programmer Type       : JTAGICE3_UPDI
         Description           : Atmel-ICE (ARM/AVR) in UPDI mode
         ICE HW version        : 0
         ICE FW version        : 1.45 (rel. 7)
         Serial number         : J41800000136
         Vtarget               : 3.15 V
         PDI/UPDI clk          : 500 kHz

avrdude: partial Family_ID returned: "AVR "
avrdude: silicon revision: 1.0
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9449 (probably 16eb14)

avrdude done.  Thank you.
```

closed 2024-06-21 by dl8dtl