avrdudes/avrdude #1826: Parallel port silently not working

taokann, 2024-06-22:
Hello, I am having trouble with my parallel "original BSD" programmer.
Everything seems to be ok on the OS level but the pins don't actually change state.
For instance the VCC pins stay tied to ground even after the call to `par_powerup` in src/par.c .

When trying to connect, I get: `avrdude error: AVR device not responding`, device being plugged or not.

My PC's parallel port is provided by a PCI 2-serial 1-parallel card. From `lspci`:

`
05:00.2 Parallel controller: MosChip Semiconductor Technology Ltd. PCIe 9912 Multi-I/O Controller (prog-if 03 [IEEE1284])
        Subsystem: Asix Electronics Corporation (Wrong ID) Device 2000
        Control: I/O+ Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Interrupt: pin C routed to IRQ 19
        Region 0: I/O ports at 3010 [size=8]
        Region 1: I/O ports at 3000 [size=8]
        Region 2: Memory at a4201000 (32-bit, non-prefetchable) [size=4K]
        Region 5: Memory at a4200000 (32-bit, non-prefetchable) [size=4K]
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
                Address: 0000000000000000  Data: 0000
        Capabilities: [78] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0-,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [80] Express (v1) Legacy Endpoint, IntMsgNum 0
                DevCap: MaxPayload 512 bytes, PhantFunc 0, Latency L0s <1us, L1 <2us
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset- TEE-IO-
                DevCtl: CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 256 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr+ NonFatalErr- FatalErr- UnsupReq+ AuxPwr- TransPend-
                LnkCap: Port #1, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <64ns, L1 unlimited
                        ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp-
                LnkCtl: ASPM Disabled; RCB 64 bytes, LnkDisable- CommClk-
                        ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1
                        TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
        Capabilities: [100 v1] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                AERCap: First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap- ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000
        Kernel driver in use: parport_serial
        Kernel modules: parport_serial
`

Related kernel modules loaded:

`
parport_serial         20480  0
parport_pc             40960  1 parport_serial
ppdev                  24576  0
parport                81920  2 parport_pc,ppdev
`

I coun't find much documentation about those modules. What I understand is that `ppdev` creates the `/dev/parport0` file (hence the name) and is used by userspace programs like `avrdude` to control the port.

Device `/dev/parport0` exists and I am a member of the `lp` group, anyways I always test with `sudo` just in case the problem is permissions-related.

stefanrueger, 2024-06-23:
Great to see a parallel printer programmer deployed in the wild. The project certainly has the ambition to still support it (and in particular `bsd`, which is named after the original author of AVRDUDE some 23 years ago) *but* it's unclear to me whether anyone in the team even has access to a a host with a parallel port. @dl8dtl? From the given description AVRDUDE could open `/dev/parport0` but later on struggled to get TPI/SPI communication with the target AVR chip going (at the time, the project called them devices). Maybe try an *older* AVRDUDE version? Maybe try an *older* OS? Here a [diagram](https://github.com/avrdudes/avrdude/blob/main/supporting-docs/avrprog.pdf) that's useful for checking the connections to the target chip.

Let us know when you know more!

taokann, 2024-06-24:
Thank you!
After some software debugging I decided to go the LED way: attach an LED at the back of the parallel port and see what happens.
I used https://yyao.ca/projects/ParallelPortLinux/ to control individual pins --- very interesting link to keep as I struggled a lot to find such a program. Plus it uses `ppdev`, the same interface used by `avrdude`. I would definitely suggest anyone having parallel-port issues to give it a go.

Thanks to this program I could realize pin 25 was pin 1, pin 24 was pin 2, you got it the connector was wired the wrong way.
In fact the PCI card is connected to the pink DB-25 connector at the back of the PC using a ribbon cable and unshrouded pins on the card. As I recently hacked around this one it ended up plugged backwards.

```
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e930b (probably t85)

avrdude: processing -U flash:r:test.hex:I
avrdude: reading flash memory ...
Reading | ################################################## | 100% 2.02 s 
avrdude: writing output file test.hex

avrdude done.  Thank you.
```

I was able to read the flash from a Tiny85.
Fun fact, it seems like power from the parallel port is enough for programming but not for running the program.

closed 2024-06-24 by taokann

dl8dtl, 2024-06-24:
> Fun fact, it seems like power from the parallel port is enough for programming but not for running the program.

Probably depends on the clock frequency (and the AVR) you are using.

Back in the old days of the "bsd" programmer, it was quite common to tie like four output pins together (through Schottky diodes) to supply the chip to program, which also means it needed to run with that MCU frequency (like 4 MHz).

But if your application then starts full speed with 16 or 20 MHz, quite understandable the few mA that these outputs could handle won't suffice.

Glad you got it working. You are back to the roots of AVRDUDE now. :-)

taokann, 2024-06-24:
Indeed my programmer pulls power from D0-D3 using Schottky diodes.
The MCU is supposed to run at 1 MHz. However after uploading a test blink program it is happily running.
So maybe the problem was with the program previously on the chip.
I will keep in mind the frequency limitation.

stefanrueger, 2024-06-24:
@taokann Thanks for coming back and sharing debug tips. Vvv useful. Glad it worked! Please tell us about the AVRDUDE version that worked (in bash `avrdude -v |& grep ersion`) so we can collect evidence that the `par` programmers are still working.

taokann, 2024-06-25:
I can't issue the command now but I build it from source on the latest commit 41781bd.
Unfortunately the Debian package has parallel support disabled at build time. I filed a bug for that: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1074229
Thank you for keeping this software alive.

stefanrueger, 2024-06-26:
Thanks @taokann! The newest version. Very good. 

I can see why you'd want parallel support enabled by default, and I have that on as well, mainly because I want to see whether new commits affect it. The thinking at the project was to not offer stuff that's not available (to avoid confusion).

taokann, 2024-06-26:
I understand.
However, I can't see the use case when a user tries to select a programmer he can't even plug into his computer.

I would prefer some being confused (before they read the manual) over making the default build unusable for some others.

Maybe an auto-detect could be performed at build? Just like it's done for SPI/GPIO. The script could have a quick look at `lsmod | grep ppdev` or even at the `/dev` directory.

The autodetect still can't be done when building for packaging, so I suggested Debian to enable the feature.

dl8dtl, 2024-06-26:
That would need to be cross-platform, Windows, *BSD – MacOS never supported parallel ports anyway.

Having it disabled by default is probably the best solution these days – your parallel port adapter is the very rare exception, so it appears feasible that you can compile AVRDUDE yourself, I think. You are on Linux, it's pretty easy to compile it there.