avrdudes/avrdude #1833: Terminal erase command

MCUdude, 2024-07-03:
Here's a thought. The `erase` command is very useful, and I often use it to _just_ wipe the EEPROM (`erase eeprom`) or flash (`erase flash`). However, `erase` alone performs a chip erase, and depending on the fuse setting, the EEPROM is either preserved or erased. This may comae as a surprise to some, as no extra argument is provided.

Wouldn't it be better if the "standard" chip erase command was something like `erase chip`, and instead print the help text when the user enter `erase`?

```
avrdude: processing -t interactive terminal
avrdude> erase 
erasing chip ...
avrdude> erase flash
Caching | ################################################## | 100% 0.01 s 
avrdude> erase eeprom
Caching | ################################################## | 100% 0.00 s 
avrdude> flush
avrdude: synching cache to device ... 
Writing | ################################################## | 100% 0.00 s 
```

stefanrueger, 2024-07-03:
Yes, I noted that the erase command developed away from being chip erase. It's a trade-off between continuity and being a tad bit more systematic.. Scripts and programs might rely on current behaviour, including AVRDUDE's factory reset that calls `erase`: https://github.com/avrdudes/avrdude/blob/41781bdd3c01ef593d090ad8ee367110e5e2dd63/src/term.c#L1587-L1588

V 8.0 would allow us to break commands, but should we?

mcuee, 2024-07-03:
> V 8.0 would allow us to break commands, but should we?

My opinion is not to do that in this case as the benefits are not that much in reality.

dl8dtl, 2024-07-04:
Agreed.

Historically, the only erase option for AVRs was a chip erase. As preserving EEPROM contents across such an erase seemed sometimes useful yet could be a potential security risk (leak of information after erasing the chip), it was implemented as a fuse defaulting to also erase the EEPROM.

AVRDUDE could then only offer the chip erase, so this is synonym with "erase".

Even by a time the AVRs itself supported more detailed erase options internally, the Atmel tools have not been supporting that for a long time.

Given the 20-year tradition of the existing behaviour, I don't think breaking it is such a good idea.

MCUdude, 2024-07-04:
> Given the 20-year tradition of the existing behaviour, I don't think breaking it is such a good idea.

Fair enough. But could it be an idea that the fuse(s) could be silently read beforehand, so we can inform the user whenever the EEPROM has been erased or not?

```
avrdude> erase
erasing flash ...
```

```
avrdude> erase
erasing flash and EEPROM ...
```

dl8dtl, 2024-07-04:
Sounds like a good idea.

stefanrueger, 2024-07-05:
>  inform the user whenever the EEPROM has been erased

Like so? PR #1836 

I don't actually know *what exactly* is deleted on a CE: `lock`? `lock bits`? `bootrow`? `usersig`? `userrow`? So think it is important to still say erasing chip (rather than flash ... or flash and eeprom ...) because this might not be the full story

dl8dtl, 2024-07-05:
> I don't actually know _what exactly_ is deleted on a CE

Flash, EEPROM (unless fused to not being erased), lock (however it is called); on a real chip erase (by hardware, as opposed to a "fake" chip erase by a bootloader), this includes the bootloader itself.
Any kind of signature is not erased, including user signature rows (on devices that have it).
Fuses (however they are called) also prevail.

stefanrueger, 2024-07-06:
... and the data sheet for AVR-DU and AVR-EB also says bootrow is erased on chip erase. Given we know all this now (including @dl8dtl's info above), dryrun/dryboot should do a better job when emulating CE. Also the terminal bootrow cache was implemented the same way as userrow at a time when we only had the ATDF and guesswork what bootrow might do. So should also be fixed. Pushed a couple of commits to do so.

PR #1836 (2024-07-07): Improve chip erase emulation for dryboot/dryrun
Fixes #1833
closed 2024-07-07 by stefanrueger