avrdudes/avrdude #1838: JTAGICE3 EDBG communication mode relies on hardcoded USBVID

askn37, 2024-07-10:
Hello everyone

I am currently creating a jtagice3_*** compatible firmware that works with the AVR-DU series. It can be recognized as USB-HID + USB-CDC using only the standard drivers for Windows and macOS without installing any special drivers. (For the time being, the goal is to support UPDI and TPI, and debugWire cannot be used.)

To meet this condition and be recognized as a programmer by AVRDUDE, it must be in EDBG (CMSIS-DAP) communication mode, at least like pkobn_updi. However, I noticed that it is hard-coded to change to EDBG only when usbvid is *0x03EB* ('ATMEL'). Changing `usbvid` in `.avrduderc` does not work, and it is always fixed to `max_xfer = USBDEV_MAX_XFER_3 (912)`. The conditional branch is in `usb_hidapi.c:173`.

I could redesign the firmware to create a dedicated endpoint that can handle 912 byte packets, but I personally don't like the hard-coded part. This wouldn't have been an issue before. All the chips with jtagice3 were "ATMEL" branded.

The AVR-DU series is a "Microchip" (MCHP) product, so I'm reluctant to use the "ATMEL" VID. I'd like to use at least *0x04DB* (MCHP) or another VID if possible.

Any ideas on how to solve this problem? It seems like I need to add a configuration option or internal flag so that the condition at `usb_hidapi.c:173` is always true.

stefanrueger, 2024-07-10:
This is up @dl8dtl's road. Joerg is currently unavailable but I am sure he'll reply.

askn37, 2024-07-11:
Additional information

Several questions came to mind while looking into this issue. All of these were found on M1 Mac + avrdude main and should be considered as separate discussions if necessary, but they all relate to the same source files.

- Disabling *hidapi* and forcing the use of *libusb* will not work properly. Interrupt-based HID devices (where the HID report descriptor forces 64byte Input/Output) will probably break. You won't even be able to open the device without at least `sudo` admin privileges. *(xplained, pkobn, pickit4)*

- `-P usb:vid:pid` will not work. Forcefully replace it with `-P usb:serialnumber` *(hidapi and libusb)*

- `-P usb:serialnumber` is limited to a maximum of 12 digits for some reason. ATMEL/MCHP products, excluding EOL, will not work properly as their S/Ns are usually 20+ digits. *(hidapi and libusb)*

stefanrueger, 2024-08-01:
> The same page erase command XMEGA_ERASE_APP_PAGE=0x04 is sent to both BOOTROW and PROGMEM of AVR-EB/DU, so it is not possible to distinguish which address space it is, meaning that the wrong area may be erased.

Thanks for alerting us to this problem, @askn37. I really wonder what the Microchip protocols do in that case. @xedbg do you know what's expected from the protocol for a bootrow page erase?

> reluctant to use the "ATMEL" VID. I'd like to use at least 0x04DB (MCHP) or another VID if possible

In absence if @dl8dtl's input I have had a look at this. I think it is vvv reasonable to also allow the Microchip VID here. This would also be hardcoded. The open routines don't seem to have access to the part `AVRPART *p` or programmer `PROGRAMMER *pgm` which makes it hard to use vid/pid settings from avrdude.conf. 

> `-P usb:serialnumber` is limited to a maximum of 12 digits

The code matches from right to left, so even one digit is normally enough (unless you have more devices with serial numbers ending in the same digit). But yes, I'll create a PR that makes this 63 digits.

Thanks @askn37 for the code snippets you provided. That makes it clearer what you are after. Ideally, provide a PR.

Have a look at the linked PR. Would that be sufficient for your purposes? It does what you suggest -P usb:vid:pid. I am unclear about the second part of your suggested changes, particularly the `fd->usb.max_xfer = 64;` line. Looks like that setting would be overwritten further below? The other setting `fd->usb.eep = 0;` won't (good)

Anyway, have a look at the PR and tell us what other changes you'd like.

> When `-vvvv` is used, the JTAG3 transmit commands are not dumped

Yes, makes sense to be able to have the transmit commands logged. `jtag.c` has 3400+ lines of code. Please provide a PR so we can consider inclusion.

askn37, 2024-08-02:
> When -vvvv is used, the JTAG3 transmit commands are not dumped

> Yes, makes sense to be able to have the transmit commands logged. `jtag.c` has 3400+ lines of code. Please provide a PR so we can consider inclusion [edit: but wait until PR #1849 has been merged - that PR will change log levels].

I'm in no rush to do it, but I have every reason and desire to work on it, so I'll wait for the opportunity.

askn37, 2024-08-02:
> I am unclear about the second part of your suggested changes, particularly the `fd->usb.max_xfer = 64;` line.

Yes, ideally you should put `fd->usb.max_xfer = 64;` at the beginning of usbhid_open(), but I don't know which known devices this affects, so I'll change the default once I'm sure.

Currently, all USB-HID devices that link at Full-Speed must have max=64 to work, if their "USB Report Descriptor" requests a report size of 64. The only exception I know of is ATMELICE3, which has a report size of 512.

closed 2024-08-03 by stefanrueger

xedbg, 2024-08-05:
> . @xedbg do you know what's expected from the protocol for a bootrow page erase?
Good question - I actually suspect that page erase for UPDI parts is just not supported (missing/forgotten)

PR #1864 (2024-08-03): Provide -P usb:vid:pid for USB hid i/f and allow long -P usb:serno
Fixes #1838

Messed up resolving conflicts with the previous PR for this. So this is a new one...

 - New port mode `-P usb:<vid>:<pid>` for USB hid interfaces (`jtag3.c`)
 - Allow 63 digits instead of 12 digit serial numbers for `-P usb:serialno` (libusb and hidapi)
 - Admit Microchip vid in addition to Atmel vid (both hard-coded)
 - Add `fd->usb.eep = 0;` setting for CMSIS-DAP programmers as dected by product string

Without the PR:
```
$ avrdude -p atmega4808 -c xplainedmini_updi -t -P usb:MICROUPDIPROGRAMMERX
avrdude error: [... cut]