avrdudes/avrdude #1841: Unable to flash using Avrdude on a 14'' M1 Macbook Pro

gauravahuja, 2024-07-15:
Hi, I am trying to flash a hex file to an ATmega8 using Avrdude. The flashing process always fails on my 14'' M1 Macbook Pro however it works flawlessly on a Linux computer. In this example I am using a urboot bootloader but I have similar issues using a stk500v2 based ISP programmer. The stk500v2 programmer works on the Linux computer but fails on the Macbook.

Following are the logs from when I used my Macbook.

Attempt 1
```
> avrdude -xdelay=100 -v -p atmega8 -C /opt/homebrew/etc/avrdude.conf -c urclock -xnometadata -b 9600 -D -P /dev/cu.usbserial-1110 -U flash:w:.pio/build/Upload_UART/firmware.hex:i

avrdude: Version 7.3-20240710 (f489dac5)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /opt/homebrew/etc/avrdude.conf
         User configuration file is /Users/gaurav/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using port            : /dev/cu.usbserial-1110
         Using programmer      : urclock
         Setting baud rate     : 9600
         AVR Part              : ATmega8
         Programming modes     : SPM, ISP, HVPP
         Programmer Type       : Urclock
         Description           : Urclock programmer for urboot bootloaders using urprotocol
         Urboot protocol for ATmega8
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9307 (probably m8)
avrdude: reading 1862 bytes for flash from input file firmware.hex
         in 1 section [0, 0x745]: 30 pages and 58 pad bytes
         preparing flash input for device bootloader
         1984 bytes in 2 sections of [0, 0x1e7f]: 31 pages and 0 pad bytes
         writing 1984 bytes to flash ...
         writing | #############                                      | 25% 0.72 s
avrdude urclock_recv() warning: programmer is not responding
         writing | ###############----------------------------------- | 29% 1.23 s
avrdude urclock_write_byte() error: bootloader does not implement bytewise write to flash
 *** failed
avrdude do_op() error: unable to write flash (rc = -1)
avrdude urclock_disable() error: protocol expects sync byte 0xa0 but got 0x00

avrdude done.  Thank you.
```

Attempt 2
```
> avrdude -xdelay=100 -v -p atmega8 -C /opt/homebrew/etc/avrdude.conf -c urclock -xnometadata -b 9600 -D -P /dev/cu.usbserial-1110 -U flash:w:.pio/build/Upload_UART/firmware.hex:i

avrdude: Version 7.3-20240710 (f489dac5)
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /opt/homebrew/etc/avrdude.conf
         User configuration file is /Users/gaurav/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using port            : /dev/cu.usbserial-1110
         Using programmer      : urclock
         Setting baud rate     : 9600
         AVR Part              : ATmega8
         Programming modes     : SPM, ISP, HVPP
         Programmer Type       : Urclock
         Description           : Urclock programmer for urboot bootloaders using urprotocol
         Urboot protocol for ATmega8
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9307 (probably m8)
avrdude: reading 1862 bytes for flash from input file firmware.hex
         in 1 section [0, 0x745]: 30 pages and 58 pad bytes
         preparing flash input for device bootloader
         1984 bytes in 2 sections of [0, 0x1e7f]: 31 pages and 0 pad bytes
         writing 1984 bytes to flash ...
         writing |                                                    | 0% 0.00 s
avrdude urclock_recv() warning: programmer is not responding
         writing | ##------------------------------------------------ | 3% 0.51 s
avrdude urclock_write_byte() error: bootloader does not implement bytewise write to flash
 *** failed
avrdude do_op() error: unable to write flash (rc = -1)
avrdude urclock_disable() error: protocol expects sync byte 0xa0 but got 0x00

avrdude done.  Thank you.
```

mcuee, 2024-07-15:
What is the USB to Serial chipset used for your bootloader and programmer? Some USB to Serial chipsets are problematic.

gauravahuja, 2024-07-17:
Hi @mcuee,

You are correct. I was using a 14 year old USB-Serial convertor. The chipset is based on CH340. I get the following details after running `system_profiler SPUSBDataType`  on my terminal.
```
USB2.0-Serial:

              Product ID: 0x7523
              Vendor ID: 0x1a86
              Version: 2.52
              Speed: Up to 12 Mb/s
              Location ID: 0x02110000 / 3
              Current Available (mA): 500
              Current Required (mA): 96
              Extra Operating Current (mA): 0
```

This week I bought two new usb-serial convertors. One of them is also a CH340 chipset, but it worked fine. The new CH340  has the following descriptor
```
USB Serial:

              Product ID: 0x7523
              Vendor ID: 0x1a86
              Version: 2.64
              Speed: Up to 12 Mb/s
              Location ID: 0x02110000 / 3
              Current Available (mA): 500
              Current Required (mA): 98
              Extra Operating Current (mA): 0
```

So it seems CH340 chipset with Version: 2.64 works but with Version: 2.52 does not.

The other one that I bought is based on CP2102, which also worked.

```
CP2102 USB to UART Bridge Controller:

              Product ID: 0xea60
              Vendor ID: 0x10c4  (Silicon Laboratories, Inc.)
              Version: 1.00
              Serial Number: 0001
              Speed: Up to 12 Mb/s
              Manufacturer: Silicon Labs
              Location ID: 0x02110000 / 8
              Current Available (mA): 500
              Current Required (mA): 100
              Extra Operating Current (mA): 0
```

closed 2024-07-17 by gauravahuja