avrdudes/avrdude #1845: JTAG2UPDI: avrdude error: cannot find programmer id jtag2updi

androot2, 2024-07-27:
In previous version 2.13 works well.

MCUdude, 2024-07-27:
There is no such thing as 2.13. Please provide more information, and an Avrdude output log.

mcuee, 2024-07-27:
> In previous version 2.13 works well.

Are you using Arduino IDE? Even then there is no version 2.13. 

And if you are using Arduino IDE, it comes with an old version of avrdude. You should use avrdude release from this repo.

stefanrueger, 2024-08-01:
Closing this as we cannot reproduce the reported problem and don't have more information for further investigations. @androot2 you can reopen any time. Here guidelines [how to report issues](https://github.com/avrdudes/avrdude/wiki/FAQ#how-to-report-an-issue-to-avrdude-project)

closed 2024-08-01 by stefanrueger