avrdudes/avrdude #1848: Correct part is not necessarily listed first when establishing connection

MCUdude, 2024-07-31:
```
 $ ./avrdude -cdryrun -patmega4809
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 1E 96 51 (ATxmega64B3, ATmega4809)

avrdude done.  Thank you.
```

I would expect the name of the target microcontroller to be listed first. And do we need to know that there are other, not relevant parts that share the same signature?

```
$ ./avrdude -cdryrun -plgt8f328p
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 1E 95 0F (ATmega328P, ATA6614Q, LGT8F328P)

avrdude done.  Thank you.
```

stefanrueger, 2024-07-31:
> do we need to know that there are other, not relevant parts that share the same signature?

I think the better question is do we need to know the signature? Whilst it is great practice to check each and every function call for errors and handle them appropriately, it isn't really necessary to report when *expected* outcomes happen.

I would argue that we should review the 344 msg_info() calls and convert some of them to higher levels (notice, notice2, ...)

Should the default info show `AVR device initialized and ready to accept instructions`? Or what the device signature is? It used to be that AVRDUDE said it's *probably(!)* the one the user specified. To me, it's a bit like the following
```
$ cat poem
cat: found poem and ready to show it on screen
cat: file pointer has address 0x706f656d00 (probably the file poem)
Humpty Dumpty sat on a wall,
Humpty Dumpty had a great fall;
All the king's horses and all the king's men
Couldn't put Humpty together again.
cat: done.  Thank you.
```

Now I *do* understand that one million things can go wrong when programming AVRs, and that programming takes time, and that messages of progress bring about a sense of reassurance, relief even. Still we could do with fewer messages on the default run thus making it easier for the user to appraise what's going on.

However, contrary to what I said earlier. I'll fight tooth and nail for keeping the sign-off 
```
avrdude done.  Thank you.
```

Without it AVRDUDE wouldn't be the same :smile:

MCUdude, 2024-07-31:
> I think the better question is do we need to know the signature?

Call me nostalgic, but I like having the signature printed when the device has been initialized. I find it convenient that I can just run `avrdude -cusbasp -pattiny13`, and immediately get _proof_ that it was able to read something from the device, and even better, identify it. But how about printing the target name that was identified first, or perhaps _only_ print the identified target name? Most ATmega328P users don't know what an ATA6614Q even is.

> I'll fight tooth and nail for keeping the sign-off

Me too! I like it just the way it is and always has been.

stefanrueger, 2024-07-31:
> get *proof* that it was able to read something from the device

You'd still have that with the same method: the *absence* of any message would confirm that to you! That's standard in UNIX:
```
$ cmp file1 file2
```
will show *nothing* when the files are the same and (like AVRDUDE) exit with 0, so you can use `&&` to run sth else on success.

> perhaps *only* print the identified target name

I'll give you that for `-c dryrun` and `-c dryboot`. Other than that the real trouble is that AVRDUDE has *no idea* what the correct part is and no way of knowing from the device signature. AVRDUDE used to hedge its bets by saying (*probably* partXYZ) giving users with a mild psychological condition of *reassurance seeking* a different set of nightmares (probably).

> and identify it

Well, you might not have! And that's the point. Imagine someone hooking up a boxed product hazarding the guess `m328p` and actually getting *wrong confirmation* it is. They might be wasting hours on something that turns out to be the automotive `ATA6614Q`.

I still think that *a lot* of `msg_info()` are debug messages in disguise. And that the project should do something about that.

stefanrueger, 2024-07-31:
> turns out to be the automotive `ATA6614Q`.

A better example would have been `LGT8F328P`, which is a readily available want-to-be clone with *very* different behaviour: no EEPROM! Different cycle times! (A pace-maker built with that might kill people)

MCUdude, 2024-07-31:
Then it would be better if the `avrdude: device signature` line wasn't printed at all _unless_the programmer detects a different signature than it expects. Then I think it should print what device it detected and its signature, as this makes it easier to help out with troubleshooting

> I still think that a lot of msg_info() are debug messages in disguise. And that the project should do something about that.

I agree. Things have gotten better, but I still think we have a way to go.

askn37, 2024-08-02:
*Something to remember:* ATMEL sometimes reuses the same signatures. For example, `0x1e 0x96 0x51` matches both `ATmega4809` and `ATxmega64B3`. Looking at the signatures alone won't tell you the difference. Luckily, in this case, you can find the correct part name by searching for one as `PM_UPDI` and the other as `PM_PDI`. However, there are some older chips where you won't know the difference. Signature issues are a pain.

stefanrueger, 2024-08-02:
> correct part name by searching for one as `PM_UPDI` and the other as `PM_PDI`

Yes, that's a good trick. AVRDUDE uses that already by only showing those parts with a matching signature that the `-c programmer` can deal with *but* the example uses `-c dryrun` which is compatible with all parts.

> Signature issues are a pain

Agree! Couldn't have said it any better :smile:

PR #1849 (2024-08-03): Update look and feel
Fixes #1848

 - Device signature only shown on error or verbosity > 0
 - Dryrun/dryboot only shows the requested part as matching
 - Also shifts most main.c info messages to notice
 
Feedback welcome.
closed 2024-08-03 by stefanrueger