avrdudes/avrdude #1858: Unable to write LOCK bit byte to JTAG3_TPI

askn37, 2024-08-02:
This was discovered during code review and has not yet been seen in production. If possible, try the following combinations:

Expected faulty PGMs:

  - jtagice3_tpi, pickit4_tpi, xplainedmini_tpi, atmelice_tpi, powerdebugger_tpi, snap_tpi

Expected faulty parts:

  - ATtiny4, ATtiny5, ATtiny9, ATtiny10, ATtiny20, ATtiny40, ATtiny102, ATtiny104

Faulty command:

  - `-U lockbits:w:0xFF:m`

Expected command output:

```
avrdude: processing -U lockbits:w:0xFF:m
avrdude: reading 1 byte for lockbits from input file 0xFF
         in 1 section [0, 0]: 1 page and 15 pad bytes
         writing 1 byte to lockbits (0xff)
avrdude jtag3_erase_tpi() error: jtag3_erase_tpi() unsupported memory: lockbits
avrdude jtag3_write_byte_tpi() error: error in communication, received status 0xffffffff
 *** failed
avrdude do_op() error: unable to write lockbits (rc = -1)
```

Reason for failure:

jtag3_erase_tpi() returns an error internally if the memory type is neither FUSE nor FLASH.
This assumption is correct, since the lock bits are prohibited from being erased by external programming.
However, in reality, each bit can only be written in one direction, from *1 to 0*, and can only be written back to *0xFF by the chip erase* command. So it is a mistake to abort with an error before writing the lock bits.

Expected fix code:

```diff
diff --git a/src/jtag3.c b/src/jtag3.c
index e9c48a04..735bb165 100644
--- a/src/jtag3.c
+++ b/src/jtag3.c
@@ -3038,11 +3038,13 @@ static int jtag3_write_byte_tpi(const PROGRAMMER *pgm, const AVRPART *p, const A
     return -1;
   }
 
+  if (mem_is_a_fuse(mem) && mem_is_flash(mem)) {
   status = jtag3_erase_tpi(pgm, p, mem, addr);
   if (status < 0) {
     pmsg_error("error in communication, received status 0x%02x\n", status);
     return -1;
   }
+  }
 
   paddr = mem->offset + addr;
```

mcuee, 2024-08-02:
@janegilruud 

Just wondering if you can review this as you added the TPI support for Microchip tools last time. Thanks.
* https://github.com/avrdudes/avrdude/pull/1205

stefanrueger, 2024-08-02:
@askn37 I agree with your analysis and suggested solution (but note it should be `||` instead of `&&` in the patch). Thanks!

closed 2024-08-03 by stefanrueger

janegilruud, 2024-08-03:
Sorry, late to the party, but it sure looks good :-)

PR #1859 (2024-08-03): Fix byte writes to TPI lockbits for jpag3.c
Fixes #1858 

As the change is vvv simple there is no need for testing, but if you want to (and have the gear) it would be
```
$ avrdude -c xplainedmini -p t104 -U ALL:r:xplainedmini-t104-bak.hex -U ALL:w:xplainedmini-t104-bak.hex
```
for testing the issue and testing against regressions.