avrdudes/avrdude #1861: Unexpected auto-erase consequences

stefanrueger, 2024-08-02:
In the good old days when the most complex command lines only consisted of a few `-U` commands auto-erase was simple. In absence of a `-e` or `-D` command AVRDUDE would check whether one of the `-U` commands was going to write to flash and, if so, auto-erase the chip at the start. Now with memory-lists and backup ability this can have unintended consequences
```
$ avrdude ... -U ALL:r:backup.hex -U ALL:w:backup.hex:I
```
actually auto-erases the chip at the beginning *before* writing the backup of the data. Neatly [demonstrated here](https://github.com/avrdudes/avrdude/pull/1859#issuecomment-2265599547)

It would have been the case that auto-erase would have done the same in the past with a line like
```
$ avrdude ... -U flash:r:backup.hex:i -U flash:w:backup.hex:i
```
but that might also have been unexpected...

I think we need a different auto-erase Strategy

stefanrueger, 2024-08-03:
@askn37 Good idea; there is already  `-T "backup <memlist> <file>[:<format>]"` so what you suggest is already possible.

This PR tries to make auto-erase less surprising for users.

closed 2024-08-03 by stefanrueger

askn37, 2024-08-03:
What bothers me is that `-U` and `-T` are run chronologically in the order they appear, while other options are run globally first. Even `-e` is run first. Putting `-U any:w` anywhere makes up for the implicit `-e`. So, with or without `-e`, you can be sure that non-`-U` global options will be run first. It will appear that way to the end user too. This is just a thought.

stefanrueger, 2024-08-03:
Yes, you are correct there is more to the `-S` option than just the functionality. The point remains, we have the backup functionality and the user can place that at the start of `-U/T` activities or anywhere else they like within these. The only thing they cannot is place *that* `-T` before `-e` or the auto-erase happens. However, they can switch off auto-erase with `-D` and they can place `-T erase` anywhere in chronological order they like. So the users *can* create a flexible ordering of tasks in complex command lines. Which makes `-S` less useful than perhaps was intended...

askn37, 2024-08-04:
I don't know what style your users prefer, but I think the -U and -T scenarios cover it all.

In that case, it's obviously unclear where the automated `-e` will be executed. In that case, I can suggest `-T aera` (it's ok to auto-erase) which allows the user to trigger the automated `-e` at any time.

dl8dtl, 2024-08-04:
While I don't see the point why someone would actually want to first read everything and then attempt to write it again (it just adds to wearout without any other use I could imagine), it would help to delay the auto-erase up until the first `-U` option is processed with write intent.

stefanrueger, 2024-08-04:
>  why someone would actually want to first read everything and then attempt to write it again

I imagined the litmus test for a programmer/part combo to be just this
```
$ avrdude -c ... -p ... -U ALL:r:backup.hex:I -U ALL:w:backup.hex:I
```
but it really needs to be that
```
$ avrdude -c ... -p ... -U ALL:r:backup.hex:I -T erase  -U ALL:w:backup.hex:I
```

> delay the auto-erase up until the first `-U` option is processed with write intent

Yes, in the good old days without `-t` and `-T "write flash ..."` this would have been the best way of handling it, but now ... this gets complicated and messy.

PR #1862 (2024-08-03): Refine auto-erase strategy
Fixes #1861 

With this PR:

#### Reading from flash then writing to it
---
No auto-erase and message about it

```
$ avrdude -c dryrun -p m328pb -xrandom -U ALL:r:m328pb.hex -U ALL:w:m328pb.hex:I

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 1E 95 16 (ATmega328PB)
avrdude: NOT auto-erasing chip as flash might need reading before writing to it

avrdude: processing -U ALL:r:m328pb.hex:I
         reading multiple memories ...
         eeprom         | ################################################## | 100% 0.00 s 
      [... cut]