avrdudes/avrdude #1884: [MacOS] Avrdude built using autotools doesn't work

MCUdude, 2024-08-12:
This regression must have been introduced recently. When building Avrdude using `./bootstrap && ./configure && make`, I'm getting this error when trying the execute the binary:

```
$ ./avrdude
dyld[17397]: symbol not found in flat namespace (_elf32_getehdr)
Abort trap: 6
```

Not sure what's causing this. Do you have any idea @stefanrueger or @ndim?

stefanrueger, 2024-08-12:
libelf depreciation in MacOS?

MCUdude, 2024-08-12:
I do have libelf installed. And the binary works if I build using build.sh

ndim, 2024-08-12:
The build.sh script uses `/opt/local` or `/opt/homebrew` or such directories to look for include files and to link libraries to. The shown call to `configure` does not.

A simple test for this hypothesis would be to call `make` with `CFLAGS=-I/opt/homebrew/include` and especially `LDFLAGS=-L/opt/homebrew/lib`

MCUdude, 2024-08-12:
I tried running

```
$ make clean

$./bootstrap

$ ./configure

$ make CFLAGS=-I/opt/homebrew/include LDFLAGS=-L/opt/homebrew/lib

```

But I'm still getting the exact same error when running `./avrdude` from the src folder

mcuee, 2024-08-12:
@MCUdude 

Your steps are wrong. The right steps are as following. 
```
$ make clean
$./bootstrap
$ CFLAGS=-I/opt/homebrew/include LDFLAGS=-L/opt/homebrew/lib ./configure
$ make 
```

If I follow your step, I can see the following configuration summary output.
```
Configuration summary:
----------------------
DON'T HAVE libelf
DON'T HAVE libusb
DON'T HAVE libusb_1_0
DON'T HAVE libftdi1
DON'T HAVE libftdi
DON'T HAVE libhid
DON'T HAVE libhidapi
DO HAVE    libreadline
DON'T HAVE libserialport
DO HAVE    pthread
DISABLED   doc
DISABLED   parport
DISABLED   linuxgpio
DISABLED   linuxspi
```

Using my steps, here is the output.
```
Configuration summary:
----------------------
DO HAVE    libelf
DO HAVE    libusb
DO HAVE    libusb_1_0
DO HAVE    libftdi1
DON'T HAVE libftdi
DON'T HAVE libhid
DO HAVE    libhidapi
DO HAVE    libreadline
DO HAVE    libserialport
DO HAVE    pthread
DISABLED   doc
DISABLED   parport
DISABLED   linuxgpio
DISABLED   linuxspi
```

mcuee, 2024-08-13:
Wiki updated.
https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-macOS#building-using-auto-tools

@MCUdude 
Please review the above Wiki page to see if I need to change anything. Thanks.

MCUdude, 2024-08-13:
That doesn't fix the issue:

```
$ ./bootstrap
+ rm -rf autom4te.cache
+ LIBTOOLIZE=libtoolize
+ glibtoolize --version
+ LIBTOOLIZE=glibtoolize
+ export LIBTOOLIZE
+ autoreconf -i

$ CFLAGS=-I/opt/homebrew/include LDFLAGS=-L/opt/homebrew/lib ./configure
checking build system type... x86_64-apple-darwin21.6.0
checking host system type... x86_64-apple-darwin21.6.0
checking target system type... x86_64-apple-darwin21.6.0
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... mkdir -p
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking versioninfo item CMAKE_PROJECT_VERSION... 7.3
checking versioninfo item CMAKE_LIBAVRDUDE_VERSION... 2.0.0
checking versioninfo item CMAKE_LIBAVRDUDE_SOVERSION... 2
checking versioninfo item GIT_COMMIT_DATE... 20240812
checking versioninfo item GIT_COMMIT_HASH... cd956e82
checking versioninfo item GIT_TAG_HASH... e599214c
checking versioninfo derived PACKAGE_VERSION... 7.3-20240812
checking versioninfo derived AVRDUDE_FULL_VERSION... 7.3-20240812 (cd956e82)
checking versioninfo derived libtool -version-info for libavrdude... 2:0:0
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking for gawk... (cached) awk
checking for a sed that does not truncate output... /usr/bin/sed
checking for bison... bison -y
checking for flex... flex
checking for lex output file root... lex.yy
checking for lex library... none needed
checking whether yytext is a pointer... yes
checking for ar... ar
checking the archiver (ar) interface... ar
checking how to print strings... printf
checking for a sed that does not truncate output... (cached) /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /Library/Developer/CommandLineTools/usr/bin/ld
checking if the linker (/Library/Developer/CommandLineTools/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 786432
checking how to convert x86_64-apple-darwin21.6.0 file names to x86_64-apple-darwin21.6.0 format... func_convert_file_noop
checking how to convert x86_64-apple-darwin21.6.0 file names to toolchain format... func_convert_file_noop
checking for /Library/Developer/CommandLineTools/usr/bin/ld option to reload object files... -r
checking for file... file
[... 209 lines cut ...]
  CC       libavrdude_la-update.lo
  CC       libavrdude_la-wiring.lo
  CC       libavrdude_la-xbee.lo
  CCLD     libavrdude.la
ld: warning: directory not found for option '-L/opt/homebrew/lib'
ld: warning: -undefined dynamic_lookup may not work with chained fixups
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-confwin.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-ppi.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-ppiwin.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-serbb_win32.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-ser_win32.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-confwin.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-ppi.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-ppiwin.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-serbb_win32.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-ser_win32.o) has no symbols
  CCLD     avrdude
ld: warning: directory not found for option '-L/opt/homebrew/lib'
ld: warning: directory not found for option '-FCoreFoundation'

$ ./avrdude
dyld[43732]: symbol not found in flat namespace (_elf32_getehdr)
Abort trap: 6

```

MCUdude, 2024-08-13:
... But I think this warning is new:

`[grep " install " /var/log/dpkg.log](ld: warning: directory not found for option '-L/opt/homebrew/lib')`

The `-FCoreFoundation` one has always been there

mcuee, 2024-08-13:
@MCUdude 

Hmm, it seems to me your Homebrew installation is broken...

I have no issues on my side.

Anyway, I have re-opened the issue.

ndim, 2024-08-14:
The CI ˋbuild.ymlˋ tests (macos, msvc, mingw) should probably test that the ˋavrdudeˋ executable` which they have just built actually runs.

The native linux builds (both cmake and automake) run the ˋDryrun_testˋ stage which should check that by running ˋ./tools/test-avrdudeˋ.

Hmmm.... now that I take a look at the output for the non-cmake case, I am not sure that stage actually runs `avrdudeˋ. It does not fail, though.

I guess I should make certain that `tools/test-avrdude` actually fails if it cannot run `avrdudeˋ, and that `tools/test-avrdude` is run for all non-crosscompile builds (i.e. add it for macos, msvc, and mingw builds).

Then I can hopefully reproduce the linking issue on the macos CI builds, and then try to fix it.

Plan B: Ask a friend to borrow his Mac for a bit.

stefanrueger, 2024-08-14:
@ndim Great plan A. 

> actually fails if it cannot run `avrdudeˋ

Pro tip: use the `-e` option to specify the path of avrdude:

```
$ echo | tools/test-avrdude -p "-c dryrun -p m328p" -e invalidpath || echo FAILED
test-avrdude: cannot execute invalidpath
FAILED
```

MCUdude, 2024-08-14:
We don't _have_ to resolve this before the 8.0 release. I'm currently building using CMmake, and I'm prioritizing testing various programmers and targets. There have been quite a few regressions discovered in the JTAG3 code base, and I hope we're able to iron all these out.

ndim, 2024-08-15:
This could be a consequence of removing the static `libavrdude.a` target, relying on libtool library `libavrdude.la` instead, which can be either shared or static.

If formerly static linking worked and now dynamic linking fails.... That could be it. Is the homebrew libelf library existing as both shared/dynamic and static, by chance?

MCUdude, 2024-08-15:
>  Is the homebrew libelf library existing as both shared/dynamic and static, by chance?

How do I check this? BTW I just tried building PR #1888 using `make`, but I'm still getting the same error.
And there are nothing in the `/opt` path. But I'm not sure there has ever been.

```
$ make
  YACC     config_gram.c
updating config_gram.h
  LEX      lexer.c
/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in .
  CC       avrdude-main.o
  CC       avrdude-whereami.o
  CC       avrdude-developer_opts.o
  CC       libavrdude_la-config_gram.lo
  CC       libavrdude_la-lexer.lo
lexer.c:2378:38: warning: comparison of integers of different signs: 'unsigned long' and 'int' [-Wsign-compare]
        if (((yy_n_chars) + number_to_move) > YY_CURRENT_BUFFER_LVALUE->yy_buf_size) {
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 warning generated.
  CC       libavrdude_la-arduino.lo
  CC       libavrdude_la-avr.lo
  CC       libavrdude_la-avr910.lo
  CC       libavrdude_la-avrcache.lo
  CC       libavrdude_la-avrftdi.lo
  CC       libavrdude_la-avrftdi_tpi.lo
  CC       libavrdude_la-avrintel.lo
  CC       libavrdude_la-avrpart.lo
  CC       libavrdude_la-avr_opcodes.lo
  CC       libavrdude_la-bitbang.lo
  CC       libavrdude_la-buspirate.lo
  CC       libavrdude_la-butterfly.lo
  CC       libavrdude_la-ch341a.lo
  CC       libavrdude_la-config.lo
  CC       libavrdude_la-confwin.lo
  CC       libavrdude_la-crc16.lo
  CC       libavrdude_la-disasm.lo
  CC       libavrdude_la-dfu.lo
  CC       libavrdude_la-dryrun.lo
  CC       libavrdude_la-fileio.lo
  CC       libavrdude_la-flip1.lo
  CC       libavrdude_la-flip2.lo
  CC       libavrdude_la-ft245r.lo
  CC       libavrdude_la-jtagmkI.lo
  CC       libavrdude_la-jtagmkII.lo
  CC       libavrdude_la-jtag3.lo
  CC       libavrdude_la-leds.lo
  CC       libavrdude_la-linuxgpio.lo
  CC       libavrdude_la-linuxspi.lo
  CC       libavrdude_la-lists.lo
  CC       libavrdude_la-micronucleus.lo
  CC       libavrdude_la-par.lo
  CC       libavrdude_la-pgm.lo
  CC       libavrdude_la-pgm_type.lo
  CC       libavrdude_la-pickit2.lo
  CC       libavrdude_la-pindefs.lo
  CC       libavrdude_la-ppi.lo
  CC       libavrdude_la-ppiwin.lo
  CC       libavrdude_la-serbb_posix.lo
  CC       libavrdude_la-serbb_win32.lo
  CC       libavrdude_la-ser_avrdoper.lo
  CC       libavrdude_la-ser_posix.lo
  CC       libavrdude_la-ser_win32.lo
  CC       libavrdude_la-serialadapter.lo
  CC       libavrdude_la-stk500.lo
  CC       libavrdude_la-stk500v2.lo
  CC       libavrdude_la-stk500generic.lo
  CC       libavrdude_la-strutil.lo
  CC       libavrdude_la-teensy.lo
  CC       libavrdude_la-term.lo
  CC       libavrdude_la-usbasp.lo
  CC       libavrdude_la-serialupdi.lo
  CC       libavrdude_la-serprog.lo
  CC       libavrdude_la-updi_link.lo
  CC       libavrdude_la-updi_state.lo
  CC       libavrdude_la-updi_readwrite.lo
  CC       libavrdude_la-updi_nvm.lo
  CC       libavrdude_la-updi_nvm_v0.lo
  CC       libavrdude_la-updi_nvm_v2.lo
  CC       libavrdude_la-updi_nvm_v3.lo
  CC       libavrdude_la-updi_nvm_v4.lo
  CC       libavrdude_la-updi_nvm_v5.lo
  CC       libavrdude_la-urclock.lo
  CC       libavrdude_la-usb_hidapi.lo
  CC       libavrdude_la-usb_libusb.lo
  CC       libavrdude_la-usbtiny.lo
  CC       libavrdude_la-update.lo
  CC       libavrdude_la-wiring.lo
  CC       libavrdude_la-xbee.lo
  CCLD     libavrdude.la
ld: warning: directory not found for option '-L/opt/homebrew/lib'
ld: warning: -undefined dynamic_lookup may not work with chained fixups
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-confwin.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-ppi.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-ppiwin.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-serbb_win32.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-ser_win32.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-confwin.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-ppi.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-ppiwin.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-serbb_win32.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libavrdude.a(libavrdude_la-ser_win32.o) has no symbols
  CCLD     avrdude
ld: warning: directory not found for option '-L/opt/homebrew/lib'
ld: warning: directory not found for option '-FCoreFoundation'

$ ./avrdude 
dyld[43872]: symbol not found in flat namespace (_elf32_getehdr)
Abort trap: 6

```

ndim, 2024-08-15:
On the GitHub macos worker, `brew list libelf` shows only a static library `libelf.a`, with no shared library:

```
2024-08-15T15:39:52.4785210Z /opt/homebrew/Cellar/libelf/0.8.13_1/INSTALL_RECEIPT.json
2024-08-15T15:39:52.4786510Z /opt/homebrew/Cellar/libelf/0.8.13_1/.brew/libelf.rb
2024-08-15T15:39:52.4787480Z /opt/homebrew/Cellar/libelf/0.8.13_1/ChangeLog
2024-08-15T15:39:52.4788550Z /opt/homebrew/Cellar/libelf/0.8.13_1/include/libelf/libelf.h
2024-08-15T15:39:52.4789750Z /opt/homebrew/Cellar/libelf/0.8.13_1/include/libelf/elf_repl.h
2024-08-15T15:39:52.4791070Z /opt/homebrew/Cellar/libelf/0.8.13_1/include/libelf/nlist.h
2024-08-15T15:39:52.4792260Z /opt/homebrew/Cellar/libelf/0.8.13_1/include/libelf/sys_elf.h
2024-08-15T15:39:52.4793490Z /opt/homebrew/Cellar/libelf/0.8.13_1/include/libelf/gelf.h
2024-08-15T15:39:52.4794570Z /opt/homebrew/Cellar/libelf/0.8.13_1/sbom.spdx.json
2024-08-15T15:39:52.4812880Z /opt/homebrew/Cellar/libelf/0.8.13_1/README
2024-08-15T15:39:52.4814010Z /opt/homebrew/Cellar/libelf/0.8.13_1/COPYING.LIB
2024-08-15T15:39:52.4815100Z /opt/homebrew/Cellar/libelf/0.8.13_1/lib/pkgconfig/libelf.pc
2024-08-15T15:39:52.4816410Z /opt/homebrew/Cellar/libelf/0.8.13_1/lib/locale/de/LC_MESSAGES/libelf.cat
2024-08-15T15:39:52.4817670Z /opt/homebrew/Cellar/libelf/0.8.13_1/lib/libelf.a
```

So `dyld` looking for symbols (`_elf32_getehdr`) and not finding them appears wrong.

ndim, 2024-08-15:
OK... let's continue this after 8.0 with finding out how the homebrew on @MCUdude's mac differs from the github worker (`brew list libelf`, and what is the prefix to which all the files from the Cellar are linked to).

Just switching to using pkg-config for finding all dependencies which provide a `.pc` file (and libelf is one of them) might already happen to bring the proper linker flags, but that is a thing for after the upcoming 8.0 release weekend.

mcuee, 2024-08-16:
As mentioned, I do not have issues with my Mac Mini M1 with Homebrew (both Apple Silicon and x86_64). I tend to think @MCUdude's homebrew installation is broken...

Maybe there are permission problems of sub-directories under '/opt/homebrew'.

mcuee, 2024-08-16:
@ndim and @MCUdude 

`build.sh` is a bit strange but somehow it works so I have never bothered to look further.

```
            # Apple M1 (may be new version of homebrew also)
            if [ -d /opt/homebrew ]
            then
                build_flags="${build_flags} -D CMAKE_C_FLAGS=-I/opt/homebrew/include -D CMAKE_EXE_LINKER_FLAGS=-L/opt/homebrew/Cellar"
            else
                build_flags="${build_flags} -D CMAKE_C_FLAGS=-I/usr/local/include -D CMAKE_EXE_LINKER_FLAGS=-L/usr/local/Cellar"
            fi
```

mcuee, 2024-08-16:
@MCUdude 

Can you try the following configure option to be in line with `build.sh`?

```
CFLAGS=-I/opt/homebrew/include LDFLAGS=-L/opt/homebrew/Cellar ./configure
```

mcuee, 2024-08-22:
Here is my build log under macOS.

```
avrdude_main/src - (main)> ./bootstrap 
+ rm -rf autom4te.cache
+ LIBTOOLIZE=libtoolize
+ glibtoolize --version
+ LIBTOOLIZE=glibtoolize
+ export LIBTOOLIZE
+ autoreconf -i

avrdude_main/src - (main)> CFLAGS=-I/opt/homebrew/include LDFLAGS=-L/opt/homebrew/lib ./configure
checking build system type... aarch64-apple-darwin23.5.0
checking host system type... aarch64-apple-darwin23.5.0
checking target system type... aarch64-apple-darwin23.5.0
checking for a BSD-compatible install... /opt/homebrew/bin/ginstall -c
checking whether sleep supports fractional seconds... yes
checking filesystem timestamp resolution... 2
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /opt/homebrew/bin/gmkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking xargs -n works... yes
checking versioninfo item CMAKE_PROJECT_VERSION... 7.3
checking versioninfo item CMAKE_LIBAVRDUDE_VERSION... 2.0.0
checking versioninfo item CMAKE_LIBAVRDUDE_SOVERSION... 2
checking versioninfo item GIT_COMMIT_DATE... 20240822
checking versioninfo item GIT_COMMIT_HASH... a9594622
checking versioninfo item GIT_TAG_HASH... e599214c
checking versioninfo derived PACKAGE_VERSION... 7.3-20240822
checking versioninfo derived AVRDUDE_FULL_VERSION... 7.3-20240822 (a9594622)
checking versioninfo derived libtool -version-info for libavrdude... 2:0:0
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking for gawk... (cached) gawk
checking for a sed that does not truncate output... /usr/bin/sed
checking for bison... bison -y
checking for flex... flex
checking for lex output file root... lex.yy
checking for lex library... none needed
checking whether yytext is a pointer... yes
checking for ar... ar
checking the archiver (ar) interface... ar
checking how to print strings... printf
checking for a sed that does not truncate output... (cached) /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /Library/Developer/CommandLineTools/usr/bin/ld
checking if the linker (/Library/Developer/CommandLineTools/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 786432
checking how to convert aarch64-apple-darwin23.5.0 file names to aarch64-apple-darwin23.5.0 format... func_convert_file_noop
checking how to convert aarch64-apple-darwin23.5.0 file names to toolchain format... func_convert_file_noop
checking for /Library/Developer/CommandLineTools/usr/bin/ld option to reload object files... -r
checking for file... file
checking for objdump... objdump
[... 194 lines cut ...]
  CC       libavrdude_la-teensy.lo
  CC       libavrdude_la-term.lo
  CC       libavrdude_la-usbasp.lo
  CC       libavrdude_la-serialupdi.lo
  CC       libavrdude_la-serprog.lo
  CC       libavrdude_la-updi_link.lo
  CC       libavrdude_la-updi_state.lo
  CC       libavrdude_la-updi_readwrite.lo
  CC       libavrdude_la-updi_nvm.lo
  CC       libavrdude_la-updi_nvm_v0.lo
  CC       libavrdude_la-updi_nvm_v2.lo
  CC       libavrdude_la-updi_nvm_v3.lo
  CC       libavrdude_la-updi_nvm_v4.lo
  CC       libavrdude_la-updi_nvm_v5.lo
  CC       libavrdude_la-urclock.lo
  CC       libavrdude_la-usb_hidapi.lo
  CC       libavrdude_la-usb_libusb.lo
  CC       libavrdude_la-usbtiny.lo
  CC       libavrdude_la-update.lo
  CC       libavrdude_la-wiring.lo
  CC       libavrdude_la-xbee.lo
  CCLD     libavrdude.la
  CCLD     avrdude
ld: warning: search path 'CoreFoundation' not found
```

mcuee, 2024-08-22:
The produced binay is also working.

```
avrdude_main/src - (main)> ./avrdude 
Usage: avrdude [options]
Options:
  -p <partno>            Specify AVR device; -p ? lists all known parts
  -p <wildcard>/<flags>  Run developer options for matched AVR devices,
                         e.g., -p ATmega328P/s or /S for part definition
  -b <baudrate>          Override RS-232 baud rate
  -B <bitclock>          Specify bit clock period (us)
  -C <config-file>       Specify location of configuration file
  -C +<config-file>      Specify additional config file, can be repeated
  -N                     Do not load config file 
  -c <programmer>        Specify programmer; -c ? and -c ?type list all
  -c <wildcard>/<flags>  Run developer options for matched programmers,
                         e.g., -c 'ur*'/s for programmer info/definition
  -A                     Disable trailing-0xff removal for file/AVR read
  -D                     Disable auto-erase for flash memory; implies -A
  -i <delay>             ISP Clock Delay [in microseconds]
  -P <port>              Connection; -P ?s or -P ?sa lists serial ones
  -r                     Reconnect to -P port after "touching" it; wait
                         400 ms for each -r; needed for some USB boards
  -F                     Override invalid signature or initial checks
  -e                     Perform a chip erase at the beginning
  -O                     Perform RC oscillator calibration (see AVR053)
  -t                     Run an interactive terminal when it is its turn
  -T <terminal cmd line> Run terminal line when it is its turn
  -U <memstr>:r|w|v:<filename>[:format]
                         Carry out memory operation when it is its turn
                         Multiple -t, -T and -U options can be specified
  -n                     Do not write to the device whilst processing -U
  -V                     Do not automatically verify during -U
  -E <exitsp>[,<exitsp>] List programmer exit specifications
  -x <extended_param>    Pass <extended_param> to programmer, see -x help
  -v                     Verbose output; -v -v for more
  -q                     Quell progress output; -q -q for less
  -l logfile             Use logfile rather than stderr for diagnostics
  -?                     Display this usage

avrdude version 7.3-20240822 (a9594622), https://github.com/avrdudes/avrdude

avrdude_main/src - (main)> otool -L .libs/avrdude 
.libs/avrdude:
	/usr/local/lib/libavrdude.2.dylib (compatibility version 3.0.0, current version 3.0.0)
	/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 5.0.0, current version 5.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/opt/homebrew/opt/libusb-compat/lib/libusb-0.1.4.dylib (compatibility version 9.0.0, current version 9.4.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 2420.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1345.100.2)
	/opt/homebrew/opt/libserialport/lib/libserialport.0.dylib (compatibility version 2.0.0, current version 2.0.0)
	/opt/homebrew/opt/hidapi/lib/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.14.0)
	/usr/lib/libedit.3.dylib (compatibility version 2.0.0, current version 3.0.0)
	/usr/lib/libncurses.5.4.dylib (compatibility version 5.4.0, current version 5.4.0)
	/opt/homebrew/opt/libftdi/lib/libftdi1.2.dylib (compatibility version 2.0.0, current version 2.5.0)

avrdude_main/src - (main)> otool -L .libs/libavrdude.dylib 
.libs/libavrdude.dylib:
	/usr/local/lib/libavrdude.2.dylib (compatibility version 3.0.0, current version 3.0.0)
	/opt/homebrew/opt/hidapi/lib/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.14.0)
	/usr/lib/libedit.3.dylib (compatibility version 2.0.0, current version 3.0.0)
	/usr/lib/libncurses.5.4.dylib (compatibility version 5.4.0, current version 5.4.0)
	/opt/homebrew/opt/libftdi/lib/libftdi1.2.dylib (compatibility version 2.0.0, current version 2.5.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1345.100.2)
```

mcuee, 2024-08-22:
BTW, homebrew formular may need a bit of change if they want the Python bindings to be installed. Somehow it does not work under my system.

```
/Users/mcuee > brew reinstall avrdude
==> Fetching avrdude
==> Downloading https://raw.githubusercontent.com/Homebrew/homebrew-core/87e96013a1516e59f51557fdc5a601a7e3a31f83/Formula/a/avrdude.rb
Already downloaded: /Users/mcuee/Library/Caches/Homebrew/downloads/d3be2565908b1a01a220498639939b4c1c3ce08d2b3aede5f3e6a7cf3d04a6b1--avrdude.rb
==> Cloning https://github.com/avrdudes/avrdude.git
Updating /Users/mcuee/Library/Caches/Homebrew/avrdude--git
==> Checking out branch main
Already on 'main'
Your branch is up to date with 'origin/main'.
HEAD is now at a9594622 Document: remove one more place where Windows parallel ports are mentioned
==> Reinstalling avrdude 
==> cmake -S . -B build/shared -DCMAKE_INSTALL_SYSCONFDIR=/opt/homebrew/etc -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_RPATH=@loader_path/../lib -DCMAKE_SHARED_LINKER_FLAGS=-Wl,-undefined,dy
==> cmake --build build/shared
==> cmake --install build/shared
Last 15 lines from /Users/mcuee/Library/Logs/Homebrew/avrdude/03.cmake:
-- Installing: /opt/homebrew/Cellar/avrdude/HEAD-a959462/lib/libavrdude.dylib
-- Installing: /opt/homebrew/Cellar/avrdude/HEAD-a959462/include/libavrdude.h
-- Installing: /opt/homebrew/Cellar/avrdude/HEAD-a959462/include/libavrdude-avrintel.h
-- Installing: /opt/homebrew/etc/avrdude.conf
-- Installing: /opt/homebrew/Cellar/avrdude/HEAD-a959462/share/man/man1/avrdude.1
-- Installing: /opt/homebrew/opt/python@3.12/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/_swig_avrdude.so
CMake Error at build/shared/src/cmake_install.cmake:119 (file):
  file INSTALL cannot copy file
  "/tmp/avrdude-20240822-18309-1scuh2/build/shared/src/_swig_avrdude.so" to
  "/opt/homebrew/opt/python@3.12/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/_swig_avrdude.so":
  Operation not permitted.
Call Stack (most recent call first):
  build/shared/cmake_install.cmake:42 (include)

READ THIS: https://docs.brew.sh/Troubleshooting

avrdude's formula was built from an unstable upstream --HEAD.
This build failure is expected behaviour.
Do not create issues about this on Homebrew's GitHub repositories.
Any opened issues will be immediately closed without response.
Do not ask for help from Homebrew or its maintainers on social media.
You may ask for help in Homebrew's discussions but are unlikely to receive a response.
Try to figure out the problem yourself and submit a fix as a pull request.
We will review it but may or may not accept it.
```

ndim, 2024-08-24:
This issue has not been resolved. The automatic closing of this issue is an unintended  byproduct of the automation.

Could someone with enough karma points reopen this?

mcuee, 2024-11-10:
@MCUdude 

Please help to try again to see if you still have issues or not.

I can not reproduct the issue and I actually believe it is more of a problem of your macOS system.

mcuee, 2025-03-08:
@MCUdude 

Just wondering if you still have the issue or not. Thanks.

MCUdude, 2025-03-12:
I recently bought a slightly used Macbook Pro (M2 Pro), and with this machine, I have no issues building with Autotools. Not sure why my old machine had problems...

stefanrueger, 2025-03-12:
I have no doubt there are problems with your (previous) Mac, @MCUdude, and it's unfortunate that we never got to the bottom of it. We don't have any more clues, and no corroborating reports from other Mac users as to what AVRDUDE could do to resolve the problems. Would it be OK to close the issue?

closed 2025-03-12 by MCUdude