avrdudes/avrdude #1890: Reading multiple memories a row results in incorrect data on the ATmega256RFR2

MCUdude, 2024-08-15:
Issue found when doing extended testing for the upcoming Avrdude v8.0 release. And this bug is most likely present with other classic AVRs with a `usersig` memory.

The signature and calibration data are incorrect when reading multiple memories at once using a JTAG3 programmer in JTAG mode.
This is true for both `-UALL:r:-:I` and `-U{eeprom,signature,{l,e,h}fuse,lock,calibration}:r:-:I`. And the incorrect data is what's stored in the `usersig` memory.

```
$ ./avrdude -cxplainedpro_jtag -patmega256rfr2 -U{eeprom,signature,{l,e,h}fuse,lock,calibration}:r:-:I

Processing -U eeprom:r:-:I
Reading eeprom memory ...
Reading | ################################################## | 100% 2.38 s 
Writing 8192 bytes to output file <stdin>
:200000005061636B206D7920626F782077697468206669766520646F7A656E206C69717531 // 00000> Pack my box with five dozen liqu eeprom
:200020006F72206A756773005061636B206D7920626F782077697468206669766520646F7F // 00020> or jugs.Pack my box with five do
:200040007A656E206C6971756F72206A756773005061636B206D7920626F782077697468F4 // 00040> zen liquor jugs.Pack my box with
:20006000206669766520646F7A656E206C6971756F72206A756773005061636B206D79203C // 00060>  five dozen liquor jugs.Pack my 
:20008000626F782077697468206669766520646F7A656E206C6971756F72206A756773009C // 00080> box with five dozen liquor jugs.
:2000A0005061636B206D7920626F782077697468206669766520646F7A656E206C69717591 // 000a0> Pack my box with five dozen liqu
:2000C0006F72206A756773005061636B206D7920626F782077697468206669766520646FDF // 000c0> or jugs.Pack my box with five do
:2000E0007A656E206C6971756F72206A756773005061636B206D7920626F78207769746854 // 000e0> zen liquor jugs.Pack my box with
:20010000206669766520646F7A656E206C6971756F72206A756773005061636B206D79209B // 00100>  five dozen liquor jugs.Pack my 
:20012000626F782077697468206669766520646F7A656E206C6971756F72206A75677300FB // 00120> box with five dozen liquor jugs.
:200140005061636B206D7920626F782077697468206669766520646F7A656E206C697175F0 // 00140> Pack my box with five dozen liqu
:200160006F72206A756773005061636B206D7920626F782077697468206669766520646F3E // 00160> or jugs.Pack my box with five do
:200180007A656E206C6971756F72206A756773005061636B206D7920626F782077697468B3 // 00180> zen liquor jugs.Pack my box with
:2001A000206669766520646F7A656E206C6971756F72206A756773005061636B206D7920FB // 001a0>  five dozen liquor jugs.Pack my 
:2001C000626F782077697468206669766520646F7A656E206C6971756F72206A756773005B // 001c0> box with five dozen liquor jugs.
:2001E0005061636B206D7920626F782077697468206669766520646F7A656E206C69717550 // 001e0> Pack my box with five dozen liqu
:200200006F72206A756773005061636B206D7920626F782077697468206669766520646F9D // 00200> or jugs.Pack my box with five do
:200220007A656E206C6971756F72206A756773005061636B206D7920626F78207769746812 // 00220> zen liquor jugs.Pack my box with
:20024000206669766520646F7A656E206C6971756F72206A756773005061636B206D79205A // 00240>  five dozen liquor jugs.Pack my 
:20026000626F782077697468206669766520646F7A656E206C6971756F72206A75677300BA // 00260> box with five dozen liquor jugs.
:200280005061636B206D7920626F782077697468206669766520646F7A656E206C697175AF // 00280> Pack my box with five dozen liqu
:2002A0006F72206A756773005061636B206D7920626F782077697468206669766520646FFD // 002a0> or jugs.Pack my box with five do
:2002C0007A656E206C6971756F72206A756773005061636B206D7920626F78207769746872 // 002c0> zen liquor jugs.Pack my box with
:2002E000206669766520646F7A656E206C6971756F72206A756773005061636B206D7920BA // 002e0>  five dozen liquor jugs.Pack my 
:20030000626F782077697468206669766520646F7A656E206C6971756F72206A7567730019 // 00300> box with five dozen liquor jugs.
:200320005061636B206D7920626F782077697468206669766520646F7A656E206C6971750E // 00320> Pack my box with five dozen liqu
:200340006F72206A756773005061636B206D7920626F782077697468206669766520646F5C // 00340> or jugs.Pack my box with five do
:200360007A656E206C6971756F72206A756773005061636B206D7920626F782077697468D1 // 00360> zen liquor jugs.Pack my box with
:20038000206669766520646F7A656E206C6971756F72206A756773005061636B206D792019 // 00380>  five dozen liquor jugs.Pack my 
:2003A000626F782077697468206669766520646F7A656E206C6971756F72206A7567730079 // 003a0> box with five dozen liquor jugs.
:2003C0005061636B206D7920626F782077697468206669766520646F7A656E206C6971756E // 003c0> Pack my box with five dozen liqu
:2003E0006F72206A756773005061636B206D7920626F782077697468206669766520646FBC // 003e0> or jugs.Pack my box with five do
:200400007A656E206C6971756F72206A756773005061636B206D7920626F78207769746830 // 00400> zen liquor jugs.Pack my box with
:20042000206669766520646F7A656E206C6971756F72206A756773005061636B206D792078 // 00420>  five dozen liquor jugs.Pack my 
:20044000626F782077697468206669766520646F7A656E206C6971756F72206A75677300D8 // 00440> box with five dozen liquor jugs.
:200460005061636B206D7920626F782077697468206669766520646F7A656E206C697175CD // 00460> Pack my box with five dozen liqu
:200480006F72206A756773005061636B206D7920626F782077697468206669766520646F1B // 00480> or jugs.Pack my box with five do
:2004A0007A656E206C6971756F72206A756773005061636B206D7920626F78207769746890 // 004a0> zen liquor jugs.Pack my box with
:2004C000206669766520646F7A656E206C6971756F72206A756773005061636B206D7920D8 // 004c0>  five dozen liquor jugs.Pack my 
:2004E000626F782077697468206669766520646F7A656E206C6971756F72206A7567730038 // 004e0> box with five dozen liquor jugs.
:200500005061636B206D7920626F782077697468206669766520646F7A656E206C6971752C // 00500> Pack my box with five dozen liqu
:200520006F72206A756773005061636B206D7920626F782077697468206669766520646F7A // 00520> or jugs.Pack my box with five do
:200540007A656E206C6971756F72206A756773005061636B206D7920626F782077697468EF // 00540> zen liquor jugs.Pack my box with
:20056000206669766520646F7A656E206C6971756F72206A756773005061636B206D792037 // 00560>  five dozen liquor jugs.Pack my 
:20058000626F782077697468206669766520646F7A656E206C6971756F72206A7567730097 // 00580> box with five dozen liquor jugs.
:2005A0005061636B206D7920626F782077697468206669766520646F7A656E206C6971758C // 005a0> Pack my box with five dozen liqu
:2005C0006F72206A756773005061636B206D7920626F782077697468206669766520646FDA // 005c0> or jugs.Pack my box with five do
:2005E0007A656E206C6971756F72206A756773005061636B206D7920626F7820776974684F // 005e0> zen liquor jugs.Pack my box with
:20060000206669766520646F7A656E206C6971756F72206A756773005061636B206D792096 // 00600>  five dozen liquor jugs.Pack my 
:20062000626F782077697468206669766520646F7A656E206C6971756F72206A75677300F6 // 00620> box with five dozen liquor jugs.
:200640005061636B206D7920626F782077697468206669766520646F7A656E206C697175EB // 00640> Pack my box with five dozen liqu
:200660006F72206A756773005061636B206D7920626F782077697468206669766520646F39 // 00660> or jugs.Pack my box with five do
:200680007A656E206C6971756F72206A756773005061636B206D7920626F782077697468AE // 00680> zen liquor jugs.Pack my box with
:2006A000206669766520646F7A656E206C6971756F72206A756773005061636B206D7920F6 // 006a0>  five dozen liquor jugs.Pack my 
:2006C000626F782077697468206669766520646F7A656E206C6971756F72206A7567730056 // 006c0> box with five dozen liquor jugs.
:2006E0005061636B206D7920626F782077697468206669766520646F7A656E206C6971754B // 006e0> Pack my box with five dozen liqu
:200700006F72206A756773005061636B206D7920626F782077697468206669766520646F98 // 00700> or jugs.Pack my box with five do
:200720007A656E206C6971756F72206A756773005061636B206D7920626F7820776974680D // 00720> zen liquor jugs.Pack my box with
[... 245 lines cut ...]
:200060005555555555555555555555555555555555555555555555555555555555555555E0 // 00060> UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU
:200080005555555555555555555555555555555555555555555555555555555555555555C0 // 00080> UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU
:2000A0005555555555555555555555555555555555555555555555555555555555555555A0 // 000a0> UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU
:2000C000555555555555555555555555555555555555555555555555555555555555555580 // 000c0> UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU
:2000E000555555555555555555555555555555555555555555555555555555555555555560 // 000e0> UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU
:2001000055555555555555555555555555555555555555555555555555555555555555553F // 00100> UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU
[... cut ...]
:200280005555555555555555555555555555555555555555555555555555555555555555BE // 00280> UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU
:2002A00055555555555555555555555555555555555555555555555555555555555555559E // 002a0> UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU
:2002C00055555555555555555555555555555555555555555555555555555555555555557E // 002c0> UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU
:2002E00055555555555555555555555555555555555555555555555555555555555555555E // 002e0> UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU
:00000001FF

Avrdude done.  Thank you.
```

askn37, 2024-08-15:
`part parent ".classic" # m64rfr2` The section contains the following comment:

```
    #####
    # Three separate flash pages
    #   - Offset 0x100 in signature row
    #   - Only programmable with JTAG or HVPP
    #   - Readable from an application
    #   - Cannot be read using ISP
    #   - Not erased by chip erase
    #
    memory "usersig"
        paged              = yes;
        size               = 768;
        page_size          = 256;
        num_pages          = 3;
        offset             = 0x100;
        min_write_delay    = 4500;
        max_write_delay    = 4500;
        mode               = 0x41;
        delay              = 50;
        blocksize          = 256;
        readsize           = 256;
    ;
```

It feels special. This is the classic part, where we have user_signatures, which is a substructure of signatures?

The affected items are signature and calibration, which are members of signatures. When user_signatures is loaded into cache memory, is it possible that it is loaded at offset 0x0000 instead of offset 0x0100?

stefanrueger, 2024-08-15:
@askn37 Yes, this is the key entry for affected parts. And your observations are valid.

Is it a cache issue? In this example `usersig` has not even been read *at all*. `eeprom` has. If I remember correctly previous tests (not mentioned here) have shown that when `signature`, `calibration` are read *before* `eeprom` or `flash` or `usersig` they read fine. As soon as `eeprom`, `flash` or `usersig` has been read subsequent reads from `signature` and `calibration` show the wrong content from usersig (possibly by the 0x100 offset @askn37  mentioned?). There is a cache in the terminal (not used with `-U`) and a cache in the jtag3.c programmer (only for eeprom and flash) which I cannot imagine producing the effect. 

It might still be that AVRDUDE requests the wrong memory type `MTYPE_xyz` for *these* classic parts.

Or it might be that it's a bug in the FW. If so, one idea would be to cache everything in the read-only `sigrow` =  `prodsig` area on `enable` and never again bother the actual physical programmer for these later. I am reluctant to spend time on this before v8.0 seeing that it's a problem that has been there for a long time affecting a small set of six parts (`avrdude -p*/CAt | grep usersig.size`)

stefanrueger, 2024-08-15:
> bug in the FW

If someone has a bare-bones breadboard with one of the affected parts (ATmega644RFR2 ATmega1284RFR2 ATmega2564RFR2 ATmega64RFR2 ATmega128RFR2 ATmega256RFR) tehy could test with another programmer (say snap) in jtag mode.

MCUdude, 2024-08-15:
> If someone has a bare-bones breadboard with one of the affected parts (ATmega644RFR2 ATmega1284RFR2 ATmega2564RFR2 ATmega64RFR2 ATmega128RFR2 ATmega256RFR) they could test with another programmer (say snap) in jtag mode.

I did this with the Power Debugger in JTAG mode, and the results were the same.

MCUdude, 2024-08-15:
@askn37 I tried changing the offset from 0x100 to 0x00, but it didn't make any difference. The results were identical.

askn37, 2024-08-15:
Hmm, looks like I need to check the parameters AVRDUDE is issuing at a lower level. `-qvvvvv` makes usbhid_send/recv show packet dumps. The key is to write the 5 `"v"`.
(It's a shame that jtag3_send doesn't have a payload dump feature.)

stefanrueger, 2024-08-15:
OK, here is the thing: Consider an m256rfr2 with a usersig memory filled with some data. Connect this to a powerdebugger in JTAG mode. Read the signature, calibration, signature, eeprom, signature in this order:

```
$ avrdude -c powerdebugger_jtag -p m256rfr2 -U{sig,cal,sig,ee,sig}:r:-:I
```
As far as I understand, the first four reads go OK and the last read gives you some usersig bytes. How can this *not* be a FW bug? How would AVRDUDE be able to *know* the contents of the usersig memory, *store* it and return instead of signature?

For above you might want to prep the `usersig` memory with unique values like so
```
$ avrdude -c powerdebugger_jtag -p m256rfr2 -Uusersig:w:$(for i in {383..0}; do printf "0x%04x\n" $i; done | paste -s -d, -):m
```

2. Verify it's there
```
$ avrdude -qq -c powerdebugger_jtag -p m256rfr2 -U usersig:r:-:I
```

Am expecting the following (I used dryrun):
```
:200000007F017E017D017C017B017A01790178017701760175017401730172017101700158 // 00000> ..~.}.|.{.z.y.x.w.v.u.t.s.r.q.p. usersig
:200020006F016E016D016C016B016A01690168016701660165016401630162016101600138 // 00020> o.n.m.l.k.j.i.h.g.f.e.d.c.b.a.`.
:200040005F015E015D015C015B015A01590158015701560155015401530152015101500118 // 00040> _.^.].\.[.Z.Y.X.W.V.U.T.S.R.Q.P.
:200060004F014E014D014C014B014A014901480147014601450144014301420141014001F8 // 00060> O.N.M.L.K.J.I.H.G.F.E.D.C.B.A.@.
:200080003F013E013D013C013B013A013901380137013601350134013301320131013001D8 // 00080> ?.>.=.<.;.:.9.8.7.6.5.4.3.2.1.0.
:2000A0002F012E012D012C012B012A012901280127012601250124012301220121012001B8 // 000a0> /...-.,.+.*.).(.'.&.%.$.#.".!. .
:2000C0001F011E011D011C011B011A01190118011701160115011401130112011101100198 // 000c0> ................................
:2000E0000F010E010D010C010B010A01090108010701060105010401030102010101000178 // 000e0> ................................
:20010000FF00FE00FD00FC00FB00FA00F900F800F700F600F500F400F300F200F100F00067 // 00100> ..~.}.|.{.z.y.x.w.v.u.t.s.r.q.p.
:20012000EF00EE00ED00EC00EB00EA00E900E800E700E600E500E400E300E200E100E00047 // 00120> o.n.m.l.k.j.i.h.g.f.e.d.c.b.a.`.
:20014000DF00DE00DD00DC00DB00DA00D900D800D700D600D500D400D300D200D100D00027 // 00140> _.^.].\.[.Z.Y.X.W.V.U.T.S.R.Q.P.
:20016000CF00CE00CD00CC00CB00CA00C900C800C700C600C500C400C300C200C100C00007 // 00160> O.N.M.L.K.J.I.H.G.F.E.D.C.B.A.@.
:20018000BF00BE00BD00BC00BB00BA00B900B800B700B600B500B400B300B200B100B000E7 // 00180> ?.>.=.<.;.:.9.8.7.6.5.4.3.2.1.0.
:2001A000AF00AE00AD00AC00AB00AA00A900A800A700A600A500A400A300A200A100A000C7 // 001a0> /...-.,.+.*.).(.'.&.%.$.#.".!. .
:2001C0009F009E009D009C009B009A009900980097009600950094009300920091009000A7 // 001c0> ................................
:2001E0008F008E008D008C008B008A00890088008700860085008400830082008100800087 // 001e0> ................................
:200200007F007E007D007C007B007A00790078007700760075007400730072007100700066 // 00200> ..~.}.|.{.z.y.x.w.v.u.t.s.r.q.p.
:200220006F006E006D006C006B006A00690068006700660065006400630062006100600046 // 00220> o.n.m.l.k.j.i.h.g.f.e.d.c.b.a.`.
:200240005F005E005D005C005B005A00590058005700560055005400530052005100500026 // 00240> _.^.].\.[.Z.Y.X.W.V.U.T.S.R.Q.P.
:200260004F004E004D004C004B004A00490048004700460045004400430042004100400006 // 00260> O.N.M.L.K.J.I.H.G.F.E.D.C.B.A.@.
:200280003F003E003D003C003B003A003900380037003600350034003300320031003000E6 // 00280> ?.>.=.<.;.:.9.8.7.6.5.4.3.2.1.0.
:2002A0002F002E002D002C002B002A002900280027002600250024002300220021002000C6 // 002a0> /...-.,.+.*.).(.'.&.%.$.#.".!. .
:2002C0001F001E001D001C001B001A001900180017001600150014001300120011001000A6 // 002c0> ................................
:2002E0000F000E000D000C000B000A00090008000700060005000400030002000100000086 // 002e0> ................................
:00000001FF
```

stefanrueger, 2024-08-15:
> It's a shame that jtag3_send doesn't have a payload dump feature.

@askn37 It's only one PR away :smile:

MCUdude, 2024-08-15:
I would prefer to use the `xplainedpro_jtag` programmer, since the ATmega256RFR2 chip is on an Xplained Pro board.
I'll make sure to have unique values in the usersig memory

I can provide a log with -vvvvv. And if you would find it useful, a wireshark capture of the entire session.

MCUdude, 2024-08-15:
Here's the output (requested by @stefanrueger):

```
$ avrdude -c xplainedpro_jtag -p m256rfr2 -t
avrdude> read usersig
Reading | ################################################## | 100% 0.02 s 
0000  7f 01 7e 01 7d 01 7c 01  7b 01 7a 01 79 01 78 01  |..~.}.|.{.z.y.x.|
0010  77 01 76 01 75 01 74 01  73 01 72 01 71 01 70 01  |w.v.u.t.s.r.q.p.|
0020  6f 01 6e 01 6d 01 6c 01  6b 01 6a 01 69 01 68 01  |o.n.m.l.k.j.i.h.|
0030  67 01 66 01 65 01 64 01  63 01 62 01 61 01 60 01  |g.f.e.d.c.b.a.`.|
0040  5f 01 5e 01 5d 01 5c 01  5b 01 5a 01 59 01 58 01  |_.^.].\.[.Z.Y.X.|
0050  57 01 56 01 55 01 54 01  53 01 52 01 51 01 50 01  |W.V.U.T.S.R.Q.P.|
0060  4f 01 4e 01 4d 01 4c 01  4b 01 4a 01 49 01 48 01  |O.N.M.L.K.J.I.H.|
0070  47 01 46 01 45 01 44 01  43 01 42 01 41 01 40 01  |G.F.E.D.C.B.A.@.|
0080  3f 01 3e 01 3d 01 3c 01  3b 01 3a 01 39 01 38 01  |?.>.=.<.;.:.9.8.|
0090  37 01 36 01 35 01 34 01  33 01 32 01 31 01 30 01  |7.6.5.4.3.2.1.0.|
00a0  2f 01 2e 01 2d 01 2c 01  2b 01 2a 01 29 01 28 01  |/...-.,.+.*.).(.|
00b0  27 01 26 01 25 01 24 01  23 01 22 01 21 01 20 01  |'.&.%.$.#.".!. .|
00c0  1f 01 1e 01 1d 01 1c 01  1b 01 1a 01 19 01 18 01  |................|
00d0  17 01 16 01 15 01 14 01  13 01 12 01 11 01 10 01  |................|
00e0  0f 01 0e 01 0d 01 0c 01  0b 01 0a 01 09 01 08 01  |.... . . . . ...|
00f0  07 01 06 01 05 01 04 01  03 01 02 01 01 01 00 01  |................|
avrdude> q

Avrdude done.  Thank you.

$ avrdude -qq -c powerdebugger_jtag -p m256rfr2 -U usersig:r:-:I
Error: no device found matching VID 0x03eb and PID list: 0x2144
Error: unable to open port usb for programmer powerdebugger_jtag
Hanss-MacBook-Pro:tools hans$ avrdude -qq -c xplainedpro_jtag -p m256rfr2 -U usersig:r:-:I
:200000007F017E017D017C017B017A01790178017701760175017401730172017101700158 // 00000> ..~.}.|.{.z.y.x.w.v.u.t.s.r.q.p. usersig
:200020006F016E016D016C016B016A01690168016701660165016401630162016101600138 // 00020> o.n.m.l.k.j.i.h.g.f.e.d.c.b.a.`.
:200040005F015E015D015C015B015A01590158015701560155015401530152015101500118 // 00040> _.^.].\.[.Z.Y.X.W.V.U.T.S.R.Q.P.
:200060004F014E014D014C014B014A014901480147014601450144014301420141014001F8 // 00060> O.N.M.L.K.J.I.H.G.F.E.D.C.B.A.@.
:200080003F013E013D013C013B013A013901380137013601350134013301320131013001D8 // 00080> ?.>.=.<.;.:.9.8.7.6.5.4.3.2.1.0.
:2000A0002F012E012D012C012B012A012901280127012601250124012301220121012001B8 // 000a0> /...-.,.+.*.).(.'.&.%.$.#.".!. .
:2000C0001F011E011D011C011B011A01190118011701160115011401130112011101100198 // 000c0> ................................
:2000E0000F010E010D010C010B010A01090108010701060105010401030102010101000178 // 000e0> ................................
:20010000FF00FE00FD00FC00FB00FA00F900F800F700F600F500F400F300F200F100F00067 // 00100> ..~.}.|.{.z.y.x.w.v.u.t.s.r.q.p.
:20012000EF00EE00ED00EC00EB00EA00E900E800E700E600E500E400E300E200E100E00047 // 00120> o.n.m.l.k.j.i.h.g.f.e.d.c.b.a.`.
:20014000DF00DE00DD00DC00DB00DA00D900D800D700D600D500D400D300D200D100D00027 // 00140> _.^.].\.[.Z.Y.X.W.V.U.T.S.R.Q.P.
:20016000CF00CE00CD00CC00CB00CA00C900C800C700C600C500C400C300C200C100C00007 // 00160> O.N.M.L.K.J.I.H.G.F.E.D.C.B.A.@.
:20018000BF00BE00BD00BC00BB00BA00B900B800B700B600B500B400B300B200B100B000E7 // 00180> ?.>.=.<.;.:.9.8.7.6.5.4.3.2.1.0.
:2001A000AF00AE00AD00AC00AB00AA00A900A800A700A600A500A400A300A200A100A000C7 // 001a0> /...-.,.+.*.).(.'.&.%.$.#.".!. .
:2001C0009F009E009D009C009B009A009900980097009600950094009300920091009000A7 // 001c0> ................................
:2001E0008F008E008D008C008B008A00890088008700860085008400830082008100800087 // 001e0> ................................
:200200007F007E007D007C007B007A00790078007700760075007400730072007100700066 // 00200> ..~.}.|.{.z.y.x.w.v.u.t.s.r.q.p.
:200220006F006E006D006C006B006A00690068006700660065006400630062006100600046 // 00220> o.n.m.l.k.j.i.h.g.f.e.d.c.b.a.`.
:200240005F005E005D005C005B005A00590058005700560055005400530052005100500026 // 00240> _.^.].\.[.Z.Y.X.W.V.U.T.S.R.Q.P.
:200260004F004E004D004C004B004A00490048004700460045004400430042004100400006 // 00260> O.N.M.L.K.J.I.H.G.F.E.D.C.B.A.@.
:200280003F003E003D003C003B003A003900380037003600350034003300320031003000E6 // 00280> ?.>.=.<.;.:.9.8.7.6.5.4.3.2.1.0.
:2002A0002F002E002D002C002B002A002900280027002600250024002300220021002000C6 // 002a0> /...-.,.+.*.).(.'.&.%.$.#.".!. .
:2002C0001F001E001D001C001B001A001900180017001600150014001300120011001000A6 // 002c0> ................................
:2002E0000F000E000D000C000B000A00090008000700060005000400030002000100000086 // 002e0> ................................
:00000001FF
Hanss-MacBook-Pro:tools hans$ avrdude -c powerdebugger_jtag -p m256rfr2 -U{sig,cal,sig,ee,sig}:r:-:I
Error: no device found matching VID 0x03eb and PID list: 0x2144
Error: unable to open port usb for programmer powerdebugger_jtag

Avrdude done.  Thank you.
Hanss-MacBook-Pro:tools hans$ avrdude -c xplainedpro_jtag -p m256rfr2 -U{sig,cal,sig,ee,sig}:r:-:I

Processing -U sig:r:-:I
Reading signature memory ...
Writing 3 bytes to output file <stdout>
:030000001EA80235                                                           // 00000> 0x1e 0xa8 0x02 signature (ATmega256RFR2)
:00000001FF

Processing -U cal:r:-:I
Reading calibration memory ...
[... 255 lines cut ...]
:201E20006F72206A756773005061636B206D7920626F782077697468206669766520646F61 // 01e20> or jugs.Pack my box with five do
[... cut ...]
:201FA000206669766520646F7A656E206C6971756F72206A756773005061636B206D7920DD // 01fa0>  five dozen liquor jugs.Pack my 
:201FC000626F782077697468206669766520646F7A656E206C6971756F72206A756773003D // 01fc0> box with five dozen liquor jugs.
:201FE0005061636B206D7920626F782077697468206669766520646F7A656E206C69717532 // 01fe0> Pack my box with five dozen liqu
:00000001FF

Processing -U sig:r:-:I
Reading signature memory ...
Writing 3 bytes to output file <stdout>
:03000000FFFEFD03                                                           // 00000> 0xff 0xfe 0xfd signature
:00000001FF

Avrdude done.  Thank you.
```

MCUdude, 2024-08-15:
And here is the full xplainedpro_jtag -vvvvv output (had to use Gist because it's too large for a single message):

https://gist.github.com/MCUdude/79cc2e2a8b73b7418d7cc7a5e028cbf1

stefanrueger, 2024-08-15:
Thanks @MCUdude. I believe the outcome of that experiments demonstrates that the bug lies squarely outside AVRDUDE. Must be a bug in the FW or the part itself (unless the `-vvvvv` output demonstrates AVRDUDE actually requests the bytes 0x100, 0x102 and 0x104 of `usersig` by mistake instead of bytes 0-2 of `signature`). The experiment also shows that caching signature, calibration etc at the beginning would allow AVRDUDE to work around this bug.

askn37, 2024-08-16:
This is a bug in the device-side firmware!

The result of removing the padding in the EDBG payload and formatting the JTAG3 payload to be more readable is shown below.

First, the first MTYPE_SIGN_JTAG gets the correct value.

```
Sending read memory command: 
Jtag3_edbg_send(): sending 12 bytes
Usbhid_send: . [80] . [11] . [00] . [10] EDBG_CMD len=16
. [0e] . [00] . [0f] . [00] seq=15
. [12] ! [21] . [00] . [b4] SCOPE_AVR MTYPE_SIGN_JTAG
. [00] . [00] . [00] . [00] addr=0
. [03] . [00] . [00] . [00] len=3
Usbhid_recv: . [80] . [01] EDBG_RSP OK
Jtag3_edbg_recv():
Usbhid_send: . [81] . [00] EDBG_RSP
Usbhid_recv: . [81] . [11] . [00] . [0a] EDBG_RSP len=10
. [0e] . [0f] . [00] seq=15
. [12] . [84] . [00] SCOPE_AVR RSP3_DATA
. [1e] . [a8] . [02] data
. [00] EOP
Jtag3_recv(): got message seqno 15 (command_sequence == 15)

Raw message:
12  84  00  1e  a8  02  00  
[AVR] Data returned:
0x00 0x1e 0xa8 0x02 0x00 

Jtag3_read_byte(.., signature, 0x1, ...)
Jtag3_read_byte(.., signature, 0x2, ...)
Device signature = 1E A8 02 (ATmega256RFR2)
```

After getting the whole EEPROM area, I get different results even though MTYPE_SIGN_JTAG is the exact same request.

```
Processing -U sig:r:-:I
Reading signature memory ...
Jtag3_read_byte(.., signature, 0x0, ...)
Sending read memory command: 
Jtag3_edbg_send(): sending 12 bytes
Usbhid_send: . [80] . [11] . [00] . [10] EDBG_CMD len=16
. [0e] . [00] . [13] . [04] seq=1043
. [12] ! [21] . [00] . [b4] SCOPE_AVR MTYPE_SIGN_JTAG
. [00] . [00] . [00] . [00] addr=0
. [03] . [00] . [00] . [00] len=3
Usbhid_recv: . [80] . [01] EDBG_RSP OK
Jtag3_edbg_recv():
Usbhid_send: . [81] . [00] EDBG_RSP
Usbhid_recv: . [81] . [11] . [00] . [0a] EDBG_RSP len=10
. [0e] . [13] . [04] seq=1043
. [12] . [84] . [00] SCOPE_AVR RSP3_DATA
. [ff] . [fe] . [fd] data
. [00] EOP
Jtag3_recv(): got message seqno 1043 (command_sequence == 1043)

Raw message:
12  84  00  ff  fe  fd  00  
[AVR] Data returned:
0x00 0xff 0xfe 0xfd 0x00 

Jtag3_read_byte(.., signature, 0x1, ...)
Jtag3_read_byte(.., signature, 0x2, ...)
Writing 3 bytes to output file <stdout>
:03000000FFFEFD03                                                           // 00000> 0xff 0xfe 0xfd signature
:00000001FF
```

My guess is that there was a buffer overflow in the firmware which corrupted the device descriptor information passed in the configuration parameters when reading the EEPROM area (8KiB).

*If this guess is correct, you might see a change if you run set parameter again before reading the signature.*

stefanrueger, 2024-08-16:
>  run set parameter again
@askn37 Cool idea! Probably best to do that after paged reads/paged writes (b/c reading flash also has the effect). Would you have a patch suggestion for @MCUdude to test?

MCUdude, 2024-08-16:
If this turns out to be a bug in the EDBG firmware (which it most likely is), we should let @xedbg know about it. @askn37 I'll give your suggested work-around a try if you provide a patch!

askn37, 2024-08-16:
@MCUdude 

> I'll give your suggested work-around a try if you provide a patch!

So let's think about a patch. It would probably look something like this:
I don't have the device in question on hand so this is all just speculation.

In jtag3_initialize:

FW release 438 is definitely the target now, so check str_starts(pgmid, "xplainedpro") and PARM3_FW_RELEASE, set the corresponding flag if applicable, and copy the send data for PARM3_DEVICEDESC to PDATA(pgm).

At least in jtag3_paged_write and jtag3_paged_load:

- After reading or writing with a different MTYPE than last time, check the corresponding flag and send mega_device_desc in PARM3_DEVICEDESC. Or maybe MTYPE irrelevant?
- jtag3_page_erase too?

In jtag3_close:

- If working memory is allocated for PDATA(pgm), release it.

xedbg, 2024-08-16:
so this is specific to that RFR2 device??

MCUdude, 2024-08-16:
> So this is specific to that RFR2 device??

@xedbg Yes, I think so. Or the entire RFR2 lineup. I've been testing [a bunch](https://github.com/avrdudes/avrdude/discussions/1654#discussioncomment-10351822) of different programmers and targets, and I've only seen this issue when reading the RFR2 device. If this is a bug in the EDBG firmware, the RFR2 is the only device that we've tested that can reproduce the issue.

If you have an Xplained Pro ATmega256RFR2 board at hand (I bought a complete Atmel ZigBit Launch Kit at finn.no for 50 NOK!), and build the latest git main, you can reproduce the issue quite easily by running:

`avrdude -cxplainedpro_jtag -patmega256rfr2 -U ALL:r:-:I`

Where ALL:r:-:I reads all memories and prints them to stdout in an Intel hex format with comments. You don't need to read all memories for it to fail, but you have to read eeprom, usersig or flash, along with signature and calibration. Reading fuses always work.

So you can also try `avrdude -cxplainedpro_jtag -patmega256rfr2 -U ALL,-flash,-usersig:r:-:I` to read all memories except for flash and usersig. It will still fail to read the signature and calibration byte.

(Speaking of bugs, we discovered a bug in the PICkit4/SNAP firmware when in PIC mode. It can't read the first 8 bytes in the SIB. See discussion and comment [here](https://github.com/avrdudes/avrdude/pull/1863#issuecomment-2274234562).

xedbg, 2024-08-16:
@MCUdude 
1. If there is a bug in the EDBG FW at this stage then I am afraid the workaround needs to be outside of that firmware :|
2. I can post a bug to the PICkit4 team on this one - maybe you should log this as a separate issue though: that thread is rather noisy now!

MCUdude, 2024-08-16:
> If there is a bug in the EDBG FW at this stage then I am afraid the workaround needs to be outside of that firmware :|

Oh no! Are they migrating away from the EDBG firmware? At least I hope they keep EDBG firmware updated for hardware like the Power Debugger and the Atmel ICE. They are way faster (at least with JTAG and UPDI) than their ARM-based competitors.

> I can post a bug to the PICkit4 team on this one - maybe you should log this as a separate issue though: that thread is rather noisy now!

I haven't created an issue about this in this repo, because it's a firmware issue we can't do anything about. We even looked at USB data using Wireshark to verify that Avrdude was not to blame for this.

xedbg, 2024-08-16:
> Oh no! Are they migrating away from the EDBG firmware? At least I hope they keep EDBG firmware updated for hardware like the Power Debugger and the Atmel ICE. They are way faster (at least with JTAG and UPDI) than their ARM-based competitors.

Well since making the nEDBG the only kits to use EDBG are ARM and that is very light on the firmware side, so there has been no need to update.  And while the Atmel-ICE and PowerDebugger share the same codebase, they are still under continuous test, which means EDBG would be very difficult to release for very little gain.

> I haven't created an issue about this in this repo, because it's a firmware issue we can't do anything about. We even looked at USB data using Wireshark to verify that Avrdude was not to blame for this.

To engage with the PICkit FW team to address this, I think it would be best to have a separate thread/issue here on GitHub.

askn37, 2024-08-16:
@MCUdude @stefanrueger 

Based on the above assumptions, I have created an experimental patch.

- The inserted code is accompanied by the `/* Experimental solution for issue #1890 */` comment label.
- The patch takes effect when the condition at line=1074 `str_starts(pgmid, "xplainedpro") && ice_rel <= 438` is met.

https://github.com/askn37/avrdude/tree/issues1890 
https://github.com/askn37/avrdude/commit/b7309e15a8e91f583d7d36e6c5ccf36b717426b5

xedbg, 2024-08-16:
> @xedbg Is there no better way to inform Microchip of suspected bugs than telling you (or creating an issue here and telling you)?

Well the correct way is via support.microchip.com

> I am with @MCUdude that we shouldn't open an issue in AVRDUDE for suspected bugs _outside_ AVRDUDE. The experiment in this [comment](https://github.com/avrdudes/avrdude/issues/1890#issuecomment-2292069178) demonstrates why we think there is a bug in the PICKit 4 FW (`signature` of the ATmega256RFR2 won't be read correctly once `eeprom` has been read using the JTAG interface).

I am confused now - is this a PICkit4 in PIC-mode bug or an EDBG bug?  The former can be fixed, the latter not so easily.

> There is also another issue that the EDBG jtag interface for classic parts does not allow to read calibration and signature beyond 3 bytes, so unable to get to the serial number in JTAG even for those classic parts where it's documented. The ISP interface in contrast has no such limitation.

I am aware of this one (our ref: FW3G-249) Its queued up for consideration with the next Atmel-ICE/Power Debugger/JTAGICE3 firmware release.  Unfortunately these are no longer very frequent.

MCUdude, 2024-08-16:
@askn37 thanks! I'll give your branch a try later today. BTW this isn't an `xplainedpro` issue. I can reproduce the issue with a PICkit4 in AVR mode, SNAP in AVR-mode, and with a Power Debugger.

> I am confused now - is this a PICkit4 in PIC-mode bug or an EDBG bug? The former can be fixed, the latter not so easily.

The bug is in the EDBG firmware. All programmers that run the EDBG firmware and are capable of JTAG programming have this issue AFAIK.

stefanrueger, 2024-08-16:
I wonder whether the bug could also be in the JTAG implementation of the part, which returns bytes stored 0x200 bytes further afield when asking for `signature` or `calibration`. After all, other parts work perfectly fine with the same programmer/EDBG FW.

xedbg, 2024-08-16:
> The bug is in the EDBG firmware. All programmers that run the EDBG firmware and are capable of JTAG programming have this issue AFAIK.

OK, got it.  (I prefer to use "EDBG-based" as they don't all run the same binary but use the same git submodule at build.)

These RFR2 parts are a little "special" - they grew this additional user row not previously on any mega/JTAG parts.  I think its actually an extension of the signature space in some way, which might explain the 'data contamination'.  I am not an expert, and don't think I can become one very quickly.  Is the correct behaviour seen on the PICkit5 implementation?

MCUdude, 2024-08-16:
>  Is the correct behavior seen on the PICkit5 implementation?

The current Avrdude PICkit5 implementation currently only supports UPDI. And I have no idea how to read multiple memories in a row using MPLABX. Does MPLABX ship with a CLI tool similar to Avrdude and atprogram?

askn37, 2024-08-16:
@MCUdude 

> The bug is in the EDBG firmware. All programmers that run the EDBG firmware and are capable of JTAG programming have this issue AFAIK.

I cannot reproduce the issue on either "Curiosity Nano" or "Xplained Mini/Nano" in my environment, so does this mean that it has nothing to do with Full-Speed ​​"nEDBG" (EDBG with 64-byte chunks) and is a phenomenon specific to High-Speed ​"​EDBG" (512-byte chunks)?

*If so, the condition narrows down to `pgm->fd.usb.max_xfer >= 512`.*

MCUdude, 2024-08-16:
@askn37 I just tried your [test branch](https://github.com/askn37/avrdude/tree/issues1890) using the Xplained Pro JTAG. Sadly, it didn't make any difference.

> I cannot reproduce the issue on either "Curiosity Nano" or "Xplained Mini/Nano" in my environment, so does this mean that it has nothing to do with Full-Speed ​​"nEDBG" (EDBG with 64-byte chunks) and is a phenomenon specific to High-Speed ​"​EDBG" (512-byte chunks)?
If so, the condition narrows down to pgm->fd.usb.max_xfer >= 512.

This might be true because there doesn't exist any JTAG-compatible EDBG-based programmer that operates with 64-byte chunks. So my theory is that the bug is present on the JTAGICE3, Atmel ICE, Power Debugger, Xplained Pro, PICkit4, and SNAP.

xedbg, 2024-08-16:
Skeptical... :) You know you can switch the Atmel-ICE and Power Debugger into a 64-byte HID mode if you want to.  I think its one of the "atprogram parameters" switches.

askn37, 2024-08-16:
@xedbg 

> You know you can switch the Atmel-ICE and Power Debugger into a 64-byte HID mode if you want to. 

Never heard of it!

It would be great if we could get and change that state using typical CMSIS-DAP commands (i.e. just 16 to 64-byte HID packets).

askn37, 2024-08-16:
@MCUdude 

> I just tried your [test branch](https://github.com/askn37/avrdude/tree/issues1890) using the Xplained Pro JTAG. Sadly, it didn't make any difference.

Got it, this branch is shelved.

Right now, if you access a memory section other than prodsig, only that section is trustworthy. A possible workaround would be to repeat ENTER_PROGMODE and LEAVE_PROGMODE for each memory section, but this is not practical, especially with a UPDI type mechanism.

MCUdude, 2024-08-17:
> for MPLAB you can try MDB

MDB seems to be for debugging only, and it seems very cumbersome to have to create a "script" just to read or write to a memory. Long story short, I didn't get MDB to work. I'll see if I can try atprogram on a Windows computer.

> Right now, if you access a memory section other than prodsig, only that section is trustworthy. A possible workaround would be to repeat ENTER_PROGMODE and LEAVE_PROGMODE for each memory section, but this is not practical, especially with a UPDI type mechanism.

If this is guaranteed to be a bug in the EDBG firmware, it feels a bit wrong to add a very impractical workaround for something that ideally should be fixed in firmware instead.

askn37, 2024-08-17:
@MCUdude 

> If this is guaranteed to be a bug in the EDBG firmware, it feels a bit wrong to add a very impractical workaround for something that ideally should be fixed in firmware instead.

I totally agree, this is not the solution.

The official consultation desk is https://microchip.my.site.com/s/ However, after using it a few times, I am dissatisfied with the extremely poor response. Even a small LIBC bug that we could reproduce in our current production using MPLAB took about 10 exchanges and over 3 months to resolve. (Even if you are a corporate customer)

Since all of my hardware other than the ATMEL-ICE and SNAP is EOL and I can't reproduce the problem in MPLAB, I'm worried this won't be officially recognized as a firmware bug.

MCUdude, 2024-08-19:
I'll see if I can reproduce the issue using atprogram.exe, the only Microchip-provided CLI tool that has a clear documentation that makes it easy to figure out how to perform basic tasks. Can't say the same for MDB.

If atprogram works with the Xplained Pro (the online docs don't mention it, but there is a lot it doesn't cover anyway), this command should do it:

```
> atprogram.exe -t xplainedpro -i JTAG -d atmega256rfr2 read --eeprom read -sg read --osccal
```
(where -sg is the signature. --signature or --sig didn't work)

mraardvark, 2024-08-19:
Correct invocation is:
-t edbg

MCUdude, 2024-08-19:
> Correct invocation is:
`-t edbg`

Got it, thanks! So if I'm able to reproduce the issue using `atprogram`, is there a chance that this bug will be looked into and hopefully fixed on your side? I know it's an edge case affecting a few microcontrollers, but it (probably) _is_  a firmware bug. For me, it's not a big deal if a new firmware version isn't released any time soon. But it would be nice if it "would be taken care of" so we can close this issue, call it a day, and tell users that the next EDBG firmware will have this issue resolved (with a TBA on when that is).

xedbg, 2024-08-19:
Yes, it looks like its in the firmware.
But its not going to be an easy fix given the age of those parts and the nature of the additional memory type.
Also - EDBG itself will probably not get new firmware at this point (standalone debuggers probably still will, and could be fixed).
Best option would probably be to reproduce it in PICkit4/5, since those are under active development.

MCUdude, 2024-08-19:
> (standalone debuggers probably still will, and could be fixed).

That would be great!

> Best option would probably be to reproduce it in PICkit4/5, since those are under active development

I can reproduce the issue with a PICKit4 in EDBG mode, but not in PIC mode. The reason is that we have only implemented support for UPDI parts at the moment.

MCUdude, 2024-08-19:
Bingo! The bug can be reproduced using `atprogram`! I'll try hooking up the Power Debugger as well just for good measure. I'll update this post once I've done the test.

<details>

<summary>atprogram.exe -t edbg -i JTAG -d atmega256rfr2 read --eeprom read -sg
read --osccal</summary>

```
>atprogram.exe -t edbg -i JTAG -d atmega256rfr2 read --eeprom read -sg
read --osccal
Firmware check OK
:100000005061636B206D7920626F78207769746826
:10001000206669766520646F7A656E206C697175FB
:100020006F72206A756773005061636B206D792071
:10003000626F782077697468206669766520646FDE
:100040007A656E206C6971756F72206A75677300CE
:100050005061636B206D7920626F782077697468D6
:10006000206669766520646F7A656E206C697175AB
:100070006F72206A756773005061636B206D792021
:10008000626F782077697468206669766520646F8E
:100090007A656E206C6971756F72206A756773007E
:1000A0005061636B206D7920626F78207769746886
:1000B000206669766520646F7A656E206C6971755B
:1000C0006F72206A756773005061636B206D7920D1
:1000D000626F782077697468206669766520646F3E
:1000E0007A656E206C6971756F72206A756773002E
:1000F0005061636B206D7920626F78207769746836
:10010000206669766520646F7A656E206C6971750A
:100110006F72206A756773005061636B206D792080
:10012000626F782077697468206669766520646FED
:100130007A656E206C6971756F72206A75677300DD
:100140005061636B206D7920626F782077697468E5
:10015000206669766520646F7A656E206C697175BA
:100160006F72206A756773005061636B206D792030
:10017000626F782077697468206669766520646F9D
:100180007A656E206C6971756F72206A756773008D
:100190005061636B206D7920626F78207769746895
:1001A000206669766520646F7A656E206C6971756A
:1001B0006F72206A756773005061636B206D7920E0
:1001C000626F782077697468206669766520646F4D
:1001D0007A656E206C6971756F72206A756773003D
:1001E0005061636B206D7920626F78207769746845
:1001F000206669766520646F7A656E206C6971751A
:100200006F72206A756773005061636B206D79208F
:10021000626F782077697468206669766520646FFC
:100220007A656E206C6971756F72206A75677300EC
:100230005061636B206D7920626F782077697468F4
:10024000206669766520646F7A656E206C697175C9
:100250006F72206A756773005061636B206D79203F
:10026000626F782077697468206669766520646FAC
:100270007A656E206C6971756F72206A756773009C
:100280005061636B206D7920626F782077697468A4
:10029000206669766520646F7A656E206C69717579
:1002A0006F72206A756773005061636B206D7920EF
:1002B000626F782077697468206669766520646F5C
:1002C0007A656E206C6971756F72206A756773004C
:1002D0005061636B206D7920626F78207769746854
:1002E000206669766520646F7A656E206C69717529
:1002F0006F72206A756773005061636B206D79209F
:10030000626F782077697468206669766520646F0B
:100310007A656E206C6971756F72206A75677300FB
:100320005061636B206D7920626F78207769746803
:10033000206669766520646F7A656E206C697175D8
:100340006F72206A756773005061636B206D79204E
:10035000626F782077697468206669766520646FBB
:100360007A656E206C6971756F72206A75677300AB
:100370005061636B206D7920626F782077697468B3
:10038000206669766520646F7A656E206C69717588
:100390006F72206A756773005061636B206D7920FE
:1003A000626F782077697468206669766520646F6B
[... 438 lines cut ...]
:101F10006F72206A756773005061636B206D792062
:101F2000626F782077697468206669766520646FCF
:101F30007A656E206C6971756F72206A75677300BF
:101F40005061636B206D7920626F782077697468C7
:101F5000206669766520646F7A656E206C6971759C
:101F60006F72206A756773005061636B206D792012
:101F7000626F782077697468206669766520646F7F
:101F80007A656E206C6971756F72206A756773006F
:101F90005061636B206D7920626F78207769746877
:101FA000206669766520646F7A656E206C6971754C
:101FB0006F72206A756773005061636B206D7920C2
:101FC000626F782077697468206669766520646F2F
:101FD0007A656E206C6971756F72206A756773001F
:101FE0005061636B206D7920626F78207769746827
:101FF000206669766520646F7A656E206C697175FC
:00000001FF

:03000000FFFEFD03
:00000001FF

:0100000000FF
:00000001FF
```

</details>

xedbg, 2024-08-20:
Don't exclude the thought that this could be a silicon bug...

MCUdude, 2024-08-20:
> Don't exclude the thought that this could be a silicon bug...

You might be on to something! I'm getting the same result when reading the memories using an AVR Dragon.

```
$ avrdude -cdragon_jtag -patmega256rfr2 -U{eeprom,signature,calibration}:r:-:I
Warning: OCDEN fuse not programmed, single-byte EEPROM updates not possible

Processing -U eeprom:r:-:I
Reading eeprom memory ...
Reading | ################################################## | 100% 4.24 s 
Writing 8192 bytes to output file <stdout>
:200000005061636B206D7920626F782077697468206669766520646F7A656E206C69717531 // 00000> Pack my box with five dozen liqu eeprom
:200020006F72206A756773005061636B206D7920626F782077697468206669766520646F7F // 00020> or jugs.Pack my box with five do
:200040007A656E206C6971756F72206A756773005061636B206D7920626F782077697468F4 // 00040> zen liquor jugs.Pack my box with
:20006000206669766520646F7A656E206C6971756F72206A756773005061636B206D79203C // 00060>  five dozen liquor jugs.Pack my 
:20008000626F782077697468206669766520646F7A656E206C6971756F72206A756773009C // 00080> box with five dozen liquor jugs.
:2000A0005061636B206D7920626F782077697468206669766520646F7A656E206C69717591 // 000a0> Pack my box with five dozen liqu
:2000C0006F72206A756773005061636B206D7920626F782077697468206669766520646FDF // 000c0> or jugs.Pack my box with five do
:2000E0007A656E206C6971756F72206A756773005061636B206D7920626F78207769746854 // 000e0> zen liquor jugs.Pack my box with
:20010000206669766520646F7A656E206C6971756F72206A756773005061636B206D79209B // 00100>  five dozen liquor jugs.Pack my 

[...]

:201F60006F72206A756773005061636B206D7920626F782077697468206669766520646F20 // 01f60> or jugs.Pack my box with five do
:201F80007A656E206C6971756F72206A756773005061636B206D7920626F78207769746895 // 01f80> zen liquor jugs.Pack my box with
:201FA000206669766520646F7A656E206C6971756F72206A756773005061636B206D7920DD // 01fa0>  five dozen liquor jugs.Pack my 
:201FC000626F782077697468206669766520646F7A656E206C6971756F72206A756773003D // 01fc0> box with five dozen liquor jugs.
:201FE0005061636B206D7920626F782077697468206669766520646F7A656E206C69717532 // 01fe0> Pack my box with five dozen liqu
:00000001FF

Processing -U signature:r:-:I
Reading signature memory ...
Writing 3 bytes to output file <stdout>
:03000000FFFEFD03                                                           // 00000> 0xff 0xfe 0xfd signature
:00000001FF

Processing -U calibration:r:-:I
Reading calibration memory ...
Writing 1 byte to output file <stdout>
:0100000000FF                                                               // 00000> 0x00 calibration
:00000001FF
Error: bad response to leave progmode command: RSP_FAILED

Avrdude done.  Thank you.
```

xedbg, 2024-08-20:
Give me a few days to have a look at the code.  If I have any luck I can hopefully send you a powerdebugger_fw.zip for some testing... (that's an "if" though)

MCUdude, 2024-08-20:
> Give me a few days to have a look at the code. If I have any luck I can hopefully send you a powerdebugger_fw.zip for some testing... (that's an "if" though)

That would be great! But since I'm able to reproduce the issue with an AVR Dragon, doesn't this confirm that this is a silicon bug?

mcuee, 2024-08-24:
I think we can close this issue in this case. We can probably document this in the Wiki FAQ section.

closed 2024-08-26 by MCUdude

xedbg, 2024-09-19:
Fix incoming.  Sorry for the delay.
Its a firmware bug that has been there a while (hence the Dragon symptoms too).
(The high-byte of the address was not being cleared for signature reads, which meant after reading something with a higher address a subsequent read of signature probably read "something else")...