avrdudes/avrdude #1898: Optiboot bigboot ATmega2560 issue

mcuee, 2024-08-22:
Some issues using Optiboot from @MCUdude's repo (Bigboot).
https://github.com/MCUdude/optiboot_flash/blob/master/bootloaders/atmega2560/16000000L/optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex
```
$ ./test-avrdude -e ../build_mingw64_nt-10.0-19045/src/avrdude.exe -d 1 -t . -p "-c arduino -P COM4 -p m2560"
Testing ../build_mingw64_nt-10.0-19045/src/avrdude.exe 7.3-20240822 (a9594622)
Prepare "-c arduino -P COM4 -p m2560" and press 'enter' or 'space' to continue. Press any other key to skip
✅   1.557 s: chip erase
✅  12.195 s: flash -U write/verify holes_rjmp_loops_262144B.hex
❌  54.405 s: flash -T write/verify holes_rjmp_loops_262144B.hex (failed command below)
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 -T "write flash ./test_files/holes_rjmp_loops_262144B.hex:a"
❌  54.449 s: flash -T write/verify rjmp_loops_for_bootloaders_262144B.hex (failed command below)
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 -T "write flash ./test_files/rjmp_loops_for_bootloaders_262144B.hex:a"
❌  54.393 s: eeprom check whether programmer can flip 0s to 1s (failed command below)
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 -Ueeprom:w:0x55:m -Ueeprom:w:0xaa:m
# ... the next test may therefore take longer
❌  54.445 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_4096B.hex (failed command below)
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 -T "write eeprom ./test_files/holes_the_five_boxing_wizards_4096B.hex:a" -T flush -T "write eeprom ./test_files/holes_pack_my_box_4096B.hex:a"
❌  54.461 s: eeprom -T write/verify lorem_ipsum_4096B.srec (failed command below)
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 -T "write eeprom ./test_files/lorem_ipsum_4096B.srec:a"
❌  54.449 s: chip erase and spot check flash is actually erased (failed command below)
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 -e -FAU flash:w:0xff:m -U flash:v:./test_files/holes_flash_0xff_262144B.hex

One or more AVRDUDE "-c arduino -P COM4 -p m2560" tests failed. Do you want to retry this particular test? (y/n): n
```
More detailed output.
https://github.com/avrdudes/avrdude/discussions/1654#discussioncomment-10405653

Same problems with avrdude7.3 release.
```
$ ./test-avrdude -e ./avrdude73.exe -c "-C ./avrdude73.conf" -d 1 -t . -p "-c arduino -P COM4 -p m2560"
Testing ./avrdude73.exe 7.3
Prepare "-c arduino -P COM4 -p m2560" and press 'enter' or 'space' to continue. Press any other key to skip
✅   1.464 s: chip erase
✅  12.022 s: flash -U write/verify holes_rjmp_loops_262144B.hex
❌  54.031 s: flash -T write/verify holes_rjmp_loops_262144B.hex (failed command below)
$ ./avrdude73.exe -qq -c arduino -P COM4 -p m2560 -T "write flash ./test_files/holes_rjmp_loops_262144B.hex:a"
❌  54.037 s: flash -T write/verify rjmp_loops_for_bootloaders_262144B.hex (failed command below)
$ ./avrdude73.exe -qq -c arduino -P COM4 -p m2560 -T "write flash ./test_files/rjmp_loops_for_bootloaders_262144B.hex:a"
❌  54.040 s: eeprom check whether programmer can flip 0s to 1s (failed command below)
$ ./avrdude73.exe -qq -c arduino -P COM4 -p m2560 -Ueeprom:w:0x55:m -Ueeprom:w:0xaa:m
# ... the next test may therefore take longer
❌  54.081 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_4096B.hex (failed command below)
$ ./avrdude73.exe -qq -c arduino -P COM4 -p m2560 -T "write eeprom ./test_files/holes_the_five_boxing_wizards_4096B.hex:a" -T flush -T "write eeprom ./test_files/holes_pack_my_box_4096B.hex:a"
❌  54.053 s: eeprom -T write/verify lorem_ipsum_4096B.srec (failed command below)
$ ./avrdude73.exe -qq -c arduino -P COM4 -p m2560 -T "write eeprom ./test_files/lorem_ipsum_4096B.srec:a"
❌  54.040 s: chip erase and spot check flash is actually erased (failed command below)
$ ./avrdude73.exe -qq -c arduino -P COM4 -p m2560 -e -FAU flash:w:0xff:m -U flash:v:./test_files/holes_flash_0xff_262144B.hex

One or more AVRDUDE "-c arduino -P COM4 -p m2560" tests failed. Do you want to retry this particular test? (y/n): n
```

mcuee, 2024-08-22:
So this is not a regression.

mcuee, 2024-08-22:
Debugging the issue by reflashing the bootloader.
So you can see the first command works. But then after that the bootloader communication will be lost.

```
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -c arduino -P COM4 -p m2560 -T "write flash ./test_files/holes_rjmp_loops_262144B.hex:a"
Caching | ################################################## | 100% 5.11 s
Synching cache to device ...
Writing | ################################################## | 100% 11.21 s

Avrdude done.  Thank you.

$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -c arduino -P COM4 -p m2560 -v
Avrdude version 7.3-20240822 (a9594622)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is C:\work\avr\avrdude_test\avrdude_main\build_mingw64_nt-10.0-19045\src\avrdude.conf

Using port            : COM4
Using programmer      : arduino
Error: programmer is not responding
Warning: attempt 1 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 2 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 3 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 4 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 5 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 6 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 7 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 8 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 9 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 10 of 10: not in sync: resp=0x00
Error: unable to open port COM4 for programmer arduino
Possible candidate serial ports are:
  -P COM3
  -P COM4 or -P ch340
Note that above ports might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.

Avrdude done.  Thank you.
```

mcuee, 2024-08-22:
@MCUdude 
Just wondering if you can reproduce the issue on your side or not. Thanks.

@stefanrueger 
Not so sure if this is a Optiboot bootloader limitation or not.

mcuee, 2024-08-22:
Bootloader dump to see if it is corrupted or not.
```
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -c usbasp -p m2560 -t
avrdude> dump flash -0x400
Reading | ################################################## | 100% 0.47 s
3fc00  02 c0 f4 c0 3c c1 11 24  84 b7 90 e8 90 93 61 00  |....<..$......a.|
3fc10  10 92 61 00 88 23 69 f0  98 2f 9a 70 92 30 49 f0  |..a..#i../.p.0I.|
3fc20  81 ff 02 c0 97 ef 94 bf  28 2e 80 e0 02 d1 0c 94  |........(..... .|
3fc30  00 00 85 e0 80 93 81 00  82 e0 80 93 c0 00 88 e1  |................|
3fc40  80 93 c1 00 80 e1 80 93  c4 00 86 e0 80 93 c2 00  |................|
3fc50  8e e0 ef d0 27 9a 84 e0  20 e3 3c ef 91 e0 30 93  |....'... .<...0.|
3fc60  85 00 20 93 84 00 96 bb  b0 9b fe cf 1f 9a a8 95  |.. .............|
3fc70  40 91 c0 00 47 fd 02 c0  81 50 89 f7 ce d0 81 34  |@...G....P.....4|
3fc80  79 f4 cb d0 c8 2f db d0  c2 38 11 f4 80 e0 04 c0  |y..../...8......|
3fc90  88 e0 c1 38 09 f0 83 e0  b9 d0 80 e1 b7 d0 ee cf  |...8 ...........|
3fca0  82 34 19 f4 84 e1 d3 d0  f8 cf 85 34 11 f4 85 e0  |.4.........4....|
3fcb0  fa cf 85 35 81 f4 b1 d0  e8 2e af d0 f8 2e 87 ff  |...5............|
3fcc0  07 c0 8b b7 81 60 8b bf  ee 0c ff 1c b8 d0 e5 cf  |.....`... ......|
3fcd0  8b b7 8e 7f f8 cf 86 35  79 f4 9f d0 8d 34 51 f4  |.......5y....4Q.|
3fce0  9c d0 cb b7 9a d0 c1 70  88 0f 8c 2b 8b bf 81 e0  |.......p...+....|
3fcf0  ae d0 cc cf 83 e0 fc cf  84 36 09 f0 46 c0 8d d0  |.........6 .F...|
avrdude> dump
Reading | ################################################## | 100% 0.46 s
3fd00  c8 2f d0 e0 dc 2f cc 27  88 d0 c8 2b 86 d0 d8 2e  |./.../.'...+....|
3fd10  5e 01 8e ef b8 1a 00 e0  12 e0 48 01 ef ef 8e 1a  |^.........H.....|
3fd20  9e 0a 7b d0 f8 01 80 83  84 01 8a 14 9b 04 a9 f7  |. {.............|
3fd30  86 d0 f5 e4 10 e0 00 e0  df 16 09 f1 50 e0 40 e0  |.......... .P.@.|
3fd40  63 e0 c7 01 53 d0 87 01  c1 2c 92 e0 d9 2e f6 01  |c...S....,......|
3fd50  41 91 51 91 6f 01 61 e0  c8 01 48 d0 0e 5f 1f 4f  |A.Q.o.a...H.._.O|
3fd60  22 97 a9 f7 50 e0 40 e0  65 e0 c7 01 3f d0 95 cf  |"...P.@.e...?...|
3fd70  60 81 c8 01 8e 0d 9f 1d  01 d1 0f 5f 1f 4f f8 01  |`.... ....._.O..|
3fd80  fe 5f c0 17 d1 07 a1 f7  88 cf 84 37 01 f5 45 d0  |._.........7..E.|
3fd90  c8 2f d0 e0 dc 2f cc 27  40 d0 c8 2b 3e d0 d8 2e  |./.../.'@..+>...|
3fda0  4e d0 87 01 f5 e4 df 12  0b c0 ce 0d df 1d c8 01  |N....... .. ....|
3fdb0  dd d0 2c d0 0f 5f 1f 4f  c0 17 d1 07 c1 f7 6d cf  |..,.._.O......m.|
3fdc0  f8 01 87 91 8f 01 22 d0  21 97 d1 f7 66 cf 85 37  |......".!...f..7|
3fdd0  39 f4 35 d0 8e e1 1a d0  88 e9 18 d0 81 e0 5c cf  |9.5...........\.|
3fde0  81 35 09 f0 73 cf 88 e0  24 d0 70 cf fc 01 0a 01  |.5 .s...$.p... .|
3fdf0  67 bf e8 95 11 24 07 b6  00 fc fd cf 66 70 29 f0  |g....$......fp).|
avrdude> dump
Reading | ################################################## | 100% 0.45 s
3fe00  45 2b 19 f4 81 e1 87 bf  e8 95 08 95 90 91 c0 00  |E+..............|
3fe10  95 ff fc cf 80 93 c6 00  08 95 80 91 c0 00 87 ff  |................|
3fe20  fc cf 80 91 c0 00 84 fd  01 c0 a8 95 80 91 c6 00  |................|
3fe30  08 95 e0 e6 f0 e0 98 e1  90 83 80 83 08 95 ed df  |................|
3fe40  80 32 19 f0 88 e0 f5 df  ff cf 84 e1 df cf cf 93  |.2..............|
3fe50  c8 2f e3 df c1 50 e9 f7  cf 91 f1 cf cf 92 df 92  |./...P..........|
3fe60  ef 92 ff 92 dc 01 cb 01  64 2f 6d 01 ee 24 ff 24  |........d/m..$.$|
3fe70  cb be a9 01 ff 90 ef 90  df 90 cf 90 b7 cf 2f 92  |............../.|
3fe80  3f 92 4f 92 5f 92 6f 92  7f 92 8f 92 9f 92 af 92  |?.O._.o.........|
3fe90  bf 92 cf 92 df 92 ef 92  ff 92 0f 93 1f 93 cf 93  |................|
3fea0  df 93 00 d0 cd b7 de b7  18 01 e9 82 6b 01 7c 01  |............k.|.|
3feb0  10 e0 00 e0 29 01 3a 01  4c 18 5d 08 6e 08 7f 08  |....).:.L.].n...|
3fec0  46 01 57 01 84 0c 95 1c  a6 1c b7 1c 02 15 13 05  |F.W.. ..........|
3fed0  31 f4 89 81 88 23 a1 f1  88 e0 ab df ff cf 30 e0  |1....#........0.|
3fee0  20 e0 43 e0 c7 01 b6 01  b9 df 80 e8 90 e0 9b 83  | .C.............|
3fef0  8a 83 ab be f4 01 27 91  36 91 c5 01 b4 01 64 19  |......'.6.....d.|
avrdude> dump
Reading | ################################################## | 100% 0.47 s
3ff00  75 09 86 09 97 09 41 e0  a9 df 8a 81 9b 81 01 97  |u . . A.........|
3ff10  9b 83 8a 83 92 e0 89 0e  91 1c a1 1c b1 1c 8a 81  |................|
3ff20  9b 81 89 2b 31 f7 30 e0  20 e0 45 e0 c7 01 b6 01  |...+1.0. .E.....|
3ff30  95 df 9f ef d9 1a e9 0a  f9 0a 0f 5f 1f 4f c0 cf  |....... . ._.O..|
3ff40  0f 90 0f 90 0f 90 df 91  cf 91 1f 91 0f 91 ff 90  |................|
3ff50  ef 90 df 90 cf 90 bf 90  af 90 9f 90 8f 90 7f 90  |................|
3ff60  6f 90 5f 90 4f 90 3f 90  2f 90 08 95 f9 99 fe cf  |o._.O.?./.......|
3ff70  92 bd 81 bd f8 9a 99 27  80 b5 08 95 26 2f f9 99  |.......'....&/..|
3ff80  fe cf 1f ba 92 bd 81 bd  20 bd 0f b6 f8 94 fa 9a  |........ .......|
3ff90  f9 9a 0f be 01 96 08 95  ff ff ff ff ff ff ff ff  |................|
3ffa0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3ffb0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3ffc0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3ffd0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3ffe0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
3fff0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff 00 08  |................|
```

mcuee, 2024-08-22:
Re-flashing the bootloader and the EEPROM related tests are ok.
```
MINGW64 /c/work/avr/avrdude_test/avrdude_main/tools
$ ./avrdude73.exe -C ./avrdude73.conf -c usbasp -p m2560 -U ./optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex
avrdude73: AVR device initialized and ready to accept instructions
avrdude73: device signature = 0x1e9801 (probably m2560)
avrdude73: Note: flash memory has been specified, an erase cycle will be performed.
           To disable this feature, specify the -D option.
avrdude73: erasing chip

avrdude73: processing -U flash:w:./optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex:i
avrdude73: reading input file ./optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex for flash
           with 922 bytes in 2 sections within [0x3fc00, 0x3ffff]
           using 4 pages and 102 pad bytes
avrdude73: writing 922 bytes flash ...
Writing | ################################################## | 100% 0.04 s
avrdude73: 922 bytes of flash written
avrdude73: verifying flash memory against ./optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex
Reading | ################################################## | 100% 0.00 s
avrdude73: 922 bytes of flash verified

avrdude73 done.  Thank you.

$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 -Ueeprom:w:0x55:m -Ueeprom:w:0xaa:m

$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 -Ueeprom:w:0x55:m -Ueeprom:w:0xaa:m && echo OK
OK

$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 -T "write eeprom ./test_files/holes_the_five_boxing_wizards_4096B.hex:a" -T flush -T "write eeprom ./test_files/holes_pack_my_box_4096B.hex:a" && echo OK
OK

$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 -T "write eeprom ./test_files/lorem_ipsum_4096B.srec:a" && echo OK
OK

$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 -e -FAU flash:w:0xff:m -U flash:v:./test_files/holes_flash_0xff_262144B.hex && echo OK
OK

$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 && echo OK
OK
```

mcuee, 2024-08-22:
The issue is with the flash related command.

After the flash related test item, the bootloader will be gone.

```
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 -T "write flash ./test_files/rjmp_loops_for_bootloaders_262144B.hex:a" &&
 echo OK
OK

$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 && echo OK
Error: programmer is not responding
Warning: attempt 1 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 2 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 3 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 4 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 5 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 6 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 7 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 8 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 9 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 10 of 10: not in sync: resp=0x00
Error: unable to open port COM4 for programmer arduino
Possible candidate serial ports are:
  -P COM3
  -P COM4 or -P ch340
Note that above ports might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.
```

mcuee, 2024-08-22:
> The issue is with the flash related command.
> 
> After the flash related test item, the bootloader will be gone.

Same here.
```
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 && echo OK
OK

$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 -Uflash:w:./test_files/holes_rjmp_loops_262144B.hex && echo OK
OK

$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -P COM4 -p m2560 && echo OK
Error: programmer is not responding
Warning: attempt 1 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 2 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 3 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 4 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 5 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 6 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 7 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 8 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 9 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 10 of 10: not in sync: resp=0x00
Error: unable to open port COM4 for programmer arduino
Possible candidate serial ports are:
  -P COM3
  -P COM4 or -P ch340
Note that above ports might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.
```

MCUdude, 2024-08-22:
Try adding the `-D` flag

mcuee, 2024-08-22:
> Try adding the `-D` flag

Same problem.

```
$ ./test-avrdude -e ../build_mingw64_nt-10.0-19045/src/avrdude.exe -d 1 -t . -p "-c arduino -D -P COM4 -p m2560"
Testing ../build_mingw64_nt-10.0-19045/src/avrdude.exe 7.3-20240822 (a9594622)
Prepare "-c arduino -D -P COM4 -p m2560" and press 'enter' or 'space' to continue. Press any other key to skip
✅   1.537 s: chip erase
✅  12.228 s: flash -U write/verify holes_rjmp_loops_262144B.hex
❌  54.308 s: flash -T write/verify holes_rjmp_loops_262144B.hex (failed command below)
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -D -P COM4 -p m2560 -T "write flash ./test_files/holes_rjmp_loops_262144B.hex:a"
❌  54.304 s: flash -T write/verify rjmp_loops_for_bootloaders_262144B.hex (failed command below)
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -D -P COM4 -p m2560 -T "write flash ./test_files/rjmp_loops_for_bootloaders_262144B.hex:a"
❌  54.359 s: eeprom check whether programmer can flip 0s to 1s (failed command below)
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -D -P COM4 -p m2560 -Ueeprom:w:0x55:m -Ueeprom:w:0xaa:m
# ... the next test may therefore take longer
❌  54.350 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_4096B.hex (failed command below)
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -D -P COM4 -p m2560 -T "write eeprom ./test_files/holes_the_five_boxing_wizards_4096B.hex:a" -T flush -T "write eeprom ./test_files/holes_pack_my_box_4096B.hex:a"
❌  54.326 s: eeprom -T write/verify lorem_ipsum_4096B.srec (failed command below)
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -D -P COM4 -p m2560 -T "write eeprom ./test_files/lorem_ipsum_4096B.srec:a"
❌  54.345 s: chip erase and spot check flash is actually erased (failed command below)
$ ../build_mingw64_nt-10.0-19045/src/avrdude.exe -qq -c arduino -D -P COM4 -p m2560 -e -FAU flash:w:0xff:m -U flash:v:./test_files/holes_flash_0xff_262144B.hex

One or more AVRDUDE "-c arduino -D -P COM4 -p m2560" tests failed. Do you want to retry this particular test? (y/n): n
```

mcuee, 2024-08-22:
Probably the bootloader can not handle hex file with holes for Flash.

stefanrueger, 2024-08-22:
> Probably the bootloader can not handle hex file with holes for Flash.

Looks like it. `test-avrdude` is designed to test this with difficult files. If only there was a better bootloader that people could use!

mcuee, 2024-08-22:
Okay, I will close this issue.

mcuee, 2024-08-22:
> > Probably the bootloader can not handle hex file with holes for Flash.
> 
> Looks like it. `test-avrdude` is designed to test this with difficult files. If only there was a better bootloader that people could use!

No worries.

@MCUdude has switched MegaCore to use urboot. The tests for Wiring Bootloader and Optiboot Bootloader for ATmega2560 are just for testing purpose to see if there are regressions.And there are no regressions. So we are good.

mcuee, 2024-08-22:
We also have the recommended bootloader section in Wiki FAQ.
https://github.com/avrdudes/avrdude/wiki/FAQ#i-have-got-the-programmer-working-now-i-want-to-try-using-bootloader-which-bootloader-should-i-use

_We recommend urboot for classic AVR chips.

For xmega, you may want to try xboot (AVR109 based)

For UPDI parts, you may want to try optiboot_x and optiboot_dx.

If you use USB AVRs, you may want to try Caterina (AVR109 based) or FLIP1/FLIP2 (Atmel extension to USB DFU)._

closed 2024-08-22 by mcuee