avrdudes/avrdude #1905: How is `version.texi` created with autotools?

stefanrueger, 2024-08-23:
I wanted avrdude.pdf to say how many parts and programmers are known to AVRDUDE, and piggy-backed this information onto the `version.texi` file in the CMake universe. I could not figure out, though, how `version.texi` is created using autotools.

dl8dtl, 2024-08-23:
```
$(srcdir)/version.texi:  $(srcdir)/stamp-vti
$(srcdir)/stamp-vti: avrdude.texi $(top_srcdir)/configure
        @(dir=.; test -f ./avrdude.texi || dir=$(srcdir); \
        set `$(SHELL) $(srcdir)/mdate-sh $$dir/avrdude.texi`; \
        echo "@set UPDATED $$1 $$2 $$3"; \
        echo "@set UPDATED-MONTH $$2 $$3"; \
        echo "@set EDITION $(VERSION)"; \
        echo "@set VERSION $(VERSION)") > vti.tmp$$$$ && \
        (cmp -s vti.tmp$$$$ $(srcdir)/version.texi \
          || (echo "Updating $(srcdir)/version.texi" && \
              cp vti.tmp$$$$ $(srcdir)/version.texi.tmp$$$$ && \
              mv $(srcdir)/version.texi.tmp$$$$ $(srcdir)/version.texi)) && \
        rm -f vti.tmp$$$$ $(srcdir)/version.texi.$$$$
        @cp $(srcdir)/version.texi $@
```
I don't know offhand, but since there is no mention in `Makefile.am`, I guess this is some automatism already built into autotools.

dl8dtl, 2024-08-23:
https://www.gnu.org/software/automake/manual/1.10/html_node/Texinfo.html

> If the .texi file @includes version.texi, then that file will be automatically generated. The file version.texi defines four Texinfo flag you can reference using @value{EDITION}, @value{VERSION}, @value{UPDATED}, and @value{UPDATED-MONTH}.

ndim, 2024-08-23:
We know how to generate parts.texi and programmers.texi and how to hook them into the build. So.... Why not generate and include a new text file?

I would generate a new data-summary.texi instead of messing with the existing version.texi generation rules. I need to check, but the version.texi generation appears to be pretty well tied into the bowels of the automake rules.

stefanrueger, 2024-08-23:
>  I need to check

Thanks. No need, though, Jörg alread did (see above) and confirmed your hunch.

And yes, `data-summary.texi` is a better name than `values.texi` that I thought of.

PR #1908 (2024-08-23): Improve Documentation
Fixes #1905, improves language around upload/download and removes long memory section in avrdude.1 that breaks the flow of the command line options section.
closed 2024-08-23 by stefanrueger