avrdudes/avrdude #1915: Missing devices in udev rules output?

ndim, 2024-08-29:
While packaging avrdude 8.0 for Fedora, I compared the old udev rules file from the package with the output of `avrdude -c '*/u'`.

One obvious thing is that there are many devices in the `avrdude -c '*/u'` list which had not been in the old rules file: Great!

However, there are also a few devices listed in the old udev file which are not in the `avrdude -c '*/u'` list (device comments from the old udev file, coming from the Debian package, exact information source unknown):

```
# at32uc3a3 DFU
SUBSYSTEM=="usb", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="2ff1", TAG+="uaccess"
# atmega16u4 DFU
SUBSYSTEM=="usb", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="2ff3", TAG+="uaccess"
# at32uc3b0/1 DFU
SUBSYSTEM=="usb", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="2ff6", TAG+="uaccess"
# at32uc3a0/1 DFU
SUBSYSTEM=="usb", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="2ff8", TAG+="uaccess"
# at89c5130/c5131 DFU
SUBSYSTEM=="usb", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="2ffd", TAG+="uaccess"
# at89c5132/c51snd1c DFU
SUBSYSTEM=="usb", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="2fff", TAG+="uaccess"
```

The common theme appears to be that these are DFU interfaces, i.e. not programmers. Not sure though why some other DFU type interfaces *are* part of the `avrdude -c '*/u'` list, and only these are not.

mcuee, 2024-08-29:
Those seem to be some odd chip except ATmega16u4.

askn37, 2024-08-30:
The list of known reserved VID:PID == 0x34b:0x2Fxx for ATMEL DFU boot programming can be easily seen in the following source code:

https://github.com/dfu-programmer/dfu-programmer/blob/master/src/arguments.c

stefanrueger, 2024-08-31:
AVRDUDE only reports udev rules for programmers it knows about. Flip1 and flip2 are DFU bootloaders, and for these the usbpid of the programmable parts is listed. AVRDUDE doesn't deal with the mentioned parts and, hence, doesn't know their pids.

Looks like the m16u4 has the wrong pid in avrdude.conf.in. I have not found an authoritative source for usbpids from Microchip (they don't seem to be in the data sheets?), so the list that @askn37 found is probably our best bet to correct the usbpids.

askn37, 2024-08-31:
> Looks like the m16u4 has the wrong pid in avrdude.conf.in.

The most reliable primary source of information about this device is the following document:

7618C–AVR–07/08 https://ww1.microchip.com/downloads/en/DeviceDoc/doc7618.pdf

PR #1919 (2024-09-01): Update usbpids in avrdude.conf
Fixes #1915 as far as `avrdude.conf` is concerned. The list in https://github.com/dfu-programmer/dfu-programmer/blob/master/src/arguments.c (thanks @askn37) was used to reliably (and semi-automatically) update avrdude.conf.

@askn37 Would you like to review? `avrdude -p '*/st' | grep usbpid` should show AVRDUDE's current understanding of `usbpid` assignments.
closed 2024-09-01 by stefanrueger