avrdudes/avrdude #1917: Error running `avrdude -c '*/u'` on s390x

ndim, 2024-08-30:
Building a package for Fedora implies not only the `x86_64` architecture, but also `s390x`.

However, running `avrdude -c '*/u'` fails there (full build log at https://kojipkgs.fedoraproject.org//work/tasks/1593/122671593/build.log).

I do not know whether this points to some portability problems in the avrdude sources which should be fixed in avrdude, or whether this just means I should not build for s390x at all.

It is obvious that nobody will use a mainframe to programm a microcontroller with, but building for that mainframe might still expose bugs in the sources. So here it is for you to draw some conclusions.

```
[...]
+ redhat-linux-build/src/avrdude -C redhat-linux-build/src/avrdude.conf -c '*/u'
Warning: invalid page size 0, ignored [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3601]
Warning: invalid page size 0, ignored [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3610]
Warning: invalid page size 0, ignored [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3618]
Warning: invalid page size 0, ignored [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3624]
Error: Common values for reduced core tinys's fuse memory location [0x820000, 0x81ffff] straddles memory section boundary 0x830000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3632]
Error: Common values for reduced core tinys's lockbits memory location [0x830000, 0x82ffff] straddles memory section boundary 0x840000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3632]
Error: Common values for reduced core tinys's signature memory location [0x840000, 0x83ffff] straddles memory section boundary 0x850000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3632]
Error: Common values for reduced core tinys's sram memory location [0x800000, 0x7fffff] straddles memory section boundary 0x810000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3632]
Warning: invalid page size 0, ignored [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3652]
Error: ATtiny4's fuse memory location [0x820000, 0x81ffff] straddles memory section boundary 0x830000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3660]
Error: ATtiny4's lockbits memory location [0x830000, 0x82ffff] straddles memory section boundary 0x840000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3660]
Error: ATtiny4's signature memory location [0x840000, 0x83ffff] straddles memory section boundary 0x850000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3660]
Error: ATtiny4's sram memory location [0x800000, 0x7fffff] straddles memory section boundary 0x810000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3660]
Error: ATtiny4's io memory location [0x800000, 0x7fffff] straddles memory section boundary 0x810000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3660]
Error: ATtiny5's fuse memory location [0x820000, 0x81ffff] straddles memory section boundary 0x830000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3676]
Error: ATtiny5's lockbits memory location [0x830000, 0x82ffff] straddles memory section boundary 0x840000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3676]
Error: ATtiny5's signature memory location [0x840000, 0x83ffff] straddles memory section boundary 0x850000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3676]
Error: ATtiny5's sram memory location [0x800000, 0x7fffff] straddles memory section boundary 0x810000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3676]
Error: ATtiny5's io memory location [0x800000, 0x7fffff] straddles memory section boundary 0x810000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3676]
Warning: invalid page size 0, ignored [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3696]
Error: ATtiny9's fuse memory location [0x820000, 0x81ffff] straddles memory section boundary 0x830000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3704]
Error: ATtiny9's lockbits memory location [0x830000, 0x82ffff] straddles memory section boundary 0x840000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3704]
Error: ATtiny9's signature memory location [0x840000, 0x83ffff] straddles memory section boundary 0x850000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3704]
Error: ATtiny9's sram memory location [0x800000, 0x7fffff] straddles memory section boundary 0x810000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3704]
Error: ATtiny9's io memory location [0x800000, 0x7fffff] straddles memory section boundary 0x810000
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/builddir/build/BUILD/avrdude-8.0-build/avrdude-8.0/redhat-linux-build/src/avrdude.conf:3704]
Error: ATtiny10's fuse memory location [0x820000, 0x81ffff] straddles memory section boundary 0x830000
[...]
```

mcuee, 2024-08-30:
I tend to think just ignore S390x as no one would probably use avrdude on an S390x machine.

But maybe it is interesting to find the root cause.

stefanrueger, 2024-08-31:
It will be the endianness of the machine. I noticed several places where the code relies on little-endian hosts, and added comments to that effect. We might want to make AVRDUDE become more general. Great to see that we can test on a s390x.

mcuee, 2024-09-01:
Closed as not-planned then.

ndim, 2024-09-03:
Would it make sense if the build aborted when it detects it is about to build avrdude for a non little endian system?

mcuee, 2024-09-03:
> Would it make sense if the build aborted when it detects it is about to build avrdude for a non little endian system?

Sounds like a good idea.

mcuee, 2024-09-03:
No so sure if there are still relevant embedded Linux system which use big endian system.
https://en.wikipedia.org/wiki/Endianness

> Some instruction set architectures are "bi-endian" and allow running software of either endianness; these include [Power ISA](https://en.wikipedia.org/wiki/Power_ISA), [SPARC](https://en.wikipedia.org/wiki/SPARC), ARM [AArch64](https://en.wikipedia.org/wiki/AArch64), [C-Sky](https://en.wikipedia.org/w/index.php?title=C-Sky&action=edit&redlink=1), and [RISC-V](https://en.wikipedia.org/wiki/RISC-V). [IBM AIX](https://en.wikipedia.org/wiki/IBM_AIX) and [IBM i](https://en.wikipedia.org/wiki/IBM_i) run in big-endian mode on bi-endian Power ISA; [Linux](https://en.wikipedia.org/wiki/Linux) originally ran in big-endian mode, but by 2019, IBM had transitioned to little-endian mode for Linux to ease the porting of Linux software from x86 to Power.[[25]](https://en.wikipedia.org/wiki/Endianness#cite_note-25)[[26]](https://en.wikipedia.org/wiki/Endianness#cite_note-26) SPARC has no relevant little-endian deployment, as both [Oracle Solaris](https://en.wikipedia.org/wiki/Oracle_Solaris) and Linux run in big-endian mode on bi-endian SPARC systems, and can be considered big-endian in practice. ARM, C-Sky, and RISC-V have no relevant big-endian deployments, and can be considered little-endian in practice.

mcuee, 2024-09-03:
@dl8dtl 

From the above Wikipedia entry, the only problem is with SPARC. I guess we do not need to worry about it and we can drop the support. Any objections?

But you mentioned that avrdude used to run okay on the Sun Sparc machine, right?

ndim, 2024-09-04:
Doesn't NetBSD run on just about anything which is Turing Complete?

Let me come up with a build time check, and we will see which builds start failing, for big endian systems on which avrdude probably has not been tested on for a long time.

dl8dtl, 2024-09-05:
> From the above Wikipedia entry, the only problem is with SPARC.

UltraSPARC is probably no longer of much interest for us (sadly). After Oracle buying Sun, they turned it into a sole high-price database server solution. (Sun also sold a number of lower-price devices.)

AFAICT, PowerPC(64) might still be a relevant thing so these days that is commonly big-endian. In FreeBSD, it's a Tier2 platform (like RISC-V, 32-bit ARMv7, and right now also i386 – to be phased out).

If it's possible to fix the build errors and warnings, I'd very much prefer that over dropping big-endian support.

If the s390x helps us with regular builds, the better. ;-)

dl8dtl, 2024-09-05:
Could we get a recent build log including the corresponding `avrdude.conf` (to see which line numbers it refers to)?

ndim, 2024-09-05:
@dl8dtl Stock avrdude-8.0.tar.gz s390x scratch build: https://koji.fedoraproject.org/koji/taskinfo?taskID=123000005 https://koji.fedoraproject.org/koji/taskinfo?taskID=123000024

That the build log will contain `cat avcdude.conf | nl -ba | sed 's|^|===|'` which can be reversed easily: Download the [build.log](https://kojipkgs.fedoraproject.org//work/tasks/24/123000024/build.log) file, then `sed -n 's|^=== *[1-9][0-9]*\t||p' build.log > avrdude.conf`.

I have attached both the [build.log.gz](https://github.com/user-attachments/files/16901062/build.log.gz) build log file and the [avrdude.conf.gz](https://github.com/user-attachments/files/16901061/avrdude.conf.gz) config file extracted as described above.

ndim, 2024-09-05:
Perhaps the qemu-s390x or qemu-mips64 or qemu-ppc64 or qemu-ppc-sparc64 system emulators could be helpful here with a Debian system image or something like that.

mcuee, 2024-09-06:
> AFAICT, PowerPC(64) might still be a relevant thing though these days that is commonly big-endian. In FreeBSD, it's a Tier2 platform (like RISC-V, 32-bit ARMv7, and right now also i386 – to be phased out).

Good point about PowerPC(64) for FreeBSD. Big Endian and Little Endian version are both offered but it seems to me Big Endian version is more popular.
https://www.freebsd.org/platforms/ppc

It is said that Linux on PPC64 now mostly use Little Endian now.

As for legacy MacOS PPC64, last version is Mac OS X 10.5.8 Leopard released on August 13, 2009. So maybe it does not matter any more. Even macports says that it is not supported.
https://ports.macports.org/port/avrdude/details/

mcuee, 2024-09-06:
NetBSD still lists PPC64, MIPS64 and SPARC64 as tier one support system. And it says both big endian (eb) and little endian (el) MIPS are supported.
https://wiki.netbsd.org/ports/

mcuee, 2024-09-06:
OpenBSD PPC64 and Sparc64 are probably big endian as well.
https://www.openbsd.org/plat.html

mcuee, 2024-09-06:
https://kernelnewbies.org/EndianIssues

Redhat Enterprise Linux -- using Little Endian for PowerPC

https://docs.redhat.com/en/documentation/red_hat_enterprise_linux/9/html/9.4_release_notes/architectures#architectures
AMD and Intel 64-bit architectures (x86-64-v2)
The 64-bit ARM architecture (ARMv8.0-A)
IBM Power Systems, Little Endian (POWER9)
64-bit IBM Z (z14)

https://docs.redhat.com/en/documentation/red_hat_enterprise_linux/8/html/8.10_release_notes/architectures
AMD and Intel 64-bit architectures
The 64-bit ARM architecture
IBM Power Systems, Little Endian
64-bit IBM Z

mcuee, 2026-05-30:
Summary from Claude

File | Issue | Severity | Safe pattern
-- | -- | -- | --
fileio.c fileio_mem_offset() | Pointer cast to uint32_t* for section boundaries | High — causes visible errors | Byte-shift read
fileio.c fileio_num() | memcpy(&val, buf, n) assumes LE | Medium — silent wrong data | Write LSB-first explicitly
config.c/config_gram.y | int offset from 0x820000 re-read as bytes | High — s390x crash | Use uint32_t, mask usage
avrcache.c | Generally uses safe shift pattern | Low — verify memcpy sites | Byte-shift only
jtagmkII.c, jtag3.c | Some struct-cast receive paths | Medium | Byte-shift all multi-byte fields
usbasp.c | Explicit LE packing in USB transfers | None — safe | Already correct
avrintel.c | buf2uint* helpers probably safe | Low — verify impl | Byte-shift only
CMakeLists.txt | No BE guard | Build hygiene | Add test_big_endian() check

The root cause is consistent throughout: the codebase was written assuming <code class="bg-text-200/5 border border-0.5 border-border-300 text-danger-000 whitespace-pre-wrap rounded-[0.4rem] px-1 py-px text-[0.9rem]">sizeof(int) == 4</code> and LE byte layout, which is true on every platform avrdude has historically been tested on (x86, ARM LE, macOS). The fix is mechanical but requires careful auditing — there is no single-line patch.

mcuee, 2026-05-30:
More from Claude

term.c — the good news and the remaining gap

The partially-fixed write path ✅ / ⚠️

The key passage from the fetched term.c source, in cmd_write():
```
c// Always write little endian (assume double and int have same endianness)
if(is_bigendian() && sd->size > 0 && (sd->type & STR_NUMBER))
    change_endian(sd->a, sd->size);
```

This is the one deliberate endian fix already in the codebase. When the terminal write command receives a multi-byte numeric literal (16-bit short, 32-bit long, float, double), it detects a BE host via is_bigendian() and reverses the byte order of sd->a before it goes into the write buffer. The result is that write flash 0 0x1234S produces [34 12] in flash on both LE and BE hosts — correct.

However there are three caveats worth flagging:
1. is_bigendian() is a runtime check, not a compile-time guarantee. It works, but it's a symptom of the wider problem — the fix was applied only at this one call site. If str_todata() in strutil.c (which parses the numeric literal into sd->a) uses native-endian memcpy internally, the data in sd->a is in host order, and only this single change_endian reversal corrects it before the write. Any other caller of str_todata() that passes numeric data to memory — and there are several in the fileio.c numeric format path — does not have the same guard.

2. The STR_FILE path in cmd_write is not covered. When sd->type == STR_FILE, the bytes are read from a pre-existing file and copied directly into the write buffer with no endian adjustment — which is correct, since the file already contains byte-ordered data. No issue here, just noting it to confirm it was considered.

3. The cmd_save / cmd_restore paths. These call fileio_segments() → fileio_mem(), which routes into fileio_num() for the m/d/h/o/b formats. The numeric write path in fileio_num() is the separate bug covered in the prior audit — the change_endian guard in cmd_write does not protect this path.

- cmd_dump / cmd_disasm ✅ Safe on BE
- cmd_dump calls hexdump_buf() which only reads raw bytes from the device and prints them individually. No multi-byte reconstruction, no endian exposure.
- cmd_disasm uses buf[j] | buf[j+1] << 8 to reassemble AVR opcodes from the byte buffer — that explicit byte-shift idiom is endian-safe. The disassembler tables in avr_opcodes.c are pure integer constants, not memory overlays, so they're safe too.
- cmd_send ⚠️ Minor
The send command takes raw hex bytes typed by the user and passes them to pgm->cmd(). The bytes are parsed one at a time as single-byte hex values, so no multi-byte assembly occurs here. Safe.

stefanrueger, 2026-06-13:
In the end the automated analysis using AI did not help *at all*. It is confident, but *wrong* and hallucinates code and bugs in functions that just are not there. So PR #2123 so far only addresses Endian issues that I had commented on in the code earlier.

mcuee, 2026-06-14:
I will shift the focus to testing using the container.

#2123 is in the right direction but there are still other issues -- I will post the testing results in PR #2123.
* #2123