avrdudes/avrdude #1920: `build.sh` enhancement ideas

mcuee, 2024-09-01:
1. It will be good to be able to define build option in the command line.

Something like the following.
```
extra_enable="-D BUILD_DOC=1 -D HAVE_LIBGPIO=1 -D HAVE_PARPORT=1 HAVE_LINUXSPI=1" ./build.sh
```

As of now, the above is not working, the user is expected to edit `build.sh` to use the function.

2. The above will also be good if the user wants to define installation prefix.

mcuee, 2024-09-01:
> The above will also be good if the user wants to define installation prefix.

Related discussion:
* https://github.com/avrdudes/avrdude/issues/1914#issuecomment-2313757420

mcuee, 2024-09-01:
Hmm, actually there is an option `-f` to pass the extra build options to CMAKE.

```

```

mcuee, 2024-09-01:
@ndim and @stefanrueger 

BTW, the following is the dependancy with TEX distribution I mentioned before.
```
[ 89%] Generating version.texi
[ 90%] Generating programmers.texi
[ 90%] Generating programmer_types.txt
[ 90%] Generating programmer_types.texi
[ 91%] Generating parts.texi
[ 91%] Generating avrstats.texi
[ 92%] Generating avrdude.info
[ 92%] Built target info
[ 93%] Generating avrdude.pdf
You don't have a working TeX binary (tex) installed anywhere in
your PATH, and texi2dvi cannot proceed without one.  If you want to use
this script, you'll need to install TeX (if you don't have it) or change
your PATH or TEX environment variable (if you do).  See the --help
output for more details.

For information about obtaining TeX, please see http://tug.org/texlive,
or do a web search for TeX and your operating system or distro.

On Debian you can install a working TeX system with
  apt-get install texlive
gmake[2]: *** [src/doc/CMakeFiles/pdf.dir/build.make:78: src/doc/avrdude.pdf] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:314: src/doc/CMakeFiles/pdf.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
Build failed.

```

mcuee, 2024-09-01:
Somehow the following does not work.
`./build.sh -f "-D HAVE_SWIG=0"` is supposed to disable the building of python bindings.

```
mcuee@PCRyzen5600G:~/build/avr/avrdude$ ./build.sh -f "-D HAVE_SWIG=0"
-- The C compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.43.0") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Found BISON: /usr/bin/bison (found version "3.8.2")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - not found
-- Looking for usb.h
-- Looking for usb.h - found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - not found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Found SWIG: /usr/bin/swig4.0 (found suitable version "4.2.0", minimum required is "4.0") found components: python 
-- Found Python3: /usr/bin/python3 (found version "3.12.3") found components: Interpreter Development Development.Module Development.Embed 
-- Configuration summary:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DO HAVE    libserialport
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- DO HAVE    swig+Python3
-- ----------------------
-- Configuring done (0.8s)
-- Generating done (0.0s)
-- Build files have been written to: /home/mcuee/build/avr/avrdude/build_linux

```

dl8dtl, 2024-09-01:
Well, to me, `build.sh` is mostly a tool to help getting things to work without memorizing too many complicated details.

So for things that appear to be frequently asked for, rather invent a new commandline option, and explain it in the `-h` output so people can start using it.

`-f` was merely intended to allow fine-tuning – which could perhaps also include options that eventually might deserve their own commandline flag. ;-)

stefanrueger, 2024-11-09:
What should we do about this? And who is taking on the work?

mcuee, 2024-11-10:
I think we can close this issue as the two sub issues are being discussed.

* https://github.com/avrdudes/avrdude/issues/1914
* https://github.com/avrdudes/avrdude/issues/1797

@ndim has some ideas for both issues.

closed 2024-11-10 by mcuee