avrdudes/avrdude #1923: DEFAULT_SPI_PORT appears to be undefined

ndim, 2024-09-08:
DEFAULT_SPI_PORT appears not to be defined anywhere:

  * cmake builds substitute the `@DEFAULT_SPI_PORT@` in `avrdude.conf` as the empty string.

  * automake builds keep the `@DEFAULT_SPI_PORT@` in `avrdude.conf` as it is.

  * `git grep DEFAULT_SPI_PORT` shows only one match: the line in `src/avrdude.conf.in`.

This does not appear to be intentional.

mcuee, 2024-09-08:
@ndim 

The default value is in the code since the following commit.
* https://github.com/avrdudes/avrdude/commit/dd1255b0c8fcf24e0cf49d04d5cda74785b450c9

Please also refer to PR #933.
* #933 

But I agree this can be a good improvement.

mcuee, 2024-09-08:
I have to specify the linuxspi port since the default value does not work.
https://github.com/avrdudes/avrdude/discussions/1654#discussioncomment-10413582

```
mcuee@raspberrypi400arm64:~/build/avrdude_git/tools $ ./test-avrdude -d 0 -p "-c linuxspi -P /dev/spidev0.1:/dev/gpiochip0:25 -p m328p"
Testing /usr/local/bin/avrdude 7.3-20240821 (c6d29a16)
Prepare "-c linuxspi -P /dev/spidev0.1:/dev/gpiochip0:25 -p m328p" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.182 s: fuse access: clear, set and read eesave fuse bit
✅   0.178 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.173 s: chip erase
✅   2.501 s: flash -U write/verify holes_rjmp_loops_32768B.hex
✅   2.479 s: flash -T write/verify holes_rjmp_loops_32768B.hex
✅   0.172 s: eeprom check whether programmer can flip 0s to 1s
✅   1.220 s: eeprom -U write/verify holes_pack_my_box_1024B.hex
✅   2.151 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_1024B.hex
✅   1.307 s: chip erase and spot check flash is actually erased
✅   0.263 s: spot check eeprom is erased, too
```

mcuee, 2024-09-08:
Current code: maybe it needs to specify the default reset pin number.
https://github.com/avrdudes/avrdude/blob/30e19f23324c9337951d6277e054f269690ce718/src/linuxspi.c#L146-L151

stefanrueger, 2024-11-09:
@ndim Well spotted. @mcuee Thanks for finding where the default is set in code.

I think we should replace the `@DEFAULT_SPI_PORT@` during the make process, perhaps with the default value `/dev/spidev0.0:/dev/gpiochip0` form the linuxspi.c code and refrain from setting it in the code.

mcuee, 2024-11-10:
> I think we should replace the `@DEFAULT_SPI_PORT@` during the make process, perhaps with the default value `/dev/spidev0.0:/dev/gpiochip0` form the linuxspi.c code and refrain from setting it in the code.

I agree.

merlinblack, 2025-01-05:
I hand edited my installed avrdude.conf to have the default of `dev/spidev0.0:/dev/gpiochip0`, and now it's working - as it did before I updated avrdude from v7 to v8 - on a Raspberry Pi 2B, connected to a ATmega328p :+1:

closed 2025-05-12 by stefanrueger