avrdudes/avrdude #1924: Arduino Nano, unable to program by USB

hcet14, 2024-09-14:
I want to program an “Arduino Nano” by its USB interface (FT232RL) with avrdude in “cmd.exe” under Wimdows 11. Avrdude runs on an USB stick. I don’t use root for “cmd.exe”.

avrdude -v -v -v -v -p m328p -c ft232r -P COM4 -b 57600 -D -F -U flash:r:D:\57600_cr_trigger_1.hex:i`

gives me

avrdude ft245r_program_enable() [ft245r.c:521] error: device is not responding to program enable; check connection
avrdude main() [main.c:1301] error: initialization failed, rc=-1

I started with avrdude-v8.0-windows-x64 and went back to avrdude-v7.2-windows-x64 due to an issue (sorry, don't remember which one it was). The issue was about the USB interface (FT232RL).

I can’t find help in the www. So I think, I do a stupid mistake! Any idea is appreciated.

mcuee, 2024-09-14:
The FT232R is just a USB to Serial converter, not a programmer.

If the bootloader is still present, then you can use `-c urclock` (recommended) or `-c arduino`.

mcuee, 2024-09-23:
Close for now. Please use discussion next time if you need help.