avrdudes/avrdude #1925: Dependancy to texi2dvi

mcuee, 2024-09-23:
Somehow we still have dependancy to texi2dvi when building documentation. Tested under Fedora 40.

```
rm -rf build_all_linux
cmake -D BUILD_DOC=1 -D DEBUG_CMAKE=1 -D CMAKE_BUILD_TYPE=RelWithDebInfo -D HAVE_PARPORT=1 -D HAVE_LINUXGPIO=1 -D HAVE_LINUXSPI=1 -B build_all_linux
cmake --build build_all_linux

mcuee@fedora:~/build/avr/avrdude$ sudo cmake --build build_all_linux --target install
[sudo] password for mcuee: 
[  0%] Built target conf
[ 42%] Built target libavrdude
[ 44%] Built target avrdude
[ 45%] Built target swig_avrdude_swig_compilation
[ 88%] Built target swig_avrdude
[ 88%] Built target avrdude_binaries
[ 92%] Built target info
[ 93%] Generating avrdude.pdf
Can't exec "texi2dvi": No such file or directory at /usr/bin/makeinfo line 1843.
[ 96%] Built target pdf
[100%] Built target html
Install the project...
-- Install configuration: "RelWithDebInfo"
-- Installing: /usr/local/bin/avrdude
-- Installing: /usr/local/bin/elf2tag
-- Installing: /usr/local/lib64/libavrdude.a
-- Installing: /usr/local/include/libavrdude.h
-- Installing: /usr/local/include/libavrdude-avrintel.h
-- Installing: /usr/local/etc/avrdude.conf
-- Installing: /usr/local/share/man/man1/avrdude.1
-- Installing: /usr/local/lib64/python3.12/site-packages/_swig_avrdude.so
-- Installing: /usr/local/lib64/python3.12/site-packages/swig_avrdude.py
-- Installing: /usr/local/share/avrdude
-- Installing: /usr/local/share/avrdude/about.ui
-- Installing: /usr/local/share/avrdude/adgui.ui
-- Installing: /usr/local/share/avrdude/askfuse.ui
-- Installing: /usr/local/share/avrdude/device.ui
-- Installing: /usr/local/share/avrdude/devinfo.ui
-- Installing: /usr/local/share/avrdude/help.ui
-- Installing: /usr/local/share/avrdude/loglevel.ui
-- Installing: /usr/local/share/avrdude/memories.ui
-- Installing: /usr/local/share/avrdude/programmer.ui
-- Up-to-date: /usr/local/share/avrdude
-- Installing: /usr/local/share/avrdude/adgui.py
-- Installing: /usr/local/share/avrdude/swigtest.py
-- Installing: /usr/local/bin/avrdude-gui
-- Installing: /usr/local/share/info/avrdude.info
CMake Error at src/doc/cmake_install.cmake:50 (file):
  file INSTALL cannot find
  "/home/mcuee/build/avr/avrdude/build_all_linux/src/doc/avrdude.pdf": No
  such file or directory.
Call Stack (most recent call first):
  cmake_install.cmake:52 (include)

gmake: *** [Makefile:100: install] Error 1
```

mcuee, 2024-09-23:
After installation of the texinfo-tex package, then it is okay.
```
mcuee@fedora:~/build/avr/avrdude$ sudo dnf install texinfo-tex
...
mcuee@fedora:~/build/avr/avrdude$ sudo cmake --build build_all_linux --target install
[  0%] Built target conf
[ 42%] Built target libavrdude
[ 44%] Built target avrdude
[ 45%] Built target swig_avrdude_swig_compilation
[ 88%] Built target swig_avrdude
[ 88%] Built target avrdude_binaries
[ 92%] Built target info
[ 93%] Generating avrdude.pdf
[ 96%] Built target pdf
[100%] Built target html
Install the project...
-- Install configuration: "RelWithDebInfo"
-- Up-to-date: /usr/local/bin/avrdude
-- Up-to-date: /usr/local/bin/elf2tag
-- Up-to-date: /usr/local/lib64/libavrdude.a
-- Up-to-date: /usr/local/include/libavrdude.h
-- Up-to-date: /usr/local/include/libavrdude-avrintel.h
-- Up-to-date: /usr/local/etc/avrdude.conf
-- Up-to-date: /usr/local/share/man/man1/avrdude.1
-- Up-to-date: /usr/local/lib64/python3.12/site-packages/_swig_avrdude.so
-- Up-to-date: /usr/local/lib64/python3.12/site-packages/swig_avrdude.py
-- Up-to-date: /usr/local/share/avrdude
-- Up-to-date: /usr/local/share/avrdude/about.ui
-- Up-to-date: /usr/local/share/avrdude/adgui.ui
-- Up-to-date: /usr/local/share/avrdude/askfuse.ui
-- Up-to-date: /usr/local/share/avrdude/device.ui
-- Up-to-date: /usr/local/share/avrdude/devinfo.ui
-- Up-to-date: /usr/local/share/avrdude/help.ui
-- Up-to-date: /usr/local/share/avrdude/loglevel.ui
-- Up-to-date: /usr/local/share/avrdude/memories.ui
-- Up-to-date: /usr/local/share/avrdude/programmer.ui
-- Up-to-date: /usr/local/share/avrdude
-- Up-to-date: /usr/local/share/avrdude/adgui.py
-- Up-to-date: /usr/local/share/avrdude/swigtest.py
-- Up-to-date: /usr/local/bin/avrdude-gui
-- Up-to-date: /usr/local/share/info/avrdude.info
-- Installing: /usr/local/share/doc/avrdude/avrdude.pdf
-- Installing: /usr/local/share/doc/avrdude/avrdude-html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_1.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_2.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_3.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_4.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_5.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_6.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_7.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_8.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_9.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_10.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_11.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_12.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_13.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_14.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_15.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_16.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_17.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_18.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_19.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_20.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_21.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_22.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_23.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_24.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_25.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_26.html
[... 8 lines cut ...]
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_35.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_36.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_37.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_38.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_39.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_40.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_41.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_42.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_43.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_44.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_45.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_46.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_47.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_48.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_49.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_50.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_51.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_52.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_53.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_54.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_55.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_56.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_toc.html
-- Installing: /usr/local/share/doc/avrdude/avrdude-html/avrdude_abt.html
```

mcuee, 2024-09-23:
@ndim 

I tend to think this is a bug after we remove the build of dvi.

dl8dtl, 2024-09-23:
Nope, it is used to produce the PDF.

`makeinfo --help` says:
```
…
  makeinfo --pdf foo.texi                write PDF using texi2dvi
                                                                      ^^^^^^^^
```

closed 2024-09-24 by mcuee