avrdudes/avrdude #1938: USB device with VID and PID not found

AndreiCherniaev, 2024-12-20:
On Ubuntu 24 x86_64 with [PICkit 4](https://www.microchip.com/en-us/development-tool/pg164140) doesn't works. Why?

AndreiCherniaev, 2024-12-20:
Because I forget to edit rules in my Linux operation system. There are how to fix steps.

How to build
```
sudo apt purge libusb-dev
sudo apt install flex bison libhidapi-dev
git clone https://github.com/avrdudes/avrdude && cd avrdude
./build.sh -j8
sudo cmake --build build_linux --target install
```
Let's try to use pickit4 in avrdude. Why "USB device with VID: 0x03eb and PID: 0x2177 not found"?
```
q@PC:~$ /usr/local/bin/avrdude -c pickit4 -vvvv
Avrdude version 8.0-20241213 (f1555aef)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/q/.avrduderc does not exist

Using port            : usb
Using programmer      : pickit4
Jtag3_open()
USB device with VID: 0x03eb and PID: 0x2177 not found
USB device with VID: 0x03eb and PID: 0x2178 not found
USB device with VID: 0x03eb and PID: 0x2179 not found
USB device with VID: 0x04d8 and PID: 0x9012 not found
USB device with VID: 0x04d8 and PID: 0x9017 not found
USB device with VID: 0x04d8 and PID: 0x9012 not found
USB device with VID: 0x04d8 and PID: 0x9017 not found
Error jtag3_open_common() jtag3.c 1735: no device found matching VID 0x04d8 and PID list: 0x2177, 0x2178, 0x2179
Error main() main.c 1450: unable to open port usb for programmer pickit4

Avrdude done.  Thank you.
```

But Linux can detect pickit4 as device with VID: 0x03eb and PID: 0x2177
```
q@PC:~$ lsusb -vvv -d 03eb:2177

Bus 003 Device 019: ID 03eb:2177 Atmel Corp. MPLAB PICkit 4 CMSIS-DAP
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x03eb Atmel Corp.
  idProduct          0x2177 
  bcdDevice            1.00
  iManufacturer           1 Microchip Technology Incorporated
  iProduct                2 MPLAB PICkit 4 CMSIS-DAP
  iSerial                 3 BUR225174055
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
[... 88 lines cut ...]
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
```

## Main point
Problem `USB device with VID and PID not found` comes because I forget to edit [rules](https://andreasrohner.at/posts/Electronics/How-to-fix-device-permissions-for-the-USBasp-programmer/#permanent_solution) to set like sudo chmod 666 /dev/bus/usb/**003**/**019**
Note: 003 and 019 depend of machine, see `lsusb` out

closed 2024-12-20 by AndreiCherniaev

AndreiCherniaev, 2024-12-22:
In case of problem like
```
(config 1) could not set config 1: Device or resource busy
Error: wrote -2 out of 912 bytes, err = error submitting URB: No such file or directory
Error: unable to send command to serial port
```
Try rebuild avrduude
```
sudo apt install libhidapi-dev
sudo apt purge  libusb-dev
u@PC:~/avrdude$ rm -Rf build_linux/*
u@PC:~/avrdude$ ./build.sh -j9
u@PC:~/avrdude$ sudo cmake --build build_linux --target install
```

mcuee, 2025-04-28:
This is probable the same as the following known issue.
* https://github.com/avrdudes/avrdude/issues/1221

mcuee, 2025-04-28:
By right, hidapi should be more preferred than libusb, so probably no need to purge libusb-dev.