avrdudes/avrdude #1940: Fails to build on Windows with recent clang

lazka, 2025-01-12:
```cpp
FAILED: src/CMakeFiles/libavrdude.dir/serbb_win32.c.obj 
C:\msys64\clangarm64\bin\clang.exe -DCONFIG_DIR=\"/clangarm64/etc\" -IC:/M/B/src/build-CLANGARM64/src -IC:/M/B/src/avrdude-8.0 -IC:/M/B/src/build-CLANGARM64 -IC:/M/B/src/avrdude-8.0/src -O2 -pipe -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wp,-D__USE_MINGW_ANSI_STDIO=1 -IC:/msys64/clangarm64/include/libusb-compat -O3 -DNDEBUG -std=gnu11 -Wall -Wextra -Wno-unused-parameter -Wno-format-zero-length -MD -MT src/CMakeFiles/libavrdude.dir/serbb_win32.c.obj -MF src\CMakeFiles\libavrdude.dir\serbb_win32.c.obj.d -o src/CMakeFiles/libavrdude.dir/serbb_win32.c.obj -c C:/M/B/src/avrdude-8.0/src/serbb_win32.c
C:/M/B/src/avrdude-8.0/src/serbb_win32.c:294:3: error: call to undeclared library function 'free' with type 'void (void *)'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  294 |   mmt_free(pgm->cookie);
      |   ^
C:/M/B/src/avrdude-8.0/src/avrdude.h:46:21: note: expanded from macro 'mmt_free'
   46 | #define mmt_free(p) free(p)
      |                     ^
C:/M/B/src/avrdude-8.0/src/serbb_win32.c:294:3: note: include the header <stdlib.h> or explicitly provide a declaration for 'free'
C:/M/B/src/avrdude-8.0/src/avrdude.h:46:21: note: expanded from macro 'mmt_free'
   46 | #define mmt_free(p) free(p)
```

lazka, 2025-01-14:
Sure, I'll have a look later, thanks.

stefanrueger, 2025-01-14:
Good, that way you get credit for your discovery

PR #1941 (2025-02-05): Fix missing stdlib.h include for free()
Building avrdude with current clang+mingw-w64 for arm64 on Windows leads to the following error:
serbb_win32.c:294:3: error: call to undeclared library function 'free' with type 'void (void *)'

Explicitely import stdlib.h for free() in avrdude.h where mmt_free() is defind to free() to fix this.

Fixes #1940
commit bca9fff7 (2025-02-05): Fix missing stdlib.h include for free()

Building avrdude with current clang+mingw-w64 for arm64 on Windows
leads to the following error:
serbb_win32.c:294:3: error: call to undeclared library function 'free' with type 'void (void *)'

Explicitely import stdlib.h for free() in avrdude.h
where mmt_free() is defind to free() to fix this.

Fixes #1940
closed 2025-02-05 by stefanrueger