avrdudes/avrdude #1946: Upload avrdude-gui artifacts for Windows

dl8dtl, 2025-01-30:
Right now, our Windows CI builds only upload `avrdude.exe` and `avrdude.conf`.

In order to spread the word, it would make sense to also distribute the GUI components. If I'm not mistaken, this ought to be

```
            build/src/_swig_avrdude.pyd
            build/src/avrdude-gui.bat
            build/src/swig_avrdude.py
            build/src/windows.rc
```

stefanrueger, 2025-03-04:
What's involved in uploading the avrdude-gui artifacts for Windows?

stefanrueger, 2025-03-04:
Thanks @mcuee. Is that something you or @dl8dtl can try out?

mcuee, 2025-03-04:
> Thanks [@mcuee](https://github.com/mcuee). Is that something you or [@dl8dtl](https://github.com/dl8dtl) can try out?

Unfortunately I do not know how to do this myself. I actually had multiple issues trying to build the Python bindings under Windows  MSVC and macOS when testing PR #https://github.com/avrdudes/avrdude/pull/1714. I was able to use Python virtual environment to work around the issues but not so sure how to use github actions to implement.

* https://github.com/avrdudes/avrdude/pull/1714

mcuee, 2025-03-04:
BTW, I tend to think the following two files may be Python version specific. So need to decide the version to use.

            build/src/_swig_avrdude.pyd
            build/src/swig_avrdude.py

The best is to package the whole thing including the Python dependancies into a Windows executible -- but again I do not know how to that myself -- I only know that can be done.

mcuee, 2025-03-08:
Related discussion:
* https://github.com/avrdudes/avrdude/issues/1914

mcuee, 2025-05-12:
I have posted the screenshots and the instruction how to get the GUI to work under Windows here.
* https://github.com/avrdudes/avrdude/pull/1971

mcuee, 2025-05-12:
Latest build log using avrdude git main and VS2022 Community Edition.

<details>
<summary>build log using avrdude git main and VS2022 Community Edition</summary>

```
C:\work\avr\avrdude>C:\work\python\py312venv\Scripts\Activate.ps1

C:\work\avr\avrdude>cmake -D DEBUG_CMAKE=1 -D CMAKE_SYSTEM_VERSION=11 -D USE_EXTERNAL_LIBS=1  -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_msvc
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 11.
-- Found FLEX: C:/work/utils/win_flex_bison-2.5.25/win_flex.exe (found version "2.6.4")
-- Found BISON: C:/work/utils/win_flex_bison-2.5.25/win_bison.exe (found version "3.8.2")
-- Found SWIG: C:/work/utils/swigwin-4.3.1/swig.exe (found suitable version "4.3.1", minimum required is "4.0") found components: python
-- Found Python3: C:/Program Files/WindowsApps/PythonSoftwareFoundation.Python.3.12_3.12.2800.0_x64__qbz5n2kfra8p0/python3.12.exe (found version "3.12.10") found components: Interpreter Development Development.Module Development.Embed
-- Fetching external libraries, please wait...
-- Using external library 'libelf'
-- Using external library 'libusb'
-- Using external library 'libhidapi'
-- Using external library 'libftdi'
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Windows-10.0.26100
-- CMAKE_SYSTEM: Windows-11
-- CMAKE_FIND_ROOT_PATH:
-- CMAKE_C_COMPILER: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.43.34808/bin/Hostx64/x64/cl.exe
-- CONFIG_DIR: C:/Program Files (x86)/avrdude/etc
-- EXTRA_VERSION_SUFFIX:
-- AVRDUDE_FULL_VERSION: 8.0-20250512 (127f40ee)
-- USE_EXTERNAL_LIBS: 1
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: 1
-- HAVE_LIBUSB: 1
-- HAVE_LIBUSB_1_0: HAVE_LIBUSB_1_0-NOTFOUND
-- HAVE_LIBUSB_WIN32: HAVE_LIBUSB_WIN32-NOTFOUND
-- HAVE_LIBHIDAPI: 1
-- HAVE_LIBFTDI: 1
-- HAVE_LIBFTDI1: HAVE_LIBFTDI1-NOTFOUND
-- HAVE_LIBREADLINE: 1
-- HAVE_LIBSERIALPORT: HAVE_LIBSERIALPORT-NOTFOUND
-- HAVE_LIBELF_H: 1
-- HAVE_LIBELF_LIBELF_H:
-- HAVE_USB_H:
-- HAVE_LUSB0_USB_H: 1
-- HAVE_LIBUSB_H:
-- HAVE_LIBUSB_1_0_LIBUSB_H:
-- HAVE_HIDAPI_HIDAPI_H: 1
-- LIBUSB_COMPAT_DIR:
-- LIBGPIODV2_FOUND:
-- LIBGPIODV1_6_FOUND:
-- LIBGPIOD_FOUND:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DON'T HAVE libusb_1_0
-- DO HAVE    libhidapi
-- DO HAVE    libftdi
-- DON'T HAVE libftdi1
-- DO HAVE    libreadline
-- DON'T HAVE libserialport
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- DO HAVE    swig+Python3
-- ----------------------
-- Configuring done (13.5s)
-- Generating done (0.2s)
-- Build files have been written to: C:/work/avr/avrdude/build_msvc

C:\work\avr\avrdude>cmake --build build_msvc --config RelWithDebInfo
[... 270 lines cut ...]
  wiring.c
  xbee.c
  lexer.c
  config_gram.c
  getopt.c
  Generating Code...
  Compiling...
  gettimeofday.c
  Generating Code...
  usleep.cpp
  readline.cpp
  Generating Code...
  libavrdude.vcxproj -> C:\work\avr\avrdude\build_msvc\src\RelWithDebInfo\libavrdude.lib
  Building Custom Rule C:/work/avr/avrdude/src/CMakeLists.txt
  main.c
  developer_opts.c
  whereami.c
  Generating Code...
     Creating library C:/work/avr/avrdude/build_msvc/src/RelWithDebInfo/avrdude.lib and object C:/work/avr/avrdude/buil
  d_msvc/src/RelWithDebInfo/avrdude.exp
  avrdude.vcxproj -> C:\work\avr\avrdude\build_msvc\src\RelWithDebInfo\avrdude.exe
  Building Custom Rule C:/work/avr/avrdude/CMakeLists.txt
```

</details>

mcuee, 2025-05-12:
More detailed build log but with a virtual env.

```
C:\Program Files\Microsoft Visual Studio\2022\Community>cd c:\work

c:\work>cd avr

c:\work\avr>.\mypy312venv\Scripts\activate.bat

(mypy312venv) c:\work\avr>pip list

(mypy312venv) c:\work\avr>pip install PySide6

(mypy312venv) c:\work\avr>cd avrdude

(mypy312venv) c:\work\avr\avrdude>cmake -D DEBUG_CMAKE=1 -D CMAKE_SYSTEM_VERSION=11 -D USE_EXTERNAL_LIBS=1  -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_msvc

(mypy312venv) c:\work\avr\avrdude>cmake --build build_msvc --config RelWithDebInfo
```

You can refer to the detailed run log below.
<details>
<summary>Detailed build log</summary>

```
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.13.2
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************

C:\Program Files\Microsoft Visual Studio\2022\Community>cd c:\work

c:\work>cd avr

c:\work\avr>.\mypy312venv\Scripts\activate.bat

(mypy312venv) c:\work\avr>pip install PySide6
Requirement already satisfied: PySide6 in .\mypy312venv\lib\site-packages (6.9.0)
Requirement already satisfied: shiboken6==6.9.0 in .\mypy312venv\lib\site-packages (from PySide6) (6.9.0)
Requirement already satisfied: PySide6-Essentials==6.9.0 in .\mypy312venv\lib\site-packages (from PySide6) (6.9.0)
Requirement already satisfied: PySide6-Addons==6.9.0 in .\mypy312venv\lib\site-packages (from PySide6) (6.9.0)

(mypy312venv) c:\work\avr>pip list
Package            Version
------------------ -------
pip                25.0.1
PySide6            6.9.0
PySide6_Addons     6.9.0
PySide6_Essentials 6.9.0
shiboken6          6.9.0

(mypy312venv) c:\work\avr>cd avrdude

(mypy312venv) c:\work\avr\avrdude>cmake -D DEBUG_CMAKE=1 -D CMAKE_SYSTEM_VERSION=11 -D USE_EXTERNAL_LIBS=1  -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_msvc
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 11.
-- The C compiler identification is MSVC 19.43.34808.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.43.34808/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.49.0.windows.1")
-- Found FLEX: C:/work/utils/win_flex_bison-2.5.25/win_flex.exe (found version "2.6.4")
-- Found BISON: C:/work/utils/win_flex_bison-2.5.25/win_bison.exe (found version "3.8.2")
-- Found SWIG: C:/work/utils/swigwin-4.3.1/swig.exe (found suitable version "4.3.1", minimum required is "4.0") found components: python
-- Found Python3: C:/work/avr/mypy312venv/Scripts/python.exe (found version "3.12.10") found components: Interpreter Development Development.Module Development.Embed
-- Fetching external libraries, please wait...
-- The CXX compiler identification is MSVC 19.43.34808.0
-- Detecting CXX compiler ABI info
[... 401 lines cut ...]
  -- Installing: C:/Program Files (x86)/avrdude/share/avrdude/memories.ui
  -- Installing: C:/Program Files (x86)/avrdude/share/avrdude/programmer.ui
  -- Up-to-date: C:/Program Files (x86)/avrdude/share/avrdude
  -- Installing: C:/Program Files (x86)/avrdude/share/avrdude/adgui.py
  -- Installing: C:/Program Files (x86)/avrdude/share/avrdude/swigtest.py
  -- Installing: C:/Program Files (x86)/avrdude/bin/avrdude-gui.bat
```

Now we can go back to the normal session to lauch the GUI.

```
(mypy312venv) c:\work\avr\avrdude>"c:\Program Files (x86)\avrdude\bin\avrdude-gui.bat"

(mypy312venv) c:\work\avr\avrdude>rem Wrapper around the AVRDUDE Qt / PySide GUI

(mypy312venv) c:\work\avr\avrdude>rem

(mypy312venv) c:\work\avr\avrdude>rem Prerequisites:

(mypy312venv) c:\work\avr\avrdude>rem * python3

(mypy312venv) c:\work\avr\avrdude>rem * Qt5 / PySide2, or Qt6 / PySide6

(mypy312venv) c:\work\avr\avrdude>"C:/work/avr/mypy312venv/Scripts/python.exe" "C:/Program Files (x86)/avrdude\share\avrdude\adgui.py"
```

mcuee, 2025-05-12:
Screenshot -- Dark theme --> main window is a bit challenging to read.

[image]

Need to highligt to read:

[image]

Other Windows:
[image]

[image]

[image]

[image]

mcuee, 2025-05-12:
@dl8dtl 

There is actually a bug in the above Python binding installation step.
```
  -- Installing: C:/work/avr/mypy312venv/_swig_avrdude.pyd
  -- Installing: C:/work/avr/mypy312venv/swig_avrdude.py
```

They should be installed to the following directory.
```
C:/work\avr/mypy312venv/Lib/site-packages
```

I just copy the files.
```
(mypy312venv) PS C:\work\avr> cd C:\work\avr\mypy312venv\Lib\site-packages\

(mypy312venv) PS C:\work\avr\mypy312venv\Lib\site-packages> ls

    Directory: C:\work\avr\mypy312venv\Lib\site-packages

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d----           5/12/2025  8:14 PM                __pycache__
d----           5/12/2025  8:00 PM                pip
d----           5/12/2025  8:00 PM                pip-25.0.1.dist-info
d----           5/12/2025  8:08 PM                PySide6
d----           5/12/2025  8:08 PM                PySide6_Addons-6.9.0.dist-info
d----           5/12/2025  8:08 PM                PySide6_Essentials-6.9.0.dist-info
d----           5/12/2025  8:08 PM                PySide6-6.9.0.dist-info
d----           5/12/2025  8:08 PM                shiboken6
d----           5/12/2025  8:08 PM                shiboken6-6.9.0.dist-info
-a---           5/12/2025  7:50 PM        9367552 _swig_avrdude.pdb
-a---           5/12/2025  7:50 PM       11007488 _swig_avrdude.pyd
-a---           5/12/2025  7:50 PM          21039 swig_avrdude.py
-a---           5/12/2025  7:50 PM           3891 windows.rc
```

mcuee, 2025-05-12:
Often you get the following errors. So I have not figured how to package a zip file for Windows user.

```
(mypy312venv) PS C:\work\avr\avrdude_gui\avrdude> ls

    Directory: C:\work\avr\avrdude_gui\avrdude

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d----           5/12/2025  8:56 PM                bin
d----           5/12/2025  8:56 PM                etc
d----           5/12/2025  8:56 PM                include
d----           5/12/2025  8:56 PM                lib
d----           5/12/2025  9:04 PM                python312
d----           5/12/2025  8:56 PM                share

(mypy312venv) PS C:\work\avr\avrdude_gui\avrdude> ls .\share\avrdude\

    Directory: C:\work\avr\avrdude_gui\avrdude\share\avrdude

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a---            4/5/2025  4:42 PM           6990 about.ui
-a---            4/5/2025  4:42 PM          74566 adgui.py
-a---            4/5/2025  4:42 PM           7336 adgui.ui
-a---            4/5/2025  4:42 PM           1800 askfuse.ui
-a---            4/5/2025  4:42 PM           4921 device.ui
-a---            4/5/2025  4:42 PM           7259 devinfo.ui
-a---            4/5/2025  4:42 PM           2284 help.ui
-a---            4/5/2025  4:42 PM           8270 loglevel.ui
-a---            4/5/2025  4:42 PM          41863 memories.ui
-a---            4/5/2025  4:42 PM           5846 programmer.ui
-a---            4/5/2025  4:42 PM           9014 swigtest.py

(mypy312venv) PS C:\work\avr\avrdude_gui\avrdude> ls .\bin

    Directory: C:\work\avr\avrdude_gui\avrdude\bin

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a---           5/12/2025  9:05 PM            200 avrdude-gui.bat
-a---           5/12/2025  8:22 PM       11006464 avrdude.exe
-a---            4/5/2025  4:42 PM           1469 elf2tag

(mypy312venv) PS C:\work\avr\avrdude_gui\avrdude> ls .\etc\

    Directory: C:\work\avr\avrdude_gui\avrdude\etc

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a---           5/12/2025  8:22 PM         908219 avrdude.conf

(mypy312venv) PS C:\work\avr\avrdude_gui\avrdude> python .\share\avrdude\adgui.py
Traceback (most recent call last):
  File "C:\work\avr\avrdude_gui\avrdude\share\avrdude\adgui.py", line 1784, in <module>
    main()
  File "C:\work\avr\avrdude_gui\avrdude\share\avrdude\adgui.py", line 1774, in main
    gui = adgui(sys.argv)
          ^^^^^^^^^^^^^^^
  File "C:\work\avr\avrdude_gui\avrdude\share\avrdude\adgui.py", line 568, in __init__
    (success, message) = avrdude_init()
                         ^^^^^^^^^^^^^^
  File "C:\work\avr\avrdude_gui\avrdude\share\avrdude\adgui.py", line 59, in avrdude_init
    p = pathlib.Path(d + "/avrdude.conf")
                     ~~^~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'

(mypy312venv) PS C:\work\avr\avrdude_gui\avrdude> .\bin\avrdude.exe -c urclock -P COM3 -b 115200 -p m2560 -v
Avrdude version 8.0-20250512 (127f40ee)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is C:\work\avr\avrdude_gui\avrdude\etc\avrdude.conf

Using port            : COM3
Using programmer      : urclock
Setting baud rate     : 115200
AVR part              : ATmega2560
Programming modes     : SPM, ISP, HVPP, JTAG
Programmer type       : Urclock
Description           : Urboot bootloaders using urprotocol
Protocol              : Urprotocol

AVR device initialized and ready to accept instructions
Device signature = 1E 98 01 (ATmega2560)

Avrdude done.  Thank you.
```

mcuee, 2025-05-17:
Side topic: avrdude-gui depends on PySide2 or PySide6. 

I am able to run avrdude-gui under Windows, macOS, Linux, FreeBSD and OpenBSD. 

NetBSD does not seem to have PySide2 or Pyside6.

Example info for OpenBSD 7.7.
```
bash-5.2$ uname -a
OpenBSD openbsdn4200.lan 7.7 GENERIC.MP#0 amd64
bash-5.2$ pkg_info -Q pyside6
debug-pyside6-6.8.2.1p0
debug-pyside6-tools-6.8.2.1p1
pyside6-6.8.2.1p0 (installed)
pyside6-tools-6.8.2.1p1
```

Under Windows and macOS, I will recommend using Python Virtual Environment. 

For most of the Linux distributions, using Virtual Env may not be required (but Ubuntu may be a problem). I tend to recommend using Python Virtual Env though.

It is not required to use Virtual Env for FreeBSD and OpenBSD,

mcuee, 2025-06-07:
> For Linux distributions, using Virtual Env may not be required (but Debian/Ubuntu may be a problem). I tend to recommend using Python Virtual Env though. I can only get the GUI to work under Debian 12 and Ubuntu 24.04 with virtual env.
> Note: for Debian 12, it seems to bundle with incomplete version of PySide2

Work-around for Debian 12 and Ubuntu 24.04, without using virtual env.
```
mcuee@PCRyzen5600G:~/build/avr/avrdude$ avrdude-gui 
Neither PySide6 (Qt6) nor PySide2 (Qt5) found, cannot create a GUI.

mcuee@PCRyzen5600G:~/build/avr/avrdude$ sudo apt install python3-pyside*
...
mcuee@PCRyzen5600G:~/build/avr/avrdude$ avrdude-gui 
QSocketNotifier: Can only be used with threads started with QThread
(now it will work)
```

Full run log.

<details>
<summary>Full run log for pyside2 installation to run avrdude-gui</summary>

```
mcuee@PCRyzen5600G:~/build/avr/avrdude$ avrdude-gui 
Neither PySide6 (Qt6) nor PySide2 (Qt5) found, cannot create a GUI.
mcuee@PCRyzen5600G:~/build/avr/avrdude$ sudo apt install python3-pyside*
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'python3-pyside2.qtscxml' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtwebsockets' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtlocation' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtgui' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtquickwidgets' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtsensors' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtwebchannel' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtqml' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtuitools' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtsql' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtsvg' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtquick' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtmultimedia' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtxml' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtquickcontrols2' for glob 'python3-pyside*'
Note, selecting 'python3-pyside' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtwebengine' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtopengl' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtremoteobjects' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qt3drender' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtscript' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtxmlpatterns' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtconcurrent' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtwidgets' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qt3dcore' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtwebenginewidgets' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtx11extras' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qt3dextras' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtpositioning' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtwebenginecore' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtserialport' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtdatavisualization' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qt3dinput' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtmultimediawidgets' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtprintsupport' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qt3danimation' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qthelp' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtcharts' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtnetwork' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtcore' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qtopenglfunctions' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qt3dlogic' for glob 'python3-pyside*'
Note, selecting 'python3-pyside2.qttest' for glob 'python3-pyside*'
[... 302 lines cut ...]
Setting up python3-pyside2.qtwebenginecore (5.15.13-1) ...
Setting up python3-pyside2.qtsql (5.15.13-1) ...
Setting up python3-pyside2.qtuitools (5.15.13-1) ...
Setting up python3-pyside2.qtopenglfunctions (5.15.13-1) ...
Setting up python3-pyside2.qthelp (5.15.13-1) ...
Setting up python3-pyside2.qtsvg (5.15.13-1) ...
Setting up python3-pyside2.qtcharts (5.15.13-1) ...
Setting up python3-pyside2.qtwebenginewidgets (5.15.13-1) ...
Setting up python3-pyside2.qt3dlogic (5.15.13-1) ...
Setting up python3-pyside2.qttest (5.15.13-1) ...
Setting up python3-pyside2.qtlocation (5.15.13-1) ...
Setting up python3-pyside2.qtmultimedia (5.15.13-1) ...
Setting up python3-pyside2.qtmultimediawidgets (5.15.13-1) ...
Setting up python3-pyside2.qt3drender (5.15.13-1) ...
Setting up python3-pyside2.qt3danimation (5.15.13-1) ...
Setting up python3-pyside2.qt3dextras (5.15.13-1) ...
Setting up python3-pyside2.qtquick (5.15.13-1) ...
Setting up python3-pyside2.qtquickcontrols2 (5.15.13-1) ...
Setting up python3-pyside2.qtquickwidgets (5.15.13-1) ...
Processing triggers for libc-bin (2.39-0ubuntu8.4) ...
mcuee@PCRyzen5600G:~/build/avr/avrdude$ avrdude-gui 
QSocketNotifier: Can only be used with threads started with QThread
```

</details>

mcuee, 2025-06-11:
@dl8dtl 

It seems to me the right tool to use is pyinstaller.
https://pyinstaller.org/en/stable/operating-mode.html

Just wondering if you can give it a try. My Python knowledge is very basic.

mcuee, 2025-06-21:
PR #2031 is an attempt to partially fix this issue.
* #2031

mcuee, 2025-06-21:
Results from PR #2031 
```
PowerShell 7.5.1
PS C:\work\avr\avrdude_bin\github_bin\avrdude-msvc-python_x64> ls

    Directory: C:\work\avr\avrdude_bin\github_bin\avrdude-msvc-python_x64

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-----           6/21/2025  6:29 AM       10514944 _swig_avrdude.pyd
-----           6/21/2025  6:28 AM            237 avrdude-gui.bat
-----           6/21/2025  6:28 AM         909679 avrdude.conf
-----           6/21/2025  6:28 AM          21094 swig_avrdude.py
-----           6/21/2025  6:28 AM           3891 windows.rc

PS C:\work\avr\avrdude_bin\github_bin\avrdude-msvc-python_x64>
```

Further improvements needed:
1. create the directory structure
2. archive the Python source files.
3. create a Python virtual environment, install PySide6
4. copy  _swig_avrdude.pyd and swig_avrdude.py to the right Python virtual environment site library
5. archive the whole Python virtual environment and avrdude files

Ultimate solution -- create an installer using things like PyInstaller.
https://pyinstaller.org/en/stable/usage.html

mcuee, 2025-06-23:
@dl8dtl 

Somehow I could not get the MSYS2 mingw build to work. I am thinking of disable MSYS2 mingw Python binding build for avrdude 8.1 release. Any objections? MSVC64 build will work.
* https://github.com/avrdudes/avrdude/discussions/1811#discussioncomment-13541527

mcuee, 2025-06-23:
Currently there is a bug that MSYS2 mingw64 build will not use its Python installation but rather Windows Store version of Python.

There is a fix from @ndim mentioned here.
* https://github.com/avrdudes/avrdude/issues/1914#issuecomment-2379125412

```
PS C:\work\avr\avrdude> git diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 6d034a6e..4574cec2 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -284,6 +284,9 @@ endif()
 # Find SWIG
 find_package(SWIG 4.0 COMPONENTS python)
 if(SWIG_FOUND)
+  # https://discourse.cmake.org/t/find-package-python3-is-not-finding-the-correct-python/10563/8
+  set(Python3_FIND_STRATEGY LOCATION)
+  set(Python3_FIND_UNVERSIONED_NAMES FIRST)
   find_package(Python3 COMPONENTS Interpreter Development)
   if(PYTHON3_FOUND)
     set(HAVE_SWIG 1)
```

However, after the fix, it still does not work.

mcuee, 2025-06-23:
The fix helps if I creat an virtual environment. You can see that now it detects the MSYS2 mingw64 version of Python.

```
MINGW64 /c/work/avr/avrdude
$ . /c/work/python/py312venv_mingw64/bin/activate
((py312venv_mingw64) )

$ which python
/c/work/python/py312venv_mingw64/bin/python
((py312venv_mingw64) )
RA-INT+XFCHEN@CR4FH94 MINGW64 /c/work/avr/avrdude
$ python --version
Python 3.12.10
((py312venv_mingw64) )

$ cat /c/work/python/py312venv_mingw64/pyvenv.cfg
home = C:/msys64/mingw64/bin
include-system-site-packages = false
version = 3.12.10
executable = C:/msys64/mingw64/bin/python.exe
command = C:/msys64/mingw64/bin/python.exe -m venv C:/work/python/py312venv_mingw64
((py312venv_mingw64) )

$ cat build_mingw64.sh
rm -rf build_mingw64
# cmake -G"MSYS Makefiles" -D DEBUG_CMAKE=1 -D CMAKE_VERBOSE_MAKEFILE=1 -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_mingw64
cmake -G"MSYS Makefiles" -D DEBUG_CMAKE=1 -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_mingw64
cmake --build build_mingw64
((py312venv_mingw64) )

$ sh build_mingw64.sh
-- The C compiler identification is GNU 15.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/mingw64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/msys64/usr/bin/git.exe (found version "2.49.0")
-- Found FLEX: C:/msys64/usr/bin/flex.exe (found version "2.6.4")
-- Found BISON: C:/msys64/usr/bin/bison.exe (found version "3.8.2")
-- Looking for libelf.h
-- Looking for libelf.h - found
-- Looking for libelf/libelf.h
-- Looking for libelf/libelf.h - found
-- Looking for usb.h
-- Looking for usb.h - not found
-- Looking for lusb0_usb.h
-- Looking for lusb0_usb.h - found
-- Looking for libusb.h
-- Looking for libusb.h - not found
-- Looking for libusb-1.0/libusb.h
-- Looking for libusb-1.0/libusb.h - found
-- Looking for hidapi/hidapi.h
-- Looking for hidapi/hidapi.h - found
-- Looking for ftdi_tcioflush
-- Looking for ftdi_tcioflush - found
-- Found SWIG: C:/msys64/mingw64/bin/swig.exe (found suitable version "4.2.1", minimum required is "4.0") found components: python
-- Found Python3: C:/work/python/py312venv_mingw64/bin/python3.exe (found version "3.12.10") found components: Interpreter Development Development.Module Development.Embed
-- Configuration summary:
-- ----------------------
-- CMAKE_HOST_SYSTEM: Windows-10.0.22631
-- CMAKE_SYSTEM: Windows-10.0.22631
-- CMAKE_FIND_ROOT_PATH:
-- CMAKE_C_COMPILER: C:/msys64/mingw64/bin/cc.exe
-- CONFIG_DIR: C:/Program Files (x86)/avrdude/etc
-- EXTRA_VERSION_SUFFIX:
-- AVRDUDE_FULL_VERSION: 8.0-20250621 (504267d9)
-- USE_EXTERNAL_LIBS: OFF
-- USE_LIBUSBWIN32: OFF
-- HAVE_LIBELF: C:/msys64/mingw64/lib/libelf.a
[... 107 lines cut ...]
-- HAVE_LIBUSB_1_0_LIBUSB_H: 1
-- HAVE_HIDAPI_HIDAPI_H: 1
-- LIBUSB_COMPAT_DIR: C:/msys64/mingw64/include/libusb-compat
-- LIBGPIODV2_FOUND:
-- LIBGPIODV1_6_FOUND:
-- LIBGPIOD_FOUND:
-- ----------------------
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DO HAVE    libserialport
-- DISABLED   doc
-- DISABLED   parport
-- DISABLED   linuxgpio
-- DISABLED   linuxspi
-- DO HAVE    swig+Python3
-- ----------------------
-- Configuring done (20.0s)
-- Generating done (0.1s)
-- Build files have been written to: C:/work/avr/avrdude/build_mingw64
```

mcuee, 2025-06-23:
But then there is a new issue.
```
$ cp ./build_mingw64/src/swig_avrdude.py /c/work/python/py312venv_mingw64/lib/python3.12/site-packages/
((py312venv_mingw64) )

$ cp ./build_mingw64/src/_swig_avrdude.pyd /c/work/python/py312venv_mingw64/lib/python3.12/site-packages/
((py312venv_mingw64) )

$ python ./src/python/adgui.py
Neither PySide6 (Qt6) nor PySide2 (Qt5) found, cannot create a GUI.
((py312venv_mingw64) )

$ pip install PySide6
ERROR: Could not find a version that satisfies the requirement PySide6 (from versions: none)

[notice] A new release of pip is available: 25.0.1 -> 25.1.1
[notice] To update, run: python.exe -m pip install --upgrade pip
ERROR: No matching distribution found for PySide6
((py312venv_mingw64) )

$ python.exe -m pip install --upgrade pip
Requirement already satisfied: pip in c:/work/python/py312venv_mingw64/lib/python3.12/site-packages (25.0.1)
Collecting pip
  Using cached pip-25.1.1-py3-none-any.whl.metadata (3.6 kB)
Using cached pip-25.1.1-py3-none-any.whl (1.8 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 25.0.1
    Uninstalling pip-25.0.1:
      Successfully uninstalled pip-25.0.1
Successfully installed pip-25.1.1
((py312venv_mingw64) )
```

mcuee, 2025-06-23:
One step further using the MSYS2 mingw64 Python.

```
$ pacman -S mingw-w64-x86_64-pyside6
...
$ python ./src/python/adgui.py
Traceback (most recent call last):
  File "C:/work/avr/avrdude/src/python/adgui.py", line 281, in <module>
    from PySide6.QtUiTools import QUiLoader
ImportError: DLL load failed while importing QtUiTools: The specified module could not be found.
```

mcuee, 2025-06-23:
> Basically the building of Python binding is working under MSYS2 mingw (with some work-arounds), but I can not get the GUI to work.

Test is similar to the previous test in PR #1714.
* https://github.com/avrdudes/avrdude/pull/1714#issuecomment-2014157727

```
MINGW64 /c/work/avr/avrdude
$ python -i ./src/python/swigtest.py
Found avrdude.conf in build_mingw64/src
>>> ad.set_msg_callback(msg_callback)
>>> ad.set_progress_callback(progress_callback)
>>> ad.cvar.verbose=2
>>> p = ad.locate_part(ad.cvar.part_list, 'm328p')
>>> ad.avr_initmem(p)
0
>>> pgm = start_programmer('usbasp', 'usb', p)
seen device from vendor >www.fischl.de<
seen product >USBasp<
auto set sck period
>>> m = ad.avr_locate_mem(p, 'signature')
>>> ad.avr_read_mem(pgm, p, m
... )
3eading  66 %
>>> ad.avr_read_mem(pgm, p, m)
3eading  66 %
>>> m.get(3)
b'\x1e\x95\x0f'
>>> m.get(3) == p.signature
True
>>> ad.fileio(ad.FIO_WRITE, "test.hex", ad.fileio_format("i"), p, "flash", -1)
32768
>>> stop_programmer(pgm)
>>> exit()
```

felias-fogg, 2025-09-24:
I just tried it out and stumbled over the following issues:
1. I had to set `Set-ExecutionPolicy Bypass` in order to activate the virtual environment.
2. When trying to start the program, I got: "did not find executable at 'C:\Users\Admin\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.13_qbz5n2kfra8p0\python.exe'"

I am not a Windows power user (actually, I usually refrain from using this system), but it seems that the Python interpreter has not been shipped with this package. And even after installing the Windows Store version of Python, nothing works. 

I use [PyInstaller](https://pyinstaller.org/en/stable/) to generate self-contained Python script executables, which usually works quite well. You only have to make sure not to generate a single file for Windows, because Windows thinks that this is a virus. So, use the default setting and create an executable and a directory.

stefanrueger, 2025-09-25:
> use the default setting and create an executable and a directory

@felias-fogg Thanks for the tip

 @mcuee Is that something that works on your various windows, and if so, could you try creating a PR?

mcuee, 2025-09-26:
> BTW, I have never encountered the issues mentioned by [@felias-fogg](https://github.com/felias-fogg). I am using Windows Store version of Python. Then I generate the virtual environment under VS2022 build environment (PowerShell or CMD).
> 
> Basically I can compile the Python bindings and GUI without issues under Windows, but I have no ideas how to use github actions to automate the process.

Please refer to the earlier comment on the process.
* https://github.com/avrdudes/avrdude/issues/1946#issuecomment-2872393037

mcuee, 2025-09-26:
The following is the full build log for a refresh installation of Python 3.13 virtual env.

```
PS C:\work\avr\avrdude> git pull
Already up to date.
PS C:\work\avr\avrdude> C:\work\python\py313venv_msstore\Scripts\Activate.ps1
(py313venv_msstore) PS C:\work\avr\avrdude> pip list
Package Version
------- -------
pip     25.1.1

(py313venv_msstore) PS C:\work\avr\avrdude> pip install PySide6
Collecting PySide6
  Downloading pyside6-6.9.2-cp39-abi3-win_amd64.whl.metadata (5.5 kB)
Collecting shiboken6==6.9.2 (from PySide6)
  Downloading shiboken6-6.9.2-cp39-abi3-win_amd64.whl.metadata (2.5 kB)
Collecting PySide6_Essentials==6.9.2 (from PySide6)
  Downloading pyside6_essentials-6.9.2-cp39-abi3-win_amd64.whl.metadata (3.8 kB)
Collecting PySide6_Addons==6.9.2 (from PySide6)
  Downloading pyside6_addons-6.9.2-cp39-abi3-win_amd64.whl.metadata (4.1 kB)
Downloading pyside6-6.9.2-cp39-abi3-win_amd64.whl (564 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 564.6/564.6 kB 14.5 MB/s eta 0:00:00
Downloading pyside6_addons-6.9.2-cp39-abi3-win_amd64.whl (160.2 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 160.2/160.2 MB 75.9 MB/s eta 0:00:00
Downloading pyside6_essentials-6.9.2-cp39-abi3-win_amd64.whl (72.6 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 72.6/72.6 MB 39.6 MB/s eta 0:00:00
Downloading shiboken6-6.9.2-cp39-abi3-win_amd64.whl (1.2 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 20.2 MB/s eta 0:00:00
Installing collected packages: shiboken6, PySide6_Essentials, PySide6_Addons, PySide6
Successfully installed PySide6-6.9.2 PySide6_Addons-6.9.2 PySide6_Essentials-6.9.2 shiboken6-6.9.2

[notice] A new release of pip is available: 25.1.1 -> 25.2
[notice] To update, run: python.exe -m pip install --upgrade pip

(py313venv_msstore) PS C:\work\avr\avrdude> python.exe -m pip install --upgrade pip
Requirement already satisfied: pip in c:\work\python\py313venv_msstore\lib\site-packages (25.1.1)
Collecting pip
  Downloading pip-25.2-py3-none-any.whl.metadata (4.7 kB)
Downloading pip-25.2-py3-none-any.whl (1.8 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.8/1.8 MB 43.1 MB/s eta 0:00:00
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 25.1.1
    Uninstalling pip-25.1.1:
      Successfully uninstalled pip-25.1.1
Successfully installed pip-25.2

(py313venv_msstore) PS C:\work\avr\avrdude> cmake -D DEBUG_CMAKE=1 -D CMAKE_SYSTEM_VERSION=11 -D USE_EXTERNAL_LIBS=1  -D CMAKE_BUILD_TYPE=RelWithDebInfo -B build_msvc
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.26100.0 to target Windows 11.
-- The C compiler identification is MSVC 19.44.35211.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.44.35207/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.51.0.windows.1")
-- Found FLEX: C:/ProgramData/chocolatey/bin/win_flex.exe (found version "2.6.3")
-- Found BISON: C:/ProgramData/chocolatey/bin/win_bison.exe (found version "2.7")
-- Found SWIG: C:/work/python/swigwin-4.2.1/swig.exe (found suitable version "4.2.1", minimum required is "4.0") found components: python
-- Found Python3: C:/work/python/py313venv_msstore/Scripts/python.exe (found version "3.13.7") found components: Interpreter Development Development.Module Development.Embed
-- Fetching external libraries, please wait...
-- The CXX compiler identification is MSVC 19.44.35211.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.44.35207/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using external library 'libelf'
-- Using external library 'libusb'
[... 330 lines cut ...]
  usbtiny.c
  update.c
  wiring.c
  xbee.c
  lexer.c
  config_gram.c
  Generating Code...
  Compiling...
  getopt.c
  gettimeofday.c
  Generating Code...
  usleep.cpp
  readline.cpp
  Generating Code...
  libavrdude.vcxproj -> C:\work\avr\avrdude\build_msvc\src\RelWithDebInfo\libavrdude.lib
  Building Custom Rule C:/work/avr/avrdude/src/CMakeLists.txt
  main.c
  developer_opts.c
  whereami.c
  Generating Code...
     Creating library C:/work/avr/avrdude/build_msvc/src/RelWithDebInfo/avrdude.lib and object C:/work/avr/avrdude/buil
  d_msvc/src/RelWithDebInfo/avrdude.exp
  avrdude.vcxproj -> C:\work\avr\avrdude\build_msvc\src\RelWithDebInfo\avrdude.exe
  Building Custom Rule C:/work/avr/avrdude/CMakeLists.txt
```

mcuee, 2025-09-26:
After the above build process, I can manually copy the Python binding and then manual run the Python GUI program. I prefer this method than the half-working installation method.

```
(py313venv_msstore) PS C:\work\avr\avrdude> cp .\build_msvc\src\RelWithDebInfo\_swig_avrdude.pyd C:\work\python\py313venv_msstore\Lib\site-packages\
(py313venv_msstore) PS C:\work\avr\avrdude> cp .\build_msvc\src\swig_avrdude.py C:\work\python\py313venv_msstore\Lib\site-packages\
(py313venv_msstore) PS C:\work\avr\avrdude> cp .\build_msvc\src\windows.rc C:\work\python\py313venv_msstore\Lib\site-packages\
(py313venv_msstore) PS C:\work\avr\avrdude> python.exe .\src\python\adgui.py
```

mcuee, 2026-06-14:
@dl8dtl and @stefanrueger 

Just wondering if you are okay to close this issue or want to keep this issue open.

The inital goal of @dl8dtl is to archive the created Python GUI binding related files, that goal has been there for a while.
https://github.com/avrdudes/avrdude/blob/main/.github/workflows/build.yml
```
      - name: Archive executables
        uses: actions/upload-artifact@v7
        with:
          name: avrdude-msvc-${{matrix.arch}}
          path: |
            build/src/avrdude.exe
            build/src/avrdude.pdb
            build/src/avrdude.conf
      - name: Move executables
        if: matrix.arch == 'x64'
        run: |
          mv build/src/${{env.BUILD_TYPE}}/_swig_avrdude.pyd build/src
      - name: Archive Python related files for x64
        if: matrix.arch == 'x64'
        uses: actions/upload-artifact@v7
        with:
          name: avrdude-msvc-python_x64
          path: |        
            build/src/avrdude.conf
            build/src/_swig_avrdude.pyd
            build/src/swig_avrdude.py
            build/src/windows.rc
            build/src/avrdude-gui.bat
```

Then I would like to create an installer for Windows to make things easier. That is much more difficult but I have some success as documented in this issue and the repo here.
https://github.com/mcuee/avrdude_python_gui_packing

I think it is possible to add the files in this repo and then use github action to build the installer. The first thing is to update adgui.py though.

If that is the goal, then we can keep this issue open.