avrdudes/avrdude #1953: AVRDUDE 8.0 can't see the programmer running the usbasp improved firmware v1.11 under Windows 11

TzOk83, 2025-02-21:
I can't make USBasp running usbasp improved firmware v1.11 to work under Windows 11.
I keep getting errors like this:
`>>>: avrdude.exe -c usbasp-clone -P usb -p m8
WARNING: Unable to detect MCU

Error: cannot find USB device with vid=0x16c0 pid=0x5dc
Error: unable to open port usb for programmer usbasp-clone

Avrdude done. Thank you.`

The device is detected as USBasp in Device Manger with ID USB\VID_16C0&PID_05DC&MI_00\9&139AF701&0&0000.

Zadig sees USBasp (interface 0) as WINUSB (v10.0.26100.1150) and USBasp (interface 1) and USBasp (Interface 2), both as HidUsb (v10.0.26100.1882) devices. I haven't tried to install any other drivers with Zadig.

MCUdude, 2025-02-21:
Try the libusbK driver though Zadig.

TzOk83, 2025-02-21:
Doesn't help, and should not be needed, as dioannidis' firmware should work with WinUSB, and it used to work on my old Win10 machine.

mcuee, 2025-02-22:
This is a known issue. Please use the mingw64 binary instead and not the MSVC binary.

* https://github.com/avrdudes/avrdude/issues/968

mcuee, 2025-02-22:
It is also documented in the release page.
https://github.com/avrdudes/avrdude/releases/tag/v8.0

mcuee, 2025-02-24:
I will close this one and keep #968.

closed 2025-02-24 by mcuee