avrdudes/avrdude #1957: How does `hvupdi_variant` 3 work? (AVR32SDnn)

stefanrueger, 2025-03-11:
The recent parts `AVR32SD20`/`28`/`32`  introduced a new way of enabling UPDI through high-voltage programming. The method is encoded in avrdude.conf's `hvupdi_variant` variable (which comes from .atdf's `HV_IMPLEMENTATION` parameter) and set to a *new* value of 3. So far AVRDUDE has used the values 0..2 (see below). The affected code is mainly located in the files `jtag3.c` (@MCUdude) and `pickit5.c` (@MX682X). Also affected is the `hvupdi_support` list of `hvupdi_variant` methods that a programmer knows about and can work with (@MCUdude).

@mraardvark, do you have more intelligence on how `HV_IMPLEMENTATION = 3` is supposed to work? So that @MX682X and @MCUdude can adapt their code. Also, which programmers know the type 3 (or are going to in a FW upgrade)?

So far known types 0..2 are mentioned here:
https://github.com/avrdudes/avrdude/blob/8c4b69ac47c3bf5d4603ca0793b8d59af2bf310e/src/libavrdude.h#L286-L288

The data sheet for the known AVR-SD parts alludes to a mechanism similar to that of AVR-DD, AVR-DU and AVR-Ex:

[image]

Unfortunately, there is no more detail in the UPDI section, contrary to what above says.

stefanrueger, 2025-03-11:
Thanks for looking into this @MX682X. Good to know. I still wonder why Microchip has seen fit to introduce a new type if it's the same procedure as for AVR-DD/AVR-DU/AVR-Ex

xedbg, 2025-03-11:
Type 3 is the "UPDI handshake".  This is in the AVR SD data sheet.  Quick summary, if you erase this part, then lock it immediately, it will refuse to boot since it will not execute 0xFFs due to safety.  It will reset immediately and try again.  If the lockbits are set then the reset includes an UPDI reset, which makes it extremely difficult to get in.
The handshake is a bit like a reset extension which makes it always respond.

mcuee, 2025-05-11:
* https://github.com/avrdudes/avrdude/discussions/1956#discussioncomment-13103107
Close this for now.

closed 2025-05-11 by mcuee