avrdudes/avrdude #1961: Backup Arduino Nano Every

mladenb, 2025-03-15:
Hi,

I've recently encountered an issue, trying to send the reset/touch signal (using 1200 baud) to my Arduino Nano Every in order to backup the binary version of the currently running sketch. After running this command:

`/usr/local/bin/avrdude -vv -p atmega4809 -c jtag2updi -r -b 115200 -P /dev/ttyACM0 -U flash:r:board.hex:i`

I got the following response:

```
Avrdude version 8.0-20250311 (8c4b69ac)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /root/.avrduderc does not exist

Touching serial port /dev/ttyACM0 at 1200 baud
Waiting for new port... using same port /dev/ttyACM0
Using port            : /dev/ttyACM0
Using programmer      : jtag2updi
Programmer baud rate  : 115200
JtagmkII_open_pdi()
JtagmkII_getsync() attempt 1 of 10: sending sign-on command: 
Error jtagmkII_recv_frame() jtagmkII.c 575: timeout
Warning jtagmkII_getsync() jtagmkII.c 650: attempt 1 of 10: sign-on command: status -1
JtagmkII_getsync() attempt 2 of 10: sending sign-on command: 
Error jtagmkII_recv_frame() jtagmkII.c 575: timeout
Warning jtagmkII_getsync() jtagmkII.c 650: attempt 2 of 10: sign-on command: status -1
JtagmkII_getsync() attempt 3 of 10: sending sign-on command: 
Error jtagmkII_recv_frame() jtagmkII.c 575: timeout
Warning jtagmkII_getsync() jtagmkII.c 650: attempt 3 of 10: sign-on command: status -1
JtagmkII_getsync() attempt 4 of 10: sending sign-on command: 
Error jtagmkII_recv_frame() jtagmkII.c 575: timeout
Warning jtagmkII_getsync() jtagmkII.c 650: attempt 4 of 10: sign-on command: status -1
JtagmkII_getsync() attempt 5 of 10: sending sign-on command: 
Error jtagmkII_recv_frame() jtagmkII.c 575: timeout
Warning jtagmkII_getsync() jtagmkII.c 650: attempt 5 of 10: sign-on command: status -1
JtagmkII_getsync() attempt 6 of 10: sending sign-on command: 
Error jtagmkII_recv_frame() jtagmkII.c 575: timeout
Warning jtagmkII_getsync() jtagmkII.c 650: attempt 6 of 10: sign-on command: status -1
JtagmkII_getsync() attempt 7 of 10: sending sign-on command: 
Error jtagmkII_recv_frame() jtagmkII.c 575: timeout
Warning jtagmkII_getsync() jtagmkII.c 650: attempt 7 of 10: sign-on command: status -1
JtagmkII_getsync() attempt 8 of 10: sending sign-on command: 
Error jtagmkII_recv_frame() jtagmkII.c 575: timeout
Warning jtagmkII_getsync() jtagmkII.c 650: attempt 8 of 10: sign-on command: status -1
JtagmkII_getsync() attempt 9 of 10: sending sign-on command: ^[[A^[[A^C
```

with a little help from Arduino forum, I've managed to successfully download the hex file to my local computer, using a python script that does that 1200 baud reset a bit different, using a small delay between opening and closing the com channel.

Here's the script, so you can see what helped to put my Arduino Nano Every into UPDI mode properly:

```
#!/usr/bin/python3
# Based on https://github.com/arduino/ArduinoCore-megaavr/issues/124#issuecomment-1330763422
#   Run this script to reset the Arduino Nano Every
#   eg. sudo python3 mcu_reset.py
#
# Note: You must install pyserial first; i.e. sudo pip3 pyserial

import serial
import os, sys, time
# total arguments
n = len(sys.argv)
if n >= 2:
    port = sys.argv[1]
else:
    port = '/dev/ttyACM0'

print("Port: {}".format(port))

#re-enable hupcl temporarily (necessary for arduino reset using serial port)
os.system('sudo /bin/stty -F {} hupcl'.format(port))

try:
    #perform Arduino Nano Every reset "handshake"
    ser = serial.Serial()
    ser.baudrate = 1200
    ser.port = port
    ser.open()
    time.sleep(0.5)
    ser.close()

except serial.SerialException:
    print("Error: serial.SerialException")
    exit()
```
I've posted this message in hope it could help you perhaps understand what could be improved with the "-r" option of the avrdude (great product btw, thank you a lot for creating it!).

Cheers

stefanrueger, 2025-03-15:
> using a small delay between opening and closing

AVRDUDE uses 400 ms for every '-r'; have you tried `-rr`?

mladenb, 2025-03-15:
yes I have, and it didn't help... I've noticed there was some longer delay with multiple `-r` but the errors were left the same...

stefanrueger, 2025-03-15:
> are `-r -r -r -r` the same as `-rrrr`? 

Yes. So all good.

I noticed that the 500 ms sleep in the python script is *before* closing the port; AVRDUDE sleeps *after* closing the port:
https://github.com/avrdudes/avrdude/blob/8c4b69ac47c3bf5d4603ca0793b8d59af2bf310e/src/serialadapter.c#L340-L349

What happens if you insert a `usleep(500*1000);` before line 340?

mladenb, 2025-03-16:
you're right on the spot! thank you for the fast thinking :) I've tried to move the line 349 `usleep(400*1000*nwaits);` up to the location after the line 338 `usleep(100);`

after that change the avrdude alone was enough to properly reset the board

thank you a lot for a precise suggestion! :)

stefanrueger, 2025-03-16:
> prolong the original `usleep(100);`

Not a good idea: That sleep is sandwiched between setting DTR/RTS and releasing it. Many boards have a reset circuit with a capacitor on that line that gets charged too much for longer sleeps; when the capacitor voltage is pushed onto Vcc by releasing DTR/RTS this in turn creates a spike above Vcc up to 2*Vcc for a fully charged cap, which in turn can make some parts enter HV programming. See related comments [here](https://github.com/avrdudes/avrdude/blob/8c4b69ac47c3bf5d4603ca0793b8d59af2bf310e/src/stk500.c#L111-L117).

 Does is help if you *add* `usleep(500*1000);` *just* before line 340 (and after line 349 that releases DTR/RTS)?

Have you said which OS/hardware you are running AVRDUDE on? Maybe our resident platform expert @mcuee has experience/an opinion on a solution.

Generally, if/when we make an intervention in code like this, we have to carefully check against regressions, ie, check whether the platforms/OSes that used to work with the old code still work with the new code.

mcuee, 2025-03-17:
Arduino Nano Every is a bit problematic as it has a buggy FW. Let me dig out it and carry out some tests later this week.

mcuee, 2025-03-23:
> Not a good idea: That sleep is sandwiched between setting DTR/RTS and releasing it. Many boards have a reset circuit with a capacitor on that line that gets charged too much for longer sleeps; when [...]

I agree.
 
> Does is help if you _add_ `usleep(500*1000);` _just_ before line 340 (and after line 349 that releases DTR/RTS)?

@mladenb 
Please help to try this out. Thanks.

> Have you said which OS/hardware you are running AVRDUDE on? Maybe our resident platform expert [@mcuee](https://github.com/mcuee) has experience/an opinion on a solution.

@mladenb 
Please to answer this question. Thanks.

> Generally, if/when we make an intervention in code like this, we have to carefully check against regressions, ie, check whether the platforms/OSes that used to work with the old code still work with the new code.

Indeed this is good.

mcuee, 2025-03-24:
Just tested avrdude git under Windows 11 23H2 (Dell Laptop) and it works fine. I will test more systems.
```
C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -C .\avrdude_git.conf -c jtag2updi -P COM4 -p m4809 -r -v
Avrdude_git version 8.0-20250311 (8c4b69ac)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude_git.conf

Touching serial port COM4 at 1200 baud
Waiting for new port... using same port COM4
Using port            : COM4
Using programmer      : jtag2updi
Programmer baud rate  : 115200
AVR part              : ATmega4809
Programming modes     : SPM, UPDI
Programmer type       : JTAGMKII_UPDI
Description           : JTAGv2 to UPDI bridge
Main MCU HW version   : 1
Main MCU FW version   : 1.07
Sec. MCU HW version   : 1
Sec. MCU FW version   : 6.07
Serial number         : 00:00:00:00:00:00
Silicon revision: 0.1

AVR device initialized and ready to accept instructions
Device signature = 1E 96 51 (ATmega4809)

Avrdude_git done.  Thank you.
```

mcuee, 2025-03-24:
No issues with my Acer Swift 3 laptop early 2021 model (Intel Core i5-1135G7 CPU).

```
mcuee@UbuntuSwift3 ~/build/avr $ avrdude -c jtag2updi -p m4809 -r -P /dev/ttyACM0 -v
Avrdude version 8.0-20250311 (8c4b69ac)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Touching serial port /dev/ttyACM0 at 1200 baud
Waiting for new port... using same port /dev/ttyACM0
Using port            : /dev/ttyACM0
Using programmer      : jtag2updi
Programmer baud rate  : 115200
AVR part              : ATmega4809
Programming modes     : SPM, UPDI
Programmer type       : JTAGMKII_UPDI
Description           : JTAGv2 to UPDI bridge
Main MCU HW version   : 1
Main MCU FW version   : 1.07
Sec. MCU HW version   : 1
Sec. MCU FW version   : 6.07
Serial number         : 00:00:00:00:00:00
Silicon revision: 0.1

AVR device initialized and ready to accept instructions
Device signature = 1E 96 51 (ATmega4809)

Avrdude done.  Thank you.
```

per1234, 2025-03-25:
> Have you said which OS

I can provide this information from the Arduino Forum post @mladenb made before opening the issue here:

https://forum.arduino.cc/t/backup-arduino-nano-every/1363777

> Ubuntu 22.04

stefanrueger, 2025-03-25:
I only have an  ATmega4808 Nano Every with a CH340 usb-tty converter, and only Ubuntu 24.04. This works:
```
$ avrdude -qq -p atmega4808 -c jtag2updi -r -P /dev/ttyUSB0 -U flash:r:board.hex:i && echo OK
OK
```

My board doesn't need the magic `-r` touch, though, so must be a different board type to @mladenb's:

```
$ avrdude -qq -p atmega4808 -c jtag2updi -P /dev/ttyUSB0 -U flash:r:board.hex:i && echo OK
OK
```

Note that I've used `/dev/ttyUSB0` and not `/dev/ttyACM0` but AVRDUDE told me I was using the wrong port:
```
$ avrdude -qq -p atmega4808 -c jtag2updi -r  -P /dev/ttyACM0 -U flash:r:board.hex:i && echo OK
OS error: cannot open port /dev/ttyACM0: No such file or directory
Error: touch_serialport() failed to open port /dev/ttyACM0 at 1200 baud
A possible candidate serial port is:
  -P /dev/ttyUSB0 or -P ch340
Note that above port might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.
```

mcuee, 2025-04-20:
Nano Every Clone (with ATmega4808 and CH340) do not need the 1200bps touch.

mcuee, 2025-05-02:
@mladenb 

My recommendation is to close this issue for now since we cannot reproduce the issue. And you have not reported back for a while. 

Any objections?

stefanrueger, 2025-05-02:
@mcuee Thanks. Maybe the Nano Every needs a firmware update? I have forgotten how to do that with the MPLAB tool, but surely there is a way? Have we documented firmware updates through Microchip somewhere in our wiki or otherwise?

mcuee, 2025-05-02:
> @mcuee Thanks. Maybe the Nano Every needs a firmware update? I have forgotten how to do that with the MPLAB tool, but surely there is a way? Have we documented firmware updates through Microchip somewhere in our wiki or otherwise?

It is not from Microchip but rather from Arduino. I consider it as a failed product from Arduino.

mcuee, 2025-05-02:
@stefanrueger 
You can see the FW has not been updated anyway for Arduino Nano Every board.
https://github.com/arduino/ArduinoCore-megaavr/tree/master/firmwares/MuxTO

People are not able to build the FW at all.
* https://github.com/arduino/ArduinoCore-megaavr/issues/103

There is supposed a workflow to build it but it never really run successfully.
* https://github.com/arduino/ArduinoCore-megaavr/pull/105

mladenb, 2025-05-03:
First of all, sorry for the slow response from my side, it's been quite a busy period for me, so I couldn't focus on this topic as much as it's needed to resolve this issue.

Let me first answer those questions.

> > Does is help if you _add_ `usleep(500*1000);` _just_ before line 340 (and after line 349 that releases DTR/RTS)?
> 
> [@mladenb](https://github.com/mladenb) Please help to try this out. Thanks.
> 

I've tried this suggestion, but it produces the same timeout as if no changes to the code have been introduced.

> > Have you said which OS/hardware you are running AVRDUDE on? Maybe our resident platform expert [@mcuee](https://github.com/mcuee) has experience/an opinion on a solution.
> 
> [@mladenb](https://github.com/mladenb) Please to answer this question. Thanks.
> 

My laptop, HP ProBook 450 G1, running Ubuntu 22.04.5 LTS

All in all, it seemed that this part was the only thing that mattered, in my tests:

```
  serial_set_dtr_rts(&fd, 1);
  usleep(100); // <--- THIS
  serial_set_dtr_rts(&fd, 0);
```

BUT... I just noticed something interesting... When I did a ``git restore .`` to reset all the changes, and rebuilt/ran the command again, it was working without the timeout... So, I thought, maybe something has changed on the board, while I was testing longer delays and decided to shut down the board to cool off and to try/repeat the tests. Sure enough, when I rebooted the board, and ran the command, the same timeout appeared... Actually, knowing the point where the command gets stuck, I've realized already it would timeout, so I pressed CTRL+C to avoid waiting. After that I ran the command again, and voila, it started working. Not just at that attempt, but every single attempt from 1st working attempt on was working without issues. As if the board switched to a different state somewhere, due to my CTRL+C... Here's the log of that run:

```
mladen@mladen-HP:~/src/avrdude$ /usr/local/bin/avrdude -vv -p atmega4809 -c jtag2updi -r -b 115200 -P /dev/ttyACM0 -U flash:r:board.hex:i
Avrdude version 8.0-20250311 (8c4b69ac)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mladen/.avrduderc does not exist

Touching serial port /dev/ttyACM0 at 1200 baud
Waiting for new port... using same port /dev/ttyACM0
Using port            : /dev/ttyACM0
Using programmer      : jtag2updi
Programmer baud rate  : 115200  
JtagmkII_open_pdi()
JtagmkII_getsync() attempt 1 of 10: sending sign-on command: ^C

mladen@mladen-HP:~/src/avrdude$ /usr/local/bin/avrdude -vv -p atmega4809 -c jtag2updi -r -b 115200 -P /dev/ttyACM0 -U flash:r:board.hex:i
Avrdude version 8.0-20250311 (8c4b69ac)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mladen/.avrduderc does not exist

Touching serial port /dev/ttyACM0 at 1200 baud
Waiting for new port... using same port /dev/ttyACM0
Using port            : /dev/ttyACM0
Using programmer      : jtag2updi
Programmer baud rate  : 115200
JtagmkII_open_pdi()
JtagmkII_getsync() attempt 1 of 10: sending sign-on command: 0x86 (29 bytes msg)
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              1.07
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.07
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
JtagmkII_getsync(): using a 298-byte device descriptor
...
```

I've tested this several times and the results are consistent, so here are the steps to reproduce:
1. Turn off the Arduino Nano Every board for at least 20-30 seconds, to cool of
2. Turn the board on
3. Run the command from my first post: ``/usr/local/bin/avrdude -vv -p atmega4809 -c jtag2updi -r -b 115200 -P /dev/ttyACM0 -U flash:r:board.hex:``
4. Make sure the command output gets stuck at "sending sign-on command", but don't wait for too long for it to actually time out
5. Press CTRL+C to stop the command execution
6. Run the same command again

The command should now run without timeout.

mladenb, 2025-05-03:
I'm not sure why this is acting out like this, I can only imagine that I've interrupted the command at the correct time, when it was waiting (charging capacitor?) and when I stopped/reran the command, it probably started with a non-empty capacitor at that point. Of course, I'm in no way an expert in electronics, but I'm just trying to make all of this make sense to me...

Anyway, I apologize in advance if it happens again that I become unresponsive on this topic. Feel free to close it and I'll reopen it if there's a feedback needed from my side and I can provide it.

Thank you once again for creating this great tool, really appreciated.

stefanrueger, 2025-05-03:
@mladenb Thanks for posting these observations. Hmm, aborting an AVRDUDE run leaves the board in a state that makes it work correctly next time... It's a mystery to me what that could be and requires a lot more digging. @mcuee has observed that the Arduino Nano Every is a fickle board that's nit easy to cater for (or reproduce errors with), so unless there is more specific info on what goes wrong (either AVRDUDE side or Arduino Nano Every board side), and what can be done about it I am still tempted to close the issue in the next week or so.

closed 2025-05-11 by stefanrueger

mcuee, 2025-05-19:
> Make sure the command output gets stuck at "sending sign-on command", but don't wait for too long for it to actually time out
Press CTRL+C to stop the command execution.

I cannot reach this step myself. The output never gets stuck.

```
C:\work\avr\avrdude_bin> .\avrdude_pr1989_clang64.exe -C .\avrdude_pr1989_clang64.conf -c jtag2updi -P COM5 -p m4809 -r -vvvv
$ 'C:\work\avr\avrdude_bin\avrdude_pr1989_clang64.exe' -C '.\avrdude_pr1989_clang64.conf' -c jtag2updi -P COM5 -p m4809 -r -vvvv
Avrdude_pr1989_clang64 version 8.0-20250518 (97039410)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is C:\work\avr\avrdude_bin\avrdude_pr1989_clang64.conf

Touching serial port COM5 at 1200 baud
Waiting for new port... using same port COM5
Using port            : COM5
Using programmer      : jtag2updi
Programmer baud rate  : 115200
JtagmkII_open_pdi()
JtagmkII_getsync()
JtagmkII_getsync() attempt 1 of 10: sending sign-on command:
JtagmkII_send(): sending 1 bytes
Ser_send: . [1b] . [00] . [00] . [01] . [00] . [00] . [00] . [0e] . [01] . [f3] . [97]
JtagmkII_recv():
Ser_recv: . [1b]
Ser_recv: . [00]
Ser_recv: . [00]
Ser_recv: . [1d]
Ser_recv: . [00]
Ser_recv: . [00]
Ser_recv: . [00]
Ser_recv: . [0e]
Ser_recv: . [86] . [01] . [01] . [07] . [01] . [01] . [01] . [07] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00]
Ser_recv: o [6f]
Ser_recv: V [56]

JtagmkII_recv(): got message seqno 0 (command_sequence == 0)
JtagmkII_recv: . [86] . [01] . [01] . [07] . [01] . [01] . [01] . [07] . [06] . [01] . [00] . [00] . [00] . [00] . [00] . [00] J [4a] T [54] A [41] G [47] I [49] C [43] E [45]   [20] m [6d] k [6b] I [49] I [49] . [00]

Raw message:
0x86 0x01 0x01 0x07 0x01 0x01 0x01 0x07 0x06 0x01 0x00 0x00 0x00 0x00 0x00 0x00
0x4a 0x54 0x41 0x47 0x49 0x43 0x45 0x20 0x6d 0x6b 0x49 0x49 0x00
Sign-on succeeded

JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              1.07
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.07
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
JtagmkII_getsync(): using a 298-byte device descriptor
JtagmkII_setparm()
JtagmkII_setparm() sending set parameter command (parm emulator_mode 0x03, 1 bytes):
JtagmkII_send(): sending 3 bytes
Ser_send: . [1b] . [01] . [00] . [03] . [00] . [00] . [00] . [0e] . [02] . [03] . [06] $ [24] 1 [31]
JtagmkII_recv():
Ser_recv: . [1b]
Ser_recv: . [01]
Ser_recv: . [00]
Ser_recv: . [01]
Ser_recv: . [00]
Ser_recv: . [00]
Ser_recv: . [00]
Ser_recv: . [0e]
[... 428 lines cut ...]
JtagmkII_close(): sending sign-off command:
JtagmkII_send(): sending 1 bytes
Ser_send: . [1b] . [11] . [00] . [01] . [00] . [00] . [00] . [0e] . [00] . [bd] \ [5c]
JtagmkII_recv():
Ser_recv: . [1b]
Ser_recv: . [11]
Ser_recv: . [00]
Ser_recv: . [01]
Ser_recv: . [00]
Ser_recv: . [00]
Ser_recv: . [00]
Ser_recv: . [0e]
Ser_recv: . [80]
Ser_recv: . [b5]
Ser_recv: . [d8]

JtagmkII_recv(): got message seqno 17 (command_sequence == 17)
JtagmkII_recv: . [80]

Raw message:
0x80
OK

Avrdude_pr1989_clang64 done.  Thank you.
```