avrdudes/avrdude #1967: Cater for new `UPDI_ENABLE_RESET_HS` method for enabling UPDI

stefanrueger, 2025-03-25:
With the advent of `AVR32SDnn` parts a new method of enabling UPDI came along.
https://github.com/avrdudes/avrdude/blob/14bfbd472e5e0428c6ba766ad9e9fb505009a261/src/libavrdude.h#L286-L289

As far as I understand, from a progamming perspective,  `UPDI_ENABLE_RESET_HS` doesn't behave any different to `UPDI_ENABLE_HV_RESET`. However, we still need to adapt the AVRDUDE source to this. `pickit5.c` has already done so.

`jpeg3.c` needs to follow suit. Is that something you'd be willing to look into @MCUdude?

MCUdude, 2025-03-26:
Here's how the HV UPDI pulse is implemented in jtag3.c

```c
    // Generate UPDI high-voltage pulse if user asks for it and hardware supports it
    if(my.use_hvupdi == true && p->hvupdi_variant != UPDI_ENABLE_ALWAYS) {
      parm[0] = PARM3_UPDI_HV_NONE;
      for(LNODEID ln = lfirst(pgm->hvupdi_support); ln; ln = lnext(ln)) {
        if(*(int *) ldata(ln) == p->hvupdi_variant) {
          pmsg_notice("sending HV pulse to targets %s pin\n", p->hvupdi_variant == UPDI_ENABLE_HV_UPDI? "UPDI": "RESET");
          parm[0] = PARM3_UPDI_HV_SIMPLE_PULSE;
          break;
        }
      }
      if(parm[0] == PARM3_UPDI_HV_NONE) {
        pmsg_error("%s does not support sending HV pulse to target %s\n", pgm->desc, p->desc);
        return -1;
      }
      if(jtag3_setparm(pgm, SCOPE_AVR, 3, PARM3_OPT_12V_UPDI_ENABLE, parm, 1) < 0)
        return -1;
    }
```

Wouldn't this work with _any_ HV UPDI variant as long as `ldata(ln) == p->hvupdi_variant`? There are no special cases for the other HV UPDI variants.

stefanrueger, 2025-03-26:
> as long as `ldata(ln) == p->hvupdi_variant`

Cool, vvv future-proof!

Is `p->hvupdi_variant == UPDI_ENABLE_HV_UPDI? "UPDI": "RESET"` still correct now that there is a fourth valid number for `p->hvupdi_variant`?

Do we need to review the `hvupdi_support` list of the programmers? That might in theory need extending to `3` for some.

Feel free to close the issue once you have convinced yourself that nothing is to do. And thanks for looking into this @MCUdude!

MCUdude, 2025-03-27:
> Do we need to review the hvupdi_support list of the programmers? That might in theory need extending to 3 for some.

Probably, but I'd like to know the following:

* Does the Power Debugger in UPDI mode support HVUPDI variant 2 (AVR-DD/DU/Ex) and 3 (AVR-Sx)?
* Is it confirmed that the PICkit4 in AVR/EDBG mode supports HV UPDI variant 2?

Maybe @xedbg knows?

xedbg, 2025-03-28:
PowerDebugger has a hardware ~~regrettable crystal-ball-failure~~ shortcoming in which the HV signal can only be multiplexed on the UPDI pin, and also has a fixed voltage, so this excludes type 2.  It can (in latest FW) do type 3.

PICkit4 in EDBG-mode is to be viewed as an interim solution, but I am fairly sure type 2 got in there, but not type 3.

stefanrueger, 2025-04-13:
@MX682X Good point. A bitmap would work, too. It's a bit involved to change that around as there is code to be rewritten for the developer options. So it might turn out to be swings and roundabouts.

mcuee, 2025-05-11:
* https://github.com/avrdudes/avrdude/discussions/1956#discussioncomment-13103107
Close this for now.

closed 2025-05-11 by mcuee