avrdudes/avrdude #1972: linuxgpio programmer do not release RESET line on failure.

Pcdwarf, 2025-04-15:
Hi, 
I think I found a minor issue with the `linuxgpio` programmer.

When running without a delay (`-i` option), the programming signals may be too fast for the MCU to respond properly, which can cause a failure — that's expected.

However, in case of such a failure, it seems that the RESET line is left low (asserted).  
On the next attempt (even with `-i 10`), the programmer doesn't release RESET high before pulling it low again.

This is problematic because if the MCU is left in an undefined or error state from the first attempt, not cycling RESET properly can cause the second attempt to fail again — for example, in my case, it results in an incorrect signature being read.

As a workaround, I had to manually release the reset line between calls using:

```bash
gpioset gpiochip0 $RESETPIN=1
```

I think avrdude should always release the RESET line on exit,
( or at least restore it to the state it was in at startup but in that case, it should ensure a proper cycle on the next invocation.)

Pcdwarf, 2025-04-15:
One other small remark:

It would be helpful if `avrdude` printed a short warning when the `-B` (`--bitclock`) option is used with the `linuxgpio` programmer since it has no effect in this context.

Right now, it silently accepts `-B` even though it does nothing, which can be confusing.

This may also apply to other bitbang-based programmers.

Something like 
```Warning: -B (bitclock) option is ignored for linuxgpio programmer. Use -i (delay) instead```

stefanrueger, 2025-04-15:
>  in case of such a failure, it seems that the RESET line is left low (asserted)

In order for us to understand the issue better: Which host hardware do you use, which Linux flavour? Which command line? So that one of us can try to replicate the issue.

Better still if you can contribute a PR seeing that you seem have a good understanding of the code: The AVRDUDE project relies on the community for code contributions, testing and bug fixes. As maintainers we do have access to a reasonable number of programmers and target MCUs but not the full gamut of some 150+ programmers, 300+ MCUs, host PCs and operating systems.

AVRDUDE has the concept of `-E` exit specifications, typically which lines should stay low/high on exit. `linuxgpio` does not utilise this concept and it might be a good idea if the project did.

Generally we review PRs from a hardware perspective, from a software perspective, a user perspective (will users be surprised by a change in AVRDUDE behaviour?)  and whether the behaviour is backward compatible (one person's bug is another person's feature).

> silently accepts -B even though it does nothing, which can be confusing

Agreed, ... but it's hard to do something about it. AVRDUDE serves 180 programmers (`avrdude -c\? |& grep = | wc -l`) that are managed by some 55 separate source code files (programmer types: `avrdude -c\*/d | grep = | cut -f4 -d\' | grep . | sort -u | wc -l`). The main program has *no idea* which of these 55 separate programmer types make use of `-B` settings in their source code and which not.   `grep -rl bitclock src | grep c$ | wc -l` seems to confirm that less than a third of the programmers actually make use of `-B` (and I am sure `-B` doesn't even make sense for a lot of programmers).

Pcdwarf, 2025-04-15:
Hi,

Thanks for your feedback.

Here's some additional context about my setup:

I'm using Raspberry Pi Compute Module 4 Rev 1.1
with Linux rpi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
custom built Avrdude version 8.0-20250414 (27057485)

Build options : 
```
-- DO HAVE    libelf
-- DO HAVE    libusb
-- DO HAVE    libusb_1_0
-- DO HAVE    libhidapi
-- DON'T HAVE libftdi
-- DO HAVE    libftdi1
-- DO HAVE    libreadline
-- DO HAVE    libserialport
-- DISABLED   doc
-- DISABLED   parport
-- ENABLED    linuxgpio
-- DO HAVE    libgpiod (1.6.3)
-- ENABLED    linuxspi
-- DON'T HAVE swig+Python3
```

And I added a custom programmer to config file

```
programmer
     id                   = "MicAltApuISP";
     desc                 = "Use the Linux sysfs interface to bitbang GPIO lines";
     type                 = "linuxgpio";
     prog_modes           = PM_ISP;
     connection_type      = linuxgpio;
     reset                = 20;
     sck                  = 18;
     sdo                  = 19;
     sdi                  = 21;
;
```

the invocation line is 

`/usr/local/bin/avrdude -c MicAltApuISP -p m328pb - i 2 -U flash:w:"${firmwarefile}.flash.bin":r `

The plan is to retry the command with a longer -i (e.g., -i 50) on failure.

oscilloscope checks showed RESET was still low after failure and that on subsequent attempts, RESET is pulled low again without being released high first. this leads to a read signature failure, likely because the MCU never got a proper RESET cycle between attempts.

I confirmed with a scope that RESET was still held low after avrdude exited on failure (and it was high before calling avrdude, (but i cannot says if it was input with pullup or output at high state)

I workedaroud that by running 
`gpioset gpiochip0 20=1 `
after each call to avrdude. 
This workaround works, but it's not intuitive nor ideal.

Side note: In the earlier sysfs-based implementation of linuxgpio, GPIOs were not always unexported properly on failure, which caused subsequent calls to fail when trying to re-export them, but I don’t recall RESET staying low. It’s possible that it was cycled again at the start of each new invocation, so the problem didn't surface then.

stefanrueger, 2025-04-22:
@MCUdude Is `-E` specification for the `linuxgpio` driver something you'd support and/or were willing to implement?

MCUdude, 2025-04-22:
I'll look into it and create a PR!

MCUdude, 2025-04-22:
@stefanrueger how about just running `pgm->close()` for all bigbanged programmers if the programming session has failed, in order to return to a known state? This will solve this issue for all bitbanged programmers, and all IO pins, not just the reset line, will return to a known state.

BTW have a look at this code snippet:

https://github.com/avrdudes/avrdude/blob/118625e18b3b902f5fa1cec9049d009526354740/src/main.c#L1637-L1640

* Why is `SERBB` in capital letters? Shouldn't it be `serbb`?
* Why would `serbb` even care about the baudrate? Isn't that the whole point of bitbanging; using the hardware handshake pins for the typical SPI/TPI interface? I can't find any mention of baudrate in any of the serbb files. (When thinking about it, it might actually be my fault from  when I added that error message for SerialUPDI programmers).

Here's my suggested fix:

```c
    if(str_eq(pgm->type, "serialupdi"))
      imsg_error(" - use -b to set lower baud rate, e.g. -b %d\n", baudrate? baudrate/2: 57600);
    else if(str_eq(pgm->type, "buspirate_bb") || str_eq(pgm->type, "linuxgpio") ||
      str_eq(pgm->type, "par") || str_eq(pgm->type, "serbb")) {
      imsg_error(" - use -i %s to set a longer delay (in microseconds) between each bit state change, e.g. -i 50\n",
        bitclock? "instead of -B": "");
      pgm->close(p);
    }
    else
      imsg_error(" - use -B to set lower the bit clock frequency, e.g. -B 125kHz\n");
```

stefanrueger, 2025-04-23:
> how about just running `pgm->close()` for all bigbanged programmers

Sounds like a *great* idea! Do you want to create a PR that can be tested?

> Why is `SERBB` in capital letters?

That's correct; the convention used to be (and still is) that the initialisation routine sets the programmer type field; see
https://github.com/avrdudes/avrdude/blob/118625e18b3b902f5fa1cec9049d009526354740/src/serbb_posix.c#L290

https://github.com/avrdudes/avrdude/blob/118625e18b3b902f5fa1cec9049d009526354740/src/serbb_win32.c#L301

Capitalisation is unusual, but the `serbb` programmer chose it that way, and we have to keep that for backward compatibility (though I wouldn't encourage that for new programmers)

MCUdude, 2025-04-24:
@stefanrueger PR #1973 is now ready for testing!