avrdudes/avrdude #1978: Regression: ATtiny13 issue with `-B32` for "Arduino AS ISP"

mcuee, 2025-05-05:
* https://github.com/MCUdude/MicroCore/issues/163#issuecomment-2850952436* https://github.com/MCUdude/MicroCore/issues/163

@MCUdude and @stefanrueger 

Hmm, I changed my mind, it seems to be a regression from avrdude 7.2 to avrdude 7.3.
`-B32` works with avrdude 7.2 release but not 7.3 release.

```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ ./avrdude72 -C ./avrdude72.conf -v -pattiny13a -cstk500v1 -B32 -P/dev/ttyUSB0 -b19200

avrdude72: Version 7.2
           Copyright the AVRDUDE authors;
           see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

           System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude72.conf
           User configuration file is /home/mcuee/.avrduderc
           User configuration file does not exist or is not a regular file, skipping

           Using Port                    : /dev/ttyUSB0
           Using Programmer              : stk500v1
           Overriding Baud Rate          : 19200
           Setting bit clk period        : 32.0
           AVR Part                      : ATtiny13A
           Chip Erase delay              : 4000 us
           RESET disposition             : dedicated
           RETRY pulse                   : SCK
           Serial program mode           : yes
           Parallel program mode         : yes
           Timeout                       : 200
           StabDelay                     : 100
           CmdexeDelay                   : 25
           SyncLoops                     : 32
           PollIndex                     : 3
           PollValue                     : 0x53
           Memory Detail                 :

                                             Block Poll               Page                       Polled
             Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
             ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
             eeprom                 65     5     4    0 no         64    4      0  4000  4000 0x00 0x00
             flash                  65    10    32    0 yes      1024   32     32  4500  4500 0x00 0x00
             lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
             hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
             lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
             signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
             calibration             0     0     0    0 no          2    1      0     0     0 0x00 0x00

           Programmer Type : STK500
           Description     : Atmel STK500 version 1.x firmware
           Hardware Version: 2
           Firmware Version: 1.18
           Topcard         : Unknown
           Vtarget         : 0.0 V
           Varef           : 0.0 V
           Oscillator      : Off
           SCK period      : 0.1 us
avrdude72: AVR device initialized and ready to accept instructions
avrdude72: device signature = 0x1e9007 (probably t13a)

avrdude72 done.  Thank you.

mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ ./avrdude73 -C ./avrdude73.conf -v -pattiny13a -cstk500v1 -B32 -P/dev/ttyUSB0 -b19200

avrdude73: Version 7.3
           Copyright the AVRDUDE authors;
           see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

           System wide configuration file is /home/mcuee/build/avr/avrdude_bin/avrdude73.conf
           User configuration file is /home/mcuee/.avrduderc
           User configuration file does not exist or is not a regular file, skipping

           Using port            : /dev/ttyUSB0
           Using programmer      : stk500v1
           Setting baud rate     : 19200
           Setting bit clk period: 32.0 us

avrdude73 stk500_setparm() error: cannot get into sync
avrdude73 stk500_set_sck_period() error: cannot set Parm_STK_SCK_DURATION
avrdude73 main() error: unable to open port /dev/ttyUSB0 for programmer stk500v1
A possible candidate serial port is:
  -P /dev/ttyUSB0 or -P ch340
Note that above port might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.

avrdude73 done.  Thank you.
```

mcuee, 2025-05-05:
Same for avrdude git. 
Without `-B32` it works fine but it does not work with `-B32`.

```
mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ avrdude -v -pattiny13a -carduino_as_isp -B32 -P/dev/ttyUSB0 -b19200
Avrdude version 8.0-20250504 (58408592)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : /dev/ttyUSB0
Using programmer      : arduino_as_isp
Programmer baud rate  : 19200
Setting bit clk period: 32.0 us

Error: cannot get into sync
Error: cannot set Parm_STK_SCK_DURATION
Error: unable to open port /dev/ttyUSB0 for programmer arduino_as_isp
A possible candidate serial port is:
  -P /dev/ttyUSB0 or -P ch340
Note that above port might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.

Avrdude done.  Thank you.

mcuee@UbuntuSwift3 ~/build/avr/avrdude_bin $ avrdude -v -pattiny13a -carduino_as_isp -P/dev/ttyUSB0 -b19200
Avrdude version 8.0-20250504 (58408592)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : /dev/ttyUSB0
Using programmer      : arduino_as_isp
Programmer baud rate  : 19200
AVR part              : ATtiny13A
Programming modes     : SPM, ISP, HVSP, debugWIRE
Programmer type       : STK500
Description           : AVR as programmer with Arduino-as-ISP FW
HW Version            : 2
FW Version            : 1.18
Topcard               : Unknown
SCK period            : 0.0 us
XTAL frequency        : 7.372800 MHz

AVR device initialized and ready to accept instructions
Device signature = 1E 90 07 (ATtiny13, ATtiny13A)

Avrdude done.  Thank you.
```

MCUdude, 2025-05-05:
Great find! The reason why I use `-B32` with MicroCore is because the ATtiny13 may run at its internal 128kHz WDT oscillator. That requires an ISP clock speed or 32kHz or less.

@mcuee are you able to roll back and figure out which commit between the Avrdude 7.3 and 7.2 release that broke `-B32`?

MCUdude, 2025-05-05:
I couldn't wait. it is commit 92c576d942aef477d942fb998d60f2f8e6de6ccd that breaks `-B32`. Confirmed on proper hardware

MCUdude, 2025-05-05:
OK, I went down the rabbit hole to figure out what's going on.

It seems like any bitclock flag will fail. But if I remove this bit of code, it works again.
https://github.com/avrdudes/avrdude/blob/58408592b99608b93ae62ac16525634b3ef9c047/src/stk500.c#L862-L865

However, the code isn't faulty. The issue is that the Arduino as ISP program is a very modest implementation of the stk500v1 protocol, and doesn't support runtime bitclock adjustments. You'll have to stick with the hardcoded bit clock value.

I'll submit a PR soon. My suggestion is to keep the check, but print a warning saying that this particular programmer doesn't support -B-

MCUdude, 2025-05-05:
I don't think it's worth submitting a PR to resolve this until we agree on what's the best way to attack this. The premise is this:

* Apparently, older Avrdude versions did not actually set the correct bit clock value, even when specified by the user. This was fixed in Avrdude 7.3
* The Arduino as ISP will crash if the user uses `-B`

Should we:
A) Warn the user and ignore `-B` if the user uses `-c arduino_as_isp`?
B) Just document that the Arduino as ISP doesn't support `-B` and move on? After all, this is a firmware bug, but the firmware is unlikely to get any updates to fix this.

mcuee, 2025-05-06:
> Should we: 
> A) Warn the user and ignore `-B` if the user uses `-c arduino_as_isp`? 
> B) Just document that the Arduino as ISP doesn't support `-B` and move on? After all, this is a firmware bug, but the firmware is unlikely to get any updates to fix this.

@MCUdude 
My preference is towards `B` because of less work. And maybe there are other `arduino_as_isp` implementation out there which may support `-B`.

MCUdude, 2025-05-06:
@stefanrueger do you have any thoughts regarding this? Try to work around or print a warning and be done with it?

stefanrueger, 2025-05-06:
> The Arduino as ISP will crash if the user uses `-B`

Looks like the `-c arduino_as_isp` programmer caters for, well, Arduinos as ISP programmers with the corresponding FW. In my view, how AVRDUDE *should* behave depends on what's out there. If the known (the true, the most used, ...) Arduino-as-ISP-FW crashes on `AVRDUDE -B ...` then AVRDUDE should *not* offer that option for that programmer. It doesn't matter if it's a bug in the FW, AVRDUDE should still try to work around that bug.

There has been another grumble recently in another threat that AVRDUDE offers `-B` in connection with `-c` programmers that don't actually utilise the `-B` option and does not warn that the programmer won't make use of the `-B` setting. Maybe there should be something in the programmer structure that asserts *Yes, I can make use `-B`*. And AVRDUDE should warn if the user sets `-B` but the programmer cannot use it (or in this case will make the FW crash).

MCUdude, 2025-05-07:
> There has been another grumble recently in another threat that AVRDUDE offers -B in connection with -c programmers that don't actually utilise the -B option and does not warn that the programmer won't make use of the -B setting. Maybe there should be something in the programmer structure that asserts Yes, I can make use of -B. And AVRDUDE should warn if the user sets -B but the programmer cannot use it (or in this case will make the FW crash).

One issue is the programmers who silently accept the -B option but don't adjust their bitclock. Another issue is when Avrdude doesn't actually sets the bitclock at all, like we saw with Avrdude versions prior to 7.3.

`-c arduino_as_isp` is the only programmer I know of that doesn't support -B.

How about we extend the `extra_features` option to include `HAS_BITCLOCK_ADJ`, and includes this on all programmers except for the `arduino_as_isp`?

closed 2025-05-12 by stefanrueger