avrdudes/avrdude #1980: terminal mode "d" command is ambiguous

dl8dtl, 2025-05-07:
```
avrdude> d flash
Error: (cmd) command d is ambiguous: dump, disasm
```
The "d" command used to be a valid abbreviation for "dump". I think it makes sense to add it explicitly as such.

stefanrueger, 2025-05-07:
Well, it's the nature of an expanding command set.

It's been like that since 8.0.

It's effort to document `d`.

There are other, equivalent cases: `r` and `re` are no longer possible owing to the new `restore`. So, terminal `help` would need to explain `d`, `r`, `re`, `dump` and `read` are the same thing. And there would *still* be the rule that the input can use a shorter non-ambiguous name except this now is from the expanded set `d`, `r`, `re`, `dump`, `read`. It all gets messy.

Summarising, it is not great practice from a documentation and maintenance perspective to introduce these special rules. 

I'd rather remove the special lex `q` than add more single-letter special cases

mcuee, 2025-05-17:
@dl8dtl 

The recommendation seems to leave as it is. Any objection to close this issue as not planned?

closed 2025-05-21 by stefanrueger