avrdudes/avrdude #1982: ser_posix.c:  ser_recv times out for low baud rates (e.g 300)

besteinm, 2025-05-10:
I am using avrdude 7.1 (on Debian 12 bookworm) with an optiboot-compatible bootloader (`avrdude -c arduino`).  At very low baud rates (e.g 300) the avrdude terminal mode command `read flash 0 10` times out:

```
$ avrdude -p avr64dd28 -P /dev/ttyUSB0 -c arduino -t -b 300

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e961b (probably avr64dd28)
avrdude>
avrdude>
avrdude> read flash 0 10

Reading | -------------------------------------------------- | 0% 5.27 s

avrdude error: programmer is not responding
avrdude error: unable to read flash page at addr 0x0000
avrdude error: (dump) error reading flash address 0x00000 of part AVR64DD28
               read operation not supported on memory type flash
avrdude>
```

with faster baud rates (e.g 9600 or 115200) the command works as expected.  Since at the low baud rate 300 the initial handshake works and the device signature bytes are received correctly, communication basically seems to work. 

Note that even though I try to read only 10 bytes from flash, avrdude always loads entire pages (512 bytes in this case), as seen with `-vvvv`.

The cause of the early timeout might be found in the file ser_posix.c, function ser_recv, and the way the timeout argument is passed to the `select` function:
 
```
static int ser_recv(const union filedescriptor *fd, unsigned char *buf, size_t buflen) {
  [...]
  timeout.tv_sec = serial_recv_timeout/1000L;
  timeout.tv_usec = (serial_recv_timeout%1000L)*1000;
  to2 = timeout;  <----------------------------------------------

  while(len < buflen) {
  reselect:
    FD_ZERO(&rfds);
    FD_SET(fd->ifd, &rfds);

    nfds = select(fd->ifd + 1, &rfds, NULL, NULL, &to2);  <---------------
    [...]
}
```

from the `select` man page:
```
On  Linux,  select()  modifies  timeout to reflect the amount of time not slept; most other implementations
do not do this.  (POSIX.1 permits either behavior.)
```

Now if on my particular system `select` modifies the timeout, then at 300 baud the timeout value might be reduced by approximately 30 ms in each iteration of the `reselect` loop.  Given page size 512, this results in a total transfer time of about 15 seconds which is above the `serial_recv_timeout` value 5000 milliseconds.  That would also explain why retrieving the signature works (only 3 bytes are transferred).

I guess the assignment `to2 = timeout;` was intended to be put inside the loop (otherwise what would be the reason to have `to2` in the first place?):

```
static int ser_recv(const union filedescriptor *fd, unsigned char *buf, size_t buflen) {
  [...]
  timeout.tv_sec = serial_recv_timeout/1000L;
  timeout.tv_usec = (serial_recv_timeout%1000L)*1000;

  while(len < buflen) {
    to2 = timeout;
  reselect:
    FD_ZERO(&rfds);
    FD_SET(fd->ifd, &rfds);

    nfds = select(fd->ifd + 1, &rfds, NULL, NULL, &to2);

```

best regards
Benoit.

stefanrueger, 2025-05-10:
> I guess the assignment `to2 = timeout;` was intended to be put inside the loop

What happens when you move this assignment into the loop? Does your 300 baud comms work OK after this intervention?

stefanrueger, 2025-05-10:
Note that most bootloaders have a watchdog timeout where $n$ seconds inactivity (typ $n=1...8$) starts the application. `-c urclock` as opposed to `-c arduino` carries out nop commands every 100 ms or so to reset the bootloader's watchdog timer. But it may well be that `-c urclock` does not work with your bootloader.

You can ensure no wait time in the terminal with piping:
```
$ echo read flash 0 10 | avrdude -p avr64dd28 -P /dev/ttyUSB0 -c arduino -t -b 300
```

stefanrueger, 2025-05-11:
I can confirm this effect using a 16-MHz ATmega328P board with an autobaud urboot bootloader. First burn the bootloader using a programmer of your choice:

```
$ avrdude -p m328p -c usbasp -U urboot:best_autobaud_ee
Reading 382 bytes for flash from input file urboot:best_autobaud_ee
Writing 382 bytes to flash
Writing | ################################################## | 100% 0.16 s 
Reading | ################################################## | 100% 0.08 s 
382 bytes of flash verified
Setting fuses for bootloader urboot:best_autobaud_ee

Avrdude done.  Thank you.
```

Then connect the board through the serial interface. At 16 MHz F_CPU autobaud catches baud rates [down to 9600 baud](https://github.com/stefanrueger/urboot/blob/main/docs/makeoptions.md#serial-communication):

```
$ avrdude -qq -c urclock -P ch340 -x showall
00000bd39209 0000-00-00 00.00  application 0 store 0 meta 1 boot 384 u8.0 weU-jPrac vector 25 (SPM_Ready) ATmega328P

$ avrdude -qq -c urclock -P ch340 -T "read flash 0 10" -b 19200
0000  3f cf 75 72 ff ff ff ff  ff ff                    |?.ur......      |

$ avrdude -qq -c urclock -P ch340 -T "read flash 0 10" -b 9600
0000  3f cf 75 72 ff ff ff ff  ff ff                    |?.ur......      |
```

At 9600 baud downloading one page of 128 bytes takes 133 ms (1280 serial bits at 8N1). And yes, the terminal uses pagewise I/O for its cache, so even if only 10 bytes are read in the terminal a page will be read from the MCU. So, setting the timeout to 100 ms should fail at 9600 baud and  be OK at 19200 baud. Indeed, limiting the timeout to 100 ms
``` diff
diff --git a/src/ser_posix.c b/src/ser_posix.c
index b71028f8..ec812f69 100644
--- a/src/ser_posix.c
+++ b/src/ser_posix.c
@@ -449,8 +449,8 @@ static int ser_recv(const union filedescriptor *fd, unsigned char *buf, size_t b
   unsigned char *p = buf;
   size_t len = 0;
 
-  timeout.tv_sec = serial_recv_timeout/1000L;
-  timeout.tv_usec = (serial_recv_timeout%1000L)*1000;
+  timeout.tv_sec = 0;
+  timeout.tv_usec = 100*1000;
   to2 = timeout;
 
   while(len < buflen) {
```

we now get

```
$ avrdude -qq -c urclock -P ch340 -T "read flash 0 10" -b 19200
0000  3f cf 75 72 ff ff ff ff  ff ff                    |?.ur......      |

$ avrdude -qq -c urclock -P ch340 -T "read flash 0 10" -b 9600
Warning: programmer is not responding
Error: protocol expects sync byte 0xa0 but got 0xff in ur_readEF()
Error: unable to read flash page at addr 0x0000
Error: (read) error reading flash address 0x00000 of part ATmega328P
Error: protocol expects sync byte 0xa0 but got 0x77
```

Moving the `to2` assignment into the while loop 
```
diff --git a/src/ser_posix.c b/src/ser_posix.c
index b71028f8..8baebdcf 100644
--- a/src/ser_posix.c
+++ b/src/ser_posix.c
@@ -449,11 +449,11 @@ static int ser_recv(const union filedescriptor *fd, unsigned char *buf, size_t b
   unsigned char *p = buf;
   size_t len = 0;
 
-  timeout.tv_sec = serial_recv_timeout/1000L;
-  timeout.tv_usec = (serial_recv_timeout%1000L)*1000;
-  to2 = timeout;
+  timeout.tv_sec = 0;
+  timeout.tv_usec = 100*1000;
 
   while(len < buflen) {
+    to2 = timeout;
   reselect:
     FD_ZERO(&rfds);
     FD_SET(fd->ifd, &rfds);
```

resolves the problem:

```
$ avrdude -qq -c urclock -P ch340 -T "read flash 0 10" -b 9600
0000  3f cf 75 72 ff ff ff ff  ff ff                    |?.ur......      |
```

So, yes, looks like this is a bug and we also have a potential PR.

stefanrueger, 2025-05-11:
Benoit, thanks for reporting and, actually, identifying the issue! It's a 20-year-old bug that only now has seen the light of day. I *think*(!) I understand what happens, and that aligns with your analysis. The difficulty we face here is that we normally want to avoid regression, ie, a negative effect of changing behaviour that someone or some software relies on.  Right now I cannot imagine what else might stop working by re-triggering the timeout with every byte read but I have learned to tread carefully. But yes, we'll carefully evaluate the PR, and, I predict apply it sometime. It might be after releasing v8.1 as the code freeze for that is imminent.

PR #1983 (2025-05-12): Retrigger `ser_posix.c` `ser_recv()` timeout
Fixes #1982.

Will need through testing against regression. Also, #1982 should be checked in other OS and this PR should be tested in POSIX-type OSes.
closed 2025-05-12 by stefanrueger