avrdudes/avrdude #1985: auto-tools build broken for OpenBSD

mcuee, 2025-05-11:
Linking step failed under OpenBSD 7.7. CMake build has no issues.

```
openbsd77vm$ make
make  all-recursive
Making all in .
  CCLD     avrdude
ld: warning: main.c(avrdude-main.o:(main)): warning: strcpy() is almost always misused, please use strlcpy()
ld: warning: main.c(avrdude-main.o:(main)): warning: strcat() is almost always misused, please use strlcat()
ld: warning: developer_opts.c(avrdude-developer_opts.o:(opsnm)): warning: sprintf() is often misused, please use snprintf()
ld: error: undefined symbol: cfg_malloc
>>> referenced by main.c
>>>               avrdude-main.o:(avrdude_message2)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_sprintf)
>>> referenced by whereami.c
>>>               avrdude-whereami.o:(wai_getExecutablePath)
>>> referenced 1 more times

ld: error: undefined symbol: init_cx
>>> referenced by main.c
>>>               avrdude-main.o:(main)

ld: error: undefined symbol: str_ends
>>> referenced by main.c
>>>               avrdude-main.o:(main)

ld: error: undefined symbol: cfg_strdup
>>> referenced by main.c
>>>               avrdude-main.o:(main)
>>> referenced by main.c
>>>               avrdude-main.o:(main)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_part)
>>> referenced 35 more times

ld: error: undefined symbol: avrdude_conf_version
>>> referenced by main.c
>>>               avrdude-main.o:(main)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_part)
>>> referenced by main.c
>>>               avrdude-main.o:(main)
>>> referenced 1 more times

ld: error: undefined symbol: default_spi
>>> referenced by main.c
>>>               avrdude-main.o:(main)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_part)

ld: error: undefined symbol: default_baudrate
>>> referenced by main.c
>>>               avrdude-main.o:(main)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_part)
>>> referenced by main.c
>>>               avrdude-main.o:(main)
>>> referenced 4 more times

ld: error: undefined symbol: default_linuxgpio
>>> referenced by main.c
>>>               avrdude-main.o:(main)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_part)
>>> referenced by main.c
>>>               avrdude-main.o:(main)

ld: error: undefined symbol: allow_subshells
>>> referenced by main.c
>>>               avrdude-main.o:(main)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_part)

ld: error: undefined symbol: str_int
>>> referenced by main.c
>>>               avrdude-main.o:(main)
>>> referenced by main.c
>>>               avrdude-main.o:(main)

ld: error: undefined symbol: str_caseeq
>>> referenced by main.c
>>>               avrdude-main.o:(main)
>>> referenced by main.c
>>>               avrdude-main.o:(main)
>>> referenced by main.c
>>>               avrdude-main.o:(main)
>>> referenced 4 more times

ld: error: undefined symbol: str_contains
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_has_subsstr_comms)

ld: error: undefined symbol: cmd_update
>>> referenced by main.c
>>>               avrdude-main.o:(main)
>>> referenced by main.c
>>>               avrdude-main.o:(main)

ld: error: undefined symbol: cfg_get_prologue
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_part)

ld: error: undefined symbol: cfg_escape
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_part)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_part)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_part)
>>> referenced 13 more times

ld: error: undefined symbol: str_eq
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_defs)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_defs)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_defs)
>>> referenced 31 more times

ld: error: undefined symbol: str_casematch
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_defs)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_part_defs)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_part_strct)
>>> referenced 4 more times

ld: error: undefined symbol: locate_programmer_type_id
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_defs)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_pgm_strct)

ld: error: undefined symbol: dev_prog_modes
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_defs)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_part_defs)
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_part_strct)
>>> referenced 1 more times

ld: error: undefined symbol: str_ccsharg
>>> referenced by main.c
>>>               avrdude-main.o:(main)

ld: error: too many errors emitted, stopping now (use --error-limit=0 to see all errors)
cc: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error 1 in . (Makefile:968 'avrdude': @echo "  CCLD    " avrdude;/bin/sh ./libtool --silent --tag=CC    --mode=link cc -Wall -Wextra -Wn...)
*** Error 1 in . (Makefile:1851 'all-recursive': @fail=;  if (target_option=k; case ${target_option-} in  ?) ;;  *) echo "am__make_running_w...)
*** Error 2 in /home/mcuee/build/avr/avrdude/src (Makefile:816 'all')

openbsd77vm$ uname -a
OpenBSD openbsd77vm.my.domain 7.7 GENERIC.MP#625 amd64
openbsd77vm$ cc -v
OpenBSD clang version 16.0.6
Target: amd64-unknown-openbsd7.7
Thread model: posix
InstalledDir: /usr/bin

openbsd77vm$ automake --version                                                                                                                                                        
automake (GNU automake) 1.16.5
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv2+: GNU GPL version 2 or later <https://gnu.org/licenses/gpl-2.0.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Tom Tromey <tromey@redhat.com>
       and Alexandre Duret-Lutz <adl@gnu.org>.

openbsd77vm$ autoconf --version                                                                                                                                                        
autoconf (GNU Autoconf) 2.71
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+/Autoconf: GNU GPL version 3 or later
<https://gnu.org/licenses/gpl.html>, <https://gnu.org/licenses/exceptions.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by David J. MacKenzie and Akim Demaille.

openbsd77vm$ libtool --version
libtool (not (GNU libtool)) 1.5.26
```

mcuee, 2025-05-11:
Full build log.

<details>

```
openbsd77vm$ ./bootstrap                                                                                                                                                               
+ rm -rf autom4te.cache
+ LIBTOOLIZE=libtoolize
+ glibtoolize --version
+ > /dev/null 
+ 2>&1 
+ export LIBTOOLIZE
+ autoreconf -i
openbsd77vm$ CFLAGS=-I/usr/local/include LDFLAGS=-L/usr/local/lib ./configure 
checking build system type... x86_64-unknown-openbsd7.7
checking host system type... x86_64-unknown-openbsd7.7
checking target system type... x86_64-unknown-openbsd7.7
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... ./auto-aux/install-sh -c -d
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking versioninfo item CMAKE_PROJECT_VERSION... 8.0
checking versioninfo item CMAKE_LIBAVRDUDE_VERSION... 3.0.0
checking versioninfo item CMAKE_LIBAVRDUDE_SOVERSION... 3
checking versioninfo item GIT_COMMIT_DATE... 20250504
checking versioninfo item GIT_COMMIT_HASH... 58408592
checking versioninfo item GIT_TAG_HASH... d8c8cf13
checking versioninfo derived PACKAGE_VERSION... 8.0-20250504
checking versioninfo derived AVRDUDE_FULL_VERSION... 8.0-20250504 (58408592)
checking versioninfo derived libtool -version-info for libavrdude... 3:0:0
checking for gcc... no
checking for cc... cc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether cc accepts -g... yes
checking for cc option to enable C11 features... none needed
checking whether cc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of cc... gcc3
checking for gawk... (cached) gawk
checking for a sed that does not truncate output... /usr/bin/sed
checking for bison... bison -y
checking for flex... flex
checking for lex output file root... lex.yy
checking for lex library... none needed
checking whether yytext is a pointer... yes
checking for ar... ar
checking the archiver (ar) interface... ar
checking how to print strings... print -r
checking for a sed that does not truncate output... (cached) /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by cc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 393216
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... no
checking how to convert x86_64-unknown-openbsd7.7 file names to x86_64-unknown-openbsd7.7 format... func_convert_file_noop
checking how to convert x86_64-unknown-openbsd7.7 file names to toolchain format... func_convert_file_noop
[... 324 lines cut ...]
>>> referenced by main.c
>>>               avrdude-main.o:(main)

ld: error: undefined symbol: jtag3_updi_initpgm
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_defs)

ld: error: undefined symbol: jtag3_tpi_initpgm
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_defs)

ld: error: undefined symbol: str_ccprintf
>>> referenced by developer_opts.c
>>>               avrdude-developer_opts.o:(dev_output_pgm_defs)
>>> referenced by main.c
>>>               avrdude-main.o:(part_ccdesc)

ld: error: too many errors emitted, stopping now (use --error-limit=0 to see all errors)
cc: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error 1 in . (Makefile:969 'avrdude': @echo "  CCLD    " avrdude;/bin/sh ./libtool --silent --tag=CC    --mode=link cc -Wall -Wextra -Wn...)
*** Error 1 in . (Makefile:1852 'all-recursive': @fail=;  if (target_option=k; case ${target_option-} in  ?) ;;  *) echo "am__make_running_w...)
*** Error 2 in /home/mcuee/build/avr/avrdude/src (Makefile:817 'all')
```

</details>

mcuee, 2025-05-11:
No issues under FreeBSD 14.2 Release and NetBSD 10.1 Release.

mcuee, 2025-05-11:
Solution found --> need to use system provided LIBTOOL.

```
openbsd76vm$ make LIBTOOL=/usr/bin/libtool 
  YACC     config_gram.c
updating config_gram.h
  LEX      lexer.c
make  all-recursive
Making all in .
  CC       avrdude-main.o
  CC       avrdude-whereami.o
  CC       avrdude-developer_opts.o
  CC       libavrdude_la-config_gram.lo
  CC       libavrdude_la-lexer.lo
  CC       libavrdude_la-arduino.lo
  CC       libavrdude_la-avr.lo
  CC       libavrdude_la-avr910.lo
  CC       libavrdude_la-avrcache.lo
  CC       libavrdude_la-avrftdi.lo
  CC       libavrdude_la-avrftdi_tpi.lo
  CC       libavrdude_la-avrintel.lo
  CC       libavrdude_la-avrpart.lo
  CC       libavrdude_la-avr_opcodes.lo
  CC       libavrdude_la-bitbang.lo
  CC       libavrdude_la-buspirate.lo
  CC       libavrdude_la-butterfly.lo
  CC       libavrdude_la-ch341a.lo
  CC       libavrdude_la-config.lo
  CC       libavrdude_la-confwin.lo
  CC       libavrdude_la-crc16.lo
  CC       libavrdude_la-disasm.lo
  CC       libavrdude_la-dfu.lo
  CC       libavrdude_la-dryrun.lo
  CC       libavrdude_la-fileio.lo
  CC       libavrdude_la-flip1.lo
  CC       libavrdude_la-flip2.lo
  CC       libavrdude_la-ft245r.lo
  CC       libavrdude_la-jtagmkI.lo
  CC       libavrdude_la-jtagmkII.lo
  CC       libavrdude_la-jtag3.lo
  CC       libavrdude_la-leds.lo
  CC       libavrdude_la-linuxgpio.lo
  CC       libavrdude_la-linuxspi.lo
  CC       libavrdude_la-lists.lo
  CC       libavrdude_la-micronucleus.lo
  CC       libavrdude_la-par.lo
  CC       libavrdude_la-pgm.lo
  CC       libavrdude_la-pgm_type.lo
  CC       libavrdude_la-pickit2.lo
  CC       libavrdude_la-pickit5_lut_dw.lo
  CC       libavrdude_la-pickit5_lut_isp.lo
  CC       libavrdude_la-pickit5_lut_jtag.lo
  CC       libavrdude_la-pickit5_lut_pdi.lo
  CC       libavrdude_la-pickit5_lut_tpi.lo
  CC       libavrdude_la-pickit5_lut_updi.lo
  CC       libavrdude_la-pickit5.lo
  CC       libavrdude_la-pindefs.lo
  CC       libavrdude_la-ppi.lo
  CC       libavrdude_la-serbb_posix.lo
  CC       libavrdude_la-serbb_win32.lo
  CC       libavrdude_la-ser_avrdoper.lo
  CC       libavrdude_la-ser_posix.lo
  CC       libavrdude_la-ser_win32.lo
  CC       libavrdude_la-serialadapter.lo
  CC       libavrdude_la-stk500.lo
  CC       libavrdude_la-stk500v2.lo
  CC       libavrdude_la-stk500generic.lo
  CC       libavrdude_la-strutil.lo
  CC       libavrdude_la-teensy.lo
  CC       libavrdude_la-term.lo
  CC       libavrdude_la-usbasp.lo
  CC       libavrdude_la-serialupdi.lo
  CC       libavrdude_la-serprog.lo
  CC       libavrdude_la-updi_link.lo
  CC       libavrdude_la-updi_state.lo
  CC       libavrdude_la-updi_readwrite.lo
  CC       libavrdude_la-updi_nvm.lo
  CC       libavrdude_la-updi_nvm_v0.lo
  CC       libavrdude_la-updi_nvm_v2.lo
  CC       libavrdude_la-updi_nvm_v3.lo
  CC       libavrdude_la-updi_nvm_v4.lo
  CC       libavrdude_la-updi_nvm_v5.lo
  CC       libavrdude_la-urbootlist.lo
  CC       libavrdude_la-urbootautogen.lo
  CC       libavrdude_la-urclock.lo
  CC       libavrdude_la-usb_hidapi.lo
  CC       libavrdude_la-usb_libusb.lo
  CC       libavrdude_la-usbtiny.lo
  CC       libavrdude_la-update.lo
  CC       libavrdude_la-wiring.lo
  CC       libavrdude_la-xbee.lo
  CCLD     libavrdude.la
  CCLD     avrdude
ld: warning: developer_opts.c(avrdude-developer_opts.o:(dev_output_pgm_defs)): warning: strcpy() is almost always misused, please use strlcpy()
ld: warning: main.c(avrdude-main.o:(main)): warning: strcat() is almost always misused, please use strlcat()
ld: warning: developer_opts.c(avrdude-developer_opts.o:(opsnm)): warning: sprintf() is often misused, please use snprintf()
```

mcuee, 2025-05-11:
Reference from curl.
* https://github.com/curl/curl/issues/5862

mcuee, 2025-05-12:
Wiki updated.
https://github.com/avrdudes/avrdude/wiki/General-Build-Instructions-Using-Auto%E2%80%90tools

closed 2025-05-12 by mcuee

stefanrueger, 2025-05-12:
> wiki updated

Is there a `make install` step as well? Or perhaps say where the binaries tend to be and/or how to call the compiled code

mcuee, 2025-05-12:
> > wiki updated
> 
> Is there a `make install` step as well? Or perhaps say where the binaries tend to be and/or how to call the compiled code

Updated.