avrdudes/avrdude #1986: avrdude hangs if valid serial port without serial programmer attached is specified

nomakewan, 2025-05-12:
Recently while helping a user they mentioned that they had accidentally selected the wrong serial port to flash and that when they did, the program seemed to hang. When I dug deeper I found that the hang was in avrdude; it was detecting a timeout initially, but then it would retry and the second try would _not_ time out, instead hanging indefinitely. Forcing it to exit worked but also left the port in an odd state; if you ran avrdude against that same port a second time, this time the first try would not time out and instead would hang indefinitely.

Here is the command I am passing to avrdude for testing:

`avrdude -V -vvvv -p m2560 -c wiring -P COM22 -b 115200 -D -U flash:w:"test.hex":i`

And here is the result (top is what happens when starting from a "bad state", bottom full text is starting from a fresh state):
[image]

I also tried the older version of avrdude built into the Arduino IDE and when selecting this port without a programmer attached the same behavior happened (the initial attempt times out, but then avrdude hangs until manually killed).

I would expect that avrdude would time out and either retry a number of times then exit with an error code, or would just exit with an error code after the first attempt fails. In either case I would expect the serial port to be properly closed before avrdude exits to prevent the port from being left in an odd state.

Thank you for any assistance you can provide.

stefanrueger, 2025-05-12:
Thanks for reporting. The team is likely to need more information such as OS, how the binary was compiled etc. Can I suggest to capture the full debug output using `-l debug.log` and [uploading the `debug.log` file](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/attaching-files) here, so we can download and analyse. Alternatively, cut and paste the complete debug log into comment box using the details markup and three-backticks section for literal display.

````
<details>
<summary> My optional headline </summary>

```
Cut and paste the full log
```

</details>
````

How long have you waited? Note the timeout can be in the order of tens of seconds... 

@mcuee Can you reproduce the issue?

mcuee, 2025-05-12:
@stefanrueger and @nomakewan 

I cannot reproduce the issue under Windows 11 24H2.

```
PS C:\work\avr\avrdude_bin> .\avrdude.exe -C .\avrdude.conf -c wiring -P COM3 -b 115200 -p m2560 -D -V -U .\Blink.ino_atmega2560_16000000L.hex
Reading 1500 bytes for flash from input file Blink.ino_atmega2560_16000000L.hex
Writing 1500 bytes to flash
Writing | ################################################## | 100% 0.22 s
1500 bytes of flash written
Avrdude done.  Thank you.

PS C:\work\avr\avrdude_bin> .\avrdude.exe -C .\avrdude.conf -c wiring -P COM100 -b 115200 -p m2560 -D -V -U .\Blink.ino_atmega2560_16000000L.hex
Error: cannot open port \\.\COM100: The system cannot find the file specified.

Error: unable to read: The handle is invalid.

Error: unable to write: sorry no info avail
Error: unable to send command to serial port
Error: unable to read: The handle is invalid.

Error: timeout
Error: unable to write: sorry no info avail
Error: unable to send command to serial port
Error: unable to read: The handle is invalid.

Error: timeout
Error: unable to write: sorry no info avail
Error: unable to send command to serial port
Error: unable to read: The handle is invalid.

Error: timeout
Error: unable to write: sorry no info avail
Error: unable to send command to serial port
Error: unable to read: The handle is invalid.

Error: timeout
Error: unable to write: sorry no info avail
Error: unable to send command to serial port
Error: unable to read: The handle is invalid.

Error: timeout
Error: unable to write: sorry no info avail
Error: unable to send command to serial port
Error: unable to read: The handle is invalid.

Error: timeout
Error: timeout communicating with programmer
Error: stk500v2_getsync() failed; try -x delay=n with some n in [-80, 100]
Error: unable to open port COM100 for programmer wiring
A possible candidate serial port is:
  -P COM3 or -P ch340
Note that above port might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.

Avrdude done.  Thank you.
```

mcuee, 2025-05-12:
@nomakewan 
BTW, it is good to abandon the wiring bootloader which is notoriously buggy. 

Example: there is a known issue with EEPROM and there is a fix since June 2018. But the fix has not been merged.
* https://github.com/arduino/ArduinoCore-avr/issues/129
* https://github.com/arduino/ArduinoCore-avr/pull/24

@MCUdude has incorporated the above EEPROM fix in his repo.
* https://github.com/MCUdude/MegaCore/issues/187

But there are still other issues. Therefore @MCUdude decided to abondan it and go with the much better urboot.
* https://github.com/MCUdude/MegaCore/issues/201
* https://github.com/arduino/ArduinoCore-avr/issues?q=is%3Aissue+is%3Aopen+stk500v2

urboot is by @stefanrueger and it is really good for classic AVRs.
https://github.com/stefanrueger/urboot

mcuee, 2025-05-12:
Still no issues from my side.

COM3 is the right port.
COM4 is another USB to Serial Adapter (FT232R).

[image]

```
(mypy312venv) PS C:\work\avr\avrdude_bin> .\avrdude80.exe -C .\avrdude80.conf -c wiring -P COM4 -b 115200 -p m2560 -D -V -U .\Blink.ino_atmega2560_16000000L.hex -v
Avrdude80 version 8.0
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is C:\work\avr\avrdude_bin\avrdude80.conf

Using port            : COM4
Using programmer      : wiring
Setting baud rate     : 115200
Error: timeout
Error: timeout
Error: timeout
Error: timeout
Error: timeout
Error: timeout
Error: timeout communicating with programmer
Error: stk500v2_getsync() failed; try -x delay=n with some n in [-80, 100]
Error: unable to open port COM4 for programmer wiring

Avrdude80 done.  Thank you.

(mypy312venv) PS C:\work\avr\avrdude_bin> .\avrdude80.exe -C .\avrdude80.conf -c wiring -P COM4 -b 115200 -p m2560 -D -V -U .\Blink.ino_atmega2560_16000000L.hex -vvvv
$ 'C:\work\avr\avrdude_bin\avrdude80.exe' -C '.\avrdude80.conf' -c wiring -P COM4 -b 115200 -p m2560 -D -V -U '.\Blink.ino_atmega2560_16000000L.hex' -vvvv
Avrdude80 version 8.0
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is C:\work\avr\avrdude_bin\avrdude80.conf

Using port            : COM4
Using programmer      : wiring
Setting baud rate     : 115200
Wiring_open(): releasing DTR/RTS
Wiring_open(): asserting DTR/RTS
Ser_send: . [1b] . [01] . [00] . [01] . [0e] . [01] . [14]
Ser_recv(): programmer is not responding
Error stk500v2_recv() stk500v2.c 692: timeout
Ser_send: . [1b] . [01] . [00] . [01] . [0e] . [01] . [14]
Ser_recv(): programmer is not responding
Error stk500v2_recv() stk500v2.c 692: timeout
Ser_send: . [1b] . [01] . [00] . [01] . [0e] . [01] . [14]
Ser_recv(): programmer is not responding
Error stk500v2_recv() stk500v2.c 692: timeout
Ser_send: . [1b] . [01] . [00] . [01] . [0e] . [01] . [14]
Ser_recv(): programmer is not responding
Error stk500v2_recv() stk500v2.c 692: timeout
Ser_send: . [1b] . [01] . [00] . [01] . [0e] . [01] . [14]
Ser_recv(): programmer is not responding
Error stk500v2_recv() stk500v2.c 692: timeout
Ser_send: . [1b] . [01] . [00] . [01] . [0e] . [01] . [14]
Ser_recv(): programmer is not responding
Error stk500v2_recv() stk500v2.c 692: timeout
Error stk500v2_getsync() stk500v2.c 766: timeout communicating with programmer
Error wiring_open() wiring.c 179: stk500v2_getsync() failed; try -x delay=n with some n in [-80, 100]
Error main() main.c 1450: unable to open port COM4 for programmer wiring

Avrdude80 done.  Thank you.
```

mcuee, 2025-05-12:
No issues with urboot either.
```
(mypy312venv) PS C:\work\avr\avrdude_bin> .\avrdude80.exe -C .\avrdude80.conf -c urclock -P COM4 -b 115200 -p m2560 -D -V -U .\Blink.ino_atmega2560_16000000L.hex -vvvv
$ 'C:\work\avr\avrdude_bin\avrdude80.exe' -C '.\avrdude80.conf' -c urclock -P COM4 -b 115200 -p m2560 -D -V -U '.\Blink.ino_atmega2560_16000000L.hex' -vvvv
Avrdude80 version 8.0
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is C:\work\avr\avrdude_bin\avrdude80.conf

Libreadline is used; avrdude -t -c urclock should work interactively
Using port            : COM4
Using programmer      : urclock
Setting baud rate     : 115200
 181 ms: enter urclock_getsync()
Ser_send: 0 [30]   [20]
Ser_recv(): programmer is not responding
 206 ms: sleeping for 32 ms
Ser_send:   [20]   [20]
Ser_recv(): programmer is not responding
 263 ms: sleeping for 64 ms
Ser_send:   [20]   [20]
Ser_recv(): programmer is not responding
 353 ms: sleeping for 128 ms
Ser_send:   [20]   [20]
Ser_recv(): programmer is not responding
 507 ms: sleeping for 256 ms
Ser_send:   [20]   [20]
Ser_recv(): programmer is not responding
 788 ms: sleeping for 256 ms
Ser_send:   [20]   [20]
Ser_recv(): programmer is not responding
1069 ms: sleeping for 256 ms
Ser_send:   [20]   [20]
Ser_recv(): programmer is not responding
1351 ms: sleeping for 256 ms
Warning urclock_getsync() urclock.c 1994: attempt 1 of 10: not in sync
Ser_send:   [20]   [20]
Ser_recv(): programmer is not responding
1633 ms: sleeping for 256 ms
Warning urclock_getsync() urclock.c 1994: attempt 2 of 10: not in sync
Ser_send:   [20]   [20]
Ser_recv(): programmer is not responding
1915 ms: sleeping for 256 ms
Warning urclock_getsync() urclock.c 1994: attempt 3 of 10: not in sync
Ser_send:   [20]   [20]
Ser_recv(): programmer is not responding
2198 ms: sleeping for 256 ms
Warning urclock_getsync() urclock.c 1994: attempt 4 of 10: not in sync
Ser_send:   [20]   [20]
Ser_recv(): programmer is not responding
2480 ms: sleeping for 256 ms
Warning urclock_getsync() urclock.c 1994: attempt 5 of 10: not in sync
Ser_send:   [20]   [20]
Ser_recv(): programmer is not responding
2762 ms: sleeping for 256 ms
Warning urclock_getsync() urclock.c 1994: attempt 6 of 10: not in sync
Ser_send:   [20]   [20]
Ser_recv(): programmer is not responding
3045 ms: sleeping for 256 ms
Warning urclock_getsync() urclock.c 1994: attempt 7 of 10: not in sync
Ser_send:   [20]   [20]
Ser_recv(): programmer is not responding
3326 ms: sleeping for 256 ms
Warning urclock_getsync() urclock.c 1994: attempt 8 of 10: not in sync
Ser_send:   [20]   [20]
Ser_recv(): programmer is not responding
3610 ms: sleeping for 256 ms
Warning urclock_getsync() urclock.c 1994: attempt 9 of 10: not in sync
Ser_send:   [20]   [20]
Ser_recv(): programmer is not responding
3891 ms: sleeping for 256 ms
Warning urclock_getsync() urclock.c 1994: attempt 10 of 10: not in sync
Ser_send:   [20]
Ser_recv(): programmer is not responding
Warning urclock_recv() urclock.c 1916: programmer is not responding; try -x strict and/or vary -x delay=100
Ser_send:   [20]
Error main() main.c 1450: unable to open port COM4 for programmer urclock

Avrdude80 done.  Thank you.
```

nomakewan, 2025-05-12:
Here is a serial port null modem emulator you can try that easily replicates the issue if you target one of the COM ports it creates:

https://sourceforge.net/projects/com0com

Perhaps it is related to being a null modem? It still does not seem ideal that there are valid configurations which can cause avrdude to hang rather than time out.

mcuee, 2025-05-12:
@nomakewan 

Unfortunately com0com no longer works under my computer due to digital signature issues.

Are you able to reproduce the issue other than using com0com?

nomakewan, 2025-05-12:
> Are you able to reproduce the issue other than using a real COM port and not a virtual serial port like com0com?

Could you clarify what you mean by "other than using a real COM port and not a virtual serial port"? What alternative do you suggest I test?

When connecting a USB serial port, in all cases avrdude does what it is supposed to (I tested an FTDI cable from Adafruit, a generic USB-C TTL adapter, and the USB-C port on an ESP32 and in all cases the timeout worked properly).

If you would like me to acquire and install a physical PCI serial port, I would be more than happy to do so (though it may take a few days).

My concern is that while a USB serial port may be working correctly, clearly there is some other type of valid port configuration in Windows which can cause avrdude to hang, and that this is not desirable.

I will also ask the user who reported this issue to us in the first place to see what serial device he had installed in his environment that caused this issue for him when he accidentally selected it as the avrdude target port. It is just odd that I was able to so easily reproduce it based solely on his report that it hung after he accidentally selected the wrong port.

stefanrueger, 2025-05-12:
@nomakewan Thanks for the logs. On the face of it it looks like AVRDUDE hangs after tracing the send buffer:
https://github.com/avrdudes/avrdude/blob/127f40ee6d74e23e241e71c8a53a67faf1ad2cea/src/ser_win32.c#L343-L351

I don't know Windows low-level routines *at all*, but these lines look like they are not meant to block. As currently you are the only person able to reproduce the observed behaviour, could you confirm AVRDUDE hangs either here
https://github.com/avrdudes/avrdude/blob/127f40ee6d74e23e241e71c8a53a67faf1ad2cea/src/ser_win32.c#L81
(that's the return of the `serial_w32SetTimeOut()`) or here
https://github.com/avrdudes/avrdude/blob/127f40ee6d74e23e241e71c8a53a67faf1ad2cea/src/ser_win32.c#L348
Maybe use a debugger or maybe sprinkle poverty-striken develeoper's print statements around these places (sth along the line of `fflush(stderr); printf("here a/b/c/d/...\n"); fflush(stdout);`), recompile and run again.

Once we have identified where AVRDUDE hangs, it's a matter of hitting the manuals and figuring out what's wrong with the code and how the code could have avoided these hangs.

And, of course, there is always the possibility that the specific serial interface came with rogue device driver code ...

nomakewan, 2025-05-12:
@stefanrueger Thank you! I have made the requested changes as follows:

```c
static BOOL serial_w32SetTimeOut(HANDLE hComPort, DWORD timeout) { // ms
  COMMTIMEOUTS ctmo;

  ZeroMemory(&ctmo, sizeof(COMMTIMEOUTS));
  ctmo.ReadIntervalTimeout = 0;
  ctmo.ReadTotalTimeoutMultiplier = 0;
  ctmo.ReadTotalTimeoutConstant = timeout;

  fflush(stderr);
  printf("before SetCommTimeouts(hComPort, &ctmo)\n");
  fflush(stdout);

  return SetCommTimeouts(hComPort, &ctmo);
}
```

```c
static int ser_send(const union filedescriptor *fd, const unsigned char *buf, size_t len) {
  if(cx->ser_serial_over_ethernet)
    return net_send(fd, buf, len);

  DWORD written;

  HANDLE hComPort = (HANDLE) fd->pfd;

  if(hComPort == INVALID_HANDLE_VALUE) {
    pmsg_error("port not open\n");
    return -1;
  }

  if(!len)
    return 0;

  if(verbose >= MSG_TRACE)
    trace_buffer(__func__, buf, len);

  serial_w32SetTimeOut(hComPort, 500);

  fflush(stderr);
  printf("if(!WriteFile(hComPort, buf, len, &written, NULL))\n");
  fflush(stdout);

  if(!WriteFile(hComPort, buf, len, &written, NULL)) {
    pmsg_error("unable to write: %s\n", "sorry no info avail"); // TODO
    return -1;
  }

  if(written != len) {
    pmsg_error("size/send mismatch\n");
    return -1;
  }

  return 0;
}
```

The result:
[image]

Hopefully this is what you were looking for?

stefanrueger, 2025-05-12:
Good. It's probably `WriteFile()` that hangs... You can verify this by inserting the following
``` diff
diff --git a/src/ser_win32.c b/src/ser_win32.c
index 8a53ad46..b4aa1c60 100644
--- a/src/ser_win32.c
+++ b/src/ser_win32.c
@@ -350,6 +350,8 @@ static int ser_send(const union filedescriptor *fd, const unsigned char *buf, si
     return -1;
   }
 
+  fflush(stderr); printf("After WriteFile()\n"); fflush(stdout);
+
   if(written != len) {
     pmsg_error("size/send mismatch\n");
     return -1;
```

OK, and if it's `WriteFile()` the next question is why. Maybe `serial_w32SetTimeOut(hComPort, 500);` needs to set some write timeouts as well if that `COMMTIMEOUTS ctmo;` has these components... That's where my knowledge ends. I don't know Microsoft low level I/O to progress this. Never used Windows...

nomakewan, 2025-05-12:
I have made the requested change, and here was the output:
[image]

(I also cleaned up the long mess debug string from earlier to just say "before writeFile" instead)

stefanrueger, 2025-05-12:
OK, it's the `WriteFile()` that hangs. Ms Google tells me that the [`COMMTIMEOUTS` structure has Write timeouts too](https://learn.microsoft.com/en-us/windows/win32/api/winbase/ns-winbase-commtimeouts). So, my hunch is to make the following intervention:
``` diff
diff --git a/src/ser_win32.c b/src/ser_win32.c
index 8a53ad46..5ec6f1c4 100644
--- a/src/ser_win32.c
+++ b/src/ser_win32.c
@@ -77,6 +77,7 @@ static BOOL serial_w32SetTimeOut(HANDLE hComPort, DWORD timeout) { // ms
   ctmo.ReadIntervalTimeout = 0;
   ctmo.ReadTotalTimeoutMultiplier = 0;
   ctmo.ReadTotalTimeoutConstant = timeout;
+  ctmo.WriteTotalTimeoutConstant = 10*timeout;
 
   return SetCommTimeouts(hComPort, &ctmo);
 }

```

Note I cannot even compile that stuff...

nomakewan, 2025-05-12:
Hah! I can't compile it either; I've been abusing Github Actions to do that for me this whole time.

I have applied your change and...!
[image]

That did it! Thank you so much for the incredibly quick fix!

stefanrueger, 2025-05-12:
> able to flash normally using the proper COM port as well

Not convinced this is fixed... nor that flashing will work for low baud rates. This *test* sets a write timeout of 5000 ms (10*500 ms) *but* a 300 baud write of 512 bytes buffer will take some 15+ seconds... I'll propose a PR

nomakewan, 2025-05-12:
_That_ I cannot test; I don't have a test device that would accept programming at incredibly low baud rates. I only have ATMEGA 2560s with STk500v2 bootloaders set to accept 115200 baud for programming. I can only say that with this change in place it does work for that use case as expected.

stefanrueger, 2025-05-12:
Thanks all round @nomakewan and @mcuee for reporting, testing and debugging.

mcuee, 2025-05-12:
> Also regarding the code signing, it would appear that version 2.2.2.0 resolves the issue according to [this post](https://sourceforge.net/p/com0com/discussion/440109/thread/ef3d18924d/#2c33) and [this post](https://docklight.de/dl_prb059/).
> 
> 2.2.2.0: https://sourceforge.net/projects/com0com/files/com0com/2.2.2.0/

Interestingly version 3.0.0.0 signed version does not work but version 2.2.2.0 signed version works. Thanks.

mcuee, 2025-05-12:
> > Are you able to reproduce the issue other than using a real COM port and not a virtual serial port like com0com?
> 
> Could you clarify what you mean by "other than using a real COM port and not a virtual serial port"? What alternative do you suggest I test?
>
> When connecting a USB serial port, in all cases avrdude does what it is supposed to (I tested an FTDI cable from Adafruit, a generic USB-C TTL adapter, a Keyspan USA-19HS, and the USB-C port on an ESP32 and in all cases the timeout worked properly).

Good to know that. Thanks.

> If you would like me to acquire and install a physical PCI serial port, I would be more than happy to do so (though it may take a few days).

No, no need. USB to TTL/Serial adapters are the main tool people are using now. PCI/PCI-E based "real" serail port are not so common even though they are available.

> My concern is that while a USB serial port may be working correctly, clearly there is some other type of valid port configuration in Windows which can cause avrdude to hang, and that this is not [...]

Thanks.

Just want to make sure, you can only reproduce the issue with com0com, right? But your user may be able to reproduce the issue with another setup. My guess is that there are something special on the users' system.

nomakewan, 2025-05-12:
> Just want to make sure, you can only reproduce the issue with com0com, right? But your user may be able to reproduce the issue with another setup. My guess is that there are something special on the users' system.

Correct, on my end I can only reproduce it using one of the COM ports created by com0com. I do not know what device(s) the user who initially reported this issue to us has in his environment but I have asked him to check his device manager and report back. As soon as I have more information about what hardware he has that triggered this initially, I will report it here.

stefanrueger, 2025-05-12:
PS: @nomakewan You might want to experiment with git main avrdude to burn an urboot bootloader
```
$ avrdude -c usbasp -p m2560 -U urboot:best_autobaud_uart0_ee_hw
```
and use that bootloader with `-c urclock`. If you are strapped for flash you might even burn a 512 byte vector bootloader using `-U urboot:best_autobaud_uart0_ee`

mcuee, 2025-05-14:
@nomakewan and @stefanrueger 

I can reproduce the issue today with a Bluetooth comport and git main.

[image]

```
C:\work\avr\avrdude_bin> .\avrdude_git.exe -c urclock -P COM6 -b 9600 -p m2560 -vvvv
$ 'C:\work\avr\avrdude_bin\avrdude_git.exe' -c urclock -P COM6 -b 9600 -p m2560 -vvvv
Avrdude_git version 8.0-20250512 (127f40ee)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is C:\work\avr\avrdude_bin\avrdude.conf

Libreadline is used; avrdude -t -c urclock should work interactively
Using port            : COM6
Using programmer      : urclock
Setting baud rate     : 9600
 319 ms: enter urclock_getsync()
Ser_send: 0 [30]   [20]

```

And good thing is that PR #1987 fixed the issue.
```
C:\work\avr\avrdude_bin> .\avrdude_pr1987.exe -c urclock -P COM6 -b 9600 -p m2560 -v
Avrdude_pr1987 version 8.0-20250513 (e9d58b5f)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is C:\work\avr\avrdude_bin\avrdude.conf
Warning: system wide configuration file version (8.0-20250512 (127f40ee))
does not match Avrdude build version (8.0-20250513 (e9d58b5f))

Using port            : COM6
Using programmer      : urclock
Setting baud rate     : 9600
Error: size/send mismatch
Error: size/send mismatch
Error: size/send mismatch
Error: size/send mismatch
Error: size/send mismatch
Error: size/send mismatch
Error: size/send mismatch
Warning: attempt 1 of 10: not in sync
Error: size/send mismatch
Warning: attempt 2 of 10: not in sync
Error: size/send mismatch
Warning: attempt 3 of 10: not in sync
Error: size/send mismatch
Warning: attempt 4 of 10: not in sync
Error: size/send mismatch
Warning: attempt 5 of 10: not in sync
Error: size/send mismatch
Warning: attempt 6 of 10: not in sync
Error: size/send mismatch
Warning: attempt 7 of 10: not in sync
Error: size/send mismatch
Warning: attempt 8 of 10: not in sync
Error: size/send mismatch
Warning: attempt 9 of 10: not in sync
Error: size/send mismatch
Warning: attempt 10 of 10: not in sync
Error: size/send mismatch
Warning: programmer is not responding; try -x strict and/or vary -x delay=100
Error: size/send mismatch
Error: unable to open port COM6 for programmer urclock
Possible candidate serial ports are:
  -P COM3
  -P COM4 or -P ch340
  -P COM6
  -P COM8
Note that above ports might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.

Avrdude_pr1987 done.  Thank you.
```

nomakewan, 2025-05-14:
Thank you for the reproduction!

The user was not yet able to confirm the hardware they were running; they made a guess at what it was but their guess seems dubious (I have never heard of a compactflash adapter being detected as a COM port), so I once again asked them to confirm using their device manager rather than making guesses.

mcuee, 2025-05-14:
> Thank you for the reproduction!

Kind of by luck that this Dell Laptop has Bluetooth COM ports.

> The user was not yet able to confirm the hardware they were running; they made a guess at what it was but their guess seems dubious (I have never heard of a compactflash adapter being detected as a [...]

Thanks for the efforts. Take your time.

stefanrueger, 2025-05-14:
> I can reproduce the issue today with a Bluetooth comport and git main

Excellent! Thank you @mcuee

PR #1987 (2025-05-21): Fix r/w timeout for Windows `ser_send()`
Fixes #1986.

First there is no bus for 20 years and then there are two buses within a couple of days (unrelated edge cases in low-level serial I/O: one for Posix in PR 1983, the other one for Windows...).

Should be thoroughly tested with serial I/O, best with large parts with large page sizes and low baud rates.
closed 2025-05-21 by stefanrueger