avrdudes/avrdude #1995: Unable to program an AVR64EA28 with Atmel ICE

gitdode, 2025-05-20:
I am running version 8.0-20250514 (0ad18e5c) on Debian Linux:

`avrdude -v -v -c atmelice_updi -p avr64ea28 -U flash:w:avr64ea.hex`

and it fails with (full log: [avrdude.log](https://github.com/user-attachments/files/20354135/avrdude.log)):

```
Jtag3_setparm()
Sending set parameter (scope 0x12, section 0, parm 0) command: 0xa0 (4 bytes msg)
Bad response to set parameter (scope 0x12, section 0, parm 0) command: 0xa0
Error main() main.c 1632: initialization failed  (rc = -1)
```

Programming an ATmega328P with `-c atmelice_isp` works just fine. Wiring should be okay (UPDI Connection v1), the green LED is on.

I am however totally new to the new AVRs like AVR64EA and all I did is connect both VDD and GND with one 100nF each.

Probably I'm missing something really basic but I don't see it?

dl8dtl, 2025-05-20:
What exactly is your wiring? I never tried an AVR*EA with the AtmelICE, but I recently used it a lot for an ATtiny204 which also programmed by UPDI. From the 6-pin connector, I have been using pin 1 for UPDI, pin 2 for VCC (which is basically what the green LED is telling you), and pin 6 for GND.

gitdode, 2025-05-20:
The wiring is exactly what you have been using, also described in the datasheet: 

[image]

dl8dtl, 2025-05-20:
Well, I tried it on a Curiosity Nano AVR128DA48 board.
I can basically confirm the AtmelICE fails on it as well.

* The onboard debugger works (`-c pkobn`)
* An external `serial_updi` cable works, too (i.e. as promised, the onboard debugger is tristated when not in use)
* Attaching the AtmelICE to the UPDI fails initialization

Not sure, maybe handling the more recent AVRs requires some particular firmware version? Mine is for sure quite old:
```
ICE HW version        : 0
ICE FW version        : 1.39 (rel. 130)
```

gitdode, 2025-05-20:
You are right - I just played with MPLAB X IDE and read the memory off the device. That upgraded the firmware from:

```
ICE HW version        : 0
ICE FW version        : 1.00 (rel. 33)
```

to:

```
ICE HW version        : 0
ICE FW version        : 1.45 (rel. 7)
```

And now programming with avrdude works 🙂

```
dode@linus:~/dev/avr64ea$ /bin/avrdude -c atmelice_updi -p avr64ea28 -U flash:w:avr64ea.hex
Reading 459 bytes for flash from input file avr64ea.hex
Writing 459 bytes to flash
Writing | ################################################## | 100% 0.13 s 
Reading | ################################################## | 100% 0.02 s 
459 bytes of flash verified

Avrdude done.  Thank you.
```

dl8dtl, 2025-05-20:
Thanks for the feedback. Would be cool to have a definitive info which firmware version is needed for which devices … Then, AVRDUDE could tell about it beforehand.

I'm closing it anyway, as the actual problem is resolved.

closed 2025-05-20 by dl8dtl

mcuee, 2025-05-20:
I have added a Wiki FAQ entry.
* https://github.com/avrdudes/avrdude/wiki/FAQ#my-microchip-tools-cannot-program-a-new-avr-part-what-can-i-do