avrdudes/avrdude #2002: Improve the Arduino as ISP experience

MCUdude, 2025-06-02:
Original discussion here https://github.com/MCUdude/MiniCore/issues/354

The Arduino as ISP programmer uses the STK500v1 protocol, and is quite popular, because many tinkerers have spare AVR-based Arduinos. Most/all tutorials on the Arduino as ISP programmer say you have to connect a capacitor (usually 10uF) between the reset pin and ground on the microcontroller board that will serve as the "Arduino as ISP" programmer. And for good reason. The DTR/RTS pulse that's being sent when opening the port resets the microcontroller (that now acts as a programmer), gets reset, and Avrdude doesn't have any additional sync attempts. You got one try, and then it will fail.

@felias-fogg's wish was to make it possible to use the Arduino as ISP programmer without the need for an additional (10uF) capacitor to suppress the DTR/RTS pulse. The problem isn't really the DTR pulse, the problem is the lack of sync attempts

This sounds doable to me, and there are two ways we can solve this particular issue:
 * We can force the DTR/RTS line high (stay at Vcc level) during the entire programming session when using the `-c arduino_as_isp`. This may work, but there is a chance there will be a tiny DTR/RTS pulse from when the port is first opened until we're able to force the pin(s) high.
 * We can embrace the auto reset functionality, and add more (10?) retry attempts for stk500v1 programmers, just like we already do for `-c arduino`. This will probably also work, but it may take longer to establish a connection, because the first try may/will fail (due to the DTR/RTS pulse), and then we'll have to wait for a new attempt.

Any thoughts? Could something like this make it into the 8.1 release?

stefanrueger, 2025-06-02:
From the top of my head: Neither `-c stk500v1` nor `-c arduino_as_isp` pluck the DTR/RTS line (only `-c arduino` does). That reset is probably effected by the OS twiddling RTS/DTR when `open()`ing the port. I wonder what's the problem with this. Isn't it good that the programmer UNO board starts from a fresh point? Is AVRDUDE too quick to start comms or too slow? I take it the UNO board plus Arduino as ISP firmware doesn't time out like bootloaders.

The difference between `-c stk500v1` and `-c arduino_as_isp` is that the latter will correctly set the EEPROM addresses, so for best Arduino as ISP experience I'd recommend `-c  arduino_as_isp`, which BTW seems to recognise a settable number of retry attempts (defaults to 10):
```
$ avrdude -c arduino_as_isp -xhelp
avrdude -c arduino_as_isp extended options:
  -x attempts=<n>   Specify the number <n> of connection retry attempts
  -x xtal=<n>[unit] Set programmer xtal frequency to <n> Hz (or kHz/MHz)
  -x help           Show this help menu and exit
```

As a side note, AVRDUDE's retry attempts rarely work when the communication is out of sync, probably owing to the 5 s default receive timeout that `-c stk500v1/arduino_as_isp` fail to overwrite with shorter timeouts (particularly when talking to a bootloader that times out, but that's not the case here).

MCUdude, 2025-06-02:
> From the top of my head: Neither -c stk500v1 nor -c arduino_as_isp pluck the DTR/RTS line (only -c arduino does). That reset is probably effected by the OS twiddling RTS/DTR when open()ing the port. I wonder what's the problem with this

I believe the problem is that the Arduino as ISP gets reset due to the DTR/RTS plucking, and the programmer is ready, Avrdude has already timed out. And this behaviour may vary depending on which OS and serial drivers are being used.

Right now, only `-c Arduino` toggles the DTR/RTS line before every sync attempt. Maybe `-c arduino_as_isp` should behave the same?

I'll flash an old UNO and observe the DTR line on a scope to see how it currently behaves on my mac

MCUdude, 2025-06-02:
I've done some more tests with an oscilloscope attached to the reset pin on the Arduino acting as the programmer.

Even though Avrdude doesn't touch the DTR/RTS line, the reset pin (measured _at_ the programmer MCU) gets a low pulse due to how the serial driver works. You can force it high, but you'll always get that low pulse, because you can't set the DTR/RTS line before establishing a connection.

However, it works surprisingly well as it is today. I can even hold the reset button on the programmer down while running avrdude, and Avrdude will still be able to communicate with it when I let go of the button:

```
$ avrdude -c arduino_as_isp -P /dev/cu.usbmodem1101 -patmega1284p -Uefuse:r:-:h
Error: programmer is not responding
Warning: attempt 1 of 10: not in sync: resp=0x00
Error: programmer is not responding
Warning: attempt 2 of 10: not in sync: resp=0x00
Reading efuse memory ...
Writing 1 byte to output file <stdout>
0xff

Avrdude done.  Thank you.
```

And it even works _as is_ when not touching the reset button at all:

```
$ avrdude -c arduino_as_isp -P /dev/cu.usbmodem1101 -patmega1284p -Uefuse:r:-:h
Reading efuse memory ...
Writing 1 byte to output file <stdout>
0xff

Avrdude done.  Thank you.
```

So we can safely conclude that there's no need for that 10uF capacitor between reset and ground on the Arduino board running the Arduino as ISP sketch.

Sorry for the noise