avrdudes/avrdude #2007: linuxgpio and linuxspi do not exit properly

mcuee, 2025-06-07:
It is strange that I can run `-c linuxgpio` and `-c linuxspi` once.

It seems to me that avrdude may not exit properly. I need to check if this is a regression or not.

```
mcuee@raspberrypi400arm64:~/build/avrdude/tools $ ./test-avrdude -v -d0 -p "-c linuxspi -p m328p"
Testing /usr/local/bin/avrdude 8.0-20250603 (4e473c04)
Prepare "-c linuxspi -p m328p" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.300 s: fuse access: clear, set and read eesave fuse bit
✅   0.315 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
✅   0.315 s: chip erase
✅   2.868 s: flash -U write/verify holes_rjmp_loops_32768B.hex
✅   2.469 s: flash -T write/verify holes_rjmp_loops_32768B.hex
✅   0.306 s: eeprom check whether programmer can flip 0s to 1s
✅   1.349 s: eeprom -U write/verify holes_pack_my_box_1024B.hex
✅   2.327 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_1024B.hex
✅   1.445 s: chip erase and spot check flash is actually erased
✅   0.351 s: spot check eeprom is erased, too

mcuee@raspberrypi400arm64:~/build/avrdude/tools $ avrdude -c linuxspi -p m328p -v
Avrdude version 8.0-20250603 (4e473c04)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file is /home/mcuee/.avrduderc

Using port            : /dev/spidev0.0:/dev/gpiochip0
Using programmer      : linuxspi
Opening /dev/spidev0.0:/dev/gpiochip0:25
Defaulting bit clock to 200 kHz
AVR part              : ATmega328P
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : linuxspi
Description           : Use Linux SPI device in /dev/spidev*
Error: AVR device not responding
Error: initialization failed  (rc = -2)
 - the programmer ISP clock is too fast for the target
 - use -B to set lower the bit clock frequency, e.g. -B 125kHz
 - use -F to override this check

Avrdude done.  Thank you.

mcuee@raspberrypi400arm64:~/build/avrdude/tools $ avrdude -c linuxgpio -p m328p -v
Avrdude version 8.0-20250603 (4e473c04)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file is /home/mcuee/.avrduderc

using libgpiod for linuxgpio
Using port            : gpiochip0
Using programmer      : linuxgpio
AVR part              : ATmega328P
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : linuxgpio
Description           : Use the Linux libgpiod interface to bitbang GPIO lines
Pin assignment        : libgpiod
  RESET  = 25
  SCK    = 11
  SDO    = 10
  SDI    = 9

AVR device initialized and ready to accept instructions
Device signature = 1E 95 0F (ATmega328P, ATA6614Q, LGT8F328P)

Avrdude done.  Thank you.

mcuee@raspberrypi400arm64:~/build/avrdude/tools $ avrdude -c linuxgpio -p m328p -v -E reset
Avrdude version 8.0-20250603 (4e473c04)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file is /home/mcuee/.avrduderc

using libgpiod for linuxgpio
Using port            : gpiochip0
Using programmer      : linuxgpio
AVR part              : ATmega328P
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : linuxgpio
Description           : Use the Linux libgpiod interface to bitbang GPIO lines
Pin assignment        : libgpiod
  RESET  = 25
  SCK    = 11
  SDO    = 10
  SDI    = 9

AVR device initialized and ready to accept instructions
Device signature = 00 3D 00
Error: expected signature for ATmega328P is 1E 95 0F
  - double check chip or use -F to carry on regardless

Avrdude done.  Thank you.
```

mcuee, 2025-06-07:
Hmm, looks like linuxgpio is broken now in my system. Reboot the Raspberry Pi 400 and it does not seem to work.

```
mcuee@raspberrypi400arm64:~/build/avrdude/tools $ avrdude -c linuxgpio -p m328p -v -U urboot_m328p.hex 
Avrdude version 8.0-20250603 (4e473c04)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file is /home/mcuee/.avrduderc

using libgpiod for linuxgpio
Using port            : gpiochip0
Using programmer      : linuxgpio
AVR part              : ATmega328P
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : linuxgpio
Description           : Use the Linux libgpiod interface to bitbang GPIO lines
Pin assignment        : libgpiod
  RESET  = 25
  SCK    = 11
  SDO    = 10
  SDI    = 9

AVR device initialized and ready to accept instructions
Device signature = 1E 95 0F (ATmega328P, ATA6614Q, LGT8F328P)
Auto-erasing chip as flash memory needs programming (-U flash:w:...)
specify the -D option to disable this feature
Erased chip
Reading 384 bytes for flash from input file urboot_m328p.hex
in 1 section [0x7e80, 0x7fff]: 3 pages and 0 pad bytes
Writing 384 bytes to flash
Writing | ################################################## | 100% 0.06 s 
Reading | ################################################## | 100% 0.05 s 
Warning: flash verification mismatch
  device 0x00 != input 0x11 at addr 0x7e80 (error)
  device 0x22 != input 0x24 at addr 0x7e81 (error)
  device 0xff != input 0xe0 at addr 0x7e87 (error)
  device 0xff != input 0x79 at addr 0x7e88 (error)
  device 0xff != input 0xd0 at addr 0x7e89 (error)
  device 0xff != input 0x21 at addr 0x7e8a (error)
  device 0xff != input 0xfe at addr 0x7e8b (error)
  device 0xff != input 0xeb at addr 0x7e8c (error)
  device 0xfe != input 0xc0 at addr 0x7e8d (error)
  device 0xff != input 0x8e at addr 0x7e8e (error)
  suppressing further verification errors
Error: flash verification mismatch

Avrdude done.  Thank you.

mcuee@raspberrypi400arm64:~/build/avrdude/tools $ avrdude -c linuxgpio -p m328p -v -U urboot_m328p.hex -E reset
Avrdude version 8.0-20250603 (4e473c04)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file is /home/mcuee/.avrduderc

using libgpiod for linuxgpio
Using port            : gpiochip0
Using programmer      : linuxgpio
AVR part              : ATmega328P
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : linuxgpio
Description           : Use the Linux libgpiod interface to bitbang GPIO lines
Pin assignment        : libgpiod
  RESET  = 25
  SCK    = 11
  SDO    = 10
  SDI    = 9

AVR device initialized and ready to accept instructions
Device signature = 1E 95 0F (ATmega328P, ATA6614Q, LGT8F328P)
Auto-erasing chip as flash memory needs programming (-U flash:w:...)
specify the -D option to disable this feature
Erased chip
Reading 384 bytes for flash from input file urboot_m328p.hex
in 1 section [0x7e80, 0x7fff]: 3 pages and 0 pad bytes
Writing 384 bytes to flash
Writing | ################################################## | 100% 0.06 s 
Reading | ################################################## | 100% 0.05 s 
Warning: flash verification mismatch
  device 0x00 != input 0x24 at addr 0x7e82 (error)
  device 0xff != input 0xb6 at addr 0x7e83 (error)
  device 0x00 != input 0x14 at addr 0x7e84 (error)
  device 0xff != input 0xbe at addr 0x7e85 (error)
  device 0xff != input 0x80 at addr 0x7e86 (error)
  device 0xff != input 0xd0 at addr 0x7e89 (error)
  device 0xff != input 0x21 at addr 0x7e8a (error)
  device 0xff != input 0xeb at addr 0x7e8c (error)
  device 0xeb != input 0xc0 at addr 0x7e8d (error)
  device 0xff != input 0x8e at addr 0x7e8e (error)
  suppressing further verification errors
Error: flash verification mismatch

Avrdude done.  Thank you.

mcuee@raspberrypi400arm64:~/build/avrdude/tools $ avrdude -c linuxspi -p m328p -v -U urboot_m328p.hex 
Avrdude version 8.0-20250603 (4e473c04)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file is /home/mcuee/.avrduderc

Using port            : /dev/spidev0.0:/dev/gpiochip0
Using programmer      : linuxspi
Opening /dev/spidev0.0:/dev/gpiochip0:25
Defaulting bit clock to 200 kHz
AVR part              : ATmega328P
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : linuxspi
Description           : Use Linux SPI device in /dev/spidev*
Error: AVR device not responding
Error: initialization failed  (rc = -2)
 - the programmer ISP clock is too fast for the target
 - use -B to set lower the bit clock frequency, e.g. -B 125kHz
 - use -F to override this check

Avrdude done.  Thank you.
```

mcuee, 2025-06-07:
Kind of strange. I tested the two recent PRs and they seemed to work fine previously.
* https://github.com/avrdudes/avrdude/pull/1977
* https://github.com/avrdudes/avrdude/pull/1973 (i mainly tested the new -E function)

mcuee, 2025-06-07:
Looks the problem is not with linuxspi but with linuxgpio.

Reboot again.
```
mcuee@raspberrypi400arm64:~/build/avrdude/tools $ avrdude -c linuxspi -p m328p -v -U urboot_m328p.hex 
Avrdude version 8.0-20250603 (4e473c04)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file is /home/mcuee/.avrduderc

Using port            : /dev/spidev0.0:/dev/gpiochip0
Using programmer      : linuxspi
Opening /dev/spidev0.0:/dev/gpiochip0:25
Defaulting bit clock to 200 kHz
AVR part              : ATmega328P
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : linuxspi
Description           : Use Linux SPI device in /dev/spidev*

AVR device initialized and ready to accept instructions
Device signature = 1E 95 0F (ATmega328P, ATA6614Q, LGT8F328P)
Auto-erasing chip as flash memory needs programming (-U flash:w:...)
specify the -D option to disable this feature
Erased chip
Reading 384 bytes for flash from input file urboot_m328p.hex
in 1 section [0x7e80, 0x7fff]: 3 pages and 0 pad bytes
Writing 384 bytes to flash
Writing | ################################################## | 100% 0.10 s 
Reading | ################################################## | 100% 0.08 s 
384 bytes of flash verified

Avrdude done.  Thank you.
mcuee@raspberrypi400arm64:~/build/avrdude/tools $ avrdude -c linuxspi -p m328p -v -U urboot_m328p.hex 
Avrdude version 8.0-20250603 (4e473c04)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file is /home/mcuee/.avrduderc

Using port            : /dev/spidev0.0:/dev/gpiochip0
Using programmer      : linuxspi
Opening /dev/spidev0.0:/dev/gpiochip0:25
Defaulting bit clock to 200 kHz
AVR part              : ATmega328P
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : linuxspi
Description           : Use Linux SPI device in /dev/spidev*

AVR device initialized and ready to accept instructions
Device signature = 1E 95 0F (ATmega328P, ATA6614Q, LGT8F328P)
Auto-erasing chip as flash memory needs programming (-U flash:w:...)
specify the -D option to disable this feature
Erased chip
Reading 384 bytes for flash from input file urboot_m328p.hex
in 1 section [0x7e80, 0x7fff]: 3 pages and 0 pad bytes
Writing 384 bytes to flash
Writing | ################################################## | 100% 0.10 s 
Reading | ################################################## | 100% 0.08 s 
384 bytes of flash verified

Avrdude done.  Thank you.

mcuee@raspberrypi400arm64:~/build/avrdude/tools $ avrdude -c linuxgpio -p m328p -v -U urboot_m328p.hex 
Avrdude version 8.0-20250603 (4e473c04)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file is /home/mcuee/.avrduderc

using libgpiod for linuxgpio
Using port            : gpiochip0
Using programmer      : linuxgpio
AVR part              : ATmega328P
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : linuxgpio
Description           : Use the Linux libgpiod interface to bitbang GPIO lines
Pin assignment        : libgpiod
  RESET  = 25
  SCK    = 11
  SDO    = 10
  SDI    = 9

AVR device initialized and ready to accept instructions
Device signature = 1E 95 0F (ATmega328P, ATA6614Q, LGT8F328P)
Auto-erasing chip as flash memory needs programming (-U flash:w:...)
specify the -D option to disable this feature
Erased chip
Reading 384 bytes for flash from input file urboot_m328p.hex
in 1 section [0x7e80, 0x7fff]: 3 pages and 0 pad bytes
Writing 384 bytes to flash
Writing | ################################################## | 100% 0.06 s 
Reading | ################################################## | 100% 0.05 s 
Warning: flash verification mismatch
  device 0x00 != input 0x14 at addr 0x7e84 (error)
  device 0xff != input 0xbe at addr 0x7e85 (error)
  device 0xff != input 0xe0 at addr 0x7e87 (error)
  device 0xff != input 0x79 at addr 0x7e88 (error)
  device 0xff != input 0xd0 at addr 0x7e89 (error)
  device 0xff != input 0x21 at addr 0x7e8a (error)
  device 0xff != input 0xfe at addr 0x7e8b (error)
  device 0xff != input 0xeb at addr 0x7e8c (error)
  device 0xff != input 0xc0 at addr 0x7e8d (error)
  device 0xff != input 0x8e at addr 0x7e8e (error)
  suppressing further verification errors
Error: flash verification mismatch

Avrdude done.  Thank you.
```

mcuee, 2025-06-07:
Just want to double confirm that linuxspi works fine.
Running test8 after reboot.
```
mcuee@raspberrypi400arm64:~/build/avrdude/tools $ ./test8 -r4 linuxspi m328p
Reading 33824 bytes for multiple memories from input file bak-linuxspi-m328p.hex
Writing 1024 bytes to eeprom
Writing | ################################################## | 100% 4.05 s 
Reading | ################################################## | 100% 0.21 s 
1024 bytes of eeprom verified
Writing 32768 bytes to flash
Writing | ################################################## | 100% 8.42 s 
Reading | ################################################## | 100% 7.16 s 
32768 bytes of flash verified
Writing 1 byte (0x62) to lfuse, 1 byte written, 1 verified
Writing 1 byte (0xDC) to hfuse, 1 byte written, 1 verified
Writing 1 byte (0xFF) to efuse, 1 byte written, 1 verified
Writing 1 byte (0xFF) to lock, 1 byte written, 1 verified

Avrdude done.  Thank you.
test8: m328p initialised

mcuee@raspberrypi400arm64:~/build/avrdude/tools $ ./test8 linuxspi m328p

Processing -U ALL:r:bak-linuxspi-m328p.hex:I
Reading multiple memories ...
 - eeprom         | ################################################## | 100% 0.21 s 
 - flash          | ################################################## | 100% 6.70 s 
 - lfuse          | ################################################## | 100% 0.00 s 
 - hfuse          | ################################################## | 100% 0.00 s 
 - efuse          | ################################################## | 100% 0.00 s 
 - lock           | ################################################## | 100% 0.00 s 
 - prodsig/sigrow | ################################################## | 100% 0.01 s 
 - sernum         | ################################################## | 100% 0.00 s 
 - signature      | ################################################## | 100% 0.00 s 
 - calibration    | ################################################## | 100% 0.00 s 
Writing 33834 bytes to output file bak-linuxspi-m328p.hex

Processing -T fact reset
performing chip erase; discarded pending writes to flash and EEPROM
Writing | ################################################## | 100% 0.45 s 
Writing | ################################################## | 100% 14.33 s 
after the next reset the part will have F_CPU = 1.000 MHz

Processing -U ALL:w:bak-linuxspi-m328p.hex:I
Reading 33824 bytes for multiple memories from input file bak-linuxspi-m328p.hex
Writing 1024 bytes to eeprom
Writing | ################################################## | 100% 4.03 s 
Reading | ################################################## | 100% 0.21 s 
1024 bytes of eeprom verified
Writing 32768 bytes to flash
Writing | ################################################## | 100% 8.34 s 
Reading | ################################################## | 100% 7.17 s 
32768 bytes of flash verified
Writing 1 byte (0x62) to lfuse, 1 byte written, 1 verified
Writing 1 byte (0xDC) to hfuse, 1 byte written, 1 verified
Writing 1 byte (0xFF) to efuse, 1 byte written, 1 verified
Writing 1 byte (0xFF) to lock, 1 byte written, 1 verified

Processing -U ALL:v:bak-linuxspi-m328p.hex:I
Verifying 33824 bytes of multiple memories against input file bak-linuxspi-m328p.hex
Reading | ################################################## | 100% 0.21 s 
1024 bytes of eeprom verified
Reading | ################################################## | 100% 6.65 s 
32768 bytes of flash verified
1 byte of lfuse verified
1 byte of hfuse verified
1 byte of efuse verified
1 byte of lock verified
24 bytes of prodsig/sigrow verified
10 bytes of sernum verified
3 bytes of signature verified
1 byte of calibration verified

Avrdude done.  Thank you.
```

mcuee, 2025-06-07:
> It may be a specific stability issue on my side. 

I tend to think this is the issue but somehow linuxspi works fine. Strange.

BTW, I tried other hex files as well.

```
mcuee@raspberrypi400arm64:~/build/avrdude/tools $ avrdude -c linuxgpio -p m328p -U urboot_m328p.hex -v
Avrdude version 8.0-20250603 (4e473c04)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file is /home/mcuee/.avrduderc

using libgpiod for linuxgpio
Using port            : gpiochip0
Using programmer      : linuxgpio
AVR part              : ATmega328P
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : linuxgpio
Description           : Use the Linux libgpiod interface to bitbang GPIO lines
Pin assignment        : libgpiod
  RESET  = 25
  SCK    = 11
  SDO    = 10
  SDI    = 9

AVR device initialized and ready to accept instructions
Device signature = 00 3D 0F
Error: expected signature for ATmega328P is 1E 95 0F
  - double check chip or use -F to carry on regardless

Avrdude done.  Thank you.

mcuee@raspberrypi400arm64:~/build/avrdude/tools $ avrdude -c linuxgpio -p m328p -U urboot_m328p.hex -v
Avrdude version 8.0-20250603 (4e473c04)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file is /home/mcuee/.avrduderc

using libgpiod for linuxgpio
Using port            : gpiochip0
Using programmer      : linuxgpio
AVR part              : ATmega328P
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : linuxgpio
Description           : Use the Linux libgpiod interface to bitbang GPIO lines
Pin assignment        : libgpiod
  RESET  = 25
  SCK    = 11
  SDO    = 10
  SDI    = 9

AVR device initialized and ready to accept instructions
Device signature = 1E 95 0F (ATmega328P, ATA6614Q, LGT8F328P)
Auto-erasing chip as flash memory needs programming (-U flash:w:...)
specify the -D option to disable this feature
Erased chip
Reading 384 bytes for flash from input file urboot_m328p.hex
in 1 section [0x7e80, 0x7fff]: 3 pages and 0 pad bytes
Writing 384 bytes to flash
Writing | ################################################## | 100% 0.06 s 
Reading | ################################################## | 100% 0.05 s 
Warning: flash verification mismatch
  device 0x00 != input 0x24 at addr 0x7e81 (error)
  device 0xff != input 0x24 at addr 0x7e82 (error)
  device 0xff != input 0xb6 at addr 0x7e83 (error)
  device 0x48 != input 0x14 at addr 0x7e84 (error)
  device 0xff != input 0xbe at addr 0x7e85 (error)
  device 0x29 != input 0x80 at addr 0x7e86 (error)
  device 0xff != input 0xd0 at addr 0x7e89 (error)
  device 0xff != input 0x21 at addr 0x7e8a (error)
  device 0xff != input 0xfe at addr 0x7e8b (error)
  device 0x42 != input 0xeb at addr 0x7e8c (error)
  suppressing further verification errors
Error: flash verification mismatch

Avrdude done.  Thank you.

```

stefanrueger, 2025-06-07:
Thanks @mcuee. This should be confirmed independently to make sure it's not flaky connections or similar.

stefanrueger, 2025-06-07:
@MCUdude Brilliant! Thanks.

mcuee, 2025-06-08:
Great. I will close this issue now.

closed 2025-06-08 by mcuee