avrdudes/avrdude #2010: avrdude issues under FreeBSD 14.2 Release

mcuee, 2025-06-09:
Known issue with CH340 and serialupdi: not an avrdude issue, but rather FreeBSD CH340 driver issue
* https://github.com/avrdudes/avrdude/discussions/1639

New issue found.
ATtiny416 Xplained Nano -- only works one time out of three times, looks like a USB driver issue with FreeBSD.
```
mcuee@freebsd14:~/build/avrdude_81_rc/tools $ avrdude -v -c xplainedmini_updi -p t416 
Avrdude version 8.0-20250608 (de2d1ad5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : xplainedmini_updi
AVR part              : ATtiny416
Programming modes     : SPM, UPDI
Programmer type       : JTAGICE3_UPDI
Description           : Atmel XplainedMini in UPDI mode
ICE HW version        : 0
ICE FW version        : 1.13 (rel. 43)
Serial number         : ATML2795042700004041

Partial Family_ID returned: "tiny"
Silicon revision: 0.1

AVR device initialized and ready to accept instructions
Device signature = 1E 92 21 (ATtiny416)

Avrdude done.  Thank you.
mcuee@freebsd14:~/build/avrdude_81_rc/tools $ avrdude -v -c xplainedmini_updi -p t416
Avrdude version 8.0-20250608 (de2d1ad5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : xplainedmini_updi
Error: unexpected reply to DAP_Info: 0x00 0x00
Error: short read, read only 64 out of 512 bytes
Jtag3_edbg_send(): unexpected response 0x00, 0x00
Error: short read, read only 64 out of 512 bytes
Jtag3_edbg_recv_frame(): unexpected response 0x00
Error: unable to open port usb for programmer xplainedmini_updi

Avrdude done.  Thank you.
mcuee@freebsd14:~/build/avrdude_81_rc/tools $ avrdude -v -c xplainedmini_updi -p t416
Avrdude version 8.0-20250608 (de2d1ad5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : xplainedmini_updi
Error: unexpected reply to DAP_Info: 0x00 0x00
Jtag3_edbg_send(): unexpected response 0x00, 0x02
Jtag3_edbg_recv_frame(): unexpected response 0x80
Error: unable to open port usb for programmer xplainedmini_updi

Avrdude done.  Thank you.
mcuee@freebsd14:~/build/avrdude_81_rc/tools $ avrdude -v -c xplainedmini_updi -p t416
Avrdude version 8.0-20250608 (de2d1ad5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : xplainedmini_updi
AVR part              : ATtiny416
Programming modes     : SPM, UPDI
Programmer type       : JTAGICE3_UPDI
Description           : Atmel XplainedMini in UPDI mode
ICE HW version        : 0
ICE FW version        : 1.13 (rel. 43)
Serial number         : ATML2795042700004041

Partial Family_ID returned: "tiny"
Silicon revision: 0.1

AVR device initialized and ready to accept instructions
Device signature = 1E 92 21 (ATtiny416)

Avrdude done.  Thank you.
mcuee@freebsd14:~/build/avrdude_81_rc/tools $ avrdude -v -c xplainedmini_updi -p t416
Avrdude version 8.0-20250608 (de2d1ad5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : xplainedmini_updi
Error: unexpected reply to DAP_Info: 0x00 0x00
Error: short read, read only 64 out of 512 bytes
Jtag3_edbg_send(): unexpected response 0x00, 0x00
Error: short read, read only 64 out of 512 bytes
Jtag3_edbg_recv_frame(): unexpected response 0x00
Error: unable to open port usb for programmer xplainedmini_updi

Avrdude done.  Thank you.
mcuee@freebsd14:~/build/avrdude_81_rc/tools $ avrdude -v -c xplainedmini_updi -p t416
Avrdude version 8.0-20250608 (de2d1ad5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : xplainedmini_updi
Error: unexpected reply to DAP_Info: 0x00 0x00
Jtag3_edbg_send(): unexpected response 0x00, 0x02
Jtag3_edbg_recv_frame(): unexpected response 0x80
Error: unable to open port usb for programmer xplainedmini_updi

Avrdude done.  Thank you.
```

mcuee, 2025-06-09:
Using `test-avrdude`.
```
mcuee@freebsd14:~/build/avrdude_81_rc/tools $ ./test-avrdude -v -p "-c xplainedmini_updi -p t416"
Testing /usr/local/bin/avrdude 8.0-20250608 (de2d1ad5)
Prepare "-c xplainedmini_updi -p t416" and press 'enter' or 'space' to continue. Press any other key to skip
✅   0.779 s: fuse access: clear, set and read eesave fuse bit
❌   0.431 s: fuse access: set eesave fusebit to delete EEPROM on chip erase (failed command below)
$ avrdude -qq -c xplainedmini_updi -p t416 -T "config eesave=ee*erased"
Error: unexpected reply to DAP_Info: 0x00 0x00
Error: short read, read only 64 out of 512 bytes
Error: short read, read only 64 out of 512 bytes
Error: unable to open port usb for programmer xplainedmini_updi
❌   0.219 s: chip erase (failed command below)
$ avrdude -qq -c xplainedmini_updi -p t416 -e -FAU flash:w:0xff:m
Error: unexpected reply to DAP_Info: 0x00 0x00
Error: unable to open port usb for programmer xplainedmini_updi
✅   1.579 s: flash -U write/verify holes_rjmp_loops_4096B.hex
✅   0.989 s: flash -T write/verify holes_rjmp_loops_4096B.hex
❌   0.433 s: eeprom check whether programmer can flip 0s to 1s (failed command below)
$ avrdude -qq -c xplainedmini_updi -p t416 -Ueeprom:w:0x55:m -Ueeprom:w:0xaa:m
Error: unexpected reply to DAP_Info: 0x00 0x00
Error: short read, read only 64 out of 512 bytes
Error: short read, read only 64 out of 512 bytes
Error: unable to open port usb for programmer xplainedmini_updi
# ... the next test may therefore take longer
❌   0.219 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_128B.hex (failed command below)
$ avrdude -qq -c xplainedmini_updi -p t416 -T "write eeprom ./test_files/holes_the_five_boxing_wizards_128B.hex:a" -T flush -T "write eeprom ./test_files/holes_pack_my_box_128B.hex:a"
Error: unexpected reply to DAP_Info: 0x00 0x00
Error: unable to open port usb for programmer xplainedmini_updi
✅   0.764 s: eeprom -T write/verify lorem_ipsum_128B.srec
❌   0.426 s: chip erase and spot check flash is actually erased (failed command below)
$ avrdude -qq -c xplainedmini_updi -p t416 -e -FAU flash:w:0xff:m -U flash:v:./test_files/holes_flash_0xff_4096B.hex
Error: unexpected reply to DAP_Info: 0x00 0x00
Error: short read, read only 64 out of 512 bytes
Error: short read, read only 64 out of 512 bytes
Error: unable to open port usb for programmer xplainedmini_updi
❌   0.217 s: usersig -T/-U write/read random_data_32B.bin (failed command below)
$ avrdude -qq -c xplainedmini_updi -p t416 -T "erase usersig; write usersig ./test_files/random_data_32B.bin" -T flush -U usersig:r:/tmp/test-avrdude.tmp.gtJB67:r -U usersig:v:/tmp/test-avrdude.tmp.gtJB67:r -T "erase usersig" -T flush -U usersig:v:./test_files/0xff_32B.hex:i
Error: unexpected reply to DAP_Info: 0x00 0x00
Error: unable to open port usb for programmer xplainedmini_updi

One or more AVRDUDE "-c xplainedmini_updi -p t416" tests failed. Do you want to retry this particular test? (y/n): n
```

mcuee, 2025-06-09:
Using test8.
```
mcuee@freebsd14:~/build/avrdude_81_rc/tools $ ./test8 -r8 xplainedmini_updi t416
Reading 4363 bytes for multiple memories from input file bak-xplainedmini_updi-t416.hex
Writing 128 bytes to eeprom
Writing | ################################################## | 100% 0.08 s 
Reading | ################################################## | 100% 0.06 s 
128 bytes of eeprom verified
Writing 4096 bytes to flash
Writing | ################################################## | 100% 2.75 s 
Reading | ################################################## | 100% 1.53 s 
4096 bytes of flash verified
Writing 10 bytes to fuses, 10 bytes written, 10 verified
Writing 1 byte (0x00) to fuse0/wdtcfg, 1 byte written, 1 verified
Writing 1 byte (0x00) to fuse1/bodcfg, 1 byte written, 1 verified
Writing 1 byte (0x7E) to fuse2/osccfg, 1 byte written, 1 verified
Writing 1 byte (0x00) to fuse4/tcd0cfg, 1 byte written, 1 verified
Writing 1 byte (0xF6) to fuse5/syscfg0, 1 byte written, 1 verified
Writing 1 byte (0xFF) to fuse6/syscfg1, 1 byte written, 1 verified
Writing 1 byte (0x00) to fuse7/append, 1 byte written, 1 verified
Writing 1 byte (0x01) to fuse8/bootend, 1 byte written, 1 verified
Writing 1 byte (0xC5) to lock, 1 byte written, 1 verified
Writing 32 bytes to userrow/usersig, 32 bytes written, 32 verified

Avrdude done.  Thank you.
test8: t416 initialised

mcuee@freebsd14:~/build/avrdude_81_rc/tools $ ./test8 xplainedmini_updi t416
Error: unexpected reply to DAP_Info: 0x00 0x00
Error: short read, read only 64 out of 512 bytes
Error: short read, read only 64 out of 512 bytes
Error: unable to open port usb for programmer xplainedmini_updi

Avrdude done.  Thank you.
-------------------------------
Re-running to create error logs

mcuee@freebsd14:~/build/avrdude_81_rc/tools $ ./test8 xplainedmini_updi t416

Processing -U ALL:r:bak-xplainedmini_updi-t416.hex:I
Reading multiple memories ...
 - eeprom          | ################################################## | 100% 0.06 s 
 - flash           | ################################################## | 100% 1.54 s 
 - fuses           | ################################################## | 100% 0.16 s 
 - fuse0/wdtcfg    | ################################################## | 100% 0.02 s 
 - fuse1/bodcfg    | ################################################## | 100% 0.02 s 
 - fuse2/osccfg    | ################################################## | 100% 0.02 s 
 - fuse4/tcd0cfg   | ################################################## | 100% 0.02 s 
 - fuse5/syscfg0   | ################################################## | 100% 0.02 s 
 - fuse6/syscfg1   | ################################################## | 100% 0.02 s 
 - fuse7/append    | ################################################## | 100% 0.02 s 
 - fuse8/bootend   | ################################################## | 100% 0.02 s 
 - lock            | ################################################## | 100% 0.02 s 
 - prodsig/sigrow  | ################################################## | 100% 0.02 s 
 - signature       | ################################################## | 100% 0.02 s 
 - sernum          | ################################################## | 100% 0.16 s 
 - osccal16        | ################################################## | 100% 0.03 s 
 - osccal20        | ################################################## | 100% 0.03 s 
 - tempsense       | ################################################## | 100% 0.03 s 
 - osc16err        | ################################################## | 100% 0.03 s 
 - osc20err        | ################################################## | 100% 0.03 s 
 - userrow/usersig | ################################################## | 100% 0.02 s 
 - sib             | ################################################## | 100% 0.00 s 
Writing 4394 bytes to output file bak-xplainedmini_updi-t416.hex

Processing -T fact reset
performing chip erase; discarded pending writes to flash and EEPROM
Caching | ################################################## | 100% 0.06 s 
Caching | ################################################## | 100% 0.00 s 
Caching | ################################################## | 100% 0.01 s 
[... 69 lines cut ...]
mcuee@freebsd14:~/build/avrdude_81_rc/tools $ ./test8 -r8 xplainedmini_updi t416
Reading 4363 bytes for multiple memories from input file bak-xplainedmini_updi-t416.hex
Writing 128 bytes to eeprom
Writing | ################################################## | 100% 0.08 s 
Reading | ################################################## | 100% 0.06 s 
128 bytes of eeprom verified
Writing 4096 bytes to flash
Writing | ################################################## | 100% 2.75 s 
Reading | ################################################## | 100% 1.53 s 
4096 bytes of flash verified
Writing 10 bytes to fuses, 10 bytes written, 10 verified
Writing 1 byte (0x00) to fuse0/wdtcfg, 1 byte written, 1 verified
Writing 1 byte (0x00) to fuse1/bodcfg, 1 byte written, 1 verified
Writing 1 byte (0x7E) to fuse2/osccfg, 1 byte written, 1 verified
Writing 1 byte (0x00) to fuse4/tcd0cfg, 1 byte written, 1 verified
Writing 1 byte (0xF6) to fuse5/syscfg0, 1 byte written, 1 verified
Writing 1 byte (0xFF) to fuse6/syscfg1, 1 byte written, 1 verified
Writing 1 byte (0x00) to fuse7/append, 1 byte written, 1 verified
Writing 1 byte (0x01) to fuse8/bootend, 1 byte written, 1 verified
Writing 1 byte (0xC5) to lock, 1 byte written, 1 verified
Writing 32 bytes to userrow/usersig, 32 bytes written, 32 verified

Avrdude done.  Thank you.
test8: t416 initialised
```

mcuee, 2025-06-09:
This is not a regression though, same issue with avrdude 8.0 release.

```
mcuee@freebsd14:~/build/avrdude_bin $ sudo lsusb
Bus /dev/usb Device /dev/ugen0.7: ID 03eb:2145 Atmel Corp. ATMEGA328P-XMINI (CDC ACM)
Bus /dev/usb Device /dev/ugen0.3: ID 16c0:05dc Van Ooijen Technische Informatica shared ID for use with libusb
Bus /dev/usb Device /dev/ugen0.2: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus /dev/usb Device /dev/ugen0.6: ID 13d3:3503 IMC Networks 
Bus /dev/usb Device /dev/ugen0.5: ID 1ea7:0064 SHARKOON Technologies GmbH 2.4GHz Wireless rechargeable vertical mouse [More&Better]
Bus /dev/usb Device /dev/ugen0.4: ID 25a7:fa61 Areson Technology Corp Elecom Co., Ltd MR-K013 Multicard Reader
Bus /dev/usb Device /dev/ugen0.1: ID 0000:0000  
mcuee@freebsd14:~/build/avrdude_bin $ sudo chmod 666 /dev/ugen0.7
mcuee@freebsd14:~/build/avrdude_bin $ ./avrdude80 -C ./avrdude80.conf  -v -c xplainedmini_updi -p t416
Avrdude80 version 8.0
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /home/mcuee/build/avrdude_bin/avrdude80.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : xplainedmini_updi
AVR part              : ATtiny416
Programming modes     : SPM, UPDI
Programmer type       : JTAGICE3_UPDI
Description           : Atmel XplainedMini in UPDI mode
ICE HW version        : 0
ICE FW version        : 1.13 (rel. 43)
Serial number         : ATML2795042700004041

Partial Family_ID returned: "tiny"
Silicon revision: 0.1

AVR device initialized and ready to accept instructions
Device signature = 1E 92 21 (ATtiny416)

Avrdude80 done.  Thank you.
mcuee@freebsd14:~/build/avrdude_bin $ ./avrdude80 -C ./avrdude80.conf  -v -c xplainedmini_updi -p t416
Avrdude80 version 8.0
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /home/mcuee/build/avrdude_bin/avrdude80.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : xplainedmini_updi
Error: unexpected reply to DAP_Info: 0x00 0x00
Error: short read, read only 64 out of 512 bytes
Jtag3_edbg_send(): unexpected response 0x00, 0x00
Error: short read, read only 64 out of 512 bytes
Jtag3_edbg_recv_frame(): unexpected response 0x00
Error: unable to open port usb for programmer xplainedmini_updi

Avrdude80 done.  Thank you.
mcuee@freebsd14:~/build/avrdude_bin $ ./avrdude80 -C ./avrdude80.conf  -v -c xplainedmini_updi -p t416
Avrdude80 version 8.0
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /home/mcuee/build/avrdude_bin/avrdude80.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : xplainedmini_updi
Error: unexpected reply to DAP_Info: 0x00 0x00
Jtag3_edbg_send(): unexpected response 0x00, 0x02
Jtag3_edbg_recv_frame(): unexpected response 0x80
Error: unable to open port usb for programmer xplainedmini_updi

Avrdude80 done.  Thank you.
mcuee@freebsd14:~/build/avrdude_bin $ ./avrdude80 -C ./avrdude80.conf  -v -c xplainedmini_updi -p t416
Avrdude80 version 8.0
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /home/mcuee/build/avrdude_bin/avrdude80.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : xplainedmini_updi
AVR part              : ATtiny416
Programming modes     : SPM, UPDI
Programmer type       : JTAGICE3_UPDI
Description           : Atmel XplainedMini in UPDI mode
ICE HW version        : 0
ICE FW version        : 1.13 (rel. 43)
Serial number         : ATML2795042700004041

Partial Family_ID returned: "tiny"
Silicon revision: 0.1

AVR device initialized and ready to accept instructions
Device signature = 1E 92 21 (ATtiny416)

Avrdude80 done.  Thank you.
```

mcuee, 2025-06-09:
@stefanrueger 

As this is not a regression and most likely not an avrdude issue but rather FreeBSD USB driver issue, this will not block avrdude 8.1 release.

stefanrueger, 2025-06-09:
Thanks for raising this problem @mcuee!

> Known issue with CH340 and serialupdi: not an avrdude issue, but rather FreeBSD CH340 driver issue

If there is nothing AVRDUDE can do, perhaps better document this under known limitations rather than opening an issue?

mcuee, 2025-06-10:
> > Known issue with CH340 and serialupdi: not an avrdude issue, but rather FreeBSD CH340 driver issue
> 
> If there is nothing AVRDUDE can do, perhaps better document this under known limitations rather than opening an issue?

Sorry for the confusion. That one with CH340 is documented. 
https://github.com/avrdudes/avrdude/wiki/Known-limitations-of-avrdude

> CH340 based serialupdi programmer does not work under FreeBSD due to the limitation of FreeBSD CH340 driver.
> https://github.com/avrdudes/avrdude/discussions/1639

This issue #2010 is about a new issue with ATtiny416 Xplained Nano under FreeBSD. I have edited the first post to make it clear.

mcuee, 2025-06-12:
Wiki updated.
https://github.com/avrdudes/avrdude/wiki/Known-limitations-of-avrdude#known-issues-for-programmers

closed 2025-06-12 by mcuee